kaloz/mwlwifi #223: 9.3.8 88W8964 Firmware

rs-se, 2017-10-09:
Does anyone know where to find to the 9.3.8 firmware for 88W8964?

All that I can seem to find is 9.3.7.

anomeome, 2017-10-09:
There is a link in this [thread](https://github.com/kaloz/mwlwifi/issues/182)

rs-se, 2017-10-09:
Thanks, I figured there was a link somewhere but missed that one.

davidc502, 2017-10-09:
yuhhaurlin,

Please commit 9.3.8 88W8964 Firmware.

Best Regards,

David

yuhhaurlin, 2017-10-09:
Yes. I will check in 9.3.0.8 tomorrow (10/11).

gufus, 2017-10-09:
Does it have fixed api for wds?

yuhhaurlin, 2017-10-09:
No.WDS needs the modification of driver. It won't be done soon. I will try to support it before 2018.

rs-se, 2017-10-09:
Latest driver and firmware is very stable, I have done stress testing on
the driver via iperf3 on both firmware and 9.3.8 works much better.

yuhhaurlin, 2017-10-09:
Good to know that.

yuhhaurlin, 2017-10-11:
Done.

closed 2017-10-11 by yuhhaurlin