kaloz/mwlwifi #224: 5G wireless error reappears

fuqiang03, 2017-10-17:
These are just my guesses
The newly updated file
It's not necessarily a wireless driven problem, but I need help
[image]
[image]
[image]

yuhhaurlin, 2017-10-18:
The firmware is gone.

yuhhaurlin, 2017-10-18:
How to let it happen?

yuhhaurlin, 2017-10-18:
Please load LEDE image from https://davidc502sis.dynamic-dns.net/. Reproduce the problem and let me know how to reproduce the code.

fuqiang03, 2017-10-18:
I rolled back some files and was ruling out other possibilities

aaron1959, 2017-10-18:
@fuqiang03 Can you please load that test image he posted so we can get this crap fixed soon.

fuqiang03, 2017-10-18:
Delete this plug-in and return to normal
Thank you very much
[image]

aaron1959, 2017-10-18:
@fuqiang03  HE won't use that information! He will only want you to load that image and log the failure.

aaron1959, 2017-10-18:
Been doing this for months

fuqiang03, 2017-10-19:
How do I do that

aaron1959, 2017-10-19:
Download the file that was posted above and install it run it under the same conditions you did to get this failure and log it in detail so that the log information from the failure can then be analyzed or whatever they do with it.

aaron1959, 2017-10-19:
Also is that thing you call a plug in something you added to some standard LEDE image, that seems to be the implication as you refer to it as a "plugin", If so I wouldn't think a plug in wouldn't  be relevant here.
If after all that is part of the standard LEDE package then it may be a broad clue but the folks from Marvell have been VERY consistent  about wanting the above procedure to be followed to proceed any further, I hope to god I'm mistaken but don't think so. :-(

fuqiang03, 2017-10-19:
I'll send the logs to the past. But it takes time

yuhhaurlin, 2017-10-24:
Image for WRT1900ACS:

LEDE upgrade image: https://drive.google.com/open?id=0B3qLWtcWB9EdX3MzZ2pTdHM4bHM
LEDE factory image: https://drive.google.com/open?id=0B3qLWtcWB9EdTTRsbGdhZ2tvVDQ
Configuration file: https://drive.google.com/open?id=0B3qLWtcWB9EdTmhVYTZaWXM0SFE

Please help to verify LEDE image (clone on 2017/10/23) with latest mwlwifi driver and firmware.

Once if you encounter "host command timeout" problem, please let me know your setting and test environment. If possible, please use one single client and iperf to reproduce the problem. I will choose the simplest way to reproduce the same problem. Thanks.

root@LEDE:/# uname -a
Linux LEDE 4.9.57 #0 SMP Sun Oct 22 21:15:11 2017 armv7l GNU/Linux

root@LEDE:/# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.4.0-20170810
firmware version: -------------
power table loaded from dts: no
firmware region code: 0x10
mac address: 14:91:82:ba:ac:e9
2g: disable
5g: enable
antenna: 4 4
irq number: 45
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e0f80000
iobase1: e1200000
tx limit: 1024
rx limit: 16384

fuqiang03, 2017-10-24:
Can I provide source code to help you find problems?

fuqiang03, 2017-10-24:
This problem has nothing to do with the driver version
Lede kernel 4.4.93 and 4.4.92 have problems

fuqiang03, 2017-10-24:
If you need source code
Give me an e-mail address
I sent the past

yuhhaurlin, 2017-10-24:
Sorry. I only takes care of problems related to mwlwifi. If you think this issue is related to specific kernel, I need to close it.

BrainSlayer, 2017-10-24:
@yuhhaurlin : its not related to any kernel version. we have seen this for months on dd-wrt, lede and any kernel which was used. no matter if 4.9 or 4.4 etc. u dont know why he wrotes this

yuhhaurlin, 2017-10-24:
@BrainSlayer This is for WRT1900ACS.

@fuqiang03 Please use the image I gave you to reproduce it.

yuhhaurlin, 2017-10-24:
Looks like fuqiang03 is leaving. I close this one first.

closed 2017-10-24 by yuhhaurlin

fuqiang03, 2017-10-26:
ddd

aaron1959, 2017-10-26:
@fuqiang03 Thank you.

fuqiang03, 2017-11-02:
错误的避免方法:
1.要排除插件的源码错误
2.make dirclean（这个个人感觉不主要）
3.删除配置文件重新编译（可以不清除编译链）
4.这个坑必须由lede背了
————————————————————————————————2017-11-02  第三次记录

fuqiang03, 2017-11-07:
关于5g无线宕机的问题
首先排除源码的错误，例如lede更新的源码。
删除配置文件在次编译，不要添加任何插件，仅添加luci。排除内核问题。
lede的编译环境可能会造成这样的错误。这个是有偶尔的。
这个情况发现多了，就会有处理经验了。
------------------------------------------------------------------------------2017-11-07