kaloz/mwlwifi #225: Crash with re-authorization

jacksonatl, 2017-10-18:
After about 2 days of operation, if I leave the network and return with the Samsung Galaxy S7 Edge, the router crashes. I was able to capture logs as the load spiked.

jacksonatl, 2017-10-18:
cat /sys/kernel/debug/ieee80211/phy1/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170810
firmware version: 0x09030007
power table loaded from dts: no
firmware region code: 0x10
mac address: 60:38:e0:b9:c6:c1
2g: enable
5g: disable
antenna: 4 4
irq number: 106
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000000
radio: disable
iobase0: e1280000
iobase1: e1500000
tx limit: 1024
rx limit: 16384

yuhhaurlin, 2017-10-18:
You use LEDE? Let me know how do you get the code. I will load it to do test.

aaron1959, 2017-10-18:
I get the exact same issue with galaxy S7  Good luck with this.

yuhhaurlin, 2017-10-18:
Please load LEDE image from https://davidc502sis.dynamic-dns.net/. Reproduce the problem and let me know how to reproduce the problem.

BrainSlayer, 2017-10-18:
@yuhhaurlin : now that you found one that uses lede you're asking him for using another lede which uses the same driver and kernel. that will not help you to reproduce the problem. please try to loan a galaxy s7 then would help you to find it.  and the steps to reproduce seem to be very simple but there is no way to describe it. simply use it with a galaxy s7 is the only step i can find here and this is also the only step someone could tell you here. he was leaving the network and entering it again. so disconnect and connecting again. this would maybe help to find it. just try to constantly disconnect and connect to it again.

yuhhaurlin, 2017-10-18:
I am pretty busy now. I hope user can load the image from this URL to make sure the problem can be reproduced. Later I can load the same image to check it. Thanks.

kubrickfr, 2017-10-18:
lol, I why should anybody trust builds by this dude? ridiculous...

yuhhaurlin, 2017-10-18:
Have a common LEDE image for debug. If you can reproduce "host command timeout" on it. I can load the same image try to reproduce the problem and hook ICE to debug it.
It looks like this problem only happened with Samsung S7?

aaron1959, 2017-10-18:
Brand new Galaxy S8 also has exact same issue so if you can borrow one of those, I told you last month about the Galaxy S7 issue.

jacksonatl, 2017-10-18:
I have installed the LEDE image as requested. It will be about 48 hours
before I can replicate the situation.

jacksonatl, 2017-10-19:
Ok, can confirm that this issue exists when running that random guy's image too. After 2 days up time, a re-authorization of a Galaxy S7 causes router to crash and reboot.

jacksonatl, 2017-10-19:
I was unable to capture logs during the last crash, but it is not hard to reproduce. It does however take exactly 2 days of up time. Reauthorization within the first 48 hours does not appear to cause problems.

jacksonatl, 2017-10-19:
Would you confirm that altering the Makefile in the LEDE build thusly will compile both the latest version and latest firmware:
PKG_VERSION:=10.3.4.0.git-20170810
PKG_RELEASE=2
PKG_SOURCE_VERSION:=9a6db695f17c0c9ec5d4602afc9c36290c3bdea1
PKG_MIRROR_HASH:=e119077b68d64e368cb9cc46bd364308db4289dc

aaron1959, 2017-10-19:
The "random guy" works for Marvell I believe and He only wants the logs NOTHING else.
If you decide not to post the logs then you are most likely wasting your time.
He has said this over and over and again.. Please RUN THE IMAGE AND RUN TO FAILURE AND THEN POST LOGS!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

aaron1959, 2017-10-19:
If you can't hook up a serial logger to your router then you will NOT be able to capture a log. Also the person who works for Marvell is the ONLY person with full access to the driver code.

aaron1959, 2017-10-19:
He has stated that he won't find a Samsung phone to test, I also have no problem reproducing these failures but he refuses my serial logs because they come from DD-WRT.

yuhhaurlin, 2017-10-19:
Thanks. I will try to reproduce the problem.

yuhhaurlin, 2017-10-19:
BTW, can you help to check if you can reproduce the same problem with stock firmware?

yuhhaurlin, 2017-10-19:
If you can get log before, why can't you get log this time?

yuhhaurlin, 2017-10-19:
However, the way to reproduce the problem:
1. Let the device run 48 hours. With or without traffic doesn't matter?
2. Use Samsung S7 to connect/disconnect with the router to let the problem happen.

jacksonatl, 2017-10-19:
Correct. Don't believe traffic levels matter but the problem does not begin
prior to 48 hours up time, I've never made it to hour 72. At some point
during day two when the cell phone leaves the primuses, upon return router
crashes. I don't have serial connection, but I can work on that over the
holiday season, so the logs are just when I'm prepared for the crash with
ssh connection and scripted dmesg and logread. If I could trigger it in
under 48 hours of up time I'd have more logs for you.

yuhhaurlin, 2017-10-19:
Thanks. Can you try to reproduce this problem with stock firmware?

jacksonatl, 2017-10-19:
That's tough, I can't really run it in production but I can try leaving it
running for two days and connecting with the phone. I'm guessing I'd have
to have a serial connection to get any meaningful logs out of stock too?

yuhhaurlin, 2017-10-19:
For me, I just needs to make sure it is "host command timeout" and I need to reproduce it in order to debug it. For stock firmware, it may lock the system. It is better to have system log.

anomeome, 2017-10-19:
@jacksonatl , that change should get the latest firmware, but either of the following:

- dmesg | grep -i firm
- cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

should give you the requisite information.

@aaron1959, I don't think that _random guy_ works for Marvell

BrainSlayer, 2017-10-19:
isnt it enough to proof it in lede? i mean, it already proofs that there is a bug. no matter if it works with stock or not. i mean it could be a mwlwifi firmware bug (stock uses the same firmware) or a bug in handling the firmware. so you may just validate one of these if you run it on stock. but with stock a standard user cannot do any logging without serial access

yuhhaurlin, 2017-10-19:
If you can reproduce this problem on stock firmware:
1. The problem is related to firmware.
2. Report to Belkin, there will be other one to check this problem. It could speed up the fix of this problem.

aaron1959, 2017-10-20:
HOLD on I have No problem running stock linksys firmware and doing this exact same test with a serial connection in putty I will make sure I have the latest linksys software and load it and start logging right away. That I can do!

aaron1959, 2017-10-20:
I'm testing now!

rs-se, 2017-10-20:
I am experiencing this exact same problem and according to david's forum over at OpenWRT there are others too. https://forum.openwrt.org/viewtopic.php?id=64949&p=145

Looks like an issue with hostapd and phy0 and I am using latest LEDE image from david's site dated October 18th with latest wifi version and firmware ( https://davidc502sis.dynamic-dns.net/releases/lede-mvebu-linksys-wrt3200acm-squashfs-sysupgrade.bin ).

```
Thu Oct 19 21:32:30 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK dc:ef:ca:87:70:8f
Thu Oct 19 21:33:59 2017 daemon.info odhcpd[1572]: Using a RA lifetime of 1800 seconds on br-lan
Thu Oct 19 21:35:00 2017 daemon.info dnscrypt-proxy[2926]: dnscrypt-proxy Refetching server certificates
Thu Oct 19 21:35:15 2017 daemon.err dnscrypt-proxy[2926]: dnscrypt-proxy Unable to retrieve server certificates
Thu Oct 19 21:37:25 2017 daemon.info odhcpd[1572]: Using a RA lifetime of 1800 seconds on br-lan
Thu Oct 19 21:37:33 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK dc:ef:ca:87:70:8f
Thu Oct 19 21:40:15 2017 daemon.info dnscrypt-proxy[2926]: dnscrypt-proxy Refetching server certificates
Thu Oct 19 21:40:30 2017 daemon.err dnscrypt-proxy[2926]: dnscrypt-proxy Unable to retrieve server certificates
Thu Oct 19 21:42:36 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK dc:ef:ca:87:70:8f
Thu Oct 19 21:45:23 2017 daemon.info odhcpd[1572]: Using a RA lifetime of 1800 seconds on br-lan
Thu Oct 19 21:45:30 2017 daemon.info dnscrypt-proxy[2926]: dnscrypt-proxy Refetching server certificates
Thu Oct 19 21:45:45 2017 daemon.err dnscrypt-proxy[2926]: dnscrypt-proxy Unable to retrieve server certificates
Thu Oct 19 21:47:39 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK dc:ef:ca:87:70:8f
Thu Oct 19 21:48:07 2017 daemon.warn dnsmasq[4365]: possible DNS-rebind attack detected: stash.atl.zillow.net
Thu Oct 19 21:50:45 2017 daemon.info dnscrypt-proxy[2926]: dnscrypt-proxy Refetching server certificates
Thu Oct 19 21:51:00 2017 daemon.err dnscrypt-proxy[2926]: dnscrypt-proxy Unable to retrieve server certificates
Thu Oct 19 21:52:42 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK dc:ef:ca:87:70:8f
Thu Oct 19 21:54:37 2017 daemon.info odhcpd[1572]: Using a RA lifetime of 1800 seconds on br-lan
Thu Oct 19 21:56:00 2017 daemon.info dnscrypt-proxy[2926]: dnscrypt-proxy Refetching server certificates
Thu Oct 19 21:56:15 2017 daemon.err dnscrypt-proxy[2926]: dnscrypt-proxy Unable to retrieve server certificates
Thu Oct 19 21:57:45 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK dc:ef:ca:87:70:8f
Thu Oct 19 21:58:37 2017 daemon.info odhcpd[1572]: Using a RA lifetime of 1800 seconds on br-lan
Thu Oct 19 22:00:23 2017 daemon.notice netifd: wan (1625): udhcpc: sending renew
Thu Oct 19 22:00:23 2017 daemon.notice netifd: wan (1625): udhcpc: lease of 172.220.117.188 obtained, lease time 28800
Thu Oct 19 22:01:15 2017 daemon.info dnscrypt-proxy[2926]: dnscrypt-proxy Refetching server certificates
Thu Oct 19 22:01:30 2017 daemon.err dnscrypt-proxy[2926]: dnscrypt-proxy Unable to retrieve server certificates
Thu Oct 19 22:02:48 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK dc:ef:ca:87:70:8f
Thu Oct 19 22:03:15 2017 daemon.info odhcpd[1572]: Using a RA lifetime of 1800 seconds on br-lan
Thu Oct 19 22:06:30 2017 daemon.info dnscrypt-proxy[2926]: dnscrypt-proxy Refetching server certificates
Thu Oct 19 22:06:45 2017 daemon.err dnscrypt-proxy[2926]: dnscrypt-proxy Unable to retrieve server certificates
Thu Oct 19 22:07:51 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK dc:ef:ca:87:70:8f
Thu Oct 19 22:08:01 2017 daemon.info hostapd: wlan0: STA e8:4e:06:47:ae:d4 IEEE 802.11: associated (aid 3)
Thu Oct 19 22:08:01 2017 daemon.notice hostapd: wlan0: AP-STA-CONNECTED e8:4e:06:47:ae:d4
Thu Oct 19 22:08:01 2017 daemon.info hostapd: wlan0: STA e8:4e:06:47:ae:d4 WPA: pairwise key handshake completed (RSN)
Thu Oct 19 22:08:01 2017 daemon.info hostapd: wlan0: STA e8:4e:06:47:ae:d4 IEEE 802.11: authenticated
Thu Oct 19 22:08:01 2017 daemon.info odhcpd[1572]: Using a RA lifetime of 1800 seconds on br-lan
Thu Oct 19 22:08:02 2017 daemon.notice odhcpd[1572]: Got DHCPv6 request
Thu Oct 19 22:08:02 2017 daemon.warn odhcpd[1572]: DHCPV6 SOLICIT IA_NA from 000100011fef71db28924a476497 on br-lan: ok 2602:100:acdc:75bc::e1a/128 fde4:7339:d619::e1a/128 
Thu Oct 19 22:08:02 2017 daemon.info dnsmasq[4365]: read /tmp/hosts/odhcpd - 0 addresses
Thu Oct 19 22:08:02 2017 daemon.info dnsmasq[4365]: read /tmp/hosts/dhcp.cfg02411c - 2 addresses
Thu Oct 19 22:08:02 2017 daemon.info dnsmasq-dhcp[4365]: read /etc/ethers - 0 addresses
Thu Oct 19 22:08:03 2017 daemon.notice odhcpd[1572]: Got DHCPv6 request
Thu Oct 19 22:08:03 2017 daemon.warn odhcpd[1572]: DHCPV6 REQUEST IA_NA from 000100011fef71db28924a476497 on br-lan: ok 2602:100:acdc:75bc::e1a/128 fde4:7339:d619::e1a/128 
Thu Oct 19 22:08:03 2017 daemon.info dnsmasq[4365]: read /tmp/hosts/odhcpd - 2 addresses
Thu Oct 19 22:08:03 2017 daemon.info dnsmasq[4365]: read /tmp/hosts/dhcp.cfg02411c - 2 addresses
Thu Oct 19 22:08:03 2017 daemon.info dnsmasq-dhcp[4365]: read /etc/ethers - 0 addresses
Thu Oct 19 22:08:58 2017 kern.info kernel: [83329.975727] ieee80211 phy0: staid 1 deleted
Thu Oct 19 22:08:58 2017 daemon.info hostapd: wlan0: STA dc:ef:ca:87:70:8f IEEE 802.11: authenticated
Thu Oct 19 22:08:58 2017 daemon.info hostapd: wlan0: STA dc:ef:ca:87:70:8f IEEE 802.11: associated (aid 1)
[... cut ...]
Thu Oct 19 22:09:40 2017 daemon.notice hostapd: nl80211: nl80211_recv_beacons->nl_recvmsgs failed: -5

root@WRT3200ACM:~# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20171011
firmware version: 0x09030008
power table loaded from dts: no
firmware region code: 0x10
mac address: 60:38:e0:b9:8d:d2
2g: disable
5g: enable
antenna: 4 4
irq number: 45
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e1000000
iobase1: e1280000
tx limit: 1024
rx limit: 16384
```

BrainSlayer, 2017-10-20:
curious. is davids version based on a unofficial driver version.?
driver version: 10.3.4.0-20171011

@yuhhaurlin do you know what this is?

yuhhaurlin, 2017-10-20:
No. It should be 10.3.4.0-20170810.

yuhhaurlin, 2017-10-20:
@re-se From your previous report, you said latest driver and firmware are very stable, right?

yuhhaurlin, 2017-10-20:
I know LEDE hostapd is updated to include fixes for KRACK. But the driver version is not correct.

yuhhaurlin, 2017-10-20:
Let Davidc502 clarify the problem of his build. If problem is still there, please let me know.  I will build LEDE from main trunk and test here. Thanks.

yuhhaurlin, 2017-10-20:
Got e-mail for updating mwlwifi driver and firmware to LEDE:

From: WildByDesign [mailto:notifications@github.com] 
Sent: Wednesday, October 18, 2017 8:32 AM
To: lede-project/source <source@noreply.github.com>
Cc: David Lin <dlin@marvell.com>; Mention <mention@noreply.github.com>
Subject: [EXT] Re: [lede-project/source] 17.01: mwlwifi: update driver to 2017-08-10 (#1425)

External Email 
________________________________________
I have been running the latest mwlwifi kernel driver (10.3.4.0-20170810) for approx. 46 days on WRT3200ACM device on trunk builds and also a user backported package for stable branch LEDE with much better stability and also utilizing multiple SSID's.
I have been running the latest firmware (9.3.0.8) for WRT3200ACM for approx. 22 days now, also with great stability and reliable uptime. This home network consists of a good mix of laptops, multiple iPhones and iPads.
Please consider this pull request for stable branch 17.01.4 release since the current driver is quite outdated and has significant stability issues.

yuhhaurlin, 2017-10-20:
I don't know if LEDE already accepted new driver and firmware. If LEDE already updated the driver and firmware and you encounter problems with the image built from the latest LEDE main trunk, please let me know. I will build code and test here.

anomeome, 2017-10-20:
Current LEDE master/trunk does not include the last firmware push. In order to differentiate what is being run, some like to change the label used in the driver so LuCI provides an indication relative to the git date.

aaron1959, 2017-10-20:
I've been running the Factory image and to this point there has been no 5 Ghz failure logging on and off with galaxy S7, 

So far the serial log doesn't show anything unusual.
I can post the serial log to this point if anyone wants to look at it.

For some reason even at 80 Mhz wide I only get 433 Mbps Rather then the 866 I get with LEDE and DD-WRT? Also my older Dell Precision laptop was connecting with no problem on 5 Ghz now it is refused connection by the linksys software. 

But let us not Get distracted we will continue to test the Linksys factory image for a couple more days. I've done this test twice in the past and reported to yuhheurlin long ago but that may have been a slightly older version of the linksys software.

lucize, 2017-10-20:
LEDE lasted me quite long until recent wireless backport, no iw crashing with

```
[ 7294.733824] ieee80211 phy0: staid 2 deleted
[ 7303.856533] ieee80211 phy0: cmd 0x9111=SetNewStation timed out
[ 7303.862400] ieee80211 phy0: return code: 0x1111
[ 7303.866948] ieee80211 phy0: timeout: 0x1111
[ 8189.715934] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
[ 8189.722063] ieee80211 phy0: return code: 0x1122
[ 8189.726611] ieee80211 phy0: timeout: 0x1122
[ 8189.730810] wlan0: failed to remove key (0, xx:xx:xx:xx:xx:xx) from hardware (-5)
[ 8193.737723] ieee80211 phy0: cmd 0x9125=BAStream timed out
[ 8193.743172] ieee80211 phy0: return code: 0x1125
[ 8193.747720] ieee80211 phy0: timeout: 0x1125
[ 8197.751696] ieee80211 phy0: cmd 0x9125=BAStream timed out
[ 8197.757127] ieee80211 phy0: return code: 0x1125
[ 8197.761677] ieee80211 phy0: timeout: 0x1125
[ 8201.765708] ieee80211 phy0: cmd 0x9125=BAStream timed out
[ 8201.771235] ieee80211 phy0: return code: 0x1125
[ 8201.775798] ieee80211 phy0: timeout: 0x1125
[ 8205.779626] ieee80211 phy0: cmd 0x9111=SetNewStation timed out
[ 8205.785536] ieee80211 phy0: return code: 0x1111
[ 8205.790084] ieee80211 phy0: timeout: 0x1111
[ 8209.793537] ieee80211 phy0: cmd 0x8050=broadcast_ssid_enable timed out
[ 8209.800238] ieee80211 phy0: return code: 0x0050
[ 8209.804796] ieee80211 phy0: timeout: 0x0050
[ 8213.809457] ieee80211 phy0: cmd 0x8127=SetInformationElements timed out
[ 8213.816106] ieee80211 phy0: return code: 0x0127
[ 8213.820653] ieee80211 phy0: timeout: 0x0127
[ 8217.825374] ieee80211 phy0: cmd 0x801c=80211RadioControl timed out
[ 8217.831613] ieee80211 phy0: return code: 0x001c
[ 8217.836161] ieee80211 phy0: timeout: 0x001c
[ 8221.841286] ieee80211 phy0: cmd 0x8126=SetFixedRate timed out
[ 8221.847060] ieee80211 phy0: return code: 0x0126
[ 8221.851619] ieee80211 phy0: timeout: 0x0126
```

aaron1959, 2017-10-20:
If you can reproduce this problem on stock firmware:

    1) The problem is related to firmware.
    2) Report to Belkin, there will be other one to check this problem. It could speed up the fix of this problem.

So it's only been 21 hours still working fine, what do we do if there is No problem with linksys software?
I will keep running it until I have four days then I will be back to the exact same place I was a couple of months ago.

yuhhaurlin, 2017-10-20:
@aaron1959 If you want to complain, there is no need for you to for test.

yuhhaurlin, 2017-10-20:
@lucize Can you show me the version？

lucize, 2017-10-20:
@yuhhaurlin this is what you need ?

```
[   10.979151] Loading modules backported from Linux version wt-2017-10-06-0-ga7a22fbbd1f0
[   10.987200] Backport generated by backports.git v4.14-rc2-1-6-gedfb595c
[   11.001123] <<Marvell Mac80211 Wireless PCIE Network Driver version 10.3.4.0-20170810>>

```

yuhhaurlin, 2017-10-20:
Looks like latest LEDE has this kind of problem, I will build the image and test here. I will also give the LEDE image for you to test at same time.

lucize, 2017-10-20:
thanks !
should I give you my build config, so I can test it without having to install anything after flashing the image ? (like wireguard,samba and a few other packages)

yuhhaurlin, 2017-10-20:
Yes. Please.

yuhhaurlin, 2017-10-20:
BTW, can you use "cat /sys/debug/kernel/ieee80211/phy0/mwlwifi/info"to show me the version information？

lucize, 2017-10-20:
I think is not enabled

```
cat /sys/debug/kernel/ieee80211/phy0/mwlwifi/info
cat: can't open '/sys/debug/kernel/ieee80211/phy0/mwlwifi/info': No such file or directory

```

lucize, 2017-10-20:
here is the config
[config.zip](https://github.com/kaloz/mwlwifi/files/1403557/config.zip)
thanks !

yuhhaurlin, 2017-10-20:
Sorry "/sys/kernel/debug/....".

lucize, 2017-10-20:
should have check it too

```
 cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170810
firmware version: 0x09030007
power table loaded from dts: no
firmware region code: 0x30
mac address: 60:38:e0:bc:36:82
2g: disable
5g: enable
antenna: 4 4
irq number: 45
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e0f80000
iobase1: e1200000
tx limit: 1024
rx limit: 16384

```

aaron1959, 2017-10-21:
This problem doesn't exist for Factory Linksys firmware.

aaron1959, 2017-10-21:
Linksys firmware is still working fine on 5 Ghz connecting and disconnecting a Samsung S7 G930P1.
Once and for all there is No problem with the stock firmware on 5 Ghz!
No More excuses.

ajmgh, 2017-10-21:
@aaron1959  You do realize that @yuhhaurlin is doing you a favor. He doesn't have to work on this. Your warranty and support are with Linksys. Linksys should have paid for an open source engineer, it looks like they choose not to and Marvell are picking up the slack.

lucize, 2017-10-21:
anyway my particular problem is on 2G too

```
[262171.109262] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
[262171.115219] ieee80211 phy0: return code: 0x001d
[262171.119859] ieee80211 phy0: timeout: 0x001d
[317806.329873] ieee80211 phy1: staid 1 deleted
[317815.457218] ieee80211 phy1: cmd 0x9111=SetNewStation timed out
[317815.463166] ieee80211 phy1: return code: 0x1111
[317815.467810] ieee80211 phy1: timeout: 0x1111
[317819.471446] ieee80211 phy1: cmd 0x8050=broadcast_ssid_enable timed out
[317819.478096] ieee80211 phy1: return code: 0x0050
[317819.482731] ieee80211 phy1: timeout: 0x0050
[317823.487372] ieee80211 phy1: cmd 0x8127=SetInformationElements timed out
[317823.494110] ieee80211 phy1: return code: 0x0127
[317823.498746] ieee80211 phy1: timeout: 0x0127
[317827.503299] ieee80211 phy1: cmd 0x801c=80211RadioControl timed out
[317827.509595] ieee80211 phy1: return code: 0x001c
[317827.514238] ieee80211 phy1: timeout: 0x001c
[317831.519233] ieee80211 phy1: cmd 0x8126=SetFixedRate timed out
[317831.525099] ieee80211 phy1: return code: 0x0126
[317831.529734] ieee80211 phy1: timeout: 0x0126
[324867.169568] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
[324867.175527] ieee80211 phy0: return code: 0x001d
[324867.180170] ieee80211 phy0: timeout: 0x001d
[324871.183883] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out

```

anomeome, 2017-10-21:
@lucize , in your comment on the PR you had stated that you though this started with the _compat_ push, just wondering if you have tried backing that commit out.

lucize, 2017-10-21:
I'll wait for @yuhhaurlin image, is not so important for the moment, anyway from what I saw, the router will reboot anyway if not, I'll do a reboot, I don't have critical adapters connected via wireless

aaron1959, 2017-10-22:
linksys firmware is still fine on 5 Ghz with Samsung Galaxy S7 G930P.The "stock" firmware does not have this problem.

aaron1959, 2017-10-22:
@ajmgh I think you're mistaken yuhhaurlin is NOT doing this out of the kindness of his heart! Do you think this guy was just sitting around and said to himself one day " hey maybe I'll go and find some project were I can help folks with some random routers" B.S. this guy represents someone and dodging the reality won't change the fact that who ever is paying him doesn't have a lot of concern with time schedules. This guy is NO volunteer!.

[167 more replies]

closed 2017-11-18 by yuhhaurlin