kaloz/mwlwifi #226: 180Mhz

p3x-robot, 2017-10-20:
Ciao!
How are you?
I am on the latest version, actually the https://github.com/eduperez/mwlwifi_LEDE/.
It is the same code by the way.

So, given the COUNTRY code and radar is working, it means I do not have to change anything, given LEDE 17.01.3 includes ```hostapad``` as well.
So I do not have to do anything and it should work?

p3x-robot, 2017-10-20:
of course it is not in the kernel or anywhere:
```
opkg remove kmod-mwifiex-sdio
opkg remove mwifiex-sdio-firmware
```

That it is why radar is working, but why is not working with the 160Mhz?

I get this error:
```text
Fri Sep  4 03:22:46 2020 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Fri Sep  4 03:22:46 2020 daemon.notice hostapd: handle_probe_req: send failed
Fri Sep  4 03:22:47 2020 daemon.notice hostapd: 20/40 MHz operation not permitted on channel pri=1 sec=5 based on overlapping BSSes
Fri Sep  4 03:22:53 2020 daemon.notice hostapd: ACS: Only VHT20/40/80 is supported now
Fri Sep  4 03:22:54 2020 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0 wasn't started
Fri Sep  4 03:22:54 2020 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Fri Sep  4 03:22:54 2020 kern.debug kernel: [   49.841253] ieee80211 phy0: change: 0x60
Fri Sep  4 03:22:54 2020 kern.info kernel: [   49.881291] device wlan0 left promiscuous mode
Fri Sep  4 03:22:54 2020 kern.info kernel: [   49.885786] br-lan: port 3(wlan0) entered disabled state
Fri Sep  4 03:22:54 2020 kern.debug kernel: [   49.943286] ieee80211 phy0: change: 0x100
Fri Sep  4 03:22:54 2020 daemon.notice hostapd: ELOOP: remaining socket: sock=21 eloop_data=0xb6f4ad00 user_data=0 handler=0x3e3fc
Fri Sep  4 03:22:54 2020 daemon.notice netifd: radio0 (1554): cat: can't open '/var/run/wifi-phy0.pid': No such file or directory
Fri Sep  4 03:22:54 2020 daemon.notice netifd: radio0 (1554): Command failed: Invalid argument
Fri Sep  4 03:22:54 2020 kern.debug kernel: [   50.121253] ieee80211 phy0: change: 0xffffffff
Fri Sep  4 03:22:54 2020 kern.info kernel: [   50.227290] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Fri Sep  4 03:22:54 2020 kern.info kernel: [   50.246941] device wlan0 entered promiscuous mode
```
So with 2.4Ghz everything works and is fast like 500Mbit.
But with 5Ghz 160Mhz, is not working already in 3200ACM, I removed packages the country is all the same, radar workings, but 160Mhz it doesn't get start up. Only 80Mhz.

yuhhaurlin, 2017-10-20:
You need to check regulatory rules "iw reg get". And for DFS channels, you need to wait 60 seconds before it working. You can use "iw dev" to check if SSID appears. The current problem is inter oprability with R7800.

closed 2017-10-20 by yuhhaurlin

p3x-robot, 2017-10-20:
But it's the same error:

```text
Fri Oct 20 11:19:38 2017 daemon.notice hostapd: ACS: Only VHT20/40/80 is supported now
Fri Oct 20 11:19:38 2017 daemon.notice hostapd: wlan0: ACS-COMPLETED freq=5620 channel=124
Fri Oct 20 11:19:38 2017 daemon.notice hostapd: wlan0: interface state ACS->HT_SCAN
Fri Oct 20 11:19:38 2017 kern.debug kernel: [ 1274.825830] ieee80211 phy0: change: 0x60
Fri Oct 20 11:19:38 2017 kern.debug kernel: [ 1274.920822] ieee80211 phy0: change: 0x100
Fri Oct 20 11:19:38 2017 kern.debug kernel: [ 1274.933881] ieee80211 phy0: change: 0x100
Fri Oct 20 11:19:38 2017 kern.debug kernel: [ 1274.946047] ieee80211 phy0: change: 0x60
Fri Oct 20 11:19:39 2017 kern.debug kernel: [ 1275.155812] ieee80211 phy0: change: 0x40
Fri Oct 20 11:19:39 2017 daemon.err hostapd: Interface initialization failed
Fri Oct 20 11:19:39 2017 daemon.notice hostapd: wlan0: interface state HT_SCAN->DISABLED
Fri Oct 20 11:19:39 2017 daemon.notice hostapd: wlan0: AP-DISABLED
Fri Oct 20 11:19:39 2017 daemon.notice hostapd: wlan0: interface state DISABLED->DISABLED
Fri Oct 20 11:19:39 2017 daemon.notice hostapd: wlan0: AP-DISABLED
Fri Oct 20 11:19:39 2017 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Fri Oct 20 11:19:39 2017 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0 wasn't started
Fri Oct 20 11:19:39 2017 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Fri Oct 20 11:19:39 2017 kern.debug kernel: [ 1275.365811] ieee80211 phy0: change: 0x60

Fri Oct 20 11:19:39 2017 kern.info kernel: [ 1275.395851] device wlan0 left promiscuous mode
Fri Oct 20 11:19:39 2017 kern.info kernel: [ 1275.400377] br-lan: port 3(wlan0) entered disabled state
Fri Oct 20 11:19:39 2017 kern.debug kernel: [ 1275.465810] ieee80211 phy0: change: 0x100
Fri Oct 20 11:19:39 2017 daemon.notice hostapd: ELOOP: remaining socket: sock=22 eloop_data=0xb6e69d00 user_data=0 hand$
Fri Oct 20 11:19:39 2017 kern.debug kernel: [ 1275.642802] ieee80211 phy0: change: 0xffffffff
Fri Oct 20 11:19:39 2017 kern.info kernel: [ 1275.749569] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
```

***device wlan0 left promiscuous mode***

Event though it says everything it says.
Exact pyh country FR (France). Still It cannot start up.

p3x-robot, 2017-10-20:
```text
root@digi:~# iw dev
phy#1
	Interface wlan1
		ifindex 11
		wdev 0x100000002
		addr 60:38:e0:bd:7a:69
		ssid patrikx3-digi-slow
		type AP
		channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
		txpower 20.00 dBm
phy#0
	Interface wlan0
		ifindex 20
		wdev 0x7
		addr 60:38:e0:bd:7a:6a
		type AP
		txpower 20.00 dBm
```

yuhhaurlin, 2017-10-20:
1. First make sure your setting is allowable by the regulatory domain.
2. If DFS channels are involved, please wait for 60 seconds.

yuhhaurlin, 2017-10-20:
Don't use ACS: daemon.notice hostapd: ACS: Only VHT20/40/80 is supported now

yuhhaurlin, 2017-10-20:
Your log tells you what happened!

yuhhaurlin, 2017-10-20:
Use WEB GUI or modify /etc/config/wireless. Please check OpenWrt WiKi.

yuhhaurlin, 2017-10-20:
Or "uci". Anyway, please check these things by yourself.

p3x-robot, 2017-10-20:
Ciao! 

How are you?
But there is now options to disable ACS:

```text
config wifi-device 'radio0'
        option type 'mac80211'
        option hwmode '11a'
        option path 'soc/soc:pcie-controller/pci0000:00/0000:00:01.0/0000:01:00.0'
        option txpower '20'
        option country 'FR'
        option channel 'auto'
        option htmode 'VHT160'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option mode 'ap'
        option ssid 'patrikx3-digi'
        option encryption 'psk2'
        option wps_pushbutton '1'
        option network 'lan'
        option key '*******'
```

[image]

BrainSlayer, 2017-10-20:
according to your screenshit "wireless network is enabled" = disable. ahm. could you please enable it?

p3x-robot, 2017-10-20:
It shows:
[image]

If I click it, it will be disabled, so to make it work I have to disable?

BrainSlayer, 2017-10-20:
try also to set a channel manually like 100. i believe the mwlwifi firmware has a hardcoded channel list and not all  combinations are supported for 160 mhz. auto channel might select a channel which isnt supported for 160 mhz operation. i tried channel 100 (but with dd-wrt) and its working

yuhhaurlin, 2017-10-20:
In US, 36 channel (160 MHz) is working too.

p3x-robot, 2017-10-20:
It works, with 100 channel, but the problem is it is changing the channel with the RADAR, I should disable the Radar Detection? Because it is working, but it is changin the channels at will (it is the radar, that is doing?)

[image]

Now it is again 36 chanell and is working, but I set to 100, but it changes.
So 160MHZ and Radar Detection is not working together or what?

yuhhaurlin, 2017-10-20:
If radar is detected, channel switch will happen and hostapd will choose channel for you.

p3x-robot, 2017-10-20:
ok, so radar detection is OK?

yuhhaurlin, 2017-10-20:
Should be.

p3x-robot, 2017-10-20:
Wow, it is awesome, so it is using 2 wifi connections in once?

[image]

But it is still not the MU-MIMO right? Or it is already? 

Look, I have 3 Wifi in 2 SSID:

[image]

This is totally werid.