kaloz/mwlwifi #227: Generic MAC80211 Wireless Controller

thagabe, 2017-10-31:
Hey @yuhhaurlin,

Recent compiles of a full disk img LEDE with various distros (LTS Ubuntu, Current Ubuntu, Arch Linux) yields an image that loads a bunch of bluetooth drivers, mwifiex-sdio,  and Marvell PCEI Controller. The issue is that i cannot create a wifi, as it will not be advertised. MWLWIFI is loaded when i check lsmod. Maybe this has something to do with update in mac80211 or cfg80211 in how it interacts with the driver.

yuhhaurlin, 2017-10-31:
You can choose what kind of drivers you want to load via "make menuconfig".

thagabe, 2017-10-31:
I know that, but default 2 drivers are selected 1) MWLWIFI 2) MWIFIEX. Last time i compiled it didnt matter that both were selected however it seems recent patches may have had some effect on which one the kernel is choosing. I will test my hypothesis

yuhhaurlin, 2017-10-31:
If you select "Profile", it will choose some default configurations. Normally you can check what your want by yourself. It is only related to build mechanism,

yuhhaurlin, 2017-10-31:
Only mwifiex has support for blue tooth. Mwlwifi for 88W8864 and 88W8964 are pure wifi.

closed 2017-10-31 by yuhhaurlin

thagabe, 2017-10-31:
https://github.com/lede-project/source/commit/471d5dc6e3c16a2bef9bde2b788b563a00357292 I believe this is the commit that is causing the problems at LEDE. Since both mwlwifi and mwifiex are selected by default on LEDE the "autoprobe" initiates the wireless controllers under the mwifiex driver rather than mwlwifi. I did not try reverting this commit, however when I manually disabled mwifiex and not compiled bluetooth in LEDE's menuconfig then "autoprobe" was able to correctly select mwlwifi driver and the corresponding config.

yuhhaurlin, 2017-10-31:
Previously the makefile will generate file "50-mwlwifi" under directory "/etc/modules.d".  I have not yet tested this patch for "autoprobe". Thanks for your information.

thagabe, 2017-10-31:
Some of the fastest/stable throughput in a while. Maybe LEDE should scrap mwified and just run mwlwifi nearly as much throughput as wire from about 10 ft. Ping seem reasonable! pretty good stuff.