kaloz/mwlwifi #229: ESP8266 Cannot connect to WRT3200ACM DD-WRT on WIFI 2.4ghz 802.11b/g/n

kaboom748, 2017-11-12:
I've found the solution for working together ESP8266 and WRT3200ACM:

1- Disable Wireless Physical Interface ath1 [2.4 Ghz TurboQAM] --> INCOMPATIBILITY

2- Use the ath2
--> AP
--> Mixed
--> Channel: Full 20mhz
--> Channel: 1 - 6 or 11

ENJOY !!!

Don't forget to set the ath0 to 5ghz only... Sometimes ESP8266 try to connect to 

802.11n is a 2.4ghz and 5ghz protocol......

closed 2017-11-12 by yuhhaurlin

rs-se, 2017-11-12:
This isn't an actual fix, this is only a workaround to get ESP8266 to use
ath2 instead of 0 or 1 but might I add this will only work at short
distances as ath2 only transmits at 1.

aaron1959, 2017-11-12:
The antenna for ath2 is a tiny internal antenna. 
I'll take a picture if you don't understand, it's not good beyond 50 feet or so.

aaron1959, 2017-11-12:
[image]