kaloz/mwlwifi #231: Constant log flow of phy1: change in 2g client mode

kubrickfr, 2017-11-15:
With the latest commit and firmware (or last "stable") I get a continuous flow of these when 2g is disabled but set-up in client mode:
```
Wed Nov 15 12:43:51 2017 kern.debug kernel: [   57.131906] ieee80211 phy1: change: 0x40
Wed Nov 15 12:43:52 2017 kern.debug kernel: [   57.281823] ieee80211 phy1: change: 0x40
Wed Nov 15 12:43:52 2017 kern.debug kernel: [   57.451823] ieee80211 phy1: change: 0x40
Wed Nov 15 12:43:52 2017 kern.debug kernel: [   57.601818] ieee80211 phy1: change: 0x40
Wed Nov 15 12:43:52 2017 kern.debug kernel: [   57.761816] ieee80211 phy1: change: 0x40
Wed Nov 15 12:43:52 2017 kern.debug kernel: [   57.911826] ieee80211 phy1: change: 0x40
Wed Nov 15 12:43:52 2017 kern.debug kernel: [   58.071827] ieee80211 phy1: change: 0x40
Wed Nov 15 12:43:53 2017 kern.debug kernel: [   58.221822] ieee80211 phy1: change: 0x40
Wed Nov 15 12:43:53 2017 kern.debug kernel: [   58.381808] ieee80211 phy1: change: 0x40
Wed Nov 15 12:43:53 2017 kern.debug kernel: [   58.531807] ieee80211 phy1: change: 0x40
Wed Nov 15 12:43:53 2017 kern.debug kernel: [   58.691811] ieee80211 phy1: change: 0x40
Wed Nov 15 12:43:53 2017 kern.debug kernel: [   58.841815] ieee80211 phy1: change: 0x40
Wed Nov 15 12:43:53 2017 kern.debug kernel: [   59.011806] ieee80211 phy1: change: 0x40
Wed Nov 15 12:43:53 2017 kern.debug kernel: [   59.161804] ieee80211 phy1: change: 0x40
Wed Nov 15 12:43:54 2017 kern.debug kernel: [   59.238814] ieee80211 phy1: change: 0x100
Wed Nov 15 12:43:54 2017 kern.debug kernel: [   60.163807] ieee80211 phy1: change: 0x100
```
```
cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170810
firmware version: 0x0903bee1
power table loaded from dts: no
firmware region code: 0x0
mac address: 60:38:e0:b6:xx:xx
2g: disable
5g: enable
antenna: 4 4
irq number: 45
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000003
radio: enable
iobase0: e0e00000
iobase1: e1080000
tx limit: 1024
rx limit: 16384
```

Is it a LEDE issue or a driver issue?

yuhhaurlin, 2017-11-15:
These messages are channel scanning.

closed 2017-11-15 by yuhhaurlin

kubrickfr, 2017-11-15:
Why is it doing it even if the interface is disabled though?

yuhhaurlin, 2017-11-15:
I just know it is channel scanning. These messages are all right.

yuhhaurlin, 2017-11-15:
Make sure you encounter and report log to me. These change messages are information related to configurations setting to driver.

kubrickfr, 2017-11-16:
Reported upstream: https://bugs.lede-project.org/index.php?do=details&task_id=1172

yuhhaurlin, 2017-11-16:
You can ignore this one.

kubrickfr, 2017-11-16:
Sometimes I wonder if you have a real attitude problem or if it's just your English that makes you come across as ill-willed and uncaring.

yuhhaurlin, 2017-11-16:
I think this is minor problem. Can you check if wpa_supplicant is still running when you disable interface?

kubrickfr, 2017-11-16:
I didn't know what these messages were, now that you told me that this is scanning, I believe you when you say this is not a driver issue, therefore I have open a bug report upstream and cross-referenced it. I do not expect you to fix LEDE problems.
I will check wpa_supplicant.

yuhhaurlin, 2017-11-16:
@kubrickfr  0x40 => Bit 6 => IEEE80211_CONF_CHANGE_CHANNEL.

enum ieee80211_conf_changed {
	IEEE80211_CONF_CHANGE_SMPS		= BIT(1),
	IEEE80211_CONF_CHANGE_LISTEN_INTERVAL	= BIT(2),
	IEEE80211_CONF_CHANGE_MONITOR		= BIT(3),
	IEEE80211_CONF_CHANGE_PS		= BIT(4),
	IEEE80211_CONF_CHANGE_POWER		= BIT(5),
	IEEE80211_CONF_CHANGE_CHANNEL		= BIT(6),
	IEEE80211_CONF_CHANGE_RETRY_LIMITS	= BIT(7),
	IEEE80211_CONF_CHANGE_IDLE		= BIT(8),
};

yuhhaurlin, 2017-11-16:
@kubrickfr Sorry. I can make sure these messages are related to "change channel". Mostly it means driver is controlled to do scanning.