kaloz/mwlwifi #232: Host command timeout

yuhhaurlin, 2017-11-18:
Please use this one to check this issue.

gsustek, 2017-11-18:
1.839606] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Sat Nov 18 09:11:21 2017 kern.info kernel: [ 2181.846545] br-lan: port 2(wlan0) entered blocking state
Sat Nov 18 09:11:21 2017 kern.info kernel: [ 2181.853200] br-lan: port 2(wlan0) entered disabled state
Sat Nov 18 09:11:21 2017 kern.info kernel: [ 2181.860103] device wlan0 entered promiscuous mode
Sat Nov 18 09:11:21 2017 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Sat Nov 18 09:11:21 2017 daemon.notice hostapd: handle_probe_req: send failed
Sat Nov 18 09:11:21 2017 daemon.notice hostapd: handle_probe_req: send failed
[ 2182.168824] ------------[ cut here ]------------
[ 2182.173515] WARNING: CPU: 1 PID: 4471 at backports-2017-11-01/net/mac80211/ieee80211_i.h:1412 sta_info_alloc+0x1ac/0x3bc [mac80211]
[ 2182.185403] Modules linked in: pppoe ppp_asyncSat Nov 18 09:11 uvcvideo:21 2017 kern.wa snd_usb_audiorn kernel: [ 218 pppox ppp_generic nf_conntrack_ipv62.168824] ------ iptable_nat ipt_REJECT ipt_MASQUERADE------[ cut here xt_time ]------------
 xt_tcpudpSat Nov 18 09:11 xt_state:21 2017 kern.wa xt_recent xt_quotarn kernel: [ 218 xt_pkttype xt_physdev2.173515] WARNIN xt_ownerG: CPU: 1 PID: 4 xt_nat471 at backports xt_multiport-2017-11-01/net/ xt_markmac80211/ieee802 xt_mac11_i.h:1412 sta_ xt_limitinfo_alloc+0x1ac xt_helper/0x3bc [mac80211 xt_conntrack]
 xt_connmark xt_connlimit xt_connbytes xt_comment xt_addrtype xt_TCPMSS xt_REDIRECT xt_LOG videobuf2_v4l2 snd_usbmidi_lib slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_netlink nf_conntrack iptable_mangle iptable_filter ip_tables input_core crc_ccitt br_netfilter videobuf2_vmalloc videobuf2_memops videobuf2_core v4l2_common videodev mwlwifi mac80211 cfg80211 compat cryptodev xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnetSat Nov 18 09:11 ip_set_hash_ipportip ip_set_hash_ipport:21 2017 kern.wa ip_set_hash_ipmarkrn kernel: [ 218 ip_set_hash_ip2.185403] Module ip_set_bitmap_ports linked in: ppp ip_set_bitmap_ipmacoe ppp_async uvc ip_set_bitmap_ipvideo snd_usb_au ip_setdio pppox ppp_ge nfnetlinkneric nf_conntra ip6t_REJECTck_ipv6 iptable_ nf_reject_ipv6nat ipt_REJECT i nf_log_ipv6pt_MASQUERADE xt nf_log_common_time xt_tcpudp  ip6table_manglext_state xt_rece ip6table_filter ip6_tablesnt xt_quota xt_p x_tableskttype xt_physde nfsdv xt_owner xt_na nfsv3t xt_multiport x nfst_mark xt_mac xt sit_limit xt_helper tunnel4 ip_tunnel xt_conntrack xt tun_connmark xt_con snd_compressnlimit xt_connby snd_pcm_osstes xt_comment x snd_mixer_osst_addrtype xt_TC snd_pcmPMSS xt_REDIRECT snd_timer xt_LOG videobuf snd_rawmidi2_v4l2 snd_usbmi snd_seq_devicedi_lib slhc nf_r snd_hwdepeject_ipv4 nf_na sndt_redirect nf_na soundcoret_masquerade_ipv lockd4 nf_conntrack_i sunrpcpv4 nf_nat_ipv4  gracenf_nat nf_log_ip dma_shared_bufferv4 nf_defrag_ipv sha256_generic6 nf_defrag_ipv4 seqiv nf_conntrack_rt jitterentropy_rngcache nf_conntra drbgck_netlink nf_co hmacnntrack iptable_ ctrmangle iptable_f ccmilter ip_tables  cbcinput_core crc_c authenccitt br_netfilte gpio_button_hotplugr videobuf2_vmal exportfsloc videobuf2_me
mops videobuf2_c[ 2182.406209] CPU: 1 PID: 4471 Comm: hostapd Not tainted 4.9.58 #0
ore v4l2_common [ 2182.413627] Hardware name: Marvell Armada 380/385 (Device Tree)
videodev mwlwifi[ 2182.420980] [<c00160dc>] (unwind_backtrace) from [<c00122dc>] (show_stack+0x10/0x14)
 mac80211 cfg802[ 2182.430137] [<c00122dc>] (show_stack) from [<c0218cac>] (dump_stack+0x7c/0x9c)
11 compat crypto[ 2182.438777] [<c0218cac>] (dump_stack) from [<c00292fc>] (__warn+0xd0/0xf0)
dev xt_set ip_se[ 2182.447069] [<c00292fc>] (__warn) from [<c00293c0>] (warn_slowpath_null+0x1c/0x24)
t_list_set
[ 2182.456105] [<c00293c0>] (warn_slowpath_null) from [<bf2251dc>] (sta_info_alloc+0x1ac/0x3bc [mac80211])
[ 2182.466597] [<bf2251dc>] (sta_info_alloc [mac80211]) from [<bf23afc4>] (ieee80211_add_station+0x80/0x118 [mac80211])
[ 2182.477208] [<bf23afc4>] (ieee80211_add_station [mac80211]) from [<bf1fdcec>] (nl80211_new_station+0x2f8/0x368 [cfg80211])
[ 2182.488326] [<bf1fdcec>] (nl80211_new_station [cfg80211]) from [<bf1e36d4>] (extack_doit+0x2c/0x70 [compat])
[ 2182.498205] [<bf1e36d4>] (extack_doit [compat]) from [<c03ca994>] (genl_rcv_msg+0x298/0x304)
[ 2182.506680] [<c03ca994>] (genl_rcv_msg) from [<c03c9d5c>] (netlink_rcv_skb+0x68/0xbc)
[ 2182.514543] [<c03c9d5c>] (netlink_rcv_skb) from [<c03ca6e8>] (genl_rcv+0x20/0x34)
[ 2182.522057] [<c03ca6e8>] (genl_rcv) from [<c03c9768>] (netlink_unicast+0x140/0x200)
[ 2182.529746] [<c03c9768>] (netlink_unicast) from [<c03c9bc4>] (netlink_sendmsg+0x2e0/0x2f8)
[ 2182.538046] [<c03c9bc4>] (netlink_sendmsg) from [<c03889f0>] (sock_sendmsg+0x14/0x24)
[ 2182.545909] [<c03889f0>] (sock_sendmsg) from [<c0388fc0>] (___sys_sendmsg+0x1d8/0x224)
[ 2182.553859] [<c0388fc0>] (___sys_sendmsg) from [<c0389d04>] (__sys_sendmsg+0x40/0x64)
[ 2182.561724] [<c0389d04>] (__sys_sendmsg) from [<c000eda0>] (ret_fast_syscall+0x0/0x3c)
[ 2182.569691] ---[ end trace dd974daeadd51b4a ]---
Sat Nov 18 09:11[ 2182.574704] ------------[ cut here ]------------
:21 2017 kern.wa[ 2182.580595] WARNING: CPU: 1 PID: 4471 at backports-2017-11-01/net/mac80211/ieee80211_i.h:1412 sta_info_alloc+0x1ac/0x3bc [mac80211]
[... cut ...]
Sat Nov 18 09:11:24 2017 daemon.notice hostapd: handle_probe_req: send failed
Sat Nov 18 09:11:24 2017 daemon.notice hostapd: handle_probe_req: send failed
Sat Nov 18 09:11:24 2017 daemon.notice hostapd: handle_probe_req: send failed
Sat Nov 18 09:11:26 2017 daemon.notice hostapd: handle_probe_req: send failed
Sat Nov 18 09:11:26 2017 daemon.notice hostapd: handle_probe_req: send failed
Sat Nov 18 09:11:26 2017 daemon.notice hostapd: handle_probe_req: send failed
Sat Nov 18 09:11:26 2017 daemon.notice hostapd: handle_probe_req: send failed

gsustek, 2017-11-18:
router after reboot can't add channel width 160.....need to change to 80 and then to 160.

root@3200acm:/# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170810
firmware version: 0x09030008
power table loaded from dts: no
firmware region code: 0x30
mac address: xx:xx:xx:xx:xx:xx
2g: disable
5g: enable
antenna: 4 4
irq number: 45
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: disable
iobase0: e1100000
iobase1: e1380000
tx limit: 1024
rx limit: 16384
root@3200acm:/# iw dev
phy#1
        Interface wlan1
                ifindex 11
                wdev 0x100000002
                addr yy:yy:yy:yy:yy:yy
                ssid 3GCDF
                type AP
                channel 11 (2462 MHz), width: 40 MHz, center1: 2452 MHz
                txpower 20.00 dBm
phy#0
        Interface wlan0
                ifindex 12
                wdev 0x2
                addr xx:xx:xx:xx:xx:xx
                type AP
                txpower 20.00 dBm

root@3200acm:/# iwinfo
wlan0     ESSID: "SDERDG"
          Access Point: yy:yy:yy:yy:yy:yy
          Mode: Master  Channel: 100 (5.500 GHz)
          Tx-Power: 20 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -100 dBm
          Bit Rate: unknown
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11nac
          Hardware: 11AB:2B40 11AB:0000 [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0

wlan1     ESSID: "3GKSJA"
          Access Point: xx:xx:xx:xx:xx:xx
          Mode: Master  Channel: 11 (2.462 GHz)
          Tx-Power: 20 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -78 dBm
          Bit Rate: unknown
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: 11AB:2B40 11AB:0000 [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy1

BrainSlayer, 2017-11-18:
@gsustek wrong mwlwifi firmware version used. please use the test mwlwifi firmware provided by @yuhhaurlin

yuhhaurlin, 2017-11-18:
Yes. I hope we can use 9.3.0.8. It looks like it is warning message. I will check it when I come to office.

gsustek, 2017-11-18:
@BrainSlayer but where? i clean and rebuild latest mwlwifi driver from repo... is there a google drive link?

BrainSlayer, 2017-11-18:
@gsustek he provided this link in another issue thread https://drive.google.com/open?id=1sEUvE_7VKSsS1d0YxYMiUZzOKwDP295V but it seems that this firmware migth also uncover other issues.

gsustek, 2017-11-18:
root@3200acm:~# find /lib -name  88W8964.bin
/lib/firmware/mwlwifi/88W8964.bin
root@3200acm:~# ls -lrt /lib/firmware/mwlwifi/88W8964.bin
-rwxr-xr-x    1 root     root        447616 Nov 17 10:14 /lib/firmware/mwlwifi/88W8964.bin
root@3200acm:~# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info 

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170810
firmware version: 0x09030008
power table loaded from dts: no
firmware region code: 0x30
mac address: 60:38:e0:bc:32:f2
2g: disable
5g: enable
antenna: 4 4
irq number: 45
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e1100000
iobase1: e1380000
tx limit: 1024
rx limit: 16384

BrainSlayer, 2017-11-18:
449304 Nov 13 08:43 88W8964.bin

see the size difference?

lucize, 2017-11-18:
@yuhhaurlin I think I understood right, to test the latest git commit and this firmware version ?

 cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

```
driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170810
firmware version: 0x09030008
power table loaded from dts: no
firmware region code: 0x30
mac address: 60:38:e0:xx:xx:xx
2g: disable
5g: enable
antenna: 4 4
irq number: 45
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e1000000
iobase1: e1280000
tx limit: 1024
rx limit: 16384

```

yuhhaurlin, 2017-11-18:
Yes. I will use it to reproduce and debug problem. Thanks.

yuhhaurlin, 2017-11-18:
BTW, if your log messages are the same as before, there is no need for you to post them to here again.

gsustek, 2017-11-18:
@BrainSlayer you are wrong.

See what @yuhhaurlin said before.

now i have the same firmware..

root@3200acm:~# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170810
firmware version: 0x0903bee1
power table loaded from dts: no
firmware region code: 0x0

BrainSlayer, 2017-11-18:
@gsustek i just saw your log above and it showed the wrong version. the new log here is correct. so i'm totally right

gsustek, 2017-11-18:
@BrainSlayer please read 3 threads above.  "lucize commented 2 hours ago " and  "yuhhaurlin commented 2 hours ago " So what is the best setup now?

yuhhaurlin, 2017-11-18:
Please use 9.3.0.8. It has been tested for a period of time. It looks like it only has this issue. It is better we focus on thus version of firmware and fix this problem. Thanks.

lucize, 2017-11-18:
so it crashed quickly, leaving the range helped, this time only two tries in getting out of range and triggered the bug, don't know if the output is the same, I'll attach as txt, this time kernel is not crashing
there is already 30 minutes from when it started, usually it takes ~10 minutes to reboot

[with_9.3.0.8.txt](https://github.com/kaloz/mwlwifi/files/1484629/with_9.3.0.8.txt)

ratsputin, 2017-11-18:
So, same crash.  Both of us left on a bike-ride for about an hour (iphone 6 [dc:41:5f:xx:xx:xx] & iphone 6+ [e0:b5:2d:xx:xx:xx] ), came back and the router had crashed within five minutes of getting home.  Built from Trunk, with the following mwlwifi version:

`PKG_SOURCE_VERSION:=466368f9454250c2bc024795600d92564553d9bb`

```
===root@orion:/usr/var/log# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170810
firmware version: 0x09030008
power table loaded from dts: no
firmware region code: 0x10
mac address: 60:38:e0:xx:xx:xx
2g: disable
5g: enable
antenna: 4 4
irq number: 45
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e0f00000
iobase1: e1180000
tx limit: 1024
rx limit: 16384
```

Here's the crashlog:
```
Time: 1511022745.707110
Modules:        nf_conntrack_ipv6@bf2d0000+15bc iptable_nat@bf2c7000+451        ipt_REJECT@bf2c3000+3ea ipt_MASQUERADE@bf2bf000+332xt_time@bf2bb000+6a6     xt_tcpudp@bf2b7000+758  xt_tcpmss@bf2b3000+44a  xt_statistic@bf2af000+3d5       xt_state@bf2ab000+349   xt_recent@bf2a6000+1960     xt_nat@bf2a2000+59d     xt_multiport@bf29e000+554       xt_mark@bf29a000+320    xt_mac@bf296000+2c7     xt_limit@bf292000+48d       xt_length@bf28e000+365  xt_hl@bf28a000+3de      xt_helper@bf286000+3cc  xt_ecn@bf282000+59c     xt_dscp@bf27e000+496        xt_conntrack@bf27a000+958       xt_connmark@bf276000+534        xt_connlimit@bf271000+10d9      xt_connbytes@bf26d000+595   xt_comment@bf269000+24b xt_TCPMSS@bf265000+a68  xt_REDIRECT@bf261000+375        xt_LOG@bf25d000+3c3     xt_HL@bf259000+598 xt_DSCP@bf255000+5ee     xt_CLASSIFY@bf251000+2cc        nf_reject_ipv4@bf24d000+783     nf_nat_redirect@bf249000+3db    nf_nat_masquerade_ipv4@bf245000+60d nf_conntrack_ipv4@bf240000+1443 nf_nat_ipv4@bf23c000+cc7        nf_nat@bf236000+263c    nf_log_ipv4@bf232000+c36    nf_defrag_ipv6@bf22b000+3214    nf_defrag_ipv4@bf227000+37c     nf_conntrack_rtcache@bf223000+a52       nf_conntrack@bf211000+cfd8  mwifiex_sdio@bf208000+5174      mwifiex@bf1cd000+3255d  iptable_mangle@bf1c9000+3fc     iptable_filter@bf1c5000+37c     ipt_ECN@bf1c1000+5a4        ip_tables@bf1bb000+2597 hid@bf1a5000+11f10      evdev@bf19f000+27a3     input_core@bf195000+5ed7        mwlwifi@bf172000+1b202      mac80211@bf104000+5f811 cfg80211@bf0ca000+31035 compat@bf0c5000+1bf0    xt_set@bf0c0000+1923    ip_set_list_set@bf0bb000+1804       ip_set_hash_netiface@bf0b2000+5189      ip_set_hash_netport@bf0a9000+51ee       ip_set_hash_netnet@bf0a0000+56d3    ip_set_hash_net@bf098000+4c42   ip_set_hash_netportnet@bf08f000+5aff    ip_set_hash_mac@bf089000+21e9   ip_set_hash_ipportnet@bf080000+5664 ip_set_hash_ipportip@bf078000+44c9      ip_set_hash_ipport@bf070000+4193        ip_set_hash_ipmark@bf069000+3f53   ip_set_hash_ip@bf062000+3e67     ip_set_bitmap_port@bf05d000+1402        ip_set_bitmap_ipmac@bf058000+168e       ip_set_bitmap_ip@bf053000+16aa      ip_set@bf049000+50c9    nfnetlink@bf045000+f5f  ip6t_REJECT@bf041000+3ec        nf_reject_ipv6@bf03d000+848     nf_log_ipv6@bf039000+cf7    nf_log_common@bf035000+90f      ip6table_mangle@bf031000+4be    ip6table_filter@bf02d000+33e    ip6_tables@bf027000+2495    x_tables@bf021000+2aac  regmap_mmio@bf01d000+9c1        ecb@bf019000+5e1        cmac@bf015000+8ad       uhci_hcd@bf00d000+4328      ohci_platform@bf009000+e88      ohci_hcd@bf000000+53fb
<6>[   16.011315] mwifiex_sdio mmc0:0001:1: mwifiex_process_cmdresp: cmd 0x242 failed during    initialization
<6>[   16.038662] mwifiex_sdio mmc0:0001:1: info: MWIFIEX VERSION: mwifiex 1.0 (15.68.7.p71)
<6>[   16.046979] mwifiex_sdio mmc0:0001:1: driver_version = mwifiex 1.0 (15.68.7.p71)
<7>[   16.313631] ieee80211 phy0: change: 0xffffffff
<6>[   16.425920] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
<7>[   16.671617] ieee80211 phy1: change: 0xffffffff
<6>[   16.807817] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
<6>[   16.814921] br-lan: port 2(wlan1) entered blocking state
<6>[   16.820299] br-lan: port 2(wlan1) entered disabled state
<6>[   16.825781] device wlan1 entered promiscuous mode
<6>[   16.830608] br-lan: port 2(wlan1) entered blocking state
<6>[   16.835983] br-lan: port 2(wlan1) entered forwarding state
<6>[   16.858021] br-lan: port 3(wlan0) entered blocking state
<6>[   16.863413] br-lan: port 3(wlan0) entered disabled state
<6>[   16.868896] device wlan0 entered promiscuous mode
<6>[   16.873653] br-lan: port 3(wlan0) entered blocking state
<6>[   16.878992] br-lan: port 3(wlan0) entered forwarding state
<6>[   16.884901] br-lan: port 3(wlan0) entered disabled state
<6>[   16.894699] br-lan: port 2(wlan1) entered disabled state
<7>[   16.901345] ieee80211 phy0: change: 0x100
<7>[   16.914039] ieee80211 phy0: change: 0x40
<7>[   16.961260] ieee80211 phy1: change: 0x100
<7>[   16.973610] ieee80211 phy1: change: 0x42
<7>[   17.152623] ieee80211 phy0: change: 0x40
<6>[   17.248763] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
<6>[   17.255284] br-lan: port 2(wlan1) entered blocking state
<6>[   17.260642] br-lan: port 2(wlan1) entered forwarding state
<7>[   17.393686] ieee80211 phy0: change: 0x40
<7>[   17.495619] ieee80211 phy0: change: 0x100
<7>[   17.508636] ieee80211 phy0: change: 0x100
<7>[   17.521597] ieee80211 phy0: change: 0x42
<7>[   17.621607] ieee80211 phy0: change: 0x40
<6>[   45.991744] ieee80211 phy1: staid 2 deleted
[... cut ...]
<4>[40803.232619] [<c03c4c5c>] (netlink_sendmsg) from [<c0384c40>] (sock_sendmsg+0x14/0x24)
<4>[40803.240483] [<c0384c40>] (sock_sendmsg) from [<c03851d0>] (___sys_sendmsg+0x184/0x228)
<4>[40803.248433] [<c03851d0>] (___sys_sendmsg) from [<c0385f6c>] (__sys_sendmsg+0x40/0x64)
<4>[40803.256298] [<c0385f6c>] (__sys_sendmsg) from [<c000ed80>] (ret_fast_syscall+0x0/0x3c)
<0>[40803.264249] Code: e5840010 e5d81004 e5c41014 e598c000 (e59c300c)
<4>[40803.270394] ---[ end trace 6e9089f656ccdd8d ]---
```

ratsputin, 2017-11-18:
I should follow that up by saying that just having one device leave/come back seems to be less-likely to cause a crash as having two devices do it at the same time.  I'll pay closer attention to this over the weekend.

gsustek, 2017-11-18:
here it is:

root@3200acm:/#
root@3200acm:/#
root@3200acm:/# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170810
firmware version: 0x0903bee1
power table loaded from dts: no
firmware region code: 0x0
mac address: 60:38:e0:aa:bb:cc
2g: disable
5g: enable
antenna: 4 4
irq number: 45
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e1100000
iobase1: e1380000
tx limit: 1024
rx limit: 16384

root@3200acm:/# iw dev
phy#1
        Interface wlan1
                ifindex 10
                wdev 0x100000002
                addr 60:38:e0:cc:bb:aa
                ssid 2G
                type AP
                channel 11 (2462 MHz), width: 40 MHz, center1: 2452 MHz
                txpower 20.00 dBm
phy#0
        Interface wlan0
                ifindex 12
                wdev 0x2
                addr 60:38:e0:aa:bb:cc
                ssid 5G
                type AP
                channel 100 (5500 MHz), width: 160 MHz, center1: 5570 MHz
                txpower 27.00 dBm
root@3200acm:/# iwinfo
wlan0     ESSID: "5G"
          Access Point: 60:38:e0:aa:bb:cc
          Mode: Master  Channel: 100 (5.500 GHz)
          Tx-Power: 27 dBm  Link Quality: 70/70
          Signal: -19 dBm  Noise: -104 dBm
          Bit Rate: 736.6 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11nac
          Hardware: 11AB:2B40 11AB:0000 [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0

wlan1     ESSID: "2G"
          Access Point: 60:38:e0:cc:bb:aa
          Mode: Master  Channel: 11 (2.462 GHz)
          Tx-Power: 20 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -65 dBm
          Bit Rate: unknown
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: 11AB:2B40 11AB:0000 [Generic MAC80211]
          TX power offset: unknown
[... 444 lines cut ...]
[32348.598093] Exception stack(0xd526dd48 to 0xd526dd90)
[32348.603165] dd40:                   c0612d00 c063db80 00000000 00000000 00000000 ffffe000
[32348.611376] dd60: 00000002 00000011 df408000 d526c000 df6b2280 c0612080 00000000 d526dd98
[32348.619585] dd80: c002d780 c002d33c 60000113 ffffffff
[32348.624659] [<c0012d4c>] (__irq_svc) from [<c002d33c>] (__do_softirq+0x80/0x234)
[32348.632086] [<c002d33c>] (__do_softirq) from [<c002d780>] (irq_exit+0x94/0xb8)
[32348.639340] [<c002d780>] (irq_exit) from [<c0062f44>] (__handle_domain_irq+0x80/0xb4)
[32348.647203] [<c0062f44>] (__handle_domain_irq) from [<c0009448>] (gic_handle_irq+0x54/0x8c)
[32348.655587] [<c0009448>] (gic_handle_irq) from [<c0012d4c>] (__irq_svc+0x6c/0x90)
[32348.663098] Exception stack(0xd526de18 to 0xd526de60)
[32348.668168] de00:                                                       c0680954 a0000013
[32348.676379] de20: 0000307e 0000307f c0680954 00000ca3 00000000 e0ab0000 de0c4800 00000000
[32348.684589] de40: df6b2280 df4c5240 00000000 d526de68 c0285de8 c046dff8 a0000013 ffffffff
[32348.692803] [<c0012d4c>] (__irq_svc) from [<c046dff8>] (_raw_spin_unlock_irqrestore+0x1c/0x20)
[32348.701453] [<c046dff8>] (_raw_spin_unlock_irqrestore) from [<c0285de8>] (uart_write_room+0x80/0xcc)
[32348.710626] [<c0285de8>] (uart_write_room) from [<c027e9c0>] (n_tty_write+0x18c/0x3e4)
[32348.718576] [<c027e9c0>] (n_tty_write) from [<c027b2e8>] (tty_write+0x21c/0x26c)
[32348.726004] [<c027b2e8>] (tty_write) from [<c00d76d0>] (__vfs_write+0x1c/0x108)
[32348.733344] [<c00d76d0>] (__vfs_write) from [<c00d82e0>] (vfs_write+0xb8/0x144)
[32348.740684] [<c00d82e0>] (vfs_write) from [<c00d8e90>] (SyS_write+0x3c/0x74)
[32348.747762] [<c00d8e90>] (SyS_write) from [<c000eda0>] (ret_fast_syscall+0x0/0x3c)
[32348.755846] Rebooting in 3 seconds..

BootROM - 1.73
Booting from NAND flash

ratsputin, 2017-11-18:
Same behavior.  We were gone for about three hours.  Came back and router immediately crashed.

```
root@orion:/usr/var/log# cat crashlog
Time: 1511039307.210338
Modules:        nf_conntrack_ipv6@bf2d0000+15bc iptable_nat@bf2c7000+451        ipt_REJECT@bf2c3000+3ea ipt_MASQUERADE@bf2bf000+332xt_time@bf2bb000+6a6     xt_tcpudp@bf2b7000+758  xt_tcpmss@bf2b3000+44a  xt_statistic@bf2af000+3d5       xt_state@bf2ab000+349   xt_recent@bf2a6000+1960     xt_nat@bf2a2000+59d     xt_multiport@bf29e000+554       xt_mark@bf29a000+320    xt_mac@bf296000+2c7     xt_limit@bf292000+48d       xt_length@bf28e000+365  xt_hl@bf28a000+3de      xt_helper@bf286000+3cc  xt_ecn@bf282000+59c     xt_dscp@bf27e000+496        xt_conntrack@bf27a000+958       xt_connmark@bf276000+534        xt_connlimit@bf271000+10d9      xt_connbytes@bf26d000+595   xt_comment@bf269000+24b xt_TCPMSS@bf265000+a68  xt_REDIRECT@bf261000+375        xt_LOG@bf25d000+3c3     xt_HL@bf259000+598 xt_DSCP@bf255000+5ee     xt_CLASSIFY@bf251000+2cc        nf_reject_ipv4@bf24d000+783     nf_nat_redirect@bf249000+3db    nf_nat_masquerade_ipv4@bf245000+60d nf_conntrack_ipv4@bf240000+1443 nf_nat_ipv4@bf23c000+cc7        nf_nat@bf236000+263c    nf_log_ipv4@bf232000+c36    nf_defrag_ipv6@bf22b000+3214    nf_defrag_ipv4@bf227000+37c     nf_conntrack_rtcache@bf223000+a52       nf_conntrack@bf211000+cfd8  mwifiex_sdio@bf208000+5174      mwifiex@bf1cd000+3255d  iptable_mangle@bf1c9000+3fc     iptable_filter@bf1c5000+37c     ipt_ECN@bf1c1000+5a4        ip_tables@bf1bb000+2597 hid@bf1a5000+11f10      evdev@bf19f000+27a3     input_core@bf195000+5ed7        mwlwifi@bf172000+1b202      mac80211@bf104000+5f811 cfg80211@bf0ca000+31035 compat@bf0c5000+1bf0    xt_set@bf0c0000+1923    ip_set_list_set@bf0bb000+1804       ip_set_hash_netiface@bf0b2000+5189      ip_set_hash_netport@bf0a9000+51ee       ip_set_hash_netnet@bf0a0000+56d3    ip_set_hash_net@bf098000+4c42   ip_set_hash_netportnet@bf08f000+5aff    ip_set_hash_mac@bf089000+21e9   ip_set_hash_ipportnet@bf080000+5664 ip_set_hash_ipportip@bf078000+44c9      ip_set_hash_ipport@bf070000+4193        ip_set_hash_ipmark@bf069000+3f53   ip_set_hash_ip@bf062000+3e67     ip_set_bitmap_port@bf05d000+1402        ip_set_bitmap_ipmac@bf058000+168e       ip_set_bitmap_ip@bf053000+16aa      ip_set@bf049000+50c9    nfnetlink@bf045000+f5f  ip6t_REJECT@bf041000+3ec        nf_reject_ipv6@bf03d000+848     nf_log_ipv6@bf039000+cf7    nf_log_common@bf035000+90f      ip6table_mangle@bf031000+4be    ip6table_filter@bf02d000+33e    ip6_tables@bf027000+2495    x_tables@bf021000+2aac  regmap_mmio@bf01d000+9c1        ecb@bf019000+5e1        cmac@bf015000+8ad       uhci_hcd@bf00d000+4328      ohci_platform@bf009000+e88      ohci_hcd@bf000000+53fb
<3>[16392.395937] ieee80211 phy0: timeout: 0x001d
<3>[16396.399347] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
<3>[16396.405213] ieee80211 phy0: return code: 0x001d
<3>[16396.409768] ieee80211 phy0: timeout: 0x001d
<3>[16400.413223] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
<3>[16400.419090] ieee80211 phy0: return code: 0x001d
<3>[16400.423645] ieee80211 phy0: timeout: 0x001d
<6>[16400.427979] mwifiex_sdio mmc0:0001:1: cmd_wait_q terminated: -512
<3>[16404.448106] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
<3>[16404.453974] ieee80211 phy0: return code: 0x001d
<3>[16404.458531] ieee80211 phy0: timeout: 0x001d
<3>[16408.461974] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
<3>[16408.467852] ieee80211 phy0: return code: 0x001d
<3>[16408.472408] ieee80211 phy0: timeout: 0x001d
<3>[16412.475849] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
<3>[16412.481717] ieee80211 phy0: return code: 0x001d
<3>[16412.486278] ieee80211 phy0: timeout: 0x001d
<3>[16416.592721] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
<3>[16416.598588] ieee80211 phy0: return code: 0x001d
<3>[16416.603141] ieee80211 phy0: timeout: 0x001d
<3>[16420.606597] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
<3>[16420.612465] ieee80211 phy0: return code: 0x001d
<3>[16420.617047] ieee80211 phy0: timeout: 0x001d
<3>[16424.621473] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
<3>[16424.627366] ieee80211 phy0: return code: 0x001d
<3>[16424.631928] ieee80211 phy0: timeout: 0x001d
<3>[16428.654348] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
<3>[16428.660217] ieee80211 phy0: return code: 0x001d
<3>[16428.664771] ieee80211 phy0: timeout: 0x001d
<3>[16432.668228] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
<3>[16432.674100] ieee80211 phy0: return code: 0x001d
<3>[16432.678677] ieee80211 phy0: timeout: 0x001d
<3>[16436.682099] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
<3>[16436.687966] ieee80211 phy0: return code: 0x001d
<3>[16436.692521] ieee80211 phy0: timeout: 0x001d
<3>[16440.696976] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
<3>[16440.702897] ieee80211 phy0: return code: 0x001d
<3>[16440.707455] ieee80211 phy0: timeout: 0x001d
<3>[16444.710851] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
<3>[16444.716718] ieee80211 phy0: return code: 0x001d
<3>[16444.721274] ieee80211 phy0: timeout: 0x001d
<3>[16448.724728] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
<3>[16448.730596] ieee80211 phy0: return code: 0x001d
<3>[16448.735151] ieee80211 phy0: timeout: 0x001d
<3>[16452.739602] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
<3>[16452.745468] ieee80211 phy0: return code: 0x001d
<3>[16452.750023] ieee80211 phy0: timeout: 0x001d
[... cut ...]
Sat Nov 18 15:02:32 2017 kern.err kernel: [16194.900272] ieee80211 phy0: timeout: 0x1122
Sat Nov 18 15:02:32 2017 kern.err kernel: [16194.904491] wlan0: failed to set key (0, e0:b5:2d:xx:xx:xx) to hardware (-5)
Sat Nov 18 15:02:32 2017 daemon.notice hostapd: wlan0: AP-STA-CONNECTED e0:b5:2d:xx:xx:xx
```
I mention this, because the AP-STA-POLL-OK sounds like it's saying that it's successfully polling that MAC address and I can guarantee you that's impossible, as both of those MAC addresses were several miles away from the router.

I don't know if that helps, but...

ratsputin, 2017-11-18:
Apologies.  I just realized there's a gap between the crash dump and the history I provided.  Here's the missing piece:
```
Sat Nov 18 15:02:36 2017 kern.err kernel: [16198.948459] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Sat Nov 18 15:02:36 2017 kern.err kernel: [16198.954327] ieee80211 phy0: return code: 0x001d
Sat Nov 18 15:02:36 2017 kern.err kernel: [16198.958888] ieee80211 phy0: timeout: 0x001d
Sat Nov 18 15:02:40 2017 kern.err kernel: [16202.962333] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Sat Nov 18 15:02:40 2017 kern.err kernel: [16202.968201] ieee80211 phy0: return code: 0x001d
Sat Nov 18 15:02:40 2017 kern.err kernel: [16202.972759] ieee80211 phy0: timeout: 0x001d
Sat Nov 18 15:02:44 2017 kern.err kernel: [16206.976210] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Sat Nov 18 15:02:44 2017 kern.err kernel: [16206.982079] ieee80211 phy0: return code: 0x001d
Sat Nov 18 15:02:44 2017 kern.err kernel: [16206.986633] ieee80211 phy0: timeout: 0x001d
Sat Nov 18 15:02:48 2017 kern.err kernel: [16210.997087] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Sat Nov 18 15:02:48 2017 kern.err kernel: [16211.002954] ieee80211 phy0: return code: 0x001d
Sat Nov 18 15:02:48 2017 kern.err kernel: [16211.007510] ieee80211 phy0: timeout: 0x001d
Sat Nov 18 15:02:52 2017 kern.err kernel: [16215.010961] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Sat Nov 18 15:02:52 2017 kern.err kernel: [16215.016828] ieee80211 phy0: return code: 0x001d
Sat Nov 18 15:02:52 2017 kern.err kernel: [16215.021382] ieee80211 phy0: timeout: 0x001d
Sat Nov 18 15:02:56 2017 kern.err kernel: [16219.024841] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Sat Nov 18 15:02:56 2017 kern.err kernel: [16219.030721] ieee80211 phy0: return code: 0x001d
Sat Nov 18 15:02:56 2017 kern.err kernel: [16219.035277] ieee80211 phy0: timeout: 0x001d
Sat Nov 18 15:03:00 2017 kern.err kernel: [16223.044715] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Sat Nov 18 15:03:00 2017 kern.err kernel: [16223.050589] ieee80211 phy0: return code: 0x001d
Sat Nov 18 15:03:00 2017 kern.err kernel: [16223.055147] ieee80211 phy0: timeout: 0x001d
Sat Nov 18 15:03:04 2017 kern.err kernel: [16227.058589] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Sat Nov 18 15:03:04 2017 kern.err kernel: [16227.064456] ieee80211 phy0: return code: 0x001d
Sat Nov 18 15:03:04 2017 kern.err kernel: [16227.069016] ieee80211 phy0: timeout: 0x001d
Sat Nov 18 15:03:08 2017 kern.err kernel: [16231.077465] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Sat Nov 18 15:03:08 2017 kern.err kernel: [16231.083351] ieee80211 phy0: return code: 0x001d
Sat Nov 18 15:03:08 2017 kern.err kernel: [16231.087906] ieee80211 phy0: timeout: 0x001d
Sat Nov 18 15:03:12 2017 kern.err kernel: [16235.094341] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Sat Nov 18 15:03:12 2017 kern.err kernel: [16235.100211] ieee80211 phy0: return code: 0x001d
Sat Nov 18 15:03:12 2017 kern.err kernel: [16235.104773] ieee80211 phy0: timeout: 0x001d
Sat Nov 18 15:03:16 2017 kern.err kernel: [16239.108216] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Sat Nov 18 15:03:16 2017 kern.err kernel: [16239.114083] ieee80211 phy0: return code: 0x001d
Sat Nov 18 15:03:16 2017 kern.err kernel: [16239.118637] ieee80211 phy0: timeout: 0x001d
Sat Nov 18 15:03:20 2017 kern.err kernel: [16243.122092] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Sat Nov 18 15:03:20 2017 kern.err kernel: [16243.127959] ieee80211 phy0: return code: 0x001d
Sat Nov 18 15:03:20 2017 kern.err kernel: [16243.132515] ieee80211 phy0: timeout: 0x001d
Sat Nov 18 15:03:20 2017 daemon.notice netifd: wan6 (1437): Command failed: Request timed out
Sat Nov 18 15:03:20 2017 daemon.notice netifd: wan6 (1437): Command failed: Request timed out
Sat Nov 18 15:03:24 2017 kern.err kernel: [16247.275962] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Sat Nov 18 15:03:24 2017 kern.err kernel: [16247.281829] ieee80211 phy0: return code: 0x001d
Sat Nov 18 15:03:24 2017 kern.err kernel: [16247.286382] ieee80211 phy0: timeout: 0x001d
Sat Nov 18 15:03:28 2017 kern.err kernel: [16251.289839] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Sat Nov 18 15:03:28 2017 kern.err kernel: [16251.295708] ieee80211 phy0: return code: 0x001d
Sat Nov 18 15:03:28 2017 kern.err kernel: [16251.300262] ieee80211 phy0: timeout: 0x001d
Sat Nov 18 15:03:32 2017 kern.err kernel: [16255.303715] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Sat Nov 18 15:03:32 2017 kern.err kernel: [16255.309582] ieee80211 phy0: return code: 0x001d
Sat Nov 18 15:03:32 2017 kern.err kernel: [16255.314136] ieee80211 phy0: timeout: 0x001d
Sat Nov 18 15:03:36 2017 kern.err kernel: [16259.332590] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Sat Nov 18 15:03:36 2017 kern.err kernel: [16259.338473] ieee80211 phy0: return code: 0x001d
Sat Nov 18 15:03:36 2017 kern.err kernel: [16259.343030] ieee80211 phy0: timeout: 0x001d
Sat Nov 18 15:03:40 2017 kern.err kernel: [16263.347468] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Sat Nov 18 15:03:40 2017 kern.err kernel: [16263.353337] ieee80211 phy0: return code: 0x001d
Sat Nov 18 15:03:40 2017 kern.err kernel: [16263.357891] ieee80211 phy0: timeout: 0x001d
Sat Nov 18 15:03:44 2017 kern.err kernel: [16267.365342] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Sat Nov 18 15:03:44 2017 kern.err kernel: [16267.371208] ieee80211 phy0: return code: 0x001d
[... cut ...]
Sat Nov 18 15:05:45 2017 kern.err kernel: [16388.222600] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Sat Nov 18 15:05:45 2017 kern.err kernel: [16388.228466] ieee80211 phy0: return code: 0x001d
Sat Nov 18 15:05:45 2017 kern.err kernel: [16388.233023] ieee80211 phy0: timeout: 0x001d
Sat Nov 18 15:05:45 2017 daemon.notice netifd: wan6 (1437): Command failed: Request timed out
Sat Nov 18 15:05:49 2017 kern.err kernel: [16392.385471] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Sat Nov 18 15:05:49 2017 kern.err kernel: [16392.391351] ieee80211 phy0: return code: 0x001d
```

wongsyrone, 2017-11-21:
The stack trace shows the crash happens when destroying ba.
```
<4>[16549.767750] [<bf178660>] (mwl_fwcmd_destroy_ba [mwlwifi]) from [<bf178b04>] (mwl_fwcmd_del_sta_streams+0x1d4/0x330 [mwlwifi])
<4>[16549.779109] [<bf178b04>] (mwl_fwcmd_del_sta_streams [mwlwifi]) from [<bf173ba4>] (mwl_mac80211_sta_remove+0x38/0x1ec [mwlwifi])
<4>[16549.790662] [<bf173ba4>] (mwl_mac80211_sta_remove [mwlwifi]) from [<bf106ca8>] (drv_sta_state+0x16c/0x184 [mac80211])
```
I wonder if this commit fixes the host command timeout issue.

https://github.com/lede-project/source/commit/6a6dc94e0c9a0533eaa9819bf3c22128b009af56

aaron1959, 2017-11-21:
I wish I knew how to compile I'd love to test it.

ratsputin, 2017-11-21:
@aaron1959 Just follow these instructions: https://lede-project.org/docs/guide-developer/quickstart-build-images

yuhhaurlin, 2017-11-21:
I try to use three clients to reproduce this problem:

1. Macbook Air (running iperf 3 TCP Tx from LAN to this client).
2. ThinkPad T430 (running YouTube to play video).
3. Samsumg S6 edge (running YouTube to play video).

However, I have not yet encountered this problem.
So the way to let this problem happen quickly is that I should move Samsung phone out of range and move it back and running YouTube at same time?

aaron1959, 2017-11-21:
You don't need to really be running any apps as far as I can tell the issue is specific with a loss of connection with very weak signal not just turning on and off I found that I can make it easier by putting the phone inside a stainless steel pan on something non conductive to keep it off the surface put the lid on it after making sure it's connected then start walking you'll have to experiment but line of sight for me was just over 30 Meters. it's the going in and out as opposed to on and off.

yuhhaurlin, 2017-11-21:
So if I only use one phone and do this kind of test, the problem will happen? I will try to move phone to shielding room and out to check if I can reproduce this problem. Thanks.

lucize, 2017-11-21:
Yes, there is no need to have traffic or lots of clients connected, most of times I only have one phone using wireless.
I'll try the LEDE commit to see if it's happening again.

yuhhaurlin, 2017-11-21:
@lucize One phone connected to AP and just let it out of range and come back again to let this problem happen? How long does it take to let this problem happen?

lucize, 2017-11-21:
@yuhhaurlin Yes, usually one there is some time when I have the other one connected, Xperia Z3 Mini and Priv, from time to time I have some Notebooks connected, but for sure it doesn't matter how long are the device connected or how much traffic, the fastest one was in about 2-3 hours, usually it takes around 2 days, but leaving the range more frequently seems to trigger it faster

yuhhaurlin, 2017-11-21:
@lucize Thanks. I will continue my current test till tomorrow. After that I will try your and aaron1959's way to reproduce the problem. BTW, can you help to disable power save mode of your phone to see if you will still encounter problem? (There is no specific way to disable power save mode of WiFi for phone, but you can disable power save mode of the phone to check if you will still encounter this problem).

lucize, 2017-11-21:
no sure if I can do that on Priv as is a locked device, I only have these options in developer mode.

[image]

yuhhaurlin, 2017-11-21:
Xperia Z3 Mini should run Android, you can disable power save mode of Android and check if you can let the problem happen via this phone.

yuhhaurlin, 2017-11-21:
As I remember you told me that Xperia Z3 Mini can also let this problem happen.

lucize, 2017-11-21:
Yes, the Priv is also an android device, I'll check later for Z3 if I can disable power save and use only 4G on Priv, but I'll try first with the latest patch from Johannes Berg

yuhhaurlin, 2017-11-21:
@lucize O.K. Thanks.

aaron1959, 2017-11-21:
using Samsung S7 Nougat both power save and normal modes both seem to trigger it the same. I notice no difference.

ratsputin, 2017-11-21:
@yuhhaurlin Regarding multiple devices triggering the issue, as mentioned before, both Saturday and Sunday we left the house (iPhone 6, 6+); Saturday for 1.5 hours, Sunday for three hours.  Both devices returned at the same time and caused a crash both days.

Yesterday, I stayed home all day, but the iPhone 6+ was gone all day (went out-of-range, then returned eight hours later).  When it returned, there was no crash.  I have, however, seen instances when the same behavior has caused a crash.  Although, now that I think about it, the period of time the one device was out-of-range was shorter (~4 hours).

I'm going to do some further testing today with my iPhone 6 and an iPad Air 2 with varying lengths of time away.

yuhhaurlin, 2017-11-21:
Thanks for your information.

ratsputin, 2017-11-21:
Okay, so I left for about 10 minutes with two wireless devices (iPhone 6, iPad Air 2), came back and no crash.  I then left for about an hour with the same two devices, intending to determine if there was a minimum amount of time before the crash will occur.  Unfortunately, at the same time I returned, the iPhone 6+ returned as well, invalidating my experiment.

The router did, however, crash again.

ratsputin, 2017-11-21:
Another small piece of information.  When I returned home and prior to getting the cmd 0x801d=MEMAddrAccess timed out, the wireless devices with me were able to establish connections to the wifi.  If, however, I disconnected and attempted to reconnect, I would receive an invalid password error.

I was, however, able to connect to the third wlan connection (the foreign one--radio2) without any issues, although it was slow to connect.

Interestingly, I've been watching these errors stream for about five minutes, but the router has not yet crashed.  Normally, it only takes a couple of minutes.

ratsputin, 2017-11-21:
@lucize I'm curious as to your crash.  You mention the devices were outside the range twice.  Is there a minimum amount of time they need to be outside of range before a crash will happen?  The minimum I've observed seems to be somewhere between five minutes and an hour.

lucize, 2017-11-21:
@ratsputin during the week I'm not so free with time to make this kind of tests, and is kind of cold outside to go for a walk :) (+2 celsius)

ratsputin, 2017-11-21:
This is very strange.  The router has been streaming
```
Tue Nov 21 12:05:03 2017 kern.err kernel: [248180.230709] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Tue Nov 21 12:05:03 2017 kern.err kernel: [248180.236666] ieee80211 phy0: return code: 0x001d
Tue Nov 21 12:05:03 2017 kern.err kernel: [248180.241314] ieee80211 phy0: timeout: 0x001d
```
For over half an hour.  I'm watching memory usage and it's climbing.  Presumably, the router will crash once memory is exhausted.  It's curious, though, the behavior changes.  This is the first time it's survived this long.

gsustek, 2017-11-21:
hi, i confirm, after the LEDE patch, 
[11381.220313] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
[11381.226188] ieee80211 phy0: return code: 0x051d
[11381.230736] ieee80211 phy0: timeout: 0x001d
Tue Nov 21 21:54:43 2017 kern.err kernel: [11381.220313] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Tue Nov 21 21:54:43 2017 kern.err kernel: [11381.226188] ieee80211 phy0: return code: 0x051d
Tue Nov 21 21:54:43 2017 kern.err kernel: [11381.230736] ieee80211 phy0: timeout: 0x001d
[11385.226206] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
[11385.232129] ieee80211 phy0: return code: 0x001d
[11385.236807] ieee80211 phy0: timeout: 0x001d
Tue Nov 21 21:54:47 2017 kern.err kernel: [11385.226206] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Tue Nov 21 21:54:47 2017 kern.err kernel: [11385.232129] ieee80211 phy0: return code: 0x001d
Tue Nov 21 21:54:47 2017 kern.err kernel: [11385.236807] ieee80211 phy0: timeout: 0x001d
[11389.219160] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
[11389.225100] ieee80211 phy0: return code: 0x001d
[11389.229704] ieee80211 phy0: timeout: 0x001d
Tue Nov 21 21:54:51 2017 kern.err kernel: [11389.219160] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Tue Nov 21 21:54:51 2017 kern.err kernel: [11389.225100] ieee80211 phy0: return code: 0x001d
Tue Nov 21 21:54:51 2017 kern.err kernel: [11389.229704] ieee80211 phy0: timeout: 0x001d
[11393.254110] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
[11393.259971] ieee80211 phy0: return code: 0x001d
[11393.264623] ieee80211 phy0: timeout: 0x001d
Tue Nov 21 21:54:55 2017 kern.err kernel: [11393.254110] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Tue Nov 21 21:54:55 2017 kern.err kernel: [11393.259971] ieee80211 phy0: return code: 0x001d
Tue Nov 21 21:54:55 2017 kern.err kernel: [11393.264623] ieee80211 phy0: timeout: 0x001d
[11397.262049] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
[11397.267910] ieee80211 phy0: return code: 0x001d
[11397.272472] ieee80211 phy0: timeout: 0x001d
Tue Nov 21 21:54:59 2017 kern.err kernel: [11397.262049] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Tue Nov 21 21:54:59 2017 kern.err kernel: [11397.267910] ieee80211 phy0: return code: 0x001d
Tue Nov 21 21:54:59 2017 kern.err kernel: [11397.272472] ieee80211 phy0: timeout: 0x001d
[11401.270129] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
[11401.276105] ieee80211 phy0: return code: 0x001d
[11401.280722] ieee80211 phy0: timeout: 0x001d
Tue Nov 21 21:55:03 2017 kern.err kernel: [11401.270129] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Tue Nov 21 21:55:03 2017 kern.err kernel: [11401.276105] ieee80211 phy0: return code: 0x001d
Tue Nov 21 21:55:03 2017 kern.err kernel: [11401.280722] ieee80211 phy0: timeout: 0x001d
[11405.285963] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
[11405.291825] ieee80211 phy0: return code: 0x001d
[11405.296392] ieee80211 phy0: timeout: 0x001d
Tue Nov 21 21:55:07 2017 kern.err kernel: [11405.285963] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Tue Nov 21 21:55:07 2017 kern.err kernel: [11405.291825] ieee80211 phy0: return code: 0x001d
Tue Nov 21 21:55:07 2017 kern.err kernel: [11405.296392] ieee80211 phy0: timeout: 0x001d
[11409.289925] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
[11409.295798] ieee80211 phy0: return code: 0x001d
[11409.300346] ieee80211 phy0: timeout: 0x001d
Tue Nov 21 21:55:11 2017 kern.err kernel: [11409.289925] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Tue Nov 21 21:55:11 2017 kern.err kernel: [11409.295798] ieee80211 phy0: return code: 0x001d
Tue Nov 21 21:55:11 2017 kern.err kernel: [11409.300346] ieee80211 phy0: timeout: 0x001d
[11413.297880] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
[11413.303754] ieee80211 phy0: return code: 0x001d
[11413.308302] ieee80211 phy0: timeout: 0x001d
Tue Nov 21 21:55:15 2017 kern.err kernel: [11413.297880] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Tue Nov 21 21:55:15 2017 kern.err kernel: [11413.303754] ieee80211 phy0: return code: 0x001d
Tue Nov 21 21:55:15 2017 kern.err kernel: [11413.308302] ieee80211 phy0: timeout: 0x001d
[11417.311826] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
[11417.317689] ieee80211 phy0: return code: 0x001d
[11417.322250] ieee80211 phy0: timeout: 0x001d
Tue Nov 21 21:55:19 2017 kern.err kernel: [11417.311826] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Tue Nov 21 21:55:19 2017 kern.err kernel: [11417.317689] ieee80211 phy0: return code: 0x001d
Tue Nov 21 21:55:19 2017 kern.err kernel: [11417.322250] ieee80211 phy0: timeout: 0x001d
[11421.321005] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
[11421.326879] ieee80211 phy0: return code: 0x001d
[11421.331428] ieee80211 phy0: timeout: 0x001d
Tue Nov 21 21:55:23 2017 kern.err kernel: [11421.321005] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Tue Nov 21 21:55:23 2017 kern.err kernel: [11421.326879] ieee80211 phy0: return code: 0x001d
Tue Nov 21 21:55:23 2017 kern.err kernel: [11421.331428] ieee80211 phy0: timeout: 0x001d
[11425.325744] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
[11425.331611] ieee80211 phy0: return code: 0x001d
[11425.336170] ieee80211 phy0: timeout: 0x001d
[... 47 lines cut ...]
[11437.788307] ddc0: d7ab0098 d54b3d80 c049df70 00000000 c06156f0 00000008 d54b3e30 00000000
[11437.796517] dde0: 00000000 c03d1330 d54b3d80 00000098 de326000 d7ab0000 00000008 c03d1730
[11437.804728] de00: d54b3e30 00000001 80000000 c03d1402 de326000 00000000 00000000 c0637240
[... cut ...]
[11437.927891] dfe0: ffffe000 c005bca8 1f46006a c063d398 00000000 000095ac 00000000 00000001
[11437.936106] [<c00710f8>] (detach_if_pending) from [<40000113>] (0x40000113)
[11437.943096] Code: e1a00003 e12fff1e e3520000 e583c000 (e58c3004)
[11437.949213] ---[ end trace c655c87a720bcdbb ]---
[11437.954511] Kernel panic - not syncing: Fatal exception in interrupt
[11439.006718] SMP: failed to stop secondary CPUs
[11439.011597] Rebooting in 3 seconds..
[11443.061142] SMP: failed to stop secondary CPUs

gsustek, 2017-11-21:
initialy after 3 hours, apple devices and even dell laptop with intel wifi couldn't connect. like pasword is wrong. i want to see what clients are connectoed to 5G, but opening the wireless menu in luci, caue na access timeout strom, and at the end, reboot.

latest commit, LEDE and mwlwifi driver:

root@3200acm:/# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170810
firmware version: 0x09030008
power table loaded from dts: no
firmware region code: 0x30
mac address: 60:38:e0:bc:32:f2
2g: disable
5g: enable
antenna: 4 4
irq number: 45
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e1100000
iobase1: e1380000
tx limit: 1024
rx limit: 16384

yuhhaurlin, 2017-11-21:
If the test result is the same as before, there is no need for you to post here again. Thanks.

ratsputin, 2017-11-21:
@yuhhaurlin What are you looking for at this point?  Right now, I'm trying to figure out how to reproduce the issue consistently, but I don't know if that will help you.

Here's a quick rundown of recent instances and the messages leading up to the first MEMAddrAccess timeout:
```
Tue Nov 14 11:18:51 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED e0:b5:2d:xx:xx:xx
Tue Nov 14 11:18:52 2017 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED e0:b5:2d:xx:xx:xx
Tue Nov 14 11:18:52 2017 kern.info kernel: [71632.036793] ieee80211 phy1: staid 7 deleted
Tue Nov 14 11:18:52 2017 kern.info kernel: [71632.202629] ieee80211 phy0: staid 3 deleted
Tue Nov 14 11:18:53 2017 daemon.notice hostapd: handle_auth_cb: STA e0:b5:2d:xx:xx:xx not found
Tue Nov 14 11:19:01 2017 kern.err kernel: [71641.323953] ieee80211 phy0: cmd 0x9111=SetNewStation timed out
Tue Nov 14 11:19:01 2017 kern.err kernel: [71641.329825] ieee80211 phy0: return code: 0x1111
Tue Nov 14 11:19:01 2017 kern.err kernel: [71641.334392] ieee80211 phy0: timeout: 0x1111
Tue Nov 14 11:19:06 2017 kern.err kernel: [71645.365712] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Tue Nov 14 11:19:06 2017 kern.err kernel: [71645.371579] ieee80211 phy0: return code: 0x001d
Tue Nov 14 11:19:06 2017 kern.err kernel: [71645.376143] ieee80211 phy0: timeout: 0x001d
```
```
Thu Nov 16 07:06:27 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK dc:41:5f:xx:xx:xx
Thu Nov 16 07:10:32 2017 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED dc:41:5f:xx:xx:xx
Thu Nov 16 07:10:32 2017 kern.info kernel: [142784.384247] ieee80211 phy1: staid 9 deleted
Thu Nov 16 07:10:32 2017 kern.info kernel: [142784.482271] ieee80211 phy0: staid 2 deleted
Thu Nov 16 07:10:36 2017 kern.err kernel: [142788.524032] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Thu Nov 16 07:10:36 2017 kern.err kernel: [142788.530263] ieee80211 phy0: return code: 0x1122
Thu Nov 16 07:10:36 2017 kern.err kernel: [142788.534905] ieee80211 phy0: timeout: 0x1122
Thu Nov 16 07:10:36 2017 kern.err kernel: [142788.539200] wlan0: failed to set key (0, dc:41:5f:xx:xx:xx) to hardware (-5)
Thu Nov 16 07:10:36 2017 daemon.notice hostapd: wlan0: AP-STA-CONNECTED dc:41:5f:xx:xx:xx
Thu Nov 16 07:10:41 2017 kern.err kernel: [142793.027931] ieee80211 phy0: cmd 0x9143=GetSeqno timed out
Thu Nov 16 07:10:41 2017 kern.err kernel: [142793.033449] ieee80211 phy0: return code: 0x1143
Thu Nov 16 07:10:41 2017 kern.err kernel: [142793.038091] ieee80211 phy0: timeout: 0x1143
Thu Nov 16 07:10:42 2017 kern.warn kernel: [142794.023994] ieee80211 phy0: can't send packet during ADDBA
Thu Nov 16 07:10:47 2017 kern.warn kernel: [142799.025161] ieee80211 phy0: can't send packet during ADDBA
Thu Nov 16 07:10:54 2017 kern.err kernel: [142805.610653] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Thu Nov 16 07:10:54 2017 kern.err kernel: [142805.616607] ieee80211 phy0: return code: 0x001d
Thu Nov 16 07:10:54 2017 kern.err kernel: [142805.621252] ieee80211 phy0: timeout: 0x001d
```
```
Thu Nov 16 19:19:11 2017 daemon.notice hostapd: wlan1: AP-STA-POLL-OK dc:41:5f:xx:xx:xx
Thu Nov 16 19:24:14 2017 daemon.notice hostapd: wlan1: AP-STA-POLL-OK dc:41:5f:xx:xx:xx
Thu Nov 16 19:27:35 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED dc:41:5f:xx:xx:xx
Thu Nov 16 19:28:43 2017 kern.err kernel: [21159.268529] ieee80211 phy1: cmd 0x9136=SetWscIE timed out
Thu Nov 16 19:28:43 2017 kern.err kernel: [21159.273965] ieee80211 phy1: return code: 0x1136
Thu Nov 16 19:28:43 2017 kern.err kernel: [21159.278523] ieee80211 phy1: timeout: 0x1136
Thu Nov 16 19:28:47 2017 kern.err kernel: [21163.283057] ieee80211 phy1: cmd 0x801c=80211RadioControl timed out
Thu Nov 16 19:28:47 2017 kern.err kernel: [21163.289279] ieee80211 phy1: return code: 0x001c
Thu Nov 16 19:28:47 2017 kern.err kernel: [21163.293834] ieee80211 phy1: timeout: 0x001c
Thu Nov 16 19:28:51 2017 kern.err kernel: [21167.297588] ieee80211 phy1: cmd 0x8126=SetFixedRate timed out
Thu Nov 16 19:28:51 2017 kern.err kernel: [21167.303400] ieee80211 phy1: return code: 0x0126
Thu Nov 16 19:28:51 2017 kern.err kernel: [21167.307964] ieee80211 phy1: timeout: 0x0126
Thu Nov 16 19:29:10 2017 kern.err kernel: [21186.972271] ieee80211 phy1: cmd 0x801d=MEMAddrAccess timed out
Thu Nov 16 19:29:10 2017 kern.err kernel: [21186.978139] ieee80211 phy1: return code: 0x001d
Thu Nov 16 19:29:10 2017 kern.err kernel: [21186.982692] ieee80211 phy1: timeout: 0x001d
```
```
Sat Nov 18 10:24:46 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK e0:b5:2d:xx:xx:xx
Sat Nov 18 10:24:47 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK dc:41:5f:xx:xx:xx
Sat Nov 18 10:25:24 2017 kern.info kernel: [40382.227610] ieee80211 phy0: staid 2 deleted
Sat Nov 18 10:25:24 2017 daemon.notice hostapd: wlan0: AP-STA-CONNECTED dc:41:5f:xx:xx:xx
Sat Nov 18 10:26:42 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED e0:b5:2d:xx:xx:xx
Sat Nov 18 10:26:42 2017 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED e0:b5:2d:xx:xx:xx
Sat Nov 18 10:26:42 2017 kern.info kernel: [40460.563366] ieee80211 phy1: staid 8 deleted
Sat Nov 18 10:26:43 2017 kern.info kernel: [40460.750375] ieee80211 phy0: staid 1 deleted
Sat Nov 18 10:26:43 2017 daemon.notice hostapd: wlan0: AP-STA-CONNECTED e0:b5:2d:xx:xx:xx
Sat Nov 18 10:26:45 2017 daemon.notice hostapd: handle_auth_cb: STA e0:b5:2d:xx:xx:xx not found
Sat Nov 18 10:27:34 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED e0:b5:2d:xx:xx:xx
[... cut ...]
Sat Nov 18 15:02:32 2017 kern.err kernel: [16194.904491] wlan0: failed to set key (0, e0:b5:2d:xx:xx:xx) to hardware (-5)
Sat Nov 18 15:02:32 2017 daemon.notice hostapd: wlan0: AP-STA-CONNECTED e0:b5:2d:xx:xx:xx
Sat Nov 18 15:02:36 2017 kern.err kernel: [16198.948459] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Sat Nov 18 15:02:36 2017 kern.err kernel: [16198.954327] ieee80211 phy0: return code: 0x001d
Sat Nov 18 15:02:36 2017 kern.err kernel: [16198.958888] ieee80211 phy0: timeout: 0x001d
```

yuhhaurlin, 2017-11-21:
It looks like this problem happens with hand held devices. I will try to reproduce it with one phone. Your test includes too many devices. You can wait for the fix and help to verify it.

Chadster766, 2017-11-22:
I only see on error in my log:

Nov 21 22:47:39 MCDEBIAN kernel: ieee80211 phy0: create ba result error 1
Nov 21 22:47:39 MCDEBIAN kernel: ieee80211 phy0: ampdu operation error code: -22

ratsputin, 2017-11-22:
@yuhhaurlin In my case, I don't use wireless for laptops and such, so my results don't necessarily indicate it's specifically handhelds.

Please keep in mind that, based on my experience, the issue seems to manifest itself more reliably if two or more devices leave wifi range for an hour or so, then return at the same time.  Issues with a single device leaving and returning seem to be much less consistent.

In the meantime, I will focus on trying to reproduce the issue reliably with a single device.

Thank you for your attention to this issue.  I really like this router with LEDE; without your efforts on this, none of this would be possible.

michaelmcmaster, 2017-11-22:
I've had this issue occur several times in the past few days on my WRT3200ACM (davidc502 r5389) and I've noticed that the "MEMAddrAccess" issue seems to start when a dual frequency (2.4 and 5GHz) device comes back into network range after several hours (touching both SSIDs).  I have not seen a single frequency device (configured for only one SSID) cause the issue.

In my most recent recreate, I can see the device hitting both phy0 and phy1, which is then immediately followed by the errors:
`
Tue Nov 21 18:01:36 2017 kern.info kernel: [61719.825738] ieee80211 phy1: staid 3 deleted
Tue Nov 21 18:01:36 2017 kern.info kernel: [61719.957752] ieee80211 phy0: staid 1 deleted
Tue Nov 21 18:01:36 2017 daemon.info hostapd: wlan0: STA f0:d7:aa:c6:2e:b4 IEEE 802.11: associated (aid 1)
Tue Nov 21 18:01:45 2017 kern.err kernel: [61729.082449] ieee80211 phy0: cmd 0x9111=SetNewStation timed out
`

Here's a little larger chunk of the same logs, for more context:
`
Tue Nov 21 17:52:30 2017 daemon.notice hostapd: wlan1: AP-STA-POLL-OK f0:d7:aa:c6:2e:b4
Tue Nov 21 17:52:36 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK f0:d7:aa:c6:2b:96
Tue Nov 21 17:52:38 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK f0:d7:aa:c6:2e:b4
Tue Nov 21 17:57:33 2017 daemon.notice hostapd: wlan1: AP-STA-POLL-OK f0:d7:aa:c6:2e:b4
Tue Nov 21 17:57:39 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK f0:d7:aa:c6:2b:96
Tue Nov 21 17:57:41 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK f0:d7:aa:c6:2e:b4
Tue Nov 21 17:58:02 2017 kern.info kernel: [61505.750462] ieee80211 phy1: staid 3 deleted
Tue Nov 21 17:58:02 2017 daemon.info hostapd: wlan1: STA f0:d7:aa:c6:2e:b4 IEEE 802.11: associated (aid 3)
Tue Nov 21 17:58:03 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED f0:d7:aa:c6:2e:b4
Tue Nov 21 17:58:03 2017 daemon.info hostapd: wlan1: STA f0:d7:aa:c6:2e:b4 WPA: pairwise key handshake completed (RSN)
Tue Nov 21 17:58:03 2017 daemon.info dnsmasq-dhcp[3162]: DHCPDISCOVER(br-lan) f0:d7:aa:c6:2e:b4
Tue Nov 21 17:58:03 2017 daemon.info dnsmasq-dhcp[3162]: DHCPOFFER(br-lan) 192.168.1.10 f0:d7:aa:c6:2e:b4
Tue Nov 21 17:58:05 2017 daemon.info dnsmasq-dhcp[3162]: DHCPDISCOVER(br-lan) f0:d7:aa:c6:2e:b4
Tue Nov 21 17:58:05 2017 daemon.info dnsmasq-dhcp[3162]: DHCPOFFER(br-lan) 192.168.1.10 f0:d7:aa:c6:2e:b4
Tue Nov 21 17:58:08 2017 daemon.info dnsmasq-dhcp[3162]: DHCPDISCOVER(br-lan) f0:d7:aa:c6:2e:b4
Tue Nov 21 17:58:08 2017 daemon.info dnsmasq-dhcp[3162]: DHCPOFFER(br-lan) 192.168.1.10 f0:d7:aa:c6:2e:b4
Tue Nov 21 17:58:08 2017 daemon.info dnsmasq-dhcp[3162]: DHCPREQUEST(br-lan) 192.168.1.10 f0:d7:aa:c6:2e:b4
Tue Nov 21 17:58:08 2017 daemon.info dnsmasq-dhcp[3162]: DHCPACK(br-lan) 192.168.1.10 f0:d7:aa:c6:2e:b4 Phone
Tue Nov 21 17:58:11 2017 daemon.info hostapd: wlan1: STA f0:d7:aa:c6:2e:b4 IEEE 802.11: authenticated
Tue Nov 21 17:58:17 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK b4:52:7e:3a:a4:04
Tue Nov 21 18:01:36 2017 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED f0:d7:aa:c6:2e:b4
Tue Nov 21 18:01:36 2017 kern.info kernel: [61719.825738] ieee80211 phy1: staid 3 deleted
Tue Nov 21 18:01:36 2017 kern.info kernel: [61719.957752] ieee80211 phy0: staid 1 deleted
Tue Nov 21 18:01:36 2017 daemon.info hostapd: wlan0: STA f0:d7:aa:c6:2e:b4 IEEE 802.11: associated (aid 1)
Tue Nov 21 18:01:45 2017 kern.err kernel: [61729.082449] ieee80211 phy0: cmd 0x9111=SetNewStation timed out
Tue Nov 21 18:01:45 2017 kern.err kernel: [61729.088310] ieee80211 phy0: return code: 0x1111
Tue Nov 21 18:01:45 2017 kern.err kernel: [61729.092866] ieee80211 phy0: timeout: 0x1111
Tue Nov 21 18:01:45 2017 daemon.info hostapd: wlan0: STA f0:d7:aa:c6:2e:b4 IEEE 802.11: deauthenticated due to local deauth request
Tue Nov 21 18:01:53 2017 kern.err kernel: [61737.110195] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Tue Nov 21 18:01:53 2017 kern.err kernel: [61737.116080] ieee80211 phy0: return code: 0x001d
Tue Nov 21 18:01:53 2017 kern.err kernel: [61737.120635] ieee80211 phy0: timeout: 0x001d
...Above three lines repeat every 4-ish seconds...
Tue Nov 21 18:02:16 2017 daemon.info hostapd: wlan1: STA f0:d7:aa:c6:2e:b4 IEEE 802.11: associated (aid 3)
Tue Nov 21 18:02:16 2017 daemon.info hostapd: wlan1: STA f0:d7:aa:c6:2e:b4 IEEE 802.11: authenticated
Tue Nov 21 18:02:17 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED f0:d7:aa:c6:2e:b4
Tue Nov 21 18:02:17 2017 daemon.info hostapd: wlan1: STA f0:d7:aa:c6:2e:b4 WPA: pairwise key handshake completed (RSN)
Tue Nov 21 18:02:17 2017 daemon.info dnsmasq-dhcp[3162]: DHCPDISCOVER(br-lan) f0:d7:aa:c6:2e:b4
Tue Nov 21 18:02:17 2017 daemon.info dnsmasq-dhcp[3162]: DHCPOFFER(br-lan) 192.168.1.10 f0:d7:aa:c6:2e:b4
Tue Nov 21 18:02:17 2017 daemon.info dnsmasq-dhcp[3162]: DHCPREQUEST(br-lan) 192.168.1.10 f0:d7:aa:c6:2e:b4
Tue Nov 21 18:02:17 2017 daemon.info dnsmasq-dhcp[3162]: DHCPACK(br-lan) 192.168.1.10 f0:d7:aa:c6:2e:b4 Phone
Tue Nov 21 18:02:23 2017 kern.err kernel: [61767.111803] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Tue Nov 21 18:02:23 2017 kern.err kernel: [61767.117667] ieee80211 phy0: return code: 0x001d
Tue Nov 21 18:02:23 2017 kern.err kernel: [61767.122224] ieee80211 phy0: timeout: 0x001d
Tue Nov 21 18:02:27 2017 kern.err kernel: [61771.125738] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Tue Nov 21 18:02:27 2017 kern.err kernel: [61771.131604] ieee80211 phy0: return code: 0x001d
Tue Nov 21 18:02:27 2017 kern.err kernel: [61771.136151] ieee80211 phy0: timeout: 0x001d
`

ratsputin, 2017-11-22:
@michaelmcmaster You bring up a good point.  Generally, our devices will be on the 5GHz band when they leave coverage, but come back in on the 2.4GHz band (as it has better range).  They will then switch to the 5GHz band at some point after being used.

wongsyrone, 2017-11-22:
I'm using 2.4G only and still the same issue. It happens ~3days of usage.

yuhhaurlin, 2017-11-22:
@wongsyrone Did you have hand held devices connected to your router?

wongsyrone, 2017-11-22:
About 4 devices:

1. Asus N53SN laptop with Gigabyte GP-WB867MD-I 802.11ac mini-pcie adapter(aka Intel ac7260)
2. Motorola Nexus 6
3. Huawei Nexus 6P
4. Huawei Honor 6Plus

yuhhaurlin, 2017-11-22:
Yes. It looks the problem is related to hand held devices. Thanks.

WildByDesign, 2017-11-22:
@wongsyrone  Interesting, I also have an Intel ac7260 card (ultrabook) on my network as well and experience the same "MEMAddrAccess" issue approx. every 2-3 days.  Also have two iPhones and two iPads.

"MEMAddrAccess" occurs on LEDE Stable and trunk.  Trunk eventually crashes, while stable keeps running with high load but does not crash.

wongsyrone, 2017-11-22:
It seems @yuhhaurlin  can use Intel ac7260 to make things easier.

yuhhaurlin, 2017-11-22:
Can you just use ac7260 to see if you will encounter problem?

[133 more replies]

closed 2017-11-30 by yuhhaurlin