kaloz/mwlwifi #234: Regulatory domain unset issue

cilix-lab, 2017-11-21:
Hi! I've been having some trouble with the latest commit and 0x0903bee1 firmware.  

I've been using Ubuntu Zesty 17.04 on the WRT32000ACM without any problem and it works great. Thing is, I updated mwlwifi to 466368f and the firmware to said version, rebuilt the kernel and flashed my router, only to find out I'm at square one with the regulatory domain issues I've seen before.  

With the previous commits/firmware, if I do `iw reg get` I get:  
```
country FR: DFS-ETSI
	(2402 - 2482 @ 40), (N/A, 20), (N/A)
	(5170 - 5250 @ 80), (N/A, 20), (N/A)
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS
	(5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
	(57000 - 66000 @ 2160), (N/A, 40), (N/A)
```

That is with CRDA REGDOMAIN set to FR, since my router is the EU version. Everything works fine and I'm not having trouble at all.  

But since I updated mwlwifi and firmware, hostapd stops working, I get lots of messeges on `journalctl` saying `Process '/sbin/crda' failed with exit code 234.` and I get `country 00: DFS-Unset` from `iw reg get`.  

I tried compiling the kernel without support for the third chip (mlan0), but it still won't work. I still get the `Process '/sbin/crda' failed with exit code 234.` messages on the logs. That happens too if I do not set REGDOMAIN in /etc/default/crda.  

I've updated mwlwifi, firmware and did a kernel update, so it might not be 100% related to mwlwifi, but if there has been any change on how the firmware manages regulatory domains, it'd be nice to know and have some feedback if possible.  

I'll do some more troubleshooting during the weekend. For now, I'll stick with the old kernel, commit and firmware.

yuhhaurlin, 2017-11-21:
LEDE and OpenWrt use regulatory domain static database. It looks like CRDA has problem.

cilix-lab, 2017-11-23:
I've been compiling the kernel with static database and it works that way. I can now confirm that the issue appeared on the latest commits of mwlwifi.  

I compiled the kernel with an older commit and previos firmware. I boot the router and get:  
```
root@UbuntuWRT:~# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170810
firmware version: 0x09030008
power table loaded from dts: no
firmware region code: 0x30
mac address: 11:22:33:44:55:aa
2g: disable
5g: enable
antenna: 4 4
irq number: 45
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e1e00000
iobase1: e2080000
tx limit: 1024
rx limit: 16384

root@UbuntuWRT:~# iw reg get
global
country FR: DFS-ETSI
	(2402 - 2482 @ 40), (N/A, 20), (N/A)
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
	(5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

phy#2
country FR: DFS-ETSI
	(2402 - 2482 @ 40), (N/A, 20), (N/A)
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
	(5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

phy#0
country FR: DFS-ETSI
	(2402 - 2482 @ 40), (N/A, 20), (N/A)
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
	(5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

phy#1
country US: DFS-FCC
	(2402 - 2472 @ 40), (N/A, 30), (N/A)
	(5170 - 5250 @ 80), (N/A, 17), (N/A), AUTO-BW
	(5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
	(5735 - 5835 @ 80), (N/A, 30), (N/A)
	(57240 - 63720 @ 2160), (N/A, 40), (N/A)
```
Note: phy#1 is sd8887.  

Everything is working fine. I update mwlwifi to latest commit and firmware, no other changes in kernel config, compile the kernel, boot the router and get:  
```
root@UbuntuWRT:~# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170810
firmware version: 0x0903bee1
power table loaded from dts: no
firmware region code: 0x0
mac address: 11:22:33:44:55:aa
2g: disable
5g: enable
antenna: 4 4
irq number: 45
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e2580000
iobase1: e2800000
tx limit: 1024
rx limit: 16384

root@UbuntuWRT:~# iw reg get
global
country 00: DFS-UNSET
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 40), (N/A, 20), (N/A), PASSIVE-SCAN
	(2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
	(5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
	(5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)

phy#2
country FR: DFS-ETSI
	(2402 - 2482 @ 40), (N/A, 20), (N/A)
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
	(5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

phy#0
country 00: DFS-UNSET
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 40), (N/A, 20), (N/A), PASSIVE-SCAN
	(2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
	(5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
	(5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)

phy#1
country US: DFS-FCC
	(2402 - 2472 @ 40), (N/A, 30), (N/A)
	(5170 - 5250 @ 80), (N/A, 17), (N/A), AUTO-BW
	(5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
	(5735 - 5835 @ 80), (N/A, 30), (N/A)
	(57240 - 63720 @ 2160), (N/A, 40), (N/A)

root@UbuntuWRT:~# journalctl -xe
...
Nov 23 13:23:52 UbuntuWRT systemd-udevd[3662]: Process '/sbin/crda' failed with exit code 234.
Nov 23 13:23:55 UbuntuWRT systemd-udevd[3665]: Process '/sbin/crda' failed with exit code 234.
Nov 23 13:23:58 UbuntuWRT systemd-udevd[3669]: Process '/sbin/crda' failed with exit code 234.
```

Evidently there's been a modification in mwlwifi driver or firmware that is causing this.

cilix-lab, 2017-11-23:
I'm now compiling the kernel with older mwlwifi and new firmware and vice versa to confirm if it's the firmware or the driver.

cilix-lab, 2017-11-23:
I can now confirm it's a firmware issue. Compiling new mwlwifi with older firmware works like a charm.  

Please, I'd really appreciate you look into this and not just randomly close the issue, since the firmware is closed source. I know you are only supporting OpenWRT/LEDE, but you can't really expect people to be OK with having firmware that conflicts with hardware included in the WRT3200ACM, meaning the SD8887 chip.  

```
# Latest commit, old firmware
root@UbuntuWRT:~# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170810
firmware version: 0x09030008
power table loaded from dts: no
firmware region code: 0x30
mac address: 11:22:33:44:55:aa
2g: disable
5g: enable
antenna: 4 4
irq number: 45
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e1e00000
iobase1: e2080000
tx limit: 1024
rx limit: 16384

root@UbuntuWRT:~# iw reg get
global
country FR: DFS-ETSI
	(2402 - 2482 @ 40), (N/A, 20), (N/A)
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
	(5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

phy#2
country FR: DFS-ETSI
	(2402 - 2482 @ 40), (N/A, 20), (N/A)
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
	(5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

phy#0
country FR: DFS-ETSI
	(2402 - 2482 @ 40), (N/A, 20), (N/A)
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
	(5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

phy#1
country US: DFS-FCC
	(2402 - 2472 @ 40), (N/A, 30), (N/A)
	(5170 - 5250 @ 80), (N/A, 17), (N/A), AUTO-BW
	(5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
	(5735 - 5835 @ 80), (N/A, 30), (N/A)
	(57240 - 63720 @ 2160), (N/A, 40), (N/A)
```

Chadster766, 2017-12-12:
@cilix-lab I'm seeing the same issue with the latest mwlwifi firmware. Have you figured out a solution for this issue?

cilix-lab, 2017-12-12:
@Chadster766 I managed to fix it removing support for CRDA from the kernel config. Now I've got `CONFIG_CFG80211_INTERNAL_REGDB=y` as always, but `# CONFIG_CFG80211_CRDA_SUPPORT is not set`. That made the trick.  

It looks like in the latest firmware, something changed in the way regulatory domain is managed and crashed CRDA when trying to set the global domain, even if it matched the chip's domain, something that didn't happen in previous firmware.  

Hope this helps.

Chadster766, 2017-12-12:
Thanks for the info. 

Since the FCC has the radio settings hard coded into the radio firmware it makes the CRDA software redundant doesn't it?

Chadster766, 2017-12-12:
Problem firmware:
```
root@MCDEBIAN:~# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20171129
firmware version: 0x09030201
power table loaded from dts: no
firmware region code: 0x0
mac address: 60:38:e0:b4:47:a2
2g: enable
5g: enable
antenna: 4 4
irq number: 50
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00010000
radio: disable
iobase0: e2900000
iobase1: e2b80000
tx limit: 1024
rx limit: 16384
```
Working firmware:
```
root@MCDEBIAN:~# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20171129
firmware version: 0x09030008
power table loaded from dts: no
firmware region code: 0x10
mac address: 60:38:e0:b4:47:a2
2g: enable
5g: enable
antenna: 4 4
irq number: 50
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e2900000
iobase1: e2b80000
tx limit: 1024
rx limit: 16384
```
I just want to backup @cilix-lab's claim that the latest firmware version: 0x09030008 is not detecting the country code properly case 5Ghz to drop and other strange issues.

Chadster766, 2017-12-12:
@yuhhaurlin did you know that for the last two to three week on the Linksys forums there is melt down occurring regarding the WRT3200ACM and WRT32x 5Ghz dropping and other strange wireless issues? Linksys is getting a really big black eye :unamused:

BrainSlayer, 2017-12-12:
@Chadster766 good for us since the mwlwifi driver is getting better and better. @yuhhaurlin  is not maintaining the driver which is used by linksys. so lets make the opensource driver as good as possible this will help the opensource scene

Chadster766, 2017-12-12:
@BrainSlayer this is true regarding the open source community but only if Linksys can continue to sell WRT open source ready routers.

How is DD-WRT getting around firmware version: 0x09030008 not properly detecting the 5Ghz country code? Isn't country code 00 restricting the universally available channel 36?

Chadster766, 2017-12-12:
The issue is in chip firmware binary which AFAIK is the same that's used in Linksys firmware. I will check with the latest Linksys firmware loaded on my WRT3200ACM and WRT32x this even.

BrainSlayer, 2017-12-12:
one thing to consider. the mwlwifi driver just supports a small sort of country codes. any other country code wont work and iw reg set must match the code programmed by the driver, but if you keep iw reg set untouched it uses the driver default. if you change it, take care to use the correct one

BrainSlayer, 2017-12-12:
@Chadster766 mmh yes, but you know what. what if we can now configure the region manually with iw? at least looks so. but havent tested the frequencies yet

Chadster766, 2017-12-12:
Could it be that the firmware bin is where the hardcoded country code exists? If so to change country codes for the WRT it might be possible to get different firmware bins from other routers purchased in different countries and apply those.

Chadster766, 2017-12-12:
@BrainSlayer changing the country code in software AFAIK (and in the past) has had no effect due to the firmware's hardcoded country code for FCC compliance.

BrainSlayer, 2017-12-12:
no. there are different codes which are taken from a on board eeprom. but the firmware is responsible to read this code and to use it proper with the hardware. the hardware itself does not handle the code itself

BrainSlayer, 2017-12-12:
this hardcoded country code is optional. the firmware returns if it has a country code or not. consider that older models did not have preprogrammed codes and where unlocked

yuhhaurlin, 2017-12-12:
I notice about the region code. I will check and let you know. That is the reason why I reopen this issue.

yuhhaurlin, 2017-12-12:
I don't know about the problem of stock firmware.

aaron1959, 2017-12-12:
@yuhhaurlin  I just looked there was a new linksys firmware released for the WRT3200ACM which seems to have a big problem with wireless clients, 
I have no idea how that applies here. 
but there it is.

yuhhaurlin, 2017-12-12:
I will report to related manager. Thanks.

yuhhaurlin, 2017-12-13:
This problem is fixed via commit https://github.com/kaloz/mwlwifi/commit/1fcef06e2ef8c03df50bc040d376b3b4282d41a0.

closed 2017-12-13 by yuhhaurlin

Chadster766, 2017-12-13:
@yuhhaurlin thank you great job :smiley:

BrainSlayer, 2017-12-13:
@yuhhaurlin it should be noted that this is caused by a firmware api change, so the new mwlwifi cannot be used with older firmwares, this would lead to a undefined behaviour again

yuhhaurlin, 2017-12-13:
@BrainSlayer This modification can work for previous firmware too. I tested both of them.

BrainSlayer, 2017-12-13:
ok

eduperez, 2017-12-13:
Updated `.ipk` packages for LEDE 17.01.4 available at:
https://github.com/eduperez/mwlwifi_LEDE/releases/tag/1fcef06

yuhhaurlin, 2017-12-15:
Firmware will base on region code to set power and allowable channels. If you set a country which is not in the region, some settings won't work.

ValCher1961, 2017-12-15:
Yes, I can't change the power, but I have access to the channel opening.

BrainSlayer, 2017-12-15:
@ValCher1961 no you wont. you just think you have but channel which arent in the preprogrammed regulatory set will be ignored by the firmware

ValCher1961, 2017-12-15:
I am using the American version of WRT3200 in Europe, with the help of blocking the region check in the driver, have access to the inclusion of 12 and 13 channels.

BrainSlayer, 2017-12-15:
no. channel 12 and 13 is shown but you cannot use them

cilix-lab, 2017-12-15:
Did anyone notice this on @ValCher1961 's output of `cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info`:  
```
chip type: 88W8964
(...)
power table loaded from dts: yes
```
Isn't the WRT3200ACM supposed to load the powertable from the EEPROM??

ValCher1961, 2017-12-15:
Believe me, it works. In the end you can check the applying patch.
[image]

ValCher1961, 2017-12-15:
@cilix-lab 
 I can not change the power, it remains the one that is registered in EEPROM

yuhhaurlin, 2017-12-15:
It looks like your WRT3200ACM doesn't program EEPROM with power table. It is very strange.

ValCher1961, 2017-12-15:
As I understand it, wrt3200 uses a power table from the EEPROM of the region for which it was originally intended.

lantis1008, 2017-12-16:
I can confirm @ValCher1961 ‘s patch works. I have been using it for a while now to use AU regulatory settings on a US 3200acm. I reworked it onto an older version of the firmware for my purposes.

ValCher1961, 2017-12-17:
@lantis1008 Thank you.

httpstorm, 2020-01-03:
While wireless regulations should be followed, the way that restrictions are actually implemented, disables features which are allowed in the region where devices were purchased.
WRT3200ACM is a good example. I purchased my device in Bulgaria:

- user region set to 🇧🇬 Bulgaria
- radio 0: mwlwifi, 5 GHz, region lock: 🇫🇷 France
- radio 1: mwlwifi, 2.4 GHz, region lock: 🇫🇷 France
- radio 2: mwlwifi, 2.4/5 GHz, region lock: 🇺🇸 USA

Due to the conflict between the wireless adapters, the system prevents usage of most 5 GHz channels. Channels 100 and above do not work. Using 160 MHz is not possible.

If anyone wants to learn how the driver notifies the system about the hard coded region of the wireless chip, and how to prevent this behaviour, the following patches might shine some light. Use for educational purpose only is advised.

WRT3200ACM mwlwifi region free, OpenWRT, kernel 4.19.91
https://pastebin.com/1LRctnm2

WRT3200ACM mwifiex region free, OpenWRT, kernel 4.19.91
https://pastebin.com/JU6mSJxC

While the user is allowed to specify any region with these patches applied, this does not break the regulation rules, since the wireless chips apply their own restrictions. For the main radios, mwlwifi does not allow TX power to be changed. EIRP on any 5 GHz channel is much less than the permitted 30 dBm in Bulgaria. The third radio which uses mwifiex allows changes to TX power, but is limited: 8-15 dBm.