kaloz/mwlwifi #235: High Ping Times

Chadster766, 2017-11-22:
```
filename:       /lib/modules/4.9.63/kernel/drivers/mwlwifi/mwlwifi.ko
license:        GPL v2
author:         Marvell Semiconductor, Inc.
version:        10.3.4.0-20170810
description:    Marvell Mac80211 Wireless PCIE Network Driver
srcversion:     3950A0A68BA183CDADC23F4
alias:          pci:v000011ABd00002B40sv*sd*bc*sc*i*
alias:          pci:v000011ABd00002B38sv*sd*bc*sc*i*
alias:          pci:v000011ABd00002A55sv*sd*bc*sc*i*
depends:
intree:         Y
vermagic:       4.9.63 SMP mod_unload ARMv7 p2v8
```

```
root@MCDEBIAN:~# ./enable-AMSDU
root@MCDEBIAN:~# ping 192.168.200.107
PING 192.168.200.107 (192.168.200.107) 56(84) bytes of data.
64 bytes from 192.168.200.107: icmp_seq=1 ttl=64 time=4356 ms
64 bytes from 192.168.200.107: icmp_seq=2 ttl=64 time=3286 ms
64 bytes from 192.168.200.107: icmp_seq=3 ttl=64 time=2247 ms
64 bytes from 192.168.200.107: icmp_seq=4 ttl=64 time=1207 ms
64 bytes from 192.168.200.107: icmp_seq=5 ttl=64 time=167 ms
64 bytes from 192.168.200.107: icmp_seq=6 ttl=64 time=1713 ms
64 bytes from 192.168.200.107: icmp_seq=7 ttl=64 time=637 ms
64 bytes from 192.168.200.107: icmp_seq=8 ttl=64 time=2118 ms
64 bytes from 192.168.200.107: icmp_seq=9 ttl=64 time=1121 ms
64 bytes from 192.168.200.107: icmp_seq=10 ttl=64 time=81.5 ms
64 bytes from 192.168.200.107: icmp_seq=11 ttl=64 time=1615 ms
64 bytes from 192.168.200.107: icmp_seq=12 ttl=64 time=539 ms
64 bytes from 192.168.200.107: icmp_seq=13 ttl=64 time=2505 ms
64 bytes from 192.168.200.107: icmp_seq=14 ttl=64 time=1510 ms
64 bytes from 192.168.200.107: icmp_seq=15 ttl=64 time=470 ms
^C
--- 192.168.200.107 ping statistics ---
17 packets transmitted, 15 received, 11% packet loss, time 16431ms
rtt min/avg/max/mdev = 81.531/1571.829/4356.331/1155.527 ms, pipe 5
```

```
root@MCDEBIAN:~# ./disable-AMSDU
root@MCDEBIAN:~# ping 192.168.200.107
PING 192.168.200.107 (192.168.200.107) 56(84) bytes of data.
64 bytes from 192.168.200.107: icmp_seq=1 ttl=64 time=2471 ms
64 bytes from 192.168.200.107: icmp_seq=2 ttl=64 time=1409 ms
64 bytes from 192.168.200.107: icmp_seq=3 ttl=64 time=370 ms
64 bytes from 192.168.200.107: icmp_seq=4 ttl=64 time=2203 ms
64 bytes from 192.168.200.107: icmp_seq=5 ttl=64 time=1127 ms
64 bytes from 192.168.200.107: icmp_seq=6 ttl=64 time=87.1 ms
64 bytes from 192.168.200.107: icmp_seq=7 ttl=64 time=2190 ms
64 bytes from 192.168.200.107: icmp_seq=8 ttl=64 time=1113 ms
64 bytes from 192.168.200.107: icmp_seq=9 ttl=64 time=73.7 ms
64 bytes from 192.168.200.107: icmp_seq=10 ttl=64 time=2318 ms
64 bytes from 192.168.200.107: icmp_seq=11 ttl=64 time=1241 ms
64 bytes from 192.168.200.107: icmp_seq=12 ttl=64 time=201 ms
64 bytes from 192.168.200.107: icmp_seq=13 ttl=64 time=18.9 ms
^C
--- 192.168.200.107 ping statistics ---
14 packets transmitted, 13 received, 7% packet loss, time 13464ms
rtt min/avg/max/mdev = 18.943/1140.557/2471.145/896.764 ms, pipe 3
```

yuhhaurlin, 2017-11-22:
Can you ping to router from your client at same time?

yuhhaurlin, 2017-11-22:
1. I only support problems related to LEDE.
2. Please ping from client to make sure this is not due to power save mode of client.
3. Make sure your environment is clean.
4.If it is really a problem, you need to give me your test setup and which client you used.

closed 2017-11-22 by yuhhaurlin

Chadster766, 2017-11-22:
Ping from the client to the router has low ping times. 

The client is a galaxy s6 and wasn't in power save mode.

Chadster766, 2017-11-22:
For comparison I will load a previous version and report results.

yuhhaurlin, 2017-11-22:
Can you test on LEDE?

Chadster766, 2017-11-22:
Yes but honestly I'm testing on something better IMO.

yuhhaurlin, 2017-11-22:
I think it is related to power save mode of WiFi of phone. You can install "ping" to your android phone. If you ping from phone to router, the ping time is pretty stable. However, if you ping from router to phone, sometimes the ping time will be long.

yuhhaurlin, 2017-11-22:
I use Samsung Galaxy S6 edge to do the test.

Chadster766, 2017-11-22:
@yuhhaurlin thank you and you are right it's the power save mode of the Wifi phone.

Testing of the latest mwlwifi version on master didn't go very well. Devices were disconnecting and sometimes not getting an IP Address. It could be the beta rootfs I'm testing so I'm comparing against the mwlwifi commit on jun-8-2017 to see if it makes a difference.

Chadster766, 2017-11-22:
I've resumed evaluating the master release.

Chadster766, 2017-11-23:
I'm not seeing the errors with the current master commit that others are reporting.

I think the new dhcpd package is doing something different because I don't remember seeing the below journalctl entries with Debian 8:

```
dhcpd[4471]: reuse_lease: lease age 75 (secs) under 25% threshold, reply with unaltered, existing lease for 192.168.200.106
```

It seems the mwlwifi is working well.