kaloz/mwlwifi #237: phy0 - 2/5G Enabled At Sametime - DDWRT v33857

rs-se, 2017-11-28:
Currently no WiFi crash but still have phy0 enabled on both 2/5G at the sametime and can't find a way to fix that.

root@WRT3200ACM:~# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170810
firmware version: 0x0903bee1
power table loaded from dts: no
firmware region code: 0x0
mac address: 60:38:e0:b9:8d:d2
2g: enable
5g: enable
antenna: 4 4
irq number: 46
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e0c00000
iobase1: e0e80000
tx limit: 1024
rx limit: 16384

Latest DD-WRT image

yuhhaurlin, 2017-11-28:
Please check READ to know how to control band selection.

closed 2017-11-28 by yuhhaurlin

BrainSlayer, 2017-11-29:
@yuhhaurlin: wrong answer. he has no control about it. i have control how to configure the dts on the precompiled images
@rs-se: you can ignore that. its cosmetic for now. will be fixed in next version. i changed the dts to disable the 2 ghz flag on phy0 and 5 ghz flag on phy1. but it wont change the behaviour, just the debugfs output

yuhhaurlin, 2017-11-29:
@BrainSlayer If you follow the README to modify DTS file, this kind of problem will be gone.

yuhhaurlin, 2017-11-29:
README shows the way to control band selection via DTS file.

BrainSlayer, 2017-11-29:
@yuhhaurlin: i know that. but i do not complain about it. it has no effect anyway since the card wont operate in 2.4 ghz even if the flag is set. i alread changed that in my source

yuhhaurlin, 2017-11-29:
For WRT3200ACM (armada-385-linksys-rango.dts):

 pcie@1,0 {
                                /* Marvell 88W8964, 5GHz-only */
                                status = "okay";

                                mwlwifi {
                                        marvell,2ghz = <0>;
                                        marvell,chainmask = <4 4>;
                                };

                        };

 pcie@2,0 {
                                /* Marvell 88W8964, 2GHz-only */
                                status = "okay";

                                mwlwifi {
                                        marvell,5ghz = <0>;
                                        marvell,chainmask = <4 4>;
                                };

                        };

BrainSlayer, 2017-11-29:
again in big capitals. I KNOW THAT ALREADY. I HAVE ALREAD CHANGED THAT IN MY DTS

yuhhaurlin, 2017-11-29:
If this is the case, it should work.

BrainSlayer, 2017-11-29:
it does work. also without that change. the user is just complaining about a small cosmetic issue which has no effect and to make him happy i alread changed it in my dts. thats all

yuhhaurlin, 2017-11-29:
No. Without this setting, user will confuse about the band setting. In WRT3200ACM, one module only works for 2.4g band and anther one only works for 5g band. However, if it is fixed for your next release, it should be all right.