kaloz/mwlwifi #238: Host command timeout still occurring when two devices return simultaneously

ratsputin, 2017-12-03:
Both of us just returned from being away for two hours.  Note, this is now happening with an iPhone 8 and an 8+ rather than a 6 and 6+ (we upgraded).

Received the following crash:

```
Time: 1512327826.231100
Modules:        nf_conntrack_ipv6@bf2cf000+15bc iptable_nat@bf2c6000+451        ipt_REJECT@bf2c2000+3ea ipt_MASQUERADE@bf2be000+332xt_time@bf2ba000+6a6     xt_tcpudp@bf2b6000+758  xt_tcpmss@bf2b2000+44a  xt_statistic@bf2ae000+3d5       xt_state@bf2aa000+349   xt_recent@bf2a5000+1960     xt_nat@bf2a1000+59d     xt_multiport@bf29d000+554       xt_mark@bf299000+320    xt_mac@bf295000+2c7     xt_limit@bf291000+48d       xt_length@bf28d000+365  xt_hl@bf289000+3de      xt_helper@bf285000+3cc  xt_ecn@bf281000+59c     xt_dscp@bf27d000+496        xt_conntrack@bf279000+958       xt_connmark@bf275000+534        xt_connlimit@bf270000+10d9      xt_connbytes@bf26c000+595   xt_comment@bf268000+24b xt_TCPMSS@bf264000+a68  xt_REDIRECT@bf260000+375        xt_LOG@bf25c000+3c3     xt_HL@bf258000+598 xt_DSCP@bf254000+5ee     xt_CLASSIFY@bf250000+2cc        nf_reject_ipv4@bf24c000+783     nf_nat_redirect@bf248000+3db    nf_nat_masquerade_ipv4@bf244000+60d nf_conntrack_ipv4@bf23f000+1443 nf_nat_ipv4@bf23b000+cc7        nf_nat@bf235000+2007    nf_log_ipv4@bf231000+c36    nf_defrag_ipv6@bf22a000+3214    nf_defrag_ipv4@bf226000+37c     nf_conntrack_rtcache@bf222000+a52       nf_conntrack@bf210000+cfd8  mwifiex_sdio@bf207000+5174      mwifiex@bf1cc000+3259d  iptable_mangle@bf1c8000+3fc     iptable_filter@bf1c4000+37c     ipt_ECN@bf1c0000+5a4        ip_tables@bf1ba000+2597 hid@bf1a4000+11f10      evdev@bf19e000+27a3     input_core@bf194000+5ed7        mwlwifi@bf172000+1aded      mac80211@bf104000+5f851 cfg80211@bf0ca000+31035 compat@bf0c5000+1bf0    xt_set@bf0c0000+1923    ip_set_list_set@bf0bb000+1804       ip_set_hash_netiface@bf0b2000+5189      ip_set_hash_netport@bf0a9000+51ee       ip_set_hash_netnet@bf0a0000+5693    ip_set_hash_net@bf098000+4c42   ip_set_hash_netportnet@bf08f000+5aff    ip_set_hash_mac@bf089000+21e9   ip_set_hash_ipportnet@bf080000+5664 ip_set_hash_ipportip@bf078000+44c9      ip_set_hash_ipport@bf070000+4193        ip_set_hash_ipmark@bf069000+3f53   ip_set_hash_ip@bf062000+3e67     ip_set_bitmap_port@bf05d000+1402        ip_set_bitmap_ipmac@bf058000+168e       ip_set_bitmap_ip@bf053000+16aa      ip_set@bf049000+50c9    nfnetlink@bf045000+f5f  ip6t_REJECT@bf041000+3ec        nf_reject_ipv6@bf03d000+848     nf_log_ipv6@bf039000+cf7    nf_log_common@bf035000+90f      ip6table_mangle@bf031000+4be    ip6table_filter@bf02d000+33e    ip6_tables@bf027000+2495    x_tables@bf021000+2aac  regmap_mmio@bf01d000+9c1        ecb@bf019000+5e1        cmac@bf015000+8ad       uhci_hcd@bf00d000+4328      ohci_platform@bf009000+e88      ohci_hcd@bf000000+53fb
<7>[59361.191643] ieee80211 phy0: change: 0x40
<7>[59361.411638] ieee80211 phy0: change: 0x40
<7>[59361.631626] ieee80211 phy0: change: 0x40
<7>[59361.841616] ieee80211 phy0: change: 0x40
<7>[59362.051607] ieee80211 phy0: change: 0x40
<7>[59362.261606] ieee80211 phy0: change: 0x40
<7>[59362.481594] ieee80211 phy0: change: 0x40
<7>[59362.701588] ieee80211 phy0: change: 0x40
<7>[59362.921574] ieee80211 phy0: change: 0x40
<7>[59363.151574] ieee80211 phy0: change: 0x40
<7>[59363.371565] ieee80211 phy0: change: 0x40
<7>[59363.591553] ieee80211 phy0: change: 0x40
<7>[59363.801543] ieee80211 phy0: change: 0x40
<7>[59364.021559] ieee80211 phy0: change: 0x40
<7>[59364.241527] ieee80211 phy0: change: 0x40
<7>[59364.461519] ieee80211 phy0: change: 0x40
<7>[59364.681512] ieee80211 phy0: change: 0x62
<7>[59364.891523] ieee80211 phy0: change: 0x40
<7>[59365.111494] ieee80211 phy0: change: 0x40
<7>[59365.331492] ieee80211 phy0: change: 0x40
<7>[59365.561480] ieee80211 phy0: change: 0x42
<6>[66218.693039] ieee80211 phy1: staid 11 deleted
<6>[76272.361812] ieee80211 phy1: staid 8 deleted
<6>[77018.867274] ieee80211 phy1: staid 8 deleted
<6>[80683.278657] ieee80211 phy1: staid 4 deleted
<6>[80687.200116] ieee80211 phy1: staid 4 deleted
<6>[81489.295977] ieee80211 phy0: staid 2 deleted
<6>[82939.877125] ieee80211 phy1: staid 8 deleted
<6>[83066.713459] ieee80211 phy1: staid 8 deleted
<6>[83387.332615] ieee80211 phy1: staid 8 deleted
<6>[86420.505360] ieee80211 phy1: staid 6 deleted
<6>[86422.422244] ieee80211 phy0: staid 4 deleted
<6>[86422.477276] ieee80211 phy0: staid 3 deleted
<6>[86422.524255] ieee80211 phy0: staid 1 deleted
<6>[87820.471679] ieee80211 phy1: staid 6 deleted
<6>[89854.265224] ieee80211 phy0: staid 2 deleted
<6>[93367.626295] ieee80211 phy1: staid 1 deleted
<6>[95426.492075] ieee80211 phy1: staid 6 deleted
<6>[96245.881600] ieee80211 phy1: staid 6 deleted
<6>[96532.906987] ieee80211 phy1: staid 6 deleted
<6>[98835.096107] ieee80211 phy1: staid 6 deleted
<6>[99520.522588] ieee80211 phy1: staid 6 deleted
<6>[99587.715146] ieee80211 phy1: staid 3 deleted
<6>[99824.629694] ieee80211 phy1: staid 3 deleted
<6>[99978.425461] ieee80211 phy1: staid 3 deleted
<6>[101967.933175] ieee80211 phy1: staid 1 deleted
<6>[105142.811011] ieee80211 phy1: staid 8 deleted
<6>[106261.244567] ieee80211 phy1: staid 1 deleted
<6>[109423.274321] ieee80211 phy1: staid 6 deleted
<6>[110545.743632] ieee80211 phy1: staid 1 deleted
<6>[110572.518249] ieee80211 phy1: staid 1 deleted
<6>[112401.476782] ieee80211 phy1: staid 3 deleted
<6>[119127.360170] ieee80211 phy1: staid 1 deleted
<6>[119660.232957] ieee80211 phy1: staid 3 deleted
<6>[126922.344777] ieee80211 phy1: staid 3 deleted
<6>[131026.048383] ieee80211 phy1: staid 6 deleted
<6>[134177.635893] ieee80211 phy1: staid 3 deleted
<6>[141436.401030] ieee80211 phy1: staid 3 deleted
<6>[148695.044541] ieee80211 phy1: staid 3 deleted
<6>[149589.378679] ieee80211 phy0: staid 1 deleted
<6>[150181.853915] ieee80211 phy1: staid 6 deleted
<6>[152627.701075] ieee80211 phy1: staid 6 deleted
<6>[153223.217029] ieee80211 phy0: staid 1 deleted
[... 143 lines cut ...]

```
root@orion:/usr/var/log# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170810
firmware version: 0x09030008
power table loaded from dts: no
firmware region code: 0x10
mac address: 60:38:e0:xx:xx:xx
2g: disable
5g: enable
antenna: 4 4
irq number: 45
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e0f00000
iobase1: e1180000
tx limit: 1024
rx limit: 16384
```

aaron1959, 2017-12-03:
You're still running the old stuff driver wise.

aaron1959, 2017-12-03:
there is new firmware as well as a new driver.

ratsputin, 2017-12-03:
@aaron1959 I see that now.  Let me update.  I hadn't been following the thread closely-enough.  Thanks.

closed 2017-12-03 by ratsputin

aaron1959, 2017-12-03:
I think there's a good chance it may change your issue for the better.

ratsputin, 2017-12-04:
Hope so!  Just upgraded the code and the firmware.  Hopefully, this fixes it!