kaloz/mwlwifi #239: 2.4GHz device at extreme range having difficulty with new firmware after 2-3 days

ratsputin, 2017-12-06:
I'm running a Rachio wireless sprinkler controller, and I updated to the 3200ACM to get it to connect reliably.  I've found that running radio1 on channel 11 at 1W seems to work reliably-enough.

The problem I'm running into, is with this newer firmware, after the device has been up for 3-4 days, it will lose connection and be unable to reestablish.  This is what I'm seeing in the logs repeatedly:

```
Wed Dec  6 11:17:24 2017 daemon.info hostapd: wlan1: STA 0c:2a:69:xx:xx:xx IEEE 802.11: associated (aid 1)
Wed Dec  6 11:17:24 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 0c:2a:69:xx:xx:xx
Wed Dec  6 11:17:24 2017 daemon.info hostapd: wlan1: STA 0c:2a:69:09:25:08 WPA: pairwise key handshake completed (RSN)
Wed Dec  6 11:17:25 2017 daemon.info hostapd: wlan1: STA 0c:2a:69:xx:xx:xx IEEE 802.11: authenticated
Wed Dec  6 11:17:35 2017 kern.info kernel: [234607.715085] ieee80211 phy1: staid 1 deleted
Wed Dec  6 11:17:35 2017 daemon.info hostapd: wlan1: STA 0c:2a:69:xx:xx:xx IEEE 802.11: associated (aid 1)
Wed Dec  6 11:17:35 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 0c:2a:69:xx:xx:xx
Wed Dec  6 11:17:35 2017 daemon.info hostapd: wlan1: STA 0c:2a:69:xx:xx:xx WPA: pairwise key handshake completed (RSN)
Wed Dec  6 11:17:37 2017 daemon.info hostapd: wlan1: STA 0c:2a:69:xx:xx:xx IEEE 802.11: authenticated
```

I wouldn't have brought this up, except for two things:

- It **seemed** to be more reliable under the previous firmware (I'm running the latest test version)
- The "fix" is simply to do a "scan" from LuCI on that radio

After the scan, the Rachio is back up-and-running reliably.  Here are the messages:

```
Wed Dec  6 11:17:54 2017 kern.info kernel: [234626.449777] ieee80211 phy1: staid 1 deleted
Wed Dec  6 11:17:54 2017 daemon.info hostapd: wlan1: STA 0c:2a:69:xx:xx:xx IEEE 802.11: associated (aid 1)
Wed Dec  6 11:17:54 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 0c:2a:69:xx:xx:xx
Wed Dec  6 11:17:54 2017 daemon.info hostapd: wlan1: STA 0c:2a:69:xx:xx:xx WPA: pairwise key handshake completed (RSN)
Wed Dec  6 11:17:55 2017 daemon.info hostapd: wlan1: STA 0c:2a:69:xx:xx:xx IEEE 802.11: authenticated
```

As you can see, no real difference, but after that, the streaming stops.

Can you provide some guidance here?  There is no channel overlap (the highest adjacent channel is 6).  I'm not sure what sort of information would be helpful here, either.

Kind regards,

Brett

yuhhaurlin, 2017-12-07:
Which version of driver you used?

ratsputin, 2017-12-07:
```
root@orion:/usr/var/log# cat /sys/kernel/debug/ieee80211/phy1/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170810
firmware version: 0x09030201
power table loaded from dts: no
firmware region code: 0x0
mac address: 60:38:e0:xx:xx:xx
2g: enable
5g: disable
antenna: 4 4
irq number: 46
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e1300000
iobase1: e1580000
tx limit: 1024
rx limit: 16384

root@orion:/usr/var/log#
```

yuhhaurlin, 2017-12-07:
Does it include "don't ack null or qos null data""?

yuhhaurlin, 2017-12-07:
Is it possible for you to use 10.3.4.0-20171129? In this way, we can make sure it is updated. So we can start to check this problem on the updated driver. Thanks.

ratsputin, 2017-12-07:
@yuhhaurlin This is the commit I'm using (I've updated package/kernel/mwlwifi/Makefile to pick this up):

```
PKG_SOURCE_VERSION:=af210f0ef79d5c416dd455747a16d76bec5912ba
```

If I'm not picking up all of the latest code using that, could someone please enlighten me as to how to do so?

With regard to your question 'Does it include "don't ack null or qos null data""?', there is not a single instance of the word "null" (case ignored) in my entire messages file for that day.

yuhhaurlin, 2017-12-07:
I will build LEDE image for you to do test.

ratsputin, 2017-12-07:
Unfortunately, I build a bunch of modules built into my image.  I can give you a copy of my .config, but otherwise, a separately-built image won't help me a lot (I use mwan3, for example).  It would be helpful if I could build an image myself.

yuhhaurlin, 2017-12-07:
You can give me your config file. If I build the image, I can update needed package with debug message later.

ratsputin, 2017-12-07:
@yuhhaurlin Thank you!  My .config is at [https://drive.google.com/open?id=1-CwuSANyq1OxES9JWDQGTtijb0x2fz6V](url)

yuhhaurlin, 2017-12-07:
I will build it tomorrow when I come to office.

kubrickfr, 2017-12-07:
@ratsputin To fetch the commit you want, not only you must set PKG_SOURCE_VERSION to the commit hash but you must also set PKG_MIRROR_HASH to an empty string (not just comment it out). Otherwise it will assume the revision you want is available as a package on the mirror.

ratsputin, 2017-12-07:
@kubrickfr Thank you so much for taking the time to answer this question!  I've been trying to figure out why my builds haven't been picking up the right code for a couple of months now.

aaron1959, 2017-12-07:
@kubrickfr Thank you from everyone.

ratsputin, 2017-12-07:
So, after a bit more investigation, it looks like the only way to really do a build that includes all of the current code is to do what's in the README.md.  I did make it a bit cleaner by just doing a git clone of the mwlwifi code.  I now have a script that automates all of this and updates the linked file from the dl directory.  Hopefully, I'll be able to build a clean version with the latest updates within the hour.

Clearly, the build system isn't designed to fetch the entire chain if you set the PKG_SOURCE_VERSION alone.  It appears that specifying a commit ID there only fetches the changes in that commit, not all of its parents.  I wondered why my driver version wasn't being updated correctly.

There really should be an easier way to get the latest updates from a package external to the core LEDE distribution without having to hack the build system as suggested in the readme.

ratsputin, 2017-12-07:
_Finally!_  Good LORD, what a pain in the ass!

```
root@orion:~# cat /sys/kernel/debug/ieee80211/phy1/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20171129
firmware version: 0x09030201
power table loaded from dts: no
firmware region code: 0x0
mac address: 60:38:e0:xx:xx:xx
2g: enable
5g: disable
antenna: 4 4
irq number: 46
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e1300000
iobase1: e1580000
tx limit: 1024
rx limit: 16384

root@orion:~#
```

So, for those who want to build their own firmware, the README.md file is a bit out-of-date and leaves out a few details.

What I ended up doing is a git clone of mwlwifi and putting it into a directory called
```
~/mwlwifi-10.3.4.0-20171011
```
That can then be updated with the following:
```
cd ~/mwlwifi-10.3.4.0-20171011
git pull
cd ..
rm -f mwlwifi-10.3.4.0-20171011.tar.xz
tar Jcvf mwlwifi-10.3.4.0-20171011.tar.xz mwlwifi-10.3.4.0-20171011/
```
The other detail is creating a link from the ~/lede/dl directory to the new tar file.  The issue you run into is that the ~/lede/dl directory won't exist until the first build.  Once the first build is done, the trunk version of mwlwifi has been downloaded and extracted, so replacing the tar file in ~/lede/dl is only half of the battle.  The extracted version has to be removed to force the build system to extract your newly-created tar file.

So, after doing your first full build, and having already created your fresh tar file with the current mwlwifi code in it:
```
cd ~/lede/dl
mv mwlwifi-10.3.4.0-20171011.tar.xz mwlwifi-10.3.4.0-20171011.tar.xz-save
ln -s ../../mwlwifi-10.3.4.0-20171011.tar.xz
```
Now, the final step is to get rid of the cached copy of the trunk code.  This is going to be LEDE-specific in this case, I assume, but in my case, I had to do:
```
rm -Rf ~/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/linux-mvebu/mwlwifi-10.3.4.0-20171011
```
Once that's done a new make will cause the updated mwlwifi code to be built into your image.

Please, if I've missed any detail or misstated anything here someone more knowledgeable let me know.  I've been hacking through this and I'm a git noob.  I've been ultra-frustrated at not being able to build trunk firmware with the latest version of this package built into it.  It appears these steps achieve this goal.

yuhhaurlin, 2017-12-08:
The link to your config file has problem.

ratsputin, 2017-12-08:
Interesting.  Apparently, the link function in github doesn't work as expected.  Here's the link: https://drive.google.com/open?id=1-CwuSANyq1OxES9JWDQGTtijb0x2fz6V

ratsputin, 2017-12-09:
So far, at 44 hours of runtime, both 5GHz and 2.4GHz have been stable.  Tomorrow will be the big test of both this issue and two devices leaving/returning simultaneously.

ratsputin, 2017-12-10:
I am pleased to say that the issue with two devices leaving/returning simultaneously and causing the router to crash appears to be fixed.

If the 2.4Ghz connections make it another 24 hours, I'll close this ticket and consider the issue resolved.

ratsputin, 2017-12-10:
Both devices just left for four hours and came back; still no crash related to the other issue.

Still no issues with 2.4GHz devices.  I'd say the latest commit is quite stable for me.  I'm going to close this issue.

closed 2017-12-10 by ratsputin

rs-se, 2017-12-10:
@ratsputin 

I don't mean to steal away your hardwork, is there any way to get insight or copies of your building scripts that you use to build LEDE for WRT3200ACM?

eduperez, 2017-12-10:
@rs-se I cannot speak on behalf of @ratsputin, but I can tell you how I manage to create my packages:

1. Follow the instructions at LEDE's wiki about "the build system", until you are capable of building the firmware yourself.
1. Use the "config.seed" file for your device to configure your build to exactly the same parameters as the official build (download it as ".config" then run "defconfig"); then reconfigure it to meet your needs.
1. Edit the "package/kernel/mwlwifi/Makefile" file in your local copy, update the package name, and point it to a different commit.
1. You can now build your updated firmware.

If you just want updated packages with the mwlwifi drivers for the stable versions of LEDE, you can grab them from my repo at https://github.com/eduperez/mwlwifi_LEDE

ratsputin, 2017-12-10:
@rs-se Sure.  It's relatively straightforward.  I build LEDE on Ubuntu.  To get set-up, just follow the instructions on https://lede-project.org/docs/guide-developer/quickstart-build-images.  That will get you a basic, generic trunk build for the WRT3200ACM.

Personally, I prefer to build the various software packages into my LEDE image, rather than adding them as software after every time I build a new image.  I don't know what the actual easiest way to accomplish this is, but I just printed a copy of all of the install packages from the Software page on my router I had up-and-running with the stock build.

Once you have the list of packages you want to install, just search from them in the make menuconfig page using "/" and add them as Built-in ([*]).  Note, there are dependencies that will be added automatically, so this is pretty time-consuming, but at the end of the day, you should have everything selected in make menuconfig that's currently added as an installed software package in your running image.

Once you've done that, save your .config file and just do a make.  If all goes well, you'll have a bin file you can upload into your router.  Here's what mine looks like:

```
wyer@ubuntu:~/lede/bin/targets/mvebu/generic$ ls -Flash
total 60M
4.0K drwxr-xr-x 3 wyer wyer 4.0K Dec  8 14:57 ./
4.0K drwxr-xr-x 3 wyer wyer 4.0K Dec  7 16:33 ../
8.0K -rw-r--r-- 1 wyer wyer 6.3K Dec  8 14:49 config.seed
8.0K -rw-r--r-- 1 wyer wyer 6.1K Dec  7 17:15 lede-mvebu-device-linksys-wrt3200acm.manifest
 18M -rw-r--r-- 1 wyer wyer  18M Dec  7 17:15 lede-mvebu-linksys-wrt3200acm-squashfs-factory.img
 13M -rw-r--r-- 1 wyer wyer  13M Dec  7 17:15 lede-mvebu-linksys-wrt3200acm-squashfs-sysupgrade.bin
8.0K -rw-r--r-- 1 wyer wyer 6.1K Dec  8 14:57 openwrt-mvebu-device-linksys-wrt3200acm.manifest
 18M -rw-r--r-- 1 wyer wyer  18M Dec  8 14:57 openwrt-mvebu-linksys-wrt3200acm-squashfs-factory.img
 13M -rw-r--r-- 1 wyer wyer  13M Dec  8 14:57 openwrt-mvebu-linksys-wrt3200acm-squashfs-sysupgrade.bin
4.0K drwxr-xr-x 2 wyer wyer 4.0K Dec  8 14:52 packages/
4.0K -rw-r--r-- 1 wyer wyer  785 Dec  8 14:57 sha256sums
wyer@ubuntu:~/lede/bin/targets/mvebu/generic$
```
If you want to use the latest code for mwlwifi, you'll need to do all of the above, then follow the instructions I posted at the beginning of this ticket.

ratsputin, 2017-12-11:
@yuhhaurlin Apparently, I spoke too soon.  I did, however, discover what seems to be causing this issue and found some more information.

The problem seems to coincide with bringing up a VMware Horizon connection to my work from a wired laptop.  The moment I initiated the Horizon connection, I noticed my phone dropped off the wifi network, saying "no internet connection".  As far as I could tell, packets from the wifi network weren't being forwarded to the Internet anymore.  This may explain why my Rachio keeps disconnecting and reconnecting; I assume it checks for Internet connectivity and if it doesn't have it, it attempts to reconnect.

The other oddity is that my phone lost its IP address and couldn't get a new one.  Note, my phone was on the 5GHz network, so both of the wireless networks exhibited the issue.

Apparently, this has nothing to do with the wifi network itself, so I have two questions:

- What does the "scan" function do and why it would it reset things such that the router starts forwarding packets from the wifi network to the other networks again once it completes?
- Where would I go from here to figure out what's going on?

Kind regards,

Brett

yuhhaurlin, 2017-12-11:
Can you use static IP and ping to router?

ratsputin, 2017-12-11:
Got it.  I'll reconfigure my phone and troubleshoot it from that angle next time it happens.  Thank you.