kaloz/mwlwifi #240: mwlwifi oops

nwmcsween, 2017-12-10:
`
[33493.226670] ------------[ cut here ]------------
[33493.231388] WARNING: CPU: 0 PID: 0 at backports-2017-11-01/net/mac80211/rx.c:4314 ieee80211_rx_napi+0x170/0x828 [mac80211]
[33493.242485] Rate marked as a VHT rate but data is invalid: MCS: 10, NSS: 2
[33493.249393] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 lz4 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY uas slhc rfcomm nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache mwifiex_sdio mwifiex lz4_decompress lz4_compress iptable_mangle iptable_filter ipt_ECN ip_tables hidp hci_uart crc_ccitt btusb btmrvl_sdio btmrvl btintel bnep bluetooth sch_cake nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow
[33493.321277]  cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress hid evdev input_core mwlwifi mac80211 cfg80211 compat xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ifb raid456 async_raid6_recov async_pq raid6_pq async_xor xor xor_neon async_memcpy async_tx raid10 raid1 raid0 md_mod regmap_mmio zram zsmalloc ecb cmac xfs libcrc32c exportfs gpio_button_hotplug
[33493.384647] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.9.65 #0
[33493.390590] Hardware name: Marvell Armada 380/385 (Device Tree)
[33493.396545] [<c0016010>] (unwind_backtrace) from [<c0012220>] (show_stack+0x10/0x14)
[33493.404326] [<c0012220>] (show_stack) from [<c0218580>] (dump_stack+0x7c/0x9c)
[33493.411581] [<c0218580>] (dump_stack) from [<c00292c8>] (__warn+0xbc/0xec)
[33493.418484] [<c00292c8>] (__warn) from [<c002932c>] (warn_slowpath_fmt+0x34/0x44)
[33493.426022] [<c002932c>] (warn_slowpath_fmt) from [<bf24cd3c>] (ieee80211_rx_napi+0x170/0x828 [mac80211])
[33493.435672] [<bf24cd3c>] (ieee80211_rx_napi [mac80211]) from [<bf2ad654>] (pcie_rx_recv_ndp+0x9e4/0xb94 [mwlwifi])
[33493.446077] [<bf2ad654>] (pcie_rx_recv_ndp [mwlwifi]) from [<c002dbcc>] (tasklet_action+0x8c/0xe8)
[33493.455075] [<c002dbcc>] (tasklet_action) from [<c002d364>] (__do_softirq+0xd0/0x204)
[33493.462937] [<c002d364>] (__do_softirq) from [<c002d71c>] (irq_exit+0x94/0xb8)
[33493.470193] [<c002d71c>] (irq_exit) from [<c0062154>] (__handle_domain_irq+0x90/0xb4)
[33493.478057] [<c0062154>] (__handle_domain_irq) from [<c0009428>] (gic_handle_irq+0x50/0x94)
[33493.486442] [<c0009428>] (gic_handle_irq) from [<c0012c8c>] (__irq_svc+0x6c/0x90)
[33493.493954] Exception stack(0xc062ff60 to 0xc062ffa8)
[33493.499026] ff60: 00000001 00000000 00000000 c001b1a0 00000000 c062e000 c0630fe4 00000001
[33493.507237] ff80: c062c168 00000000 c062ffb8 00000001 00000000 c062ffb0 c000f808 c000f80c
[33493.515446] ffa0: 60000013 ffffffff
[33493.518950] [<c0012c8c>] (__irq_svc) from [<c000f80c>] (arch_cpu_idle+0x2c/0x38)
[33493.526381] [<c000f80c>] (arch_cpu_idle) from [<c005b4a4>] (cpu_startup_entry+0xf0/0x19c)
[33493.534596] [<c005b4a4>] (cpu_startup_entry) from [<c05e8c54>] (start_kernel+0x39c/0x420)
[33493.542823] ---[ end trace 95a45d7a8279bb79 ]---

`

yuhhaurlin, 2017-12-10:
This is warning message, I think the system can still work. I will check and remove this warning.

yuhhaurlin, 2017-12-12:
Can you let me know which version of driver you used? I check current code, this kind of warning message should not happen.

yuhhaurlin, 2017-12-12:
I close this issue. If you still encounter this problem on current driver, please let me know.

closed 2017-12-12 by yuhhaurlin

marcust, 2018-06-09:
@yuhhaurlin 
This still seems to happen (running a LEDE snapshot from https://davidc502sis.dynamic-dns.net/)
```
Linux version 4.14.44 (davidc502@pc2) (gcc version 7.3.0 (OpenWrt GCC 7.3.0 r7093-4fdc6ca31b)) #0 SMP Fri Jun 1 19:01:03 2018
<<Marvell Mac80211 Wireless PCIE Network Driver version 10.3.4.0-20180330>>
```
Don't know how often that happens, have seen it for the first time today:
```
[425979.639526] ------------[ cut here ]------------
[425979.644349] WARNING: CPU: 0 PID: 0 at backports-2017-11-01/net/mac80211/rx.c:4355 ieee80211_rx_napi+0x160/0x9cc [mac80211]
[425979.655545] Rate marked as a VHT rate but data is invalid: MCS: 12, NSS: 2
[425979.662535] Modules linked in: pppoe ppp_async pppox ppp_generic nf_nat_pptp nf_conntrack_pptp nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_policy xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm slhc rfcomm nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_conntrack_ipv4 nf_nat_ipv4 nf_nat_h323 nf_nat_amanda nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_tftp
[425979.734591]  nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_irc nf_conntrack_h323 nf_conntrack_broadcast ts_kmp nf_conntrack_amanda iptable_mangle iptable_filter ipt_ah ipt_ECN ip_tables hidp hci_uart crc_ccitt btusb btmrvl_sdio btmrvl btintel bnep bluetooth fuse sch_cake em_nbyte cls_basic sch_prio sch_dsmark sch_pie sch_gred sch_teql act_ipt sch_multiq em_text em_meta sch_codel sch_sfq sch_fq act_police em_cmp sch_red act_connmark nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress hid evdev input_core mwlwifi mac80211 cfg80211 compat cryptodev xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac
[425979.805776]  ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables msdos bonding ifb sit tunnel4 ip_tunnel tun vfat fat ntfs nls_utf8 nls_iso8859_1 nls_cp437 ecdh_generic sha512_generic sha256_generic md5 kpp hmac ecb cmac authenc ohci_pci uhci_hcd ohci_platform ohci_hcd gpio_button_hotplug
[425979.850943] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.14.44 #0
[425979.857062] Hardware name: Marvell Armada 380/385 (Device Tree)
[425979.863108] [<c010f5c0>] (unwind_backtrace) from [<c010b2c8>] (show_stack+0x10/0x14)
[425979.870978] [<c010b2c8>] (show_stack) from [<c061baa8>] (dump_stack+0x88/0x9c)
[425979.878321] [<c061baa8>] (dump_stack) from [<c0121e00>] (__warn+0xe4/0x100)
[425979.885400] [<c0121e00>] (__warn) from [<c0121e54>] (warn_slowpath_fmt+0x38/0x48)
[425979.893035] [<c0121e54>] (warn_slowpath_fmt) from [<bf2abe98>] (ieee80211_rx_napi+0x160/0x9cc [mac80211])
[425979.902789] [<bf2abe98>] (ieee80211_rx_napi [mac80211]) from [<bf316e78>] (pcie_rx_recv_ndp+0x790/0xcb0 [mwlwifi])
[425979.913292] [<bf316e78>] (pcie_rx_recv_ndp [mwlwifi]) from [<c0126268>] (tasklet_action+0x94/0x10c)
[425979.922466] [<c0126268>] (tasklet_action) from [<c0101628>] (__do_softirq+0xe0/0x240)
[425979.930418] [<c0101628>] (__do_softirq) from [<c01265b0>] (irq_exit+0xb0/0xd8)
[425979.937764] [<c01265b0>] (irq_exit) from [<c015d58c>] (__handle_domain_irq+0x9c/0xac)
[425979.945717] [<c015d58c>] (__handle_domain_irq) from [<c0101464>] (gic_handle_irq+0x5c/0x90)
[425979.954192] [<c0101464>] (gic_handle_irq) from [<c010bf4c>] (__irq_svc+0x6c/0x90)
[425979.961794] Exception stack(0xc0a01f50 to 0xc0a01f98)
[425979.966953] 1f40:                                     00000001 00000000 00000000 c0114c00
[425979.975255] 1f60: ffffe000 c0a03c70 c0a03c24 00000000 00000000 414fc091 00000000 00000000
[425979.983556] 1f80: c0a01f98 c0a01fa0 c010847c c0108480 60000013 ffffffff
[425979.990288] [<c010bf4c>] (__irq_svc) from [<c0108480>] (arch_cpu_idle+0x34/0x38)
[425979.997804] [<c0108480>] (arch_cpu_idle) from [<c01544d8>] (do_idle+0xdc/0x19c)
[425980.005232] [<c01544d8>] (do_idle) from [<c01547f4>] (cpu_startup_entry+0x18/0x1c)
[425980.012926] [<c01547f4>] (cpu_startup_entry) from [<c0900d4c>] (start_kernel+0x470/0x47c)
[425980.021239] ---[ end trace 4cbf4e2366dface4 ]---
```