kaloz/mwlwifi #241: MU-MIMO for 88W8964.

yuhhaurlin, 2017-12-14:
Create this issue to track this feature.

adamcarter1, 2017-12-14:
Excited for this!

ad019, 2017-12-14:
@yuhhaurlin before Christmas this year?

yuhhaurlin, 2017-12-14:
Company will shutdown one week after Christmas. There is only one week before Christmas. I don't think it can be done within one week.

s-pimenta, 2017-12-14:
Take your time... We have enough of instability...

s-pimenta, 2017-12-16:
160MHz will also be available?

yuhhaurlin, 2017-12-16:
160 MHz working now.

s-pimenta, 2017-12-16:
Nice... I will buy the new Intel Wireless-AC 9260 https://www.intel.com/content/www/us/en/wireless-products/dual-band-wireless-ac-9260-brief.html to test 160Mhz

p3x-robot, 2017-12-27:
Yes 160 Mhz have been working for a few months for me. Just MU-MIMO is not enabled and as I know needs code done.
Look:
[image]

p3x-robot, 2017-12-27:
[image]

p3x-robot, 2017-12-27:
I use the kmod from here, the latest Kaloz always LEDE latest stable now 17.01.4:
https://github.com/eduperez/mwlwifi_LEDE/

My CDN:
https://cdn.corifeus.com/lede/17.01.4/targets/mvebu/generic/packages/

https://cdn.corifeus.com/lede/17.01.4/targets/mvebu/generic/packages/kmod-iwlwifi_4.4.92%2B2017-01-31-3_arm_cortex-a9_vfpv3.ipk

aaron1959, 2017-12-27:
@p3x-robot   all your connections are @ 80 Mhz.
Do you actually have any functioning at 160 Mhz Rates? 
Your post does not show this.

p3x-robot, 2017-12-27:
That's how it works right now. Once MU-MIMO is done, it will work. I guess 160Mhz only work with MU-MIMO. I set to 160Mhz, but the ```Kaloz``` is not done. But the rates of latest commits are very fast and stable, never instable. Besides it works with the ```Radar``` as well. Once the working latest commit is implemented with ```MU-MIMO```, it will work with 160 Mhz. That's all could do. :)

aaron1959, 2017-12-27:
VHT 160 has absolutely NOTHING to do with MU-MIMO!

p3x-robot, 2017-12-27:
I think my 2 Androids don't understand 160Mhz, I guess, either my ```wifi workstation```, they are max 80Mhz.
But MU-MIMO will work later. The settings are correct.

aaron1959, 2017-12-27:
VHT 160 works right now But ONLY between devices which are capable of VHT 160.  
It seems that you are confusing a couple of things which have no relation to each other.

p3x-robot, 2017-12-27:
Ok, so I understand that mu mimo beamforming has nothing to do with frequencies. Thanks.

aaron1959, 2017-12-27:
The Androids are ONLY VHT 80 capable, don't you understand this? 
You will NEVER connect to them with VHT 160!!!!

p3x-robot, 2017-12-27:
Yes, that's why said. My Android, neither my wifi are max 80 Mhz, too new. :)

aaron1959, 2017-12-27:
BY that I mean **your particular Android devices**, 
In the near future yes the next gen devices  MAY have VHT 160 but not NOW.

p3x-robot, 2017-12-27:
But  MU-MIMO will work out of box once Kaloz is done and Radar is good and the 3200ACM 160Mhz is working as well.
On 1900ACS it has only max 80Mhz. I think that doesn't have MU-MIMO neither.
Only 3200ACM has 160Mhz. That's the future. :)

p3x-robot, 2017-12-27:
yes, exactly.

p3x-robot, 2017-12-27:
Waiting it for so much ! :)

GillyMoMo, 2017-12-27:
[image]

An example of the VHT 160. It would more or less like this.

thagabe, 2017-12-28:
@GillyMoMo What device is that? 

@p3x-robot Pls, stapppp you are mixing up things left and right.

p3x-robot, 2017-12-28:
@thagabe sorry, this is a discussion, it's like the force awakens... :) Happy new year dude.

p3x-robot, 2017-12-28:
@GillyMoMo right, your wifi device can use 160 mhz, mines can't :) but though when MU-MIMO is done, that will be awesome.

p3x-robot, 2017-12-28:
I only got 1200AC WIFI device on my workstation, at least 2 x 866 Mbit :)

aaron1959, 2017-12-28:
@thagabe I'm hoping it can be attributed to a language issue, otherwise someone is prone to random spurious emissions.
Nothing to be learned by that. :-(

thagabe, 2017-12-29:
@aaron1959 I do believe it has to do with language issues. However, the fact that he said he had 160 MHz and showed a bunch of screen shots that showed he didn't + now showing that he has a 1200AC which DOES NOT have 160 MHz makes me believe he doesnt know what he is talking about lol

aaron1959, 2017-12-29:
Roger that.

GillyMoMo, 2017-12-29:
@thegabe I have two 3200ACMs. They are bridged. I also have a 1900AC v1 and v2.

thagabe, 2017-12-29:
@GillyMoMo WDS bridge? That would explain the 160 MHz. I still do not own a devices that can connect via 160 Mhz to device. I'm using a 3200ACM as the main router and my old 1900ACv1 as a DumbAP but wired.

thagabe, 2017-12-29:
@p3x-robot Also Kaloz is the openwrt maintainer of mwlwifi but the developers behind the driver is yuhhaurlin who is a Marvell employee/contractor.

GillyMoMo, 2017-12-31:
@thagabe Yes, I do have a WDS bridge. It works pretty good. At first I was having to restart from time to time, but now that it's stable it works just fine now.

bbogey, 2018-01-25:
Thinking to get wrt3200acm. MU-MIMO coming anytime soon?

aaron1959, 2018-01-25:
They can't even make it function as a basic wifi router yet I doubt that MU-MIMO is going to happen before the router is even fully functional, right now all firmware and drivers are still in alpha testing if you read these Git hub you will know.

yuhhaurlin, 2018-01-25:
@aaron1959 Did you still encounter problems? If you did not have problems, why you said basic function of this router is not fully functional? Right now, only problems with Google problem of Andriod phone and group rekey for WDS AP are under tracking. Most of users should use the router without problems.

yuhhaurlin, 2018-01-25:
MU-MIMO should be added later. In fact, mac80211 only supports MU-MIMO for client mode. I wonder which mac80211open source router supports MU-MIMO for AP mode?

adamcarter1, 2018-02-27:
Now that everything is fixed, mu-mimo full steam ahead ;)

thagabe, 2018-03-05:
@yuhhaurlin Ping

p3x-robot, 2018-03-05:
Hey @yuhhaurlin @kaloz!

How are you?

As for the MU-MIMO... Is there any way to progress with the LinkSys WRT 3200ACM, I would love to code it, but is there any way to find specs, docs and descriptions to do it?

It is such a shame that only Linksys / Marvell / Benwin employee/contractor can do it. It is supposed to be open source, but given I have no idea to find the info, how can we progress? 

It is soon an aged router and still, the MU-MIMO is not working, but this is why I bought this router.

I would love to work on the code. So much.

Is there any way? Or only Marvell / Linksys, etc. employees ???

Some docs would be so great to find out the specs and implement this.

Thanks,
Patrik

thagabe, 2018-03-05:
He isn't wrong. Specifications for Wireless AX are coming out and soon early prototypes will be released. mwlwifi is, for a lack of a better word, chugging along. It has some unfinished or dirty workarounds for older models that are left while focusing on newer models which is still not feature complete. Most of Q1'18 has been debugging rather than implementing. I do realize this is not easy but @p3x-robot is right, more people can and would get involved but everything is shrouded in secrecy.

p3x-robot, 2018-03-05:
So basically, from Linksys (or whoever overtaking Benwin or something like that, don't remember the name something with B) is just cheating totally and is not open source at all, closed and the programmers like @yuhhaurlin just debugging, hacking and cracking. So so so sad!!!

p3x-robot, 2018-03-05:
By the way, when I look at all the routers, for something, I still think Linksys WRT is the best router (horsepower, features and stability), some routers don't even have 5GHZ and basically, my 3200ACM and 1900ACS with the latest mwlwifi is rock solid.

On Android, my fastest was 400Mbit down (out of 1Gbit) and 200Mbit up (out of 200Mbit) and we have like 50 routers in the aparment building (including 4 of mine all Wifi both 2.4 and 5 ghz in 2 rooms, small area ).

Oh, yeah and it works totally with LEDE (4 kilo) / OpenWrt (5 kilo) from my own cdn:
http://cdn.corifeus.com fo show!

**So besides, props up for you guys for helping making this best router in the world with this Wifi!!!**

:+1: :100: :1234: :1st_place_medal:

yuhhaurlin, 2018-03-06:
Start to work on this feature.

adamcarter1, 2018-03-31:
Progress?

ad019, 2018-04-10:
@yuhhaurlin any eta for this?

adamcarter1, 2018-04-11:
@ad019 i think we will have this all latched up by the time 802.11ax comes out.

ad019, 2018-04-11:
¯\_(ツ)_/¯

p3x-robot, 2018-04-20:
no news still.. you will see in the commits if something changed or they have notified on this page. 

if nothing changed in the commits or new info on below, there is still unimplemented.

ad019, 2018-04-20:
Could be that Foxconn acquisition of Belkin has led to change of priorities between Marvell and Belkin.

adamcarter1, 2018-04-20:
True.

yuhhaurlin, 2018-04-24:
Sorry. I am busy on other jobs recently. I come back to work on this feature now.

davidc502, 2018-04-26:
Good to see you back yuhhaurlin. 

Thanks for your hard work.

p3x-robot, 2018-05-03:
this is true, AX is coming soon, and MI-MIMO will be never happen, i can sense this.
both the actual specs and the actual implementation - not happening at all.

we stuck with this, but i guess if i had a 160MHZ client, i could use 2600Gb as i understand, but anyway, i can sense this kaloz mwlwifi will not happen at all, like years will go on as:
* this is gonna happen
* 6 months later - this is gonna happen
* 1 year later - this is gonna happen
* this sucks ....
* 2 years later => AX .... new router...

i guess this is belkin or the latest owner whatever the name of the company.... you name it... this is not linksys (cisco small child => linksys), no no no, this is something asian company now... not USA design...

:-1:  :1234:  :-1:

BrainSlayer, 2018-05-03:
mmh its not about belkin. its about marvell, the chipset vendor. linksys/belkin does not develop the drivers. marvell has a mi-mimo supporting driver which is propertiery. personally i dont understand why its usefull to maintain 2 drivers. one is propertiery and one is opensource but less featured due the small human resources offered for it. but consider that its the same for qca with ath10k. the chipsets are mu-mimo capable, but ath10k does not support it, but the propertiery driver does. but at least its easier to get access to qca's internal driver sources with some efforts. and for sure we dont need to talk about broadcom. this all makes no sense for me. it makes no sense for marvells business and it makes no sense for the driver quality which is the only thing visible to end users

p3x-robot, 2018-05-03:
i guess, it looks like mu-mimo is not wirespread at all. plus they provide to work with like AC. imagine openwrt has so many packages, you can do anything you want, companies want to pay for sure (including marvell god damn...)

that is what it is about 1000000%

BrainSlayer, 2018-05-03:
it is widespread for all vendor provided firmwares, but the benefits arent directly visible for the end user. mu-mimo feature has also just measureable performance gain for multiple clients. thats nothing you will notice at home. for sure you're able to get the propertiery driver from marvell if you start a business discussion with them and if they see some benefit in it you get it. i got them for older chipsets and i did not have to pay any money for it. from my experience companies always try to protect intellectual properties which are included in these special drivers. but if i review the propertiery drivers of qca, marvell or broadcom or mediatek which are heavily complex and bringing own wireless stacks. i can't find any argument for it. thats why i say that this all makes no sense at all.  all companies putting alot of effort in developing own wireless stacks which replace mac80211 and there seem to work alot of people on these drivers. if they would spend this power in using existing technology like mac80211 they would even reduce development time.

p3x-robot, 2018-05-03:
* so basically, `openwrt` is using an `mac80211` wifi interface for all routers (including `mwlwifi`). 
* the marvell linksys wrt3200acm `chipset` on the original router firmware is not using `mac80211` at all, but the marvell wifi developed, and it looks like `mwlwifi` is an interface to make it work with `mac80211`?
* so even, if i talk to them and i could sell a binary `mac80211` interfaced driver and give the the royality, i would still have to learn both marvell and `mac80211` and then create this interface, so something i do not have time to do it...

looks like not happening for me either, and why would marvell help create an `open source` `mac80211` interface/driver, why?

besides linksys said it is `open source ready`, which is true, they never said it is `mu-mimo open source ready` ...

so the doors are as they are... `open source` open, `mu-mimo` closed ...

p3x-robot, 2018-05-03:
@BrainSlayer so you looks like live in Germany? Right? So in germany, do you have laptop + desktop + tablet + phone, they have all `MU-MIMO`? Because it looks like very rarely as I sense. I guess we just want this, but very few people have `MU-MIMO` appliancies actually, rarely, or wrong?

What is weird, why people do not like this dual core ARM 1.8 Ghz CPU and 512MB DDR3 with USB3 on LEDE, supports even `MariaDB 10` including `NodeJs 10` and all `GCC`, latest `Python`, `PHP`, etc ... , I guess, it should have an `nvme` ssd by default, I would say `WRT3200ACM` is garbage, but is not!  How can it be so bad??? I love it like hell!!!

https://www.techspot.com/news/66672-linksys-latest-wrt-router-combines-open-source-firmware.html

*This cut*:
While Linksys has certainly built an impressive router on paper, it’s worth noting that the WRT3200ACM will only as fast as the clients connected to it. As it stands now there are no Tri-band 160 laptops or phones currently on the market, and there are very few that support MU-MIMO technology, so you’re essentially getting bragging rights and hoping those technologies become widespread in the near future.

The Linksys AC3200 WRT Gigabit MU-MIMO Wi-Fi Router is available now via the Linksys web store and Amazon priced at a hefty $280.

[16 more replies]

closed 2018-05-03 by yuhhaurlin