kaloz/mwlwifi #242: WDS: No traffic after group key handshake

eduperez, 2017-12-14:
Main router: WRT3200ACM running LEDE 17.01.4 plus updated (commit 843d00c) mwlwifi drivers.
Wireless extender: WDR4300 running LEDE 17.01.4

I have been running these two devices using "relayd" for a couple of months, and the connection was fast and rock solid. I recently changed to WDS using recent drives, and have noticed that traffic stops completely when the AP router performs a "group key handshake".

These are the logs at the AP when the STA is powered on and a connection is established:

```
Thu Dec 14 22:56:49 2017 daemon.info hostapd: wlan_5g: STA c0:4a:00:fc:2c:2a IEEE 802.11: associated (aid 2)
Thu Dec 14 22:56:49 2017 daemon.notice hostapd: wlan_5g: AP-STA-CONNECTED c0:4a:00:fc:2c:2a
Thu Dec 14 22:56:49 2017 daemon.info hostapd: wlan_5g: STA c0:4a:00:fc:2c:2a WPA: pairwise key handshake completed (RSN)
Thu Dec 14 22:56:49 2017 kern.info kernel: [ 3810.237052] device wlan_5g.sta2 entered promiscuous mode
Thu Dec 14 22:56:49 2017 kern.info kernel: [ 3810.242669] br-lan: port 5(wlan_5g.sta2) entered forwarding state
Thu Dec 14 22:56:49 2017 kern.info kernel: [ 3810.248859] br-lan: port 5(wlan_5g.sta2) entered forwarding state
Thu Dec 14 22:56:51 2017 daemon.info dnsmasq-dhcp[4212]: DHCPDISCOVER(br-lan) c0:4a:00:fc:2c:28 
Thu Dec 14 22:56:51 2017 daemon.info dnsmasq-dhcp[4212]: DHCPOFFER(br-lan) 192.168.1.253 c0:4a:00:fc:2c:28 
Thu Dec 14 22:56:51 2017 daemon.info hostapd: wlan_5g: STA c0:4a:00:fc:2c:2a IEEE 802.11: authenticated
Thu Dec 14 22:56:51 2017 kern.info kernel: [ 3812.246134] br-lan: port 5(wlan_5g.sta2) entered forwarding state
Thu Dec 14 22:56:54 2017 daemon.info dnsmasq-dhcp[4212]: DHCPDISCOVER(br-lan) c0:4a:00:fc:2c:28 
Thu Dec 14 22:56:54 2017 daemon.info dnsmasq-dhcp[4212]: DHCPOFFER(br-lan) 192.168.1.253 c0:4a:00:fc:2c:28 
Thu Dec 14 22:56:56 2017 daemon.info dnsmasq-dhcp[4212]: DHCPDISCOVER(br-lan) b8:27:eb:b2:77:4a 
Thu Dec 14 22:56:56 2017 daemon.info dnsmasq-dhcp[4212]: DHCPOFFER(br-lan) 192.168.1.1 b8:27:eb:b2:77:4a 
Thu Dec 14 22:56:57 2017 daemon.info dnsmasq-dhcp[4212]: DHCPDISCOVER(br-lan) c0:4a:00:fc:2c:28 
Thu Dec 14 22:56:57 2017 daemon.info dnsmasq-dhcp[4212]: DHCPOFFER(br-lan) 192.168.1.253 c0:4a:00:fc:2c:28 
Thu Dec 14 22:56:58 2017 daemon.info dnsmasq-dhcp[4212]: DHCPREQUEST(br-lan) 192.168.1.253 c0:4a:00:fc:2c:28 
Thu Dec 14 22:56:58 2017 daemon.info dnsmasq-dhcp[4212]: DHCPACK(br-lan) 192.168.1.253 c0:4a:00:fc:2c:28 bridge
```
Then traffic flows normally and everything works as expected; some time later these logs appear:
```
Thu Dec 14 23:03:45 2017 daemon.info hostapd: wlan_2g: STA 40:9f:38:5e:1f:18 WPA: group key handshake completed (RSN)
Thu Dec 14 23:03:46 2017 daemon.info hostapd: wlan_ext2g: STA 60:03:08:87:7b:49 WPA: group key handshake completed (RSN)
Thu Dec 14 23:03:47 2017 daemon.info hostapd: wlan_5g: STA 80:00:6e:4c:94:a3 WPA: group key handshake completed (RSN)
Thu Dec 14 23:03:50 2017 daemon.notice hostapd: wlan_5g: AP-STA-DISCONNECTED c0:4a:00:fc:2c:2a
Thu Dec 14 23:03:52 2017 kern.debug kernel: [ 4232.606886] ieee80211 phy0: staid 2 deleted
Thu Dec 14 23:03:52 2017 daemon.info hostapd: wlan_5g: STA c0:4a:00:fc:2c:2a IEEE 802.11: associated (aid 2)
Thu Dec 14 23:03:52 2017 daemon.notice hostapd: wlan_5g: AP-STA-CONNECTED c0:4a:00:fc:2c:2a
Thu Dec 14 23:03:52 2017 daemon.info hostapd: wlan_5g: STA c0:4a:00:fc:2c:2a WPA: pairwise key handshake completed (RSN)
Thu Dec 14 23:04:01 2017 kern.debug kernel: [ 4241.893217] ieee80211 phy0: staid 2 deleted
Thu Dec 14 23:04:01 2017 daemon.info hostapd: wlan_5g: STA c0:4a:00:fc:2c:2a IEEE 802.11: authenticated
```

And traffic halts completely; however, after a few minutes I see these messages:
```
Thu Dec 14 23:18:52 2017 daemon.notice hostapd: wlan_5g: AP-STA-DISCONNECTED c0:4a:00:fc:2c:2a
Thu Dec 14 23:18:52 2017 daemon.info hostapd: wlan_5g: STA c0:4a:00:fc:2c:2a IEEE 802.11: disassociated due to inactivity
Thu Dec 14 23:18:52 2017 daemon.info hostapd: wlan_5g: STA c0:4a:00:fc:2c:2a IEEE 802.11: associated (aid 2)
Thu Dec 14 23:18:52 2017 daemon.notice hostapd: wlan_5g: AP-STA-CONNECTED c0:4a:00:fc:2c:2a
Thu Dec 14 23:18:52 2017 daemon.info hostapd: wlan_5g: STA c0:4a:00:fc:2c:2a WPA: pairwise key handshake completed (RSN)
Thu Dec 14 23:18:58 2017 daemon.info hostapd: wlan_5g: STA c0:4a:00:fc:2c:2a IEEE 802.11: authenticated
Thu Dec 14 23:19:05 2017 daemon.notice hostapd: wlan_2g: AP-STA-DISCONNECTED e0:cb:1d:1d:d7:3b
Thu Dec 14 23:19:05 2017 daemon.info hostapd: wlan_2g: STA e0:cb:1d:1d:d7:3b IEEE 802.11: disassociated
Thu Dec 14 23:19:05 2017 kern.debug kernel: [ 5145.886931] ieee80211 phy1: staid 3 deleted
```
And traffic beings to flow again.

BrainSlayer, 2017-12-14:
how is your group key configured on ap side and how is your group key configured on station side. you missed something very important. your hostapd and your wpa supplicant config. a wrong group key config leads to non working connections (or just unicast but no broadcast/multicast is working)

yuhhaurlin, 2017-12-15:
I will check it. You can disable group rekey as temporary solution. Thanks.

eduperez, 2017-12-15:
@BrainSlayer Perhaps I am missing something... but I have not configured any group key; I just have an AP configured using "psk-mixed+aes" and WDS active on the main router, and a STA also configured with WDS active on the "wireless extender".

BrainSlayer, 2017-12-15:
group key cipher its what i mean. not the key itself.

eduperez, 2017-12-15:
With group re-keying disabled, traffic has been flowing for a while, and everything seems to work as expected.

eduperez, 2018-01-18:
Problem still present at driver 10.3.4.0-20180118 with firmware 9.3.2.4.

yuhhaurlin, 2018-01-18:
Yes. This one is still under working

gufus, 2018-07-14:
@eduperez

Sure, I'll try your work-around.

How do I dissable group key handshake?

eduperez, 2018-07-14:
@gufus Just add a `"option wpa_group_rekey 0"` line in your "`/etc/config/wifi`" file on the AP, under the proper "`config wifi-iface`" section.

gufus, 2018-07-15:
@eduperez 

Thannks!

 WDS STA sttays up now :--)