kaloz/mwlwifi #244: 5G cannot change to channel other than 36?

jacksonatl, 2017-12-16:
Trying to change the channel causes router crash and is reset back to channel 36.

cat /sys/kernel/debug/ieee80211/phy1/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20171214
firmware version: 0x09030202
power table loaded from dts: no
firmware region code: 0x10
mac address: 60:38:e0:b9:c6:c1
2g: enable
5g: disable
antenna: 4 4
irq number: 106
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000000
radio: disable
iobase0: e1280000
iobase1: e1500000
tx limit: 1024
rx limit: 16384

cat /etc/openwrt_release 
DISTRIB_ID='LEDE'
DISTRIB_RELEASE='17.01-SNAPSHOT'
DISTRIB_REVISION='r3785-ceea0ac25d'
DISTRIB_CODENAME='reboot'
DISTRIB_TARGET='mvebu/generic'
DISTRIB_ARCH='arm_cortex-a9_vfpv3'
DISTRIB_DESCRIPTION='LEDE Reboot 17.01-SNAPSHOT r3785-ceea0ac25d'
DISTRIB_TAINTS='no-all mklibs'

cilix-lab, 2017-12-16:
Try `iw reg get`. If you get something like “DFS-Unset”, it’s a regulatory issue which can be fixed by removing the sd8887 driver (mwifiex, I think).

jacksonatl, 2017-12-16:
iw reg get
global
country US: DFS-FCC
    (2402 - 2472 @ 40), (N/A, 30), (N/A)
    (5170 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
    (5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
    (5490 - 5730 @ 160), (N/A, 23), (0 ms), DFS
    (5735 - 5835 @ 80), (N/A, 30), (N/A)
    (57240 - 63720 @ 2160), (N/A, 40), (N/A)

phy#2
country US: DFS-FCC
    (2402 - 2472 @ 40), (N/A, 30), (N/A)
    (5170 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
    (5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
    (5490 - 5730 @ 160), (N/A, 23), (0 ms), DFS
    (5735 - 5835 @ 80), (N/A, 30), (N/A)
    (57240 - 63720 @ 2160), (N/A, 40), (N/A)

phy#1
country US: DFS-FCC
    (2402 - 2472 @ 40), (N/A, 30), (N/A)
    (5170 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
    (5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
    (5490 - 5730 @ 160), (N/A, 23), (0 ms), DFS
    (5735 - 5835 @ 80), (N/A, 30), (N/A)
    (57240 - 63720 @ 2160), (N/A, 40), (N/A)

phy#0
country US: DFS-FCC
    (2402 - 2472 @ 40), (N/A, 30), (N/A)
    (5170 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
    (5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
    (5490 - 5730 @ 160), (N/A, 23), (0 ms), DFS
    (5735 - 5835 @ 80), (N/A, 30), (N/A)
    (57240 - 63720 @ 2160), (N/A, 40), (N/A)

s-pimenta, 2017-12-16:
This also happened to me...

Just go to: Network -> Wireless -> radio0 -> Advanced Settings
 "Country Code" and set your country... and now will be able to use DFS channels...

https://forum.openwrt.org/viewtopic.php?pid=369553#p369553

jacksonatl, 2017-12-16:
It's already set to US

s-pimenta, 2017-12-16:
Did you reboot the router then?

I changed mine to PT - Portugal and it worked (it was to 00 - World)

s-pimenta, 2017-12-16:
Are you using the image from davidc502?

jacksonatl, 2017-12-16:
Yes, I have rebooted but it was never not US.
Image is:cat /etc/openwrt_version
r3785-ceea0ac25d
root@router:~# cat /etc/openwrt_release
DISTRIB_ID='LEDE'
DISTRIB_RELEASE='17.01-SNAPSHOT'
DISTRIB_REVISION='r3785-ceea0ac25d'
DISTRIB_CODENAME='reboot'
DISTRIB_TARGET='mvebu/generic'
DISTRIB_ARCH='arm_cortex-a9_vfpv3'
DISTRIB_DESCRIPTION='LEDE Reboot 17.01-SNAPSHOT r3785-ceea0ac25d'
DISTRIB_TAINTS='no-all mklibs'

cilix-lab, 2017-12-16:
Doesn’t seem regulatory related.  
I’m not familiar with LEDE, anymore, but you can set the channel to something other than 36, copy the config file LEDE generates and then use it to debug hostapd with `hostapd -d <config-file>` so we get an idea why it crashes.  
Maybe there’s an easier way someone else can come up with.

yuhhaurlin, 2017-12-16:
This is impossible.

closed 2017-12-16 by yuhhaurlin

jacksonatl, 2017-12-17:
What is impossible?

aaron1959, 2017-12-17:
Probably the idea that it has anything to do with mwlwifi.

yuhhaurlin, 2017-12-17:
Mwlwifi doesn't have this problem if regulatory domain is all right.