kaloz/mwlwifi #245: Wifi speed drops after update from 0x09030201 to 0x09030202

jacksonatl, 2017-12-16:
cat /sys/kernel/debug/ieee80211/phy1/mwlwifi/info
driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20171129
firmware version: 0x09030201
power table loaded from dts: no
firmware region code: 0x0
mac address: 60:38:e0:b9:c6:c1
2g: enable
5g: disable
antenna: 4 4
irq number: 106
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000000
radio: disable
iobase0: e1280000
iobase1: e1500000
tx limit: 1024
rx limit: 16384

iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[ 4] local 192.168.1.101 port 55990 connected to 192.168.1.1 port 5201
[ ID] Interval Transfer Bandwidth Retr Cwnd
[ 4] 0.00-1.00 sec 11.7 MBytes 97.9 Mbits/sec 0 506 KBytes
[ 4] 1.00-2.01 sec 20.7 MBytes 173 Mbits/sec 0 817 KBytes
[ 4] 2.01-3.01 sec 21.5 MBytes 181 Mbits/sec 0 953 KBytes
[ 4] 3.01-4.00 sec 21.9 MBytes 185 Mbits/sec 0 1001 KBytes
[ 4] 4.00-5.01 sec 22.0 MBytes 182 Mbits/sec 0 1001 KBytes
[ 4] 5.01-6.00 sec 21.0 MBytes 178 Mbits/sec 0 1001 KBytes
[ 4] 6.00-7.02 sec 22.0 MBytes 182 Mbits/sec 0 1001 KBytes
[ 4] 7.02-8.00 sec 21.1 MBytes 180 Mbits/sec 0 1001 KBytes
[ 4] 8.00-9.00 sec 21.5 MBytes 180 Mbits/sec 0 1001 KBytes
[ 4] 9.00-10.01 sec 21.2 MBytes 178 Mbits/sec 0 1001 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval Transfer Bandwidth Retr
[ 4] 0.00-10.01 sec 205 MBytes 172 Mbits/sec 0 sender
[ 4] 0.00-10.01 sec 203 MBytes 170 Mbits/sec receiver

cat /sys/kernel/debug/ieee80211/phy1/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20171214
firmware version: 0x09030202
power table loaded from dts: no
firmware region code: 0x10
mac address: 60:38:e0:b9:c6:c1
2g: enable
5g: disable
antenna: 4 4
irq number: 106
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000000
radio: disable
iobase0: e1280000
iobase1: e1500000
tx limit: 1024
rx limit: 16384

iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  4] local 192.168.1.101 port 59664 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bandwidth       Retr  Cwnd
[  4]   0.00-1.00   sec  3.23 MBytes  27.0 Mbits/sec    1   69.3 KBytes       
[  4]   1.00-2.00   sec  3.12 MBytes  26.2 Mbits/sec    0   97.6 KBytes       
[  4]   2.00-3.00   sec  3.18 MBytes  26.7 Mbits/sec    0    120 KBytes       
[  4]   3.00-4.00   sec  3.16 MBytes  26.5 Mbits/sec    0    139 KBytes       
[  4]   4.00-5.00   sec  3.16 MBytes  26.5 Mbits/sec    0    156 KBytes       
[  4]   5.00-6.00   sec  3.14 MBytes  26.4 Mbits/sec    0    170 KBytes       
[  4]   6.00-7.00   sec  3.16 MBytes  26.6 Mbits/sec    0    184 KBytes       
[  4]   7.00-8.00   sec  3.14 MBytes  26.3 Mbits/sec    0    197 KBytes       
[  4]   8.00-9.00   sec  3.15 MBytes  26.4 Mbits/sec    0    208 KBytes       
[  4]   9.00-10.00  sec  3.16 MBytes  26.6 Mbits/sec    0    256 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-10.00  sec  31.6 MBytes  26.5 Mbits/sec    1             sender
[  4]   0.00-10.00  sec  31.4 MBytes  26.3 Mbits/sec                  receiver

yuhhaurlin, 2017-12-16:
9.3.2.2 is almost the same as 9.3.2.1.

yuhhaurlin, 2017-12-16:
I don't trust your test.

closed 2017-12-16 by yuhhaurlin

jacksonatl, 2017-12-17:
What test would you trust? The speed difference between the two is very
apparent.

yuhhaurlin, 2017-12-17:
1. We have done throughput test.
2. 9.3.2.2 is only added one API for WDS.

yuhhaurlin, 2017-12-17:
On other word, 9.3.2.2 is almost the same as 9.3.2.1. If you do not use WDS, they are the same firmware.

yuhhaurlin, 2017-12-17:
Especially you did not specify your setting. If you do the test for different bandwidth, you will get different result.