kaloz/mwlwifi #246: Wireless printer using bonjour stops working on WRT3200ACM

ad019, 2017-12-17:
#243 Opening this new ticket to track the problem.

daloki83, 2017-12-21:
From my side also affected devices:
AVM Repeater 300E
Instar IN-6014HD (Webcam)
iPhone 5s (just from saying)

Tested on firmware to 9.3.2.2 // 9.3.2.1 // 9.3.0.8 with driver 10.3.4.0-20171214

The log looks like this when it happens:
Dec 20 17:32:07 LEDE123 dnsmasq-dhcp[2051]: DHCPDISCOVER(br-lan) b8:27:eb:2e:71:5b
Dec 20 17:32:07 LEDE123 dnsmasq-dhcp[2051]: DHCPOFFER(br-lan) 192.169.178.19 b8:27:eb:2e:71:5b
Dec 20 17:32:12 LEDE123 dnsmasq-dhcp[2051]: DHCPDISCOVER(br-lan) b8:27:eb:2e:71:5b
Dec 20 17:32:12 LEDE123 dnsmasq-dhcp[2051]: DHCPOFFER(br-lan) 192.169.178.19 b8:27:eb:2e:71:5b
Dec 20 17:32:17 LEDE123 dnsmasq-dhcp[2051]: DHCPDISCOVER(br-lan) b8:27:eb:2e:71:5b
Dec 20 17:32:17 LEDE123 dnsmasq-dhcp[2051]: DHCPOFFER(br-lan) 192.169.178.19 b8:27:eb:2e:71:5b
Dec 20 17:32:22 LEDE123 dnsmasq-dhcp[2051]: DHCPDISCOVER(br-lan) b8:27:eb:2e:71:5b
Dec 20 17:32:22 LEDE123 dnsmasq-dhcp[2051]: DHCPOFFER(br-lan) 192.169.178.19 b8:27:eb:2e:71:5b
Dec 20 17:32:27 LEDE123 dnsmasq-dhcp[2051]: DHCPDISCOVER(br-lan) b8:27:eb:2e:71:5b
Dec 20 17:32:27 LEDE123 dnsmasq-dhcp[2051]: DHCPOFFER(br-lan) 192.169.178.19 b8:27:eb:2e:71:5b
Dec 20 17:32:32 LEDE123 dnsmasq-dhcp[2051]: DHCPDISCOVER(br-lan) b8:27:eb:2e:71:5b
Dec 20 17:32:32 LEDE123 dnsmasq-dhcp[2051]: DHCPOFFER(br-lan) 192.169.178.19 b8:27:eb:2e:71:5b
Dec 20 17:33:37 LEDE123 dnsmasq-dhcp[2051]: DHCPDISCOVER(br-lan) b8:27:eb:2e:71:5b
Dec 20 17:33:37 LEDE123 dnsmasq-dhcp[2051]: DHCPOFFER(br-lan) 192.169.178.19 b8:27:eb:2e:71:5b

daloki83, 2017-12-21:
Next device:
VU+ Wireless USB Adapter https://www.amazon.de/VU-Wireless-Adapter-incl-Setup/dp/B00AFX847S
Nothing in the log files...Ping to the device (from router) is possible. DHCP Lease time renews as expected. But not possible to access the device. After reboot (the USB device) everything works fine.

yuhhaurlin, 2017-12-21:
Your case is very strange. After the issue for new firmware is fixed, you can try to run it. There is no need for you to put more devices here.

daloki83, 2017-12-21:
Okay, i'll wait. Just one note: I was trying to provoke the failure and it seems to be that it occurs much faster/more often when you set the DHCP Lease Time to "2m".
Maybe this will help.

yuhhaurlin, 2018-01-02:
@daloki83 Please check issue #243 to get LEDE image or modify driver to dump DHCP packets on mwlwifi to help debug. Thanks.

yuhhaurlin, 2018-01-02:
@daloki83 Please put your test result and discussion to issue #243.

daloki83, 2018-01-02:
OK, if you tell me how to modify the driver ?

Btw. setting DHCP Lease to 24h + cron reboot every 24h everything is fine. (Just as a workaround for the moment)

yuhhaurlin, 2018-01-02:
Check issue #243.

daloki83, 2018-01-02:
Hmm yep....i still have no idea how to patch the driver...please explain/provide and .ipkg or anything else...otherwise i can't support you. Sorry, i'm not a dev/programmer

jbsky, 2018-01-02:
Hi,

With 8864, I also have the problem.

daloki83, 2018-01-03:
@yuhhaurlin could you share the patched .ipk file ? I never implemented patches and my tries to compile yours doesn't look like it has worked. Still at the same driver (without the "-debug" tag). Where can i find the dump ?

yuhhaurlin, 2018-01-03:
@daloki83 You need to modify "#define PCIE_DRV_VERSION "10.3.4.0-20171214"" of dev.h to change driver version. It will dump to console or you can get it via logread. If you still have problem,  you can load the LEDE image I posted.

eduperez, 2018-01-04:
@daloki83 You can download updated .ipk files for recent LEDE releases from here:
https://github.com/eduperez/mwlwifi_LEDE/releases

ad019, 2018-01-05:
@yuhhaurlin I have captured traffic from the printer. Sending it across to you on you marvell email address.

ad019, 2018-01-05:
@yuhhaurlin HP have released a new driver and firmware. It looks to have solved the issue. I did not have printer not found problem for 2-3 hours that I tested. The change log shows following:

**Fix and enhancements:**
Improvements for WIFI robustness and reliability
Firmware release fixes Industry-wide Wi-Fi WPA2 Standard Vulnerabilities and Execution of arbitrary code or Denial of Service (CVE 2017-9765)
Addresses an intermittent connectivity issue when using Secure AirPrint
Miscellaneous bug fixes

Thank you for your time and effort on this.

closed 2018-01-05 by yuhhaurlin