kaloz/mwlwifi #247: WDS: ARP traffic being dropped

eduperez, 2017-12-19:
Main router: WRT3200ACM running LEDE 17.01.4 plus updated (commit 843d00c) mwlwifi drivers.
Wireless extender: WDR4300 running LEDE 17.01.4

Main router configured as WDS-AP, wireless extender configured as WDS-STA.
Wireless and wired clients connected to both devices.

Wired and wireless clients connected to wireless extender can reach main router and navigate internet.
Main router can reach wireless extender and clients connected to wireless extender.

Wired and wireless clients connected to main router cannot reach wireless extender or wired clients connected to wireless extender, ARP tables show "incomplete". However, after wireless extender pings wired clients connected to main router, then wired clients connected to main router can reach wireless extender, and ARP tables show correct IP address.

Rebooting devices does not seem to fix the issue.

This setup worked when wireless extender was configured using relayd instead of WDS.

yuhhaurlin, 2018-01-18:
Please help to check https://github.com/kaloz/mwlwifi/commit/91709d67f6b21551ed36a06ce12ca8ca1c92f806. I think it fixes this issue.

yuhhaurlin, 2018-01-18:
I close this one first. If you still have problem, please let me know.

closed 2018-01-18 by yuhhaurlin

eduperez, 2018-01-18:
I will test this tonight and report back. But remember that I noticed this
issue when the WRT3200ACM was running as WDS-AP. The patch seems to be
related to WDS-STA only. Thanks!

yuhhaurlin, 2018-01-18:
No. The patch is for WDS AP. WDS STA mode does not have this issue.

Skilmatix, 2018-01-18:
How do I test this? Reinstall 9.3..2.4?

yuhhaurlin, 2018-01-18:
I put a version to mwlwifi.

eduperez, 2018-01-18:
@yuhhaurlin So far, WDS-STA seems to be working perfectly, the I problem reported no longer happens, and traffic flows in both directions. Many thanks!

@Skilmatix If you are using LEDE 17.01.4, you can download the updated package that I have tested here from https://github.com/eduperez/mwlwifi_LEDE/releases/tag/a149180