kaloz/mwlwifi #248: WRT3200ACM not connected mode -> no signal strength

Beaudetf, 2017-12-22:
Hi,

I'm tring to dump the signal level of all not connected clients and AP around my WRT3200ACM, but I'm only able to have the signal level when the client is connected. 

Do I missed something ? Is it something planned ?

I just installed : 10.3.4.0-20171214-843d00c from https://github.com/eduperez/mwlwifi_LEDE/releases/tag/843d00c

thanks for your great job !

aaron1959, 2017-12-22:
I've never heard of that before. As far as I know there is only signal strength reported for connected clients.
For any router.

Beaudetf, 2017-12-22:
Hi,

I have another router with Qualcomm atheros chipset and no problem to get client and AP RSSI while they are not connected. I use kismet-server to have a pcapdump.

thanks

yuhhaurlin, 2017-12-22:
Sorry. It is impossible to get receive signal of a client if client is not connected.

closed 2017-12-22 by yuhhaurlin

yuhhaurlin, 2017-12-22:
I think what you need is wireless sniffer. You can install WireShark to Macbook Pro (3x3) Or Macbook Air (2x2) to capture wireless packets. You can check signal of received packet to know signal of the sending station.

Beaudetf, 2017-12-22:
Exactly, I want to use many routeurs as wireless sniffers.

I first try with the linksys WRT3200ACM and with kismet to have a pcapdump. Now you say it's impossible to get the signal strength, I'll need to find another router. Have you any suggestion ?

thanks for your support

Chadster766, 2017-12-22:
With McDebian on the WRT3200ACM you can use the mlan (3rd radio) for this application.

Beaudetf, 2017-12-22:
Thanks, yes, the third radio work because it use another driver (mwifiex), but this way I don't have any info on 2,4 GHz.

Chadster766, 2017-12-22:
The 3rd radio is dual band AFAIK

Beaudetf, 2017-12-22:
Really ! Let me check that, thanks !

BrainSlayer, 2017-12-22:
in monitor mode you might see clients if they are active scanning or connected to another ap but the signal will always be inaccurate since beacons are transmitted with different power settings than standard packets. so what you're requesting is monitor mode and nothing else

Beaudetf, 2017-12-22:
Exactly, and it seem that the driver mwifiex on the third radio of WRT3200 doesnt support monitor mode. and mwlwifi in monitor mode doesnt give the signal of the beacons (not connected device)

thanks for you support.

Chadster766, 2017-12-22:
I see, the 3rd radio can only be used to scan for APs then?

```
root@MCDEBIAN:~# iw dev mlan0 scan|grep -i SSID
        SSID: McWifi-24_Ext
        SSID: McWifi-24
        Extended capabilities: SSID List, 6
        SSID: McWifi_AC
        Extended capabilities: SSID List, 6
        SSID: McWifi_AC_Ext
```

yuhhaurlin, 2017-12-22:
It needs to enhance monitor mode and handle slow data with promiscuous mode. I have no plan to do that.