kaloz/mwlwifi #250: 802.11w

schm0, 2017-12-28:
Hi!

Is the 802.11w extension supposed to work properly?
Can you have a look at this thread please
https://forum.lede-project.org/t/wifi-problems-client-connects-inconsistently/9944/14
thank you.

yuhhaurlin, 2018-01-02:
802.11w should work. I close this one first.

yuhhaurlin, 2018-01-02:
I will test it later to make sure it works.

yuhhaurlin, 2018-01-02:
It looks like even though you select wpad (full), the build configuration does not support 802.11w. I just complete the build and I will test it later. However, you can check if you build in 802.11w for hostapd first.

schm0, 2018-01-02:
A happy new year =)
Thank you for looking into this.

I think my build has 802.11w support build in cause my interfaces come up fine?
Why is the rsn capabilities field all zero in the beacons but not in the 3/4 msg of the 4 way handshake?

yuhhaurlin, 2018-01-02:
These things are done by hostapd.

schm0, 2018-01-02:
Doesnt hostapd query the driver for some things?

yuhhaurlin, 2018-01-02:
No. Driver only needs to take care of key setting. I check the code, it should work.

yuhhaurlin, 2018-01-02:
However, I will test it.

yuhhaurlin, 2018-01-03:
RSN IE of beacon and probe response does not include "group management cipher suite". I will check and fix it. Thanks for your information.

yuhhaurlin, 2018-01-03:
RSN capability is all right in my test case, but it lacks "group management cipher suite". I will complete the test and update fix once if the problem is fixed.

schm0, 2018-01-03:
Thank you again ;)

yuhhaurlin, 2018-01-04:
It looks like if your use default "group management cipher suite", hostapd won't append this to RSN IE. I will test and make sure all issues are gone, but RSN IE is correct.

schm0, 2018-01-04:
So this is an hostapd issue?
Is there an easy way to fix?

yuhhaurlin, 2018-01-04:
It should be all right for Hostapd to omit announcement of default group management cipher suite. I still check it.

yuhhaurlin, 2018-01-05:
1. You need to make sure your client supports MFP. If it supports MFP, there is no problem for the connection to WRT3200ACM.
2. However, the data path has problem. I will try to fix it.

Station  (on wlan0)
        rx bytes:       6466
        rx packets:     67
        tx bytes:       20975
        tx packets:     196
        tx retries:     0
        tx failed:      0
        rx drop misc:   5
        signal:         -43 dBm
        signal avg:     -39 dBm
        tx bitrate:     6.0 MBit/s
        rx bitrate:     300.0 MBit/s VHT-MCS 7 40MHz short GI VHT-NSS 2
        authorized:     yes
        authenticated:  yes
        associated:     yes
        preamble:       long
        WMM/WME:        yes
        MFP:            yes
        TDLS peer:      no
        DTIM period:    2
        beacon interval:100
        short slot time:yes
        connected time: 17 seconds

schm0, 2018-01-05:
I have a wrt1200ac.

I have to devices i can test with.
One is a samsung s5 with lineage os.

In optional mode the behavior is the same as in the thread i linked.
The device will not connect because rsn compatibility field is different.
However after the 802.11w timeout timer is reached the device connects.
In required mode no connection at all. 

My other device is an wireless network cam.
In required mode the device doesn't connect at all.
In optional mode the device connects immediately. 
Somehow the fallback here is more faster.

Did you manage to fix the blank rsn ccompatibility field?

yuhhaurlin, 2018-01-05:
I have no blank run capability problem. It will reflect required and optional mode.

yuhhaurlin, 2018-01-05:
I only have data path problem. I think 8864 should be all right. After I fix data path problem of 8964, I will try 8864.

yuhhaurlin, 2018-01-05:
BTW, group management cipher suite and MFP capability of RSN IE capability field is controlled by hostapd. If your configuration is correct, hostapd will add related bits to RSN IE capability.

schm0, 2018-01-05:
yuhhaurlin sorry that i wasted your time.
it seems like group_mgmt_cipher has to be set to something.
with the help from jow on the lede/openwrt irc chat i added one line of code (/lib/netifd/hostapd.sh) to set the group_mgmt_cipher to the "default" AES-128-CMAC.
it works now.

thank you very much for your afford.

yuhhaurlin, 2018-01-05:
All right. It looks like 8864 does not have data path problem. I close this issue. I will continue to fix data path problem of 8964.

closed 2018-01-05 by yuhhaurlin

schm0, 2018-03-04:
With 802.11w enabled.
Arp resolution stops working on my clients.
The Broadcast are not reaching the clients.
Only works for a brief moment after boot.
No im not sure if this a bug from the router or from the clients.