kaloz/mwlwifi #252: LEDE /rom/lib/firmware/mwlwifi/88W8964.bin problem

p3x-robot, 2018-01-04:
Hey!

How are you?

Don't you think, that the router actually is using the ```/rom/lib/firmware/mwlwifi/88W8964.bin``` firmware for the wifi?

What is werid is that I have overriden the **LEDE** ```package/kernel/mwlwifi/Makefile```, still in the rom, there is a different wifi firmware, why is that?

I see it is there in ```/lib/firmware/mwlwifi/```, but on booting, it uses the wifi firmware from the rom, not the storage. Don't you think?
  
I have is like overriden like here https://github.com/patrikx3/lede-insomnia/blob/master/patches/source/package/kernel/mwlwifi/Makefile

How come the ```rom``` and the storage is different firmware ```.bin``` file?

p3x-robot, 2018-01-04:
```text
root@digi:/lib/firmware/mwlwifi# ll /rom/lib/firmware/mwlwifi/ && ll /lib/firmware/mwlwifi/
drwxr-xr-x    2 root     root            99 Dec  6 09:38 .
drwxr-xr-x    3 root     root            30 Dec  6 09:38 ..
-rwxr-xr-x    1 root     root        118776 Oct 18 10:54 88W8864.bin
-rwxr-xr-x    1 root     root        489932 Oct 18 10:54 88W8897.bin
-rwxr-xr-x    1 root     root        449316 Oct 18 10:54 88W8964.bin
-rw-r--r--    1 root     root          2140 Oct 18 10:54 Marvell_license.txt
drwxr-xr-x    1 root     root          4096 Dec 14 20:48 .
drwxr-xr-x    1 root     root          4096 Oct 18 10:54 ..
-rwxr-xr-x    1 root     root        118776 Oct 18 10:54 88W8864.bin
-rwxr-xr-x    1 root     root        489932 Oct 18 10:54 88W8897.bin
-rwxr-xr-x    1 root     root        449420 Oct 18 10:54 88W8964.bin
-rw-r--r--    1 root     root          2140 Oct 18 10:54 Marvell_license.txt
```
The ```88W8964.bin``` is different.

anomeome, 2018-01-04:
You appear to loading a ipk post build image. The one from the forum, i presume, so this would be expected.

closed 2018-01-04 by yuhhaurlin

p3x-robot, 2018-01-04:
Hey!
How are you?
Nope, as I said I overridden the package ```package/kernel/mwlwifi/Makefile```, and I create a LEDE image with the image builder. In the LEDE there is only one mwlwifi Makefile and that is correct and the ipk is correct and is installed with the kmod as well when I install the router firmware, still the rom and the storage is different, where is the different is coming??? Don't you know?

Thanks,
Patrik

p3x-robot, 2018-01-04:
What is weird is that is not the wifi firmware from LEDE 17.01.4, still from the rom and the storage is different, or is it because the rom and the storage format (ext4 vs rom, that is why is different?)

Chadster766, 2018-01-04:
@p3x-robot if you want to know for sure which Marvell firmware is loaded run the below command:
```
cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info
```

p3x-robot, 2018-01-05:
ciao, thanks, it works, now I see now, take care!