kaloz/mwlwifi #253: 3200acm-eu very bad 2.4ghz Wifi performance and coverage

Kherby, 2018-01-09:
Im using LEDE/OpenWRT with my WRT3200acm-EU.
My problem is that the 2.4ghz performance and coverage is "very bad".
Even my old Asus RT-N16 outperforms this device...
I've tried david's build (r5398 and the latest, r5621).
At the moment im running a build of hnyman with the lastest mwlwifi drivers (lede-r5645-7aa15953e1-20171230-1916).
Btw, Official Linksys Firmware does the same bad job in terms in performance and coverage/range.

For example i'm able to download with 35-40mbps with my N16 where the 3200acm is only able to archieve 12-15mbps. (~10-12m away from the router, same placement)
And in my garden i do get a decent signal to watch videos on the internet (13-15mbps) with my RT-N16 and the WRT3200 wont even connect or will loose the signal right after connecting.
One time i was able to hold a connection and the download speed was 1-3mbps !
I get -54 dBm signal quality when i’m right next to the router (0.5m).
**Could this be a sign of a faulty Wifi Module ?**

```
root@OpenWrt:~# iw phy1 info
Wiphy phy1
        max # scan SSIDs: 4
        max scan IEs length: 2242 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports T-DLS.
        Available Antennas: TX 0 RX 0
        Supported interface modes:
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
        Band 1:
                Capabilities: 0x186f
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX HT20 SGI
                        RX HT40 SGI
                        No RX STBC
                        Max AMSDU length: 7935 bytes
                        DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 4 usec (0x05)
                HT TX/RX MCS rate indexes supported: 0-23, 32
                VHT Capabilities (0x33837976):
                        Max MPDU length: 11454
                        Supported Channel Width: 160 MHz
                        RX LDPC
                        short GI (80 MHz)
                        short GI (160/80+80 MHz)
                        SU Beamformer
                        SU Beamformee
                        RX antenna pattern consistency
                        TX antenna pattern consistency
                VHT RX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: MCS 0-9
                        4 streams: not supported
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT RX highest supported: 0 Mbps
                VHT TX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: MCS 0-9
                        4 streams: not supported
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT TX highest supported: 0 Mbps
                Frequencies:
                        * 2412 MHz [1] (20.0 dBm)
                        * 2417 MHz [2] (20.0 dBm)
                        * 2422 MHz [3] (20.0 dBm)
                        * 2427 MHz [4] (20.0 dBm)
                        * 2432 MHz [5] (20.0 dBm)
                        * 2437 MHz [6] (20.0 dBm)
                        * 2442 MHz [7] (20.0 dBm)
                        * 2447 MHz [8] (20.0 dBm)
                        * 2452 MHz [9] (20.0 dBm)
                        * 2457 MHz [10] (20.0 dBm)
                        * 2462 MHz [11] (20.0 dBm)
                        * 2467 MHz [12] (20.0 dBm)
                        * 2472 MHz [13] (20.0 dBm)
                        * 2484 MHz [14] (disabled)
        valid interface combinations:
                 * #{ AP } <= 16, #{ managed } <= 1,
                   total <= 16, #channels <= 1, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz, 160 MHz }

        HT Capability overrides:
                 * MCS: ff ff ff ff ff ff ff ff ff ff
                 * maximum A-MSDU length
                 * supported channel width
                 * short GI for 40 MHz
                 * max A-MPDU length exponent
                 * min MPDU start spacing
```

```
root@OpenWrt:~# cat /sys/kernel/debug/ieee80211/phy1/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20171214
firmware version: 0x09030202
power table loaded from dts: no
firmware region code: 0x30
mac address: 60:38:e0:*
2g: enable
5g: disable
antenna: 4 4
irq number: 46
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e1480000
iobase1: e1700000
tx limit: 1024
rx limit: 16384
```

Should i return my WRT3200acm/ ask for a replacement ?
Or is this routers 2.4 coverage/performance really that bad ?
As i said, my 8 years old Asus N16 does by far a better job in terms of 2.4ghz.
But i didn't bought the 3200acm with the intension to run another wifi router just to get a useable signal in my garden...
At the moment im really dissapointed with the Marvel Wifi and im kinda dissapointed with Linksys/Belkin.
This isn't a 30$ router... i paid 200eur for this unit !

ajmgh, 2018-01-09:
From what I could tell this router's 2.4GHz is just particularly bad. I also have the WRT3200ACM-EU model and the performance is really bad on stock and mwlwifi. Maybe the power tables are wrong or just particularly low. All but one of my devices support 5GHz ac so it's not enough for me to replace it.

Kherby, 2018-01-09:
Well the problem is that the 5ghz range/coverage is very limited compared to 2.4ghz.
Im not able to recieve a 5ghz signal in my garden and in front of my house.
So i really need a decent 2.4ghz signal.
Imho it's such a shame that my 8 years old RT-N16 outperforms this device !

To be honest i kinda regret the purchase of this device... it's really not a great "wifi" router at this time.

yuhhaurlin, 2018-01-10:
The driver and firmware has passed range vs rate test. I can't check your environment.

closed 2018-01-10 by yuhhaurlin

yuhhaurlin, 2018-01-10:
[image]

yuhhaurlin, 2018-01-10:
[image]

yuhhaurlin, 2018-01-10:
In real world, hostapd will check if 2.4g can run as 40 MHz. Normally it will roll back to 20 MHz even though you set 40 MHz.

Kherby, 2018-01-10:
> The driver and firmware have passed range vs rate test. I can't check your environment.

Thanks for your reply but you dont need to check my enviorement. 
As i said my other 8 years old broadcom based router is running fine @2,4ghz and the coverage is way better.
I dont think the bad coverage and performance is directly related to the firmware or driver, it's more the Wifi module itself. There are just issues with the WRT3200acm, which has this Marvel Wifi chipset...

Could you please explain to me why my Asus RT-N16 does a way better job with coverage and performance @2.4ghz ? 
I've tested this with several devices and with several chipsets (Qualcom/Atheros, Intel and broadcom).
I'm not a expert but in my opinion this is a proof that my Marvel wifi module is **1. very weak in general** or **2. faulty**... If it's related to the driver or firmware i would say you did a bad job @2.4ghz.

Or is there anything i'm missing here ???

thagabe, 2018-01-10:
The hardware itself is worse. Asus rt-n56u > WRT3200ACM on 2.4GHz but 5GHz is much better than many other 5GHz

yuhhaurlin, 2018-01-10:
From RvR test, I can't find problem of 2.4g. I can't guarantee it won't have problem with specific clients. Due to you have same problem with stock firmware, please check with Belkin.

yuhhaurlin, 2018-01-10:
BTW, maybe you can try 5g band.

Kherby, 2018-01-10:
@yuhhaurlin
Yes i do have the same coverage/performance problems with the stock firmware.
And it's really sad that the Hardware is so bad @2.4ghz. I need a decent 2.4ghz signal for my garden and another room. 5ghz wont help me here.
By the way the general 5ghz performance isn't so great either with the lastest driver... I dont have any reboot problems so far but network access stopped a few times (had to reconnect the Wifi) and DFS isn't working with the build that im currently using.

Im still thinking about a replacement but if 2.4ghz with the WRT3200acm is really that bad in general i need a complete new router.
I dont want to run two router at the same time to be able to get a decent 2.4ghz signal.
Do you think Linksys official High-Gain Antennas could help me with my coverage/performance problem ?

yuhhaurlin, 2018-01-10:
1. If your client is 11ac, 5g band can run to get good throughput (running as 80 MHz). And there should be less noise on 5g band.
2. What kind of stop problem? It won't work when you run application or it needs to reconnect after your client idle for a while?
3. DFS should be all right.
4. About High-Gain Antenna, you need to check with Belkin.

yuhhaurlin, 2018-01-10:
For DFS, you need to wait for 60 seconds to make sure CAC doen't detect radar singal.
If you use "iw dev" and it will display SSID of the wireless interface, it means the interface can work now.

Kherby, 2018-01-10:
> What kind of stop problem? It won't work when you run application or it needs to reconnect after your client idle for a while?

It happend while i was using the web gui... but it happend another time when i was surfing normaly on the internet (not sure if the GUI was open at this time).
> For DFS, you need to wait for 60 seconds to make sure CAC doen't detect radar singal.

I think it's related to the build that i'm currently using.
DFS has worked decent with david's build that i was using before but with david's build i had other problems with memory leaking and reboots.
I will change to another build later on and check everything again but i guess it wont help me with my 2.4ghz problem.
Hopefully future driver/firmware from you will improve 2.4ghz coverage/signal. #fingers crossed
And i really hope that my Wifi module isn't damaged, i have no idea how i could check it...

yuhhaurlin, 2018-01-10:
1. Normally client reconnects to AP pretty quickly. If there are traffic running between AP and client, AP won't disconnect client.
2. If RvR test has problem, we will check the problem. But just like what I posted, RvR test result is good.

yuhhaurlin, 2018-01-10:
Memory problem is fixed long time ago.

s-pimenta, 2018-01-10:
@Kherby I also noticed the 2.4GHz is quite weak (but almost the same compared to other cheaper routers)... but I think this can be solved (for me) to put some High Gain antennae, since I live in a apartment, and have High Gain antennae will spread more the signal horizontally.

**Regarding your problem**: to make shure the WRT3200ACM is worse... Put the router on the **same spot** as the Asus, put the antenna in the **same orientation**, and use the **same 2.4GHz channel**.... This way you can really compare...

The WRT3200ACM has the advantage to swap antennae and the Asus don't...

The WRT3200ACM "should" be better in receiving... since a chipset with more Spatial Streams are more sensitive to ear signals...

rs-se, 2018-01-10:
@Kherby <https://github.com/kherby> I got the high gain antennas and for a
single level apt it works great for 2.4g but also stabilizes performance of
5g at farther distances for me. Belkin reccomends these antennas for only
single level homes but in the instructions it states to leave antennas
straight up for single level home and for multi-level it recommends the
antennas at a different angle.

Kherby, 2018-01-10:
>  Put the router on the same spot as the Asus, put the antenna in the same orientation, and use the same 2.4GHz channel.... This way you can really compare...

I did that, exactly as you discribed. Same channel, same place, same Antenna postion. Only 1 Wifi enabled at the time.
WRT3200acm is not only worse in signal quality/coverage it's also worse in performance/troughput.
Ping never goes under 2ms (not complaining much about this) where the Asus will be 1ms 99% of the time.
With my old Asus N16 i can watch Netflix HD at my garden (Tablet, Laptop, Galaxy S7) and also in front of my house, where the WRT3200acm (if the 3200acm does connect at all) only gets 1-3mbps with just 1 signal bar at my garden (Asus= 13-15mbps with 2-3 signal bars) and no signal in front of my house.
Everything is first floor (garden, bed room), Antennas straight up.
Can you recommend any programs to test and log performance/coverage ?
I was about to buy these High-Gain Antennas but to be honest im very dissapointed with Linksys/Belkin and i dont want to give them more money atm.
I allready paid 200eur for this device and the High-Gain Antennas would cost me another 60eur...

> The WRT3200ACM has the advantage to swap antennae and the Asus don't...
> The WRT3200ACM "should" be better in receiving... since a chipset with more Spatial Streams are more sensitive to ear signals...

It's also possible to swap Antennas with my RT-N16 but there was no need for me to do so.
I always had a sufficent wifi singal were i needed it, which is not the case with my new WRT3200acm....
Tomorrow i will get a old RT-N66u of my friend, i will also test it against the WRT3200acm @2.4ghz.
It's kinda driving me crazy ;)

s-pimenta, 2018-01-12:
@Kherby if the Asus can detach the antennas and swap with the WRT3200ACM and see if the Asus have signal better or worst... You can try this...

It's good to make this test to be reported/logged the problem... 

Wifi Signal analyser you can use "WiFiAnalyser" from Android Play Store... can give suggestion for the best channel/less crowded...

You don't need to pay 60 bucks for antennae... you can go to Ebay much much cheapper... those Linksys antennae are overpriced...