kaloz/mwlwifi #254: 5GHz wifi stops working (cmd 0x9125=BAStream timed out)

uk3, 2018-01-14:
Hello,
I switched to LEDE 17.01.4 yesterday (form some 18moths old openwrt snapshot) on my Linksys WRT1200AC. 

After a while (it seems under heavy load) the 5GHz wifi stops working. First hint in the logs is:
```
[  848.364358] ieee80211 phy0: cmd 0x9125=BAStream timed out
[  848.369794] ieee80211 phy0: return code: 0x1125
[  848.374343] ieee80211 phy0: timeout: 0x1125
```
I tried the stock 17.01.4 LEDE packages as well as the ones provided here: https://github.com/eduperez/mwlwifi_LEDE/releases/tag/0e38a82
Same results.

Any Ideas on how to debug this? (It did not happen on my old OpenWRT Snapshot).
Cheers, Ulrich

pmcduff, 2018-01-14:
Try these builds:
[https://davidc502sis.dynamic-dns.net/](url)

yuhhaurlin, 2018-01-15:
Please give version information first. If the problem still exists with the latest version of driver, please let me know your test environment and setting, the way to reproduce the problem.

uk3, 2018-01-15:
Update: Today it was the 2GHz that stopped to function.
Currently running:
```
~# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.4.0-20171214
firmware version: 0x0702091a
power table loaded from dts: yes
firmware region code: 0x0
```

yuhhaurlin, 2018-01-15:
Please let me know your setting and the way to reproduce it.

uk3, 2018-01-16:
It was again on the 2.4 GHz interface tonight.
I don't know what it might have triggered. What was running was a timemachine backup from my mac to my synology over that specific wlan (which I think was the case at most of the occurances).

First message in dmesg:
```
[20272.070215] ieee80211 phy1: cmd 0x9125=BAStream timed out
[20272.075677] ieee80211 phy1: return code: 0x1125
[20272.080256] ieee80211 phy1: timeout: 0x1125
```

```
~# cat /sys/kernel/debug/ieee80211/phy1/mwlwifi/info

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.4.0-20171214
firmware version: 0x0702091a
power table loaded from dts: yes
firmware region code: 0x0
mac address: 00:25:9c:13:c9:86
2g: enable
5g: disable
antenna: 2 2
irq number: 105
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e1280000
iobase1: e1400000
tx limit: 768
rx limit: 64
qe trigger number: 1012448
```

Config:
```
wireless.radio1=wifi-device
wireless.radio1.type='mac80211'
wireless.radio1.hwmode='11g'
wireless.radio1.path='soc/soc:pcie-controller/pci0000:00/0000:00:02.0/0000:02:00.0'
wireless.radio1.htmode='HT40'
wireless.radio1.txpower='20'
wireless.radio1.channel='2'
wireless.radio1.country='00'
wireless.@wifi-iface[1]=wifi-iface
wireless.@wifi-iface[1].device='radio1'
wireless.@wifi-iface[1].network='lan'
wireless.@wifi-iface[1].mode='ap'
wireless.@wifi-iface[1].ssid='<removed>'
wireless.@wifi-iface[1].encryption='psk2+ccmp'
wireless.@wifi-iface[1].macaddr='c2:56:27:b8:d8:32'
wireless.@wifi-iface[1].key=<removed>
```

yuhhaurlin, 2018-01-16:
WRT1200 is old device. Can you assign county code first before test? If the problem still happened, I will prepare a LEDE image for you to do test and check what I can do. Thanks.

uk3, 2018-01-16:
Hello,
I removed the CC to test if it had any influence. Before that the Country-Code 'AT' was assigned. Changing that did not make any difference.

yuhhaurlin, 2018-01-16:
All right. I will build LEDE image for you to do test (maybe next week).

uk3, 2018-01-16:
I can check the driver/firmware combination of the openwrt snapshot i had been running before (~18 Months old) if it helps. The problem never occurred there.

yuhhaurlin, 2018-01-16:
I have no idea now. However, if you can give me more information, it will help me to check the problem. Thanks.

stratosnn, 2018-01-17:
Hey,
Same issue with stable LEDE 17.01.04 on WRT1900ACS with "stable" and latest versions of driver. I am happy to test with debug image if needed!

```
[  578.515796] ieee80211 phy1: cmd 0x9125=BAStream timed out
[  578.521230] ieee80211 phy1: return code: 0x1125
[  578.525794] ieee80211 phy1: timeout: 0x1125
[  614.473363] ieee80211 phy1: cmd 0x9122=UpdateEncryption timed out
[  614.479531] ieee80211 phy1: return code: 0x1122
[  614.484095] ieee80211 phy1: timeout: 0x1122
[  614.488308] wlan1: failed to remove key (0, 34:51:c9:f1:8a:eb) from hardware (-5)

Linux version 4.4.92

# cat /sys/kernel/debug/ieee80211/phy1/mwlwifi/info

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.4.0-20170606
firmware version: 0x0702091a
power table loaded from dts: yes
firmware region code: 0x0
mac address: 00:25:9c:13:b7:c5
2g: enable
5g: disable
antenna: 4 4
irq number: 105
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000003
radio: enable
iobase0: e1300000
iobase1: e1480000
tx limit: 768
rx limit: 64
qe trigger number: 33963

```

uk3, 2018-01-17:
Hi,
It happened again with phy1 (2.4Ghz).
No obvious trigger and nothing in the logs at the time.
But after that, hostapd seems to be confused about the interfaces, because it now seems to think, that both are 5Ghz or something:

```
Wed Jan 17 19:32:04 2018 daemon.notice hostapd: wlan0: DFS-RADAR-DETECTED freq=5600 ht_enabled=0 chan_offset=0 chan_width=3 cf1=5610 cf2=0
Wed Jan 17 19:32:04 2018 daemon.notice hostapd: wlan0: DFS-NEW-CHANNEL freq=5180 chan=36 sec_chan=1
Wed Jan 17 19:32:04 2018 daemon.notice hostapd: wlan1: DFS-RADAR-DETECTED freq=5600 ht_enabled=0 chan_offset=0 chan_width=3 cf1=5610 cf2=0
```
wlan1 should not have or show a freq=5600 ....

And now the router is still in the broken state - if I should check for any debug or other output.
Cheers, Ulrich.

uk3, 2018-01-18:
Hello,
I booted the old release now - where I never had the problem. Driver is the following:
```
driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.0.17-20160324
firmware version: 0x07020902
mac address: 00:25:9c:13:c9:86
2g: enable
5g: disable
antenna: 2 2
irq number: 102
iobase0: e1600000
iobase1: e1800000
tx limit: 768
rx limit: 64
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
qe trigger number: 4347
mfg mode: false
```
Package in the old version is:
```
kmod-mwlwifi - 4.1.23+10.3.0.17-20160324-1
```

But there is also a different firmware in the old version:
```
cf822344477440a9cf89c64e8e8a36fa  88W8864.bin
 ```
In the current version it is
```
b87cd3c1deb5204b69fad2ba6152c5d4  88W8864.bin
```

The other thing I noticed: Appearently 802.11ac (5GHz) does not work with my MacBook with the new driver. The old one gives me 
```
526.6 Mbit/s, 80MHz, VHT-MCS 6, VHT-NSS 2
526.6 Mbit/s, 80MHz, VHT-MCS 6, VHT-NSS 2
```
Whereas on the new one I only get 54Mbit/s

yuhhaurlin, 2018-01-18:
Please test LEDE image with latest driver:

Factory: https://drive.google.com/open?id=1oamYj4na6bGx1EitFrZM1PX-lqYGCn9e
Upgrade: https://drive.google.com/open?id=1DuRzk41r5W9HQ8Su6OFBSRaF8TivMvsD

uk3, 2018-01-18:
Ok, the ac-Problem seems to be related to different Results in Radar detection and arbitrary channel changes ...

uk3, 2018-01-18:
I am currently testing with the newest driver but the older firmware - to check if the problem maybe lies within the FW.
After that I'll try the image you provided.
Cheers, Ulrich.

stratosnn, 2018-01-19:
@yuhhaurlin Would you mind building for WRT1900ACS? I can help with testing and collecting debug info.

yuhhaurlin, 2018-01-19:
All right, I will build it next week. Thanks.

uk3, 2018-01-20:
Just a quick update:
I have been running the old firmware (cf822344477440a9cf89c64e8e8a36fa  88W8864.bin) for 48h now without a Problem (which is far longer than anything else I tried before).
So is it possible that the issue is Firmware-related and not driver-related?
Cheers, Ulrich

yuhhaurlin, 2018-01-20:
Can you try the image I posted? Thanks.

yuhhaurlin, 2018-01-22:
WRT1900ACS:

Factory: https://drive.google.com/open?id=195bicILpJs-WzkdZNOc_sxG1CSeqavgS
Upgrade: https://drive.google.com/open?id=1CloGZ9nSxk_EBNchdhWsollA98HsTDof

yuhhaurlin, 2018-01-22:
What is the test result for LEDE image? If you can reproduce the problem, I will send out debug driver to log firmware commands. Thanks.

yuhhaurlin, 2018-01-25:
Please let me know your test result.

uk3, 2018-01-25:
Hello,

I'll test this weekend - did not come around to do it earlier.
But: I am still running the old Firmware for the wifi system and had no
issue for about a week now.
So I'd say the problem is related to something that changed in the
Firmware.....

Cheers, Ulrich.

yuhhaurlin, 2018-01-25:
Please reproduce it on the LEDE image I posted, so I can try to log firmware commands and check how to debug this problem. Thanks.

stratosnn, 2018-01-26:
Hi,

I could not reproduce the error with the LEDE image you have provided for wrt1900acs. I wonder if it has something to do with the kernel version or particular set of kernel modules loaded.

yuhhaurlin, 2018-01-26:
I build the LEDE image on 20171023. LEDE and OpenWRT are  merged. Maybe I can build the updated OpenWrt image and check the result. I think you can give me your configuration file.

yuhhaurlin, 2018-02-21:
Build OpenWrt (commit 9595a9e kernel: fix return value of usb_xhci_needs_pci_reset, got on 2018/2/21):

WRT1900ACS:
Factory: https://drive.google.com/open?id=1FQPoQ-q-N_BFY1aIHca-fo2nv8vCBIaQ
Upgrade: https://drive.google.com/open?id=1oGTawggwVejNLnFrw6X7eUMrlaYvEiQa

I disable PM, CPU FREQ and CPU IDLE, you can check /proc/config.gz to confirm that.
Please help to test this image to check if you can still encounter this problem. If you encounter this problem, please let me know how to reproduce it.

thagabe, 2018-02-21:
@yuhhaurlin Can you post your diffconfig? I want to go and try to disable PM, CPU FREQ and CPU IDLE. Especially now that the next kernel is queued up and I would like to debug 4.14 before it is the default (especially after that nasty bug that affected mamba on 4.9)

yuhhaurlin, 2018-02-22:
@thagabe Here you are: https://drive.google.com/open?id=1NQkG79MrdNeR4X6tmErtJAgTMkTSVyAd.

yuhhaurlin, 2018-02-26:
I close this issue first. If you still encounter this issue for updated OpenWrt version, please let me know which commit of OpenWrt is used by you. Thanks.

closed 2018-02-26 by yuhhaurlin