kaloz/mwlwifi #256: 9.3.2.4 is buggy and hangs in wds sta mode

BrainSlayer, 2018-01-16:
same issue like with the test firmware. just sometimes (1 out of 20 boots) it does not hang. firmware is buggy or is not 100% compatible with mwlwifi

yuhhaurlin, 2018-01-16:
Yes. I will check it and previous WDS bugs.

yuhhaurlin, 2018-01-17:
@BrainSlayer I just try 9.3.2.4 with latest driver for wds sta mode, it works well. I reboot my WRT3200ACM five times and it can connect to WDS AP without problem.

The LEDE image I used:
Factory: https://drive.google.com/open?id=16FljmKJVelx4H3Nis3ESFEwnc6zYOqe6
Upgrade: https://drive.google.com/open?id=1Py3lo5TQB31i6aDlQeYWlOlhYlYRsRjO

root@LEDE:/# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20180117
firmware version: 0x09030204
power table loaded from dts: no
firmware region code: 0x10
mac address: 
2g: disable
5g: enable
antenna: 4 4
irq number: 45
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00010000
radio: enable
iobase0: e0f00000
iobase1: e1180000
tx limit: 1024
rx limit: 16384

yuhhaurlin, 2018-01-17:
BTW, 10.3.4.0-20180117 is the latest commit, but I change the version number. It should be the same as previous 10.3.4.0-20171214 (Adding dhcp dump functions and debugfs file coredump will not change behavior of driver).

I will check other WDS problems first. Please let me know if you still have problem. Thanks.

BrainSlayer, 2018-01-17:
i will try to to get you more infos about the sporadic hanging problem to prevent coincidences

yuhhaurlin, 2018-01-17:
I test here. Basically WDS AP and STA work well for 9.3.2.4. However ARP problem is still there, I will try to fix it first, then I will check group rekey problem.
From my side, 9.3.2.4 works as previous firmware for WDS.
If you can let me know how to reproduce this issue, I will check it. Thanks.

jacksonatl, 2018-01-17:
cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20171214
firmware version: 0x09030204
power table loaded from dts: no
firmware region code: 0x10
mac address: 60:38:e0:b9:c6:c2
2g: disable
5g: enable
antenna: 4 4
irq number: 105
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e0e80000
iobase1: e1100000
tx limit: 1024
rx limit: 16384

This is what I get when enabling wifi:

Wed Jan 17 12:07:36 2018 daemon.notice netifd: radio0 (10870): command failed: Not supported (-95)
Wed Jan 17 12:07:36 2018 daemon.err hostapd: Configuration file: /var/run/hostapd-phy0.conf
Wed Jan 17 12:07:36 2018 daemon.err hostapd: HT (IEEE 802.11n) with WPA/WPA2 requires CCMP/GCMP to be enabled, disabling HT capabilities
Wed Jan 17 12:07:36 2018 daemon.err hostapd: VHT (IEEE 802.11ac) with WPA/WPA2 requires CCMP/GCMP to be enabled, disabling VHT capabilities
Wed Jan 17 12:07:37 2018 kern.debug kernel: [  186.526804] ieee80211 phy0: change: 0xffffffff
Wed Jan 17 12:07:37 2018 kern.info kernel: [  186.636082] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Wed Jan 17 12:07:37 2018 kern.info kernel: [  186.644665] device wlan0 entered promiscuous mode
Wed Jan 17 12:07:37 2018 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Wed Jan 17 12:07:37 2018 daemon.notice hostapd: ACS: Automatic channel selection started, this may take a bit
Wed Jan 17 12:07:37 2018 kern.debug kernel: [  186.653061] ieee80211 phy0: change: 0x100
Wed Jan 17 12:07:37 2018 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->ACS
Wed Jan 17 12:07:37 2018 daemon.notice hostapd: wlan0: ACS-STARTED 
Wed Jan 17 12:07:37 2018 kern.debug kernel: [  186.666137] ieee80211 phy0: change: 0x40
Wed Jan 17 12:07:37 2018 kern.debug kernel: [  186.869891] ieee80211 phy0: change: 0x40
Wed Jan 17 12:07:37 2018 kern.debug kernel: [  187.079800] ieee80211 phy0: change: 0x40
Wed Jan 17 12:07:37 2018 kern.debug kernel: [  187.289792] ieee80211 phy0: change: 0x40
Wed Jan 17 12:07:38 2018 kern.debug kernel: [  187.499800] ieee80211 phy0: change: 0x40
Wed Jan 17 12:07:38 2018 kern.debug kernel: [  187.709795] ieee80211 phy0: change: 0x40
Wed Jan 17 12:07:38 2018 kern.debug kernel: [  187.919797] ieee80211 phy0: change: 0x40
Wed Jan 17 12:07:38 2018 kern.debug kernel: [  188.129797] ieee80211 phy0: change: 0x40
Wed Jan 17 12:07:38 2018 kern.debug kernel: [  188.329797] ieee80211 phy0: change: 0x40
Wed Jan 17 12:07:39 2018 kern.debug kernel: [  188.529798] ieee80211 phy0: change: 0x40
Wed Jan 17 12:07:39 2018 kern.debug kernel: [  188.739799] ieee80211 phy0: change: 0x40
Wed Jan 17 12:07:39 2018 kern.debug kernel: [  188.949802] ieee80211 phy0: change: 0x40
Wed Jan 17 12:07:39 2018 kern.debug kernel: [  189.159802] ieee80211 phy0: change: 0x40
Wed Jan 17 12:07:39 2018 kern.debug kernel: [  189.369802] ieee80211 phy0: change: 0x40
Wed Jan 17 12:07:40 2018 kern.debug kernel: [  189.579802] ieee80211 phy0: change: 0x40
Wed Jan 17 12:07:40 2018 kern.debug kernel: [  189.789805] ieee80211 phy0: change: 0x40
Wed Jan 17 12:07:40 2018 kern.debug kernel: [  189.999808] ieee80211 phy0: change: 0x40
Wed Jan 17 12:07:40 2018 kern.debug kernel: [  190.209806] ieee80211 phy0: change: 0x40
Wed Jan 17 12:07:40 2018 kern.debug kernel: [  190.419806] ieee80211 phy0: change: 0x40
Wed Jan 17 12:07:41 2018 kern.debug kernel: [  190.619813] ieee80211 phy0: change: 0x40
Wed Jan 17 12:07:41 2018 kern.debug kernel: [  190.819810] ieee80211 phy0: change: 0x60
Wed Jan 17 12:07:41 2018 kern.debug kernel: [  191.019810] ieee80211 phy0: change: 0x40
Wed Jan 17 12:07:41 2018 kern.debug kernel: [  191.229810] ieee80211 phy0: change: 0x40
Wed Jan 17 12:07:42 2018 kern.debug kernel: [  191.439852] ieee80211 phy0: change: 0x40
Wed Jan 17 12:07:42 2018 kern.debug kernel: [  191.649860] ieee80211 phy0: change: 0x60
Wed Jan 17 12:07:42 2018 kern.debug kernel: [  191.748829] ieee80211 phy0: change: 0x100
Wed Jan 17 12:07:42 2018 kern.debug kernel: [  191.761883] ieee80211 phy0: change: 0x100
Wed Jan 17 12:07:42 2018 kern.debug kernel: [  191.774830] ieee80211 phy0: change: 0x40
Wed Jan 17 12:07:42 2018 kern.debug kernel: [  191.979825] ieee80211 phy0: change: 0x40
Wed Jan 17 12:07:42 2018 kern.debug kernel: [  192.189822] ieee80211 phy0: change: 0x40
Wed Jan 17 12:07:42 2018 kern.debug kernel: [  192.399817] ieee80211 phy0: change: 0x40
Wed Jan 17 12:07:43 2018 kern.debug kernel: [  192.609822] ieee80211 phy0: change: 0x40
Wed Jan 17 12:07:43 2018 kern.debug kernel: [  192.819820] ieee80211 phy0: change: 0x40
[... 146 lines cut ...]
Wed Jan 17 12:08:01 2018 kern.debug kernel: [  210.659930] ieee80211 phy0: change: 0x40
Wed Jan 17 12:08:01 2018 kern.debug kernel: [  210.869931] ieee80211 phy0: change: 0x40
Wed Jan 17 12:08:01 2018 kern.debug kernel: [  211.079930] ieee80211 phy0: change: 0x40
Wed Jan 17 12:08:01 2018 kern.debug kernel: [  211.279930] ieee80211 phy0: change: 0x60
Wed Jan 17 12:08:02 2018 kern.debug kernel: [  211.479948] ieee80211 phy0: change: 0x40
Wed Jan 17 12:08:02 2018 kern.debug kernel: [  211.689954] ieee80211 phy0: change: 0x40
Wed Jan 17 12:08:02 2018 kern.debug kernel: [  211.899967] ieee80211 phy0: change: 0x40
Wed Jan 17 12:08:02 2018 daemon.notice hostapd: wlan0: ACS-COMPLETED freq=5580 channel=116
Wed Jan 17 12:08:02 2018 daemon.notice hostapd: wlan0: interface state ACS->HT_SCAN
[... cut ...]
Wed Jan 17 12:08:03 2018 kern.debug kernel: [  212.759979] ieee80211 phy0: change: 0x100
Wed Jan 17 12:08:03 2018 kern.debug kernel: [  212.773016] ieee80211 phy0: change: 0x42
Wed Jan 17 12:08:03 2018 kern.debug kernel: [  212.873955] ieee80211 phy0: change: 0x40
Wed Jan 17 12:08:11 2018 daemon.notice hostapd: handle_probe_req: send failed
Wed Jan 17 12:08:11 2018 daemon.notice hostapd: handle_probe_req: send failed
Wed Jan 17 12:08:21 2018 daemon.notice hostapd: handle_probe_req: send failed
Wed Jan 17 12:08:21 2018 daemon.notice hostapd: handle_probe_req: send failed

yuhhaurlin, 2018-01-17:
Please use CCMP only.

yuhhaurlin, 2018-01-18:
I test WDS STA mode for a period of time. It works well. I resolve ARP problem. Currently only group rekey has problem (no matter for WDS AP or WDS STA mode). However, if WRT3200ACM works as STA mode, it will reconnect to WDS AP after failure of group rekey.
I close this one first. If you still encounter this issue, please let me know and show me the way to reproduce it.

closed 2018-01-18 by yuhhaurlin

yuhhaurlin, 2018-01-18:
@BrainSlayer Did you still have problem with firmware 9.3.2.4? I want to put a version to mwlwifi and let user check if there is still connection issue.

yuhhaurlin, 2018-01-18:
@BrainSlayer I put a version to mwlwifi, please let me know if dd-WRT has problems with this version of code. Thanks.

aaron1959, 2018-01-18:
I can test new one in dd-wrt when available.

yuhhaurlin, 2018-01-18:
@BrainSlayer Please check issue #247, WDS STA mode is all right and ARP problem for WDS AP is gone.

yuhhaurlin, 2018-01-18:
BTW, problem of @jacksonatl  is gone due to setting of ieee80211w. From current test result, 10.3.4.0-20180118 + Firmware 9.3.2.4 only has re-connection problem with Andriod phone  (I check what kind of device now). If you still have WDS  STA problem with 9.3.2.4, please let me know.

BrainSlayer, 2018-01-19:
will check as soon as i can. sorry was busy these days

yuhhaurlin, 2018-01-19:
@BrainSlayer It is all right. Thanks for your help.