kaloz/mwlwifi #257: WRT3200ACM wifi keeps dropping

lewisdiamond, 2018-01-17:
I keep losing wifi connection, that's all I see in `dmesg`.

```
[173487.033291] ieee80211 phy1: staid 4 deleted
[174150.410226] ieee80211 phy1: staid 3 deleted
[174195.626504] ieee80211 phy1: staid 7 deleted
[174250.819774] ieee80211 phy1: staid 5 deleted
[174317.199712] ieee80211 phy0: staid 4 deleted
[174353.421092] ieee80211 phy1: staid 1 deleted
[174378.379691] ieee80211 phy1: staid 3 deleted
[174487.052141] ieee80211 phy1: staid 4 deleted
[174523.205569] ieee80211 phy1: staid 3 deleted
[174556.895044] ieee80211 phy1: staid 3 deleted
[174639.068726] ieee80211 phy0: staid 4 deleted
[174648.070156] ieee80211 phy1: staid 4 deleted
[174664.629318] ieee80211 phy1: staid 5 deleted
[174689.847066] ieee80211 phy1: staid 4 deleted
[174696.644816] ieee80211 phy0: staid 4 deleted
[174707.593696] ieee80211 phy1: staid 1 deleted
[174734.503244] ieee80211 phy0: staid 4 deleted
[174745.444223] ieee80211 phy1: staid 1 deleted
[174791.204436] ieee80211 phy1: staid 1 deleted
[174831.479830] ieee80211 phy1: staid 1 deleted
[174858.802282] ieee80211 phy1: staid 5 deleted
[175009.138036] ieee80211 phy1: staid 1 deleted
[175016.537921] ieee80211 phy1: staid 4 deleted
[175029.362322] ieee80211 phy1: staid 1 deleted
[175039.792566] ieee80211 phy1: staid 1 deleted
[175070.390097] ieee80211 phy1: staid 3 deleted
[175472.112884] ieee80211 phy1: staid 4 deleted
[175526.551060] ieee80211 phy0: staid 3 deleted

```

yuhhaurlin, 2018-01-17:
Please test with latest driver and firmware. Please give more detail information. I close this one first.

closed 2018-01-17 by yuhhaurlin

lewisdiamond, 2018-01-17:
@yuhhaurlin Ok I was using the latest as of right before your last commit. I'll try again tonight.

PHLAK, 2018-01-21:
@lewisdiamond How did the latest version work for you? I recently got a WRT3200ACM and have been having the same issue.

lewisdiamond, 2018-01-22:
@PHLAK I don't know yet I'll have to switch back to this device for wifi. I'll update when I have tested.

HZ89, 2019-02-25:
i have same issue but only on 2G interface