kaloz/mwlwifi #258: 2018-01-18 release Still broken

jacksonatl, 2018-01-18:
I changed to CCMP only, not that a functioning driver should crash if TKIP is enabled anyways...

cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20180118
firmware version: 0x09030204
power table loaded from dts: no
firmware region code: 0x10
mac address: 60:38:e0:b9:c6:c2
2g: disable
5g: enable
antenna: 4 4
irq number: 105
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e0e80000
iobase1: e1100000
tx limit: 1024
rx limit: 16384

logread |grep hostapd
Thu Jan 18 09:15:17 2018 daemon.err hostapd: Configuration file: /var/run/hostapd-phy0.conf
Thu Jan 18 09:15:17 2018 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Thu Jan 18 09:15:17 2018 daemon.notice hostapd: ACS: Automatic channel selection started, this may take a bit
Thu Jan 18 09:15:17 2018 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->ACS
Thu Jan 18 09:15:17 2018 daemon.notice hostapd: wlan0: ACS-STARTED 
Thu Jan 18 09:15:18 2018 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Thu Jan 18 09:15:18 2018 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Thu Jan 18 09:15:18 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 09:15:18 2018 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Thu Jan 18 09:15:18 2018 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Thu Jan 18 09:15:18 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 09:20:01 2018 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Thu Jan 18 09:20:01 2018 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Thu Jan 18 09:20:01 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 09:20:01 2018 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Thu Jan 18 09:20:01 2018 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Thu Jan 18 09:20:01 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 09:20:02 2018 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Thu Jan 18 09:20:02 2018 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Thu Jan 18 09:20:02 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 09:20:02 2018 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Thu Jan 18 09:20:02 2018 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Thu Jan 18 09:20:02 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 09:20:02 2018 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Thu Jan 18 09:20:02 2018 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Thu Jan 18 09:20:02 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 09:20:02 2018 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Thu Jan 18 09:20:02 2018 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Thu Jan 18 09:20:02 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 09:20:05 2018 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Thu Jan 18 09:20:05 2018 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Thu Jan 18 09:20:05 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 09:20:11 2018 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Thu Jan 18 09:20:11 2018 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Thu Jan 18 09:20:11 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 09:20:11 2018 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Thu Jan 18 09:20:11 2018 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Thu Jan 18 09:20:11 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 09:20:11 2018 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Thu Jan 18 09:20:11 2018 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Thu Jan 18 09:20:11 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 09:20:11 2018 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Thu Jan 18 09:20:11 2018 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Thu Jan 18 09:20:11 2018 daemon.notice hostapd: handle_probe_req: send failed
[... 42 lines cut ...]
Thu Jan 18 09:20:57 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 09:20:57 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 09:21:01 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 09:21:01 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 09:21:07 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 09:21:07 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 09:21:11 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 09:21:11 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 09:21:17 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 09:21:17 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 09:21:21 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 09:21:21 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 09:21:22 2018 daemon.notice hostapd: wlan0: DFS-CAC-COMPLETED success=1 freq=5260 ht_enabled=0 chan_offset=0 chan_width=3 cf1=5290 cf2=0
Thu Jan 18 09:21:22 2018 daemon.err hostapd: Using interface wlan0 with hwaddr 60:38:e0:b9:c6:c2 and ssid "Pope Alley"
Thu Jan 18 09:21:22 2018 daemon.notice hostapd: wlan0: interface state DFS->ENABLED
Thu Jan 18 09:21:22 2018 daemon.notice hostapd: wlan0: AP-ENABLED 
Thu Jan 18 09:21:29 2018 daemon.info hostapd: wlan0: STA 84:16:f9:09:1a:2c IEEE 802.11: associated (aid 1)
Thu Jan 18 09:21:34 2018 daemon.info hostapd: wlan0: STA 98:de:d0:11:45:43 IEEE 802.11: associated (aid 1)
Thu Jan 18 09:21:34 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 98:de:d0:11:45:43
Thu Jan 18 09:21:34 2018 daemon.info hostapd: wlan0: STA 98:de:d0:11:45:43 RADIUS: starting accounting session 39EAE271556AC817
Thu Jan 18 09:21:34 2018 daemon.info hostapd: wlan0: STA 98:de:d0:11:45:43 WPA: pairwise key handshake completed (RSN)
Thu Jan 18 09:21:39 2018 daemon.info hostapd: wlan0: STA 3c:a9:f4:be:52:9c IEEE 802.11: associated (aid 2)
Thu Jan 18 09:21:39 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 3c:a9:f4:be:52:9c
Thu Jan 18 09:21:39 2018 daemon.info hostapd: wlan0: STA 3c:a9:f4:be:52:9c RADIUS: starting accounting session 0236193D9ECA1EC6
Thu Jan 18 09:21:39 2018 daemon.info hostapd: wlan0: STA 3c:a9:f4:be:52:9c WPA: pairwise key handshake completed (RSN)

yuhhaurlin, 2018-01-18:
1. From your previous log, HT and VHT are disable due to CAMP is not used.
2. Please check READ to resolve DFS problem.
3. From the log, I can't find crash information.

yuhhaurlin, 2018-01-18:
What the problem you have? From the log, client is connected.

jacksonatl, 2018-01-18:
No clients can connect, all get authentication error. I have filtered out the kernel errors but can provide if helpful. 
As mentioned, I have changed to CCMP only for this bug report.
I do not understand your previous comment about DFS problem, this exact setup (except was TKIP Only) was working prior to 1/16/2018 release.

yuhhaurlin, 2018-01-18:
Can you set specific channel first?

yuhhaurlin, 2018-01-18:
Please give me your configuration file. Basically if you use TKIP, HT and VHT can't work.
1. Set specific channel first.
2. Use ACS and use "iw dev" to check final channel    setting.
3. Use "logread | grep hostapd" to check log from hostapd.

jacksonatl, 2018-01-18:
Using CCMP only security. Set to channel 140. Still no clients can connect.

# iw dev
phy#2
	Interface mlan0
		ifindex 8
		wdev 0x200000001
		addr 60:38:e0:b9:c6:c3
		type managed
		txpower 0.00 dBm
phy#0
	Interface wlan0
		ifindex 91
		wdev 0x25
		addr 60:38:e0:b9:c6:c2
		ssid Pope Alley
		type AP
		channel 140 (5700 MHz), width: 80 MHz, center1: 5690 MHz
		txpower 23.00 dBm

Logread after enabling wifi:
Thu Jan 18 10:06:38 2018 daemon.notice netifd: radio0 (7112): command failed: Not supported (-95)
Thu Jan 18 10:06:39 2018 daemon.err hostapd: Configuration file: /var/run/hostapd-phy0.conf
Thu Jan 18 10:06:39 2018 kern.debug kernel: [  116.699599] ieee80211 phy0: change: 0xffffffff
Thu Jan 18 10:06:39 2018 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Thu Jan 18 10:06:39 2018 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->HT_SCAN
Thu Jan 18 10:06:39 2018 kern.info kernel: [  116.807788] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Thu Jan 18 10:06:39 2018 kern.info kernel: [  116.814597] device wlan0 entered promiscuous mode
Thu Jan 18 10:06:39 2018 kern.debug kernel: [  116.819741] ieee80211 phy0: change: 0x100
Thu Jan 18 10:06:39 2018 kern.debug kernel: [  116.832620] ieee80211 phy0: change: 0x40
Thu Jan 18 10:06:39 2018 kern.debug kernel: [  117.032597] ieee80211 phy0: change: 0x40
Thu Jan 18 10:06:39 2018 daemon.notice hostapd: wlan0: interface state HT_SCAN->DFS
Thu Jan 18 10:06:39 2018 daemon.notice hostapd: wlan0: DFS-CAC-START freq=5700 chan=140 sec_chan=1, width=1, seg0=138, seg1=0, cac_time=60s
Thu Jan 18 10:06:39 2018 kern.debug kernel: [  117.232599] ieee80211 phy0: change: 0x40
Thu Jan 18 10:06:39 2018 kern.debug kernel: [  117.328615] ieee80211 phy0: change: 0x100
Thu Jan 18 10:06:39 2018 kern.debug kernel: [  117.341668] ieee80211 phy0: change: 0x100
Thu Jan 18 10:06:39 2018 kern.debug kernel: [  117.354715] ieee80211 phy0: change: 0x42
Thu Jan 18 10:06:40 2018 kern.debug kernel: [  117.456617] ieee80211 phy0: change: 0x40
Thu Jan 18 10:06:43 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:06:43 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:06:43 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:06:43 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:06:53 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:06:53 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:06:53 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:06:53 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:03 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:03 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:03 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:03 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:13 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:13 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:13 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:13 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:23 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:23 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:23 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:23 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:33 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:33 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:33 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:33 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:40 2018 kern.debug kernel: [  177.762923] ieee80211 phy0: change: 0x40
Thu Jan 18 10:07:40 2018 kern.debug kernel: [  177.861924] ieee80211 phy0: change: 0x100
Thu Jan 18 10:07:40 2018 daemon.notice hostapd: wlan0: DFS-CAC-COMPLETED success=1 freq=5700 ht_enabled=0 chan_offset=0 chan_width=3 cf1=5690 cf2=0
Thu Jan 18 10:07:40 2018 daemon.err hostapd: Using interface wlan0 with hwaddr 60:38:e0:b9:c6:c2 and ssid "Pope Alley"
Thu Jan 18 10:07:40 2018 kern.debug kernel: [  177.921587] ieee80211 phy0: change: 0x100
Thu Jan 18 10:07:40 2018 kern.debug kernel: [  177.932917] ieee80211 phy0: change: 0x42
Thu Jan 18 10:07:40 2018 kern.debug kernel: [  178.031924] ieee80211 phy0: change: 0x40
Thu Jan 18 10:07:40 2018 kern.info kernel: [  178.210944] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Thu Jan 18 10:07:40 2018 kern.info kernel: [  178.217385] br-lan: port 2(wlan0) entered forwarding state
Thu Jan 18 10:07:40 2018 kern.info kernel: [  178.222918] br-lan: port 2(wlan0) entered forwarding state
Thu Jan 18 10:07:40 2018 daemon.notice hostapd: wlan0: interface state DFS->ENABLED
Thu Jan 18 10:07:40 2018 daemon.notice hostapd: wlan0: AP-ENABLED 
Thu Jan 18 10:07:40 2018 daemon.notice netifd: Network device 'wlan0' link is up
Thu Jan 18 10:07:42 2018 kern.info kernel: [  180.222917] br-lan: port 2(wlan0) entered forwarding state
Thu Jan 18 10:07:44 2018 daemon.info hostapd: wlan0: STA 84:16:f9:09:1a:2c IEEE 802.11: associated (aid 1)
Thu Jan 18 10:07:44 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 84:16:f9:09:1a:2c
[... cut ...]
Thu Jan 18 10:07:44 2018 daemon.info hostapd: wlan0: STA 84:16:f9:09:1a:2c WPA: pairwise key handshake completed (RSN)
Thu Jan 18 10:07:44 2018 daemon.info hostapd: wlan0: STA 98:de:d0:11:45:43 IEEE 802.11: associated (aid 2)
Thu Jan 18 10:07:44 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 98:de:d0:11:45:43
Thu Jan 18 10:07:44 2018 daemon.info hostapd: wlan0: STA 98:de:d0:11:45:43 RADIUS: starting accounting session AB352D0F4E0D9DA4
Thu Jan 18 10:07:44 2018 daemon.info hostapd: wlan0: STA 98:de:d0:11:45:43 WPA: pairwise key handshake completed (RSN)

yuhhaurlin, 2018-01-18:
Phy 2 is not mwlwifi.

yuhhaurlin, 2018-01-18:
Please check README to remove it.

yuhhaurlin, 2018-01-18:
BTW, from the log, I can't find problem.

jacksonatl, 2018-01-18:
Ok removed mwifiex packages and rebooted. Still no clients can connect. As I mentioned this same router and configuration (other than changing TKIP to CCMP) "worked" prior to 1/16/2018 release. Performance was poor which is why I upgrade to each release you push. 

Thu Jan 18 10:06:38 2018 daemon.notice netifd: radio0 (7112): command failed: Not supported (-95)
Thu Jan 18 10:06:39 2018 daemon.err hostapd: Configuration file: /var/run/hostapd-phy0.conf
Thu Jan 18 10:06:39 2018 kern.debug kernel: [  116.699599] ieee80211 phy0: change: 0xffffffff
Thu Jan 18 10:06:39 2018 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Thu Jan 18 10:06:39 2018 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->HT_SCAN
Thu Jan 18 10:06:39 2018 kern.info kernel: [  116.807788] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Thu Jan 18 10:06:39 2018 kern.info kernel: [  116.814597] device wlan0 entered promiscuous mode
Thu Jan 18 10:06:39 2018 kern.debug kernel: [  116.819741] ieee80211 phy0: change: 0x100
Thu Jan 18 10:06:39 2018 kern.debug kernel: [  116.832620] ieee80211 phy0: change: 0x40
Thu Jan 18 10:06:39 2018 kern.debug kernel: [  117.032597] ieee80211 phy0: change: 0x40
Thu Jan 18 10:06:39 2018 daemon.notice hostapd: wlan0: interface state HT_SCAN->DFS
Thu Jan 18 10:06:39 2018 daemon.notice hostapd: wlan0: DFS-CAC-START freq=5700 chan=140 sec_chan=1, width=1, seg0=138, seg1=0, cac_time=60s
Thu Jan 18 10:06:39 2018 kern.debug kernel: [  117.232599] ieee80211 phy0: change: 0x40
Thu Jan 18 10:06:39 2018 kern.debug kernel: [  117.328615] ieee80211 phy0: change: 0x100
Thu Jan 18 10:06:39 2018 kern.debug kernel: [  117.341668] ieee80211 phy0: change: 0x100
Thu Jan 18 10:06:39 2018 kern.debug kernel: [  117.354715] ieee80211 phy0: change: 0x42
Thu Jan 18 10:06:40 2018 kern.debug kernel: [  117.456617] ieee80211 phy0: change: 0x40
Thu Jan 18 10:06:43 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:06:43 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:06:43 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:06:43 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:06:53 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:06:53 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:06:53 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:06:53 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:03 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:03 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:03 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:03 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:13 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:13 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:13 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:13 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:23 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:23 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:23 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:23 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:33 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:33 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:33 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:33 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:07:40 2018 kern.debug kernel: [  177.762923] ieee80211 phy0: change: 0x40
Thu Jan 18 10:07:40 2018 kern.debug kernel: [  177.861924] ieee80211 phy0: change: 0x100
Thu Jan 18 10:07:40 2018 daemon.notice hostapd: wlan0: DFS-CAC-COMPLETED success=1 freq=5700 ht_enabled=0 chan_offset=0 chan_width=3 cf1=5690 cf2=0
Thu Jan 18 10:07:40 2018 daemon.err hostapd: Using interface wlan0 with hwaddr 60:38:e0:b9:c6:c2 and ssid "Pope Alley"
Thu Jan 18 10:07:40 2018 kern.debug kernel: [  177.921587] ieee80211 phy0: change: 0x100
Thu Jan 18 10:07:40 2018 kern.debug kernel: [  177.932917] ieee80211 phy0: change: 0x42
Thu Jan 18 10:07:40 2018 kern.debug kernel: [  178.031924] ieee80211 phy0: change: 0x40
Thu Jan 18 10:07:40 2018 kern.info kernel: [  178.210944] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Thu Jan 18 10:07:40 2018 kern.info kernel: [  178.217385] br-lan: port 2(wlan0) entered forwarding state
Thu Jan 18 10:07:40 2018 kern.info kernel: [  178.222918] br-lan: port 2(wlan0) entered forwarding state
Thu Jan 18 10:07:40 2018 daemon.notice hostapd: wlan0: interface state DFS->ENABLED
Thu Jan 18 10:07:40 2018 daemon.notice hostapd: wlan0: AP-ENABLED 
Thu Jan 18 10:07:40 2018 daemon.notice netifd: Network device 'wlan0' link is up
Thu Jan 18 10:07:42 2018 kern.info kernel: [  180.222917] br-lan: port 2(wlan0) entered forwarding state
Thu Jan 18 10:07:44 2018 daemon.info hostapd: wlan0: STA 84:16:f9:09:1a:2c IEEE 802.11: associated (aid 1)
Thu Jan 18 10:07:44 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 84:16:f9:09:1a:2c
Thu Jan 18 10:07:44 2018 daemon.info hostapd: wlan0: STA 84:16:f9:09:1a:2c RADIUS: starting accounting session E696F7BFD50A6A24
[... cut ...] :f9:09:1a:2c WPA: pairwise key handshake completed (RSN)
Thu Jan 18 10:07:44 2018 daemon.info hostapd: wlan0: STA 98:de:d0:11:45:43 IEEE 802.11: associated (aid 2)
Thu Jan 18 10:07:44 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 98:de:d0:11:45:43
Thu Jan 18 10:07:44 2018 daemon.info hostapd: wlan0: STA 98:de:d0:11:45:43 RADIUS: starting accounting session AB352D0F4E0D9DA4
Thu Jan 18 10:07:44 2018 daemon.info hostapd: wlan0: STA 98:de:d0:11:45:43 WPA: pairwise key handshake completed (RSN)

jacksonatl, 2018-01-18:
Logs of windows laptop trying to connect:
Thu Jan 18 10:21:06 2018 daemon.info dnsmasq-dhcp[5247]: read /etc/ethers - 0 addresses
Thu Jan 18 10:21:12 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:21:12 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:21:15 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:21:15 2018 daemon.notice hostapd: handle_probe_req: send failed
Thu Jan 18 10:21:20 2018 kern.debug kernel: [   76.640799] ieee80211 phy0: change: 0x40
Thu Jan 18 10:21:21 2018 kern.debug kernel: [   76.738811] ieee80211 phy0: change: 0x100
Thu Jan 18 10:21:21 2018 daemon.notice hostapd: wlan0: DFS-CAC-COMPLETED success=1 freq=5700 ht_enabled=0 chan_offset=0 chan_width=3 cf1=5690 cf2=0
Thu Jan 18 10:21:21 2018 daemon.err hostapd: Using interface wlan0 with hwaddr 60:38:e0:b9:c6:c2 and ssid "Pope Alley"
Thu Jan 18 10:21:21 2018 kern.debug kernel: [   76.797592] ieee80211 phy0: change: 0x100
Thu Jan 18 10:21:21 2018 kern.debug kernel: [   76.810643] ieee80211 phy0: change: 0x42
Thu Jan 18 10:21:21 2018 kern.debug kernel: [   76.908802] ieee80211 phy0: change: 0x40
Thu Jan 18 10:21:21 2018 kern.info kernel: [   77.087801] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Thu Jan 18 10:21:21 2018 kern.info kernel: [   77.094248] br-lan: port 2(wlan0) entered forwarding state
Thu Jan 18 10:21:21 2018 kern.info kernel: [   77.099776] br-lan: port 2(wlan0) entered forwarding state
Thu Jan 18 10:21:21 2018 daemon.notice hostapd: wlan0: interface state DFS->ENABLED
Thu Jan 18 10:21:21 2018 daemon.notice hostapd: wlan0: AP-ENABLED 
Thu Jan 18 10:21:21 2018 daemon.notice netifd: Network device 'wlan0' link is up
Thu Jan 18 10:21:23 2018 kern.info kernel: [   79.090741] br-lan: port 2(wlan0) entered forwarding state
Thu Jan 18 10:21:23 2018 daemon.info hostapd: wlan0: STA 98:de:d0:11:45:43 IEEE 802.11: associated (aid 1)
Thu Jan 18 10:21:23 2018 kern.debug kernel: [   79.276462] ieee80211 phy0: staid 1 deleted
Thu Jan 18 10:21:27 2018 daemon.info hostapd: wlan0: STA 84:16:f9:09:1a:2c IEEE 802.11: associated (aid 1)
Thu Jan 18 10:21:27 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 84:16:f9:09:1a:2c
Thu Jan 18 10:21:27 2018 daemon.info hostapd: wlan0: STA 84:16:f9:09:1a:2c RADIUS: starting accounting session 6F11FC5551F9BCCE
Thu Jan 18 10:21:27 2018 daemon.info hostapd: wlan0: STA 84:16:f9:09:1a:2c WPA: pairwise key handshake completed (RSN)
Thu Jan 18 10:30:15 2018 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 84:16:f9:09:1a:2c
Thu Jan 18 10:30:15 2018 daemon.info hostapd: wlan0: STA 84:16:f9:09:1a:2c IEEE 802.11: disassociated due to inactivity
Thu Jan 18 10:30:19 2018 kern.err kernel: [  614.920253] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Thu Jan 18 10:30:19 2018 kern.err kernel: [  614.926386] ieee80211 phy0: return code: 0x1122
Thu Jan 18 10:30:19 2018 kern.err kernel: [  614.930942] ieee80211 phy0: timeout: 0x1122
Thu Jan 18 10:30:19 2018 kern.err kernel: [  614.935143] wlan0: failed to remove key (0, 84:16:f9:09:1a:2c) from hardware (-5)
Thu Jan 18 10:30:19 2018 daemon.info hostapd: wlan0: STA 84:16:f9:09:1a:2c IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Thu Jan 18 10:30:23 2018 kern.err kernel: [  618.940789] ieee80211 phy0: cmd 0x9125=BAStream timed out
Thu Jan 18 10:30:23 2018 kern.err kernel: [  618.946214] ieee80211 phy0: return code: 0x1125
Thu Jan 18 10:30:23 2018 kern.err kernel: [  618.950761] ieee80211 phy0: timeout: 0x1125
Thu Jan 18 10:30:27 2018 kern.err kernel: [  622.954732] ieee80211 phy0: cmd 0x9125=BAStream timed out
Thu Jan 18 10:30:27 2018 kern.err kernel: [  622.960156] ieee80211 phy0: return code: 0x1125
Thu Jan 18 10:30:27 2018 kern.err kernel: [  622.964714] ieee80211 phy0: timeout: 0x1125
Thu Jan 18 10:30:31 2018 kern.err kernel: [  626.978046] ieee80211 phy0: cmd 0x9111=SetNewStation timed out
Thu Jan 18 10:30:31 2018 kern.err kernel: [  626.983918] ieee80211 phy0: return code: 0x1111
Thu Jan 18 10:30:31 2018 kern.err kernel: [  626.988465] ieee80211 phy0: timeout: 0x1111
Thu Jan 18 10:30:35 2018 kern.err kernel: [  630.992612] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Thu Jan 18 10:30:35 2018 kern.err kernel: [  630.998472] ieee80211 phy0: return code: 0x001d
Thu Jan 18 10:30:35 2018 kern.err kernel: [  631.003038] ieee80211 phy0: timeout: 0x001d
Thu Jan 18 10:30:39 2018 kern.err kernel: [  634.993631] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Thu Jan 18 10:30:39 2018 kern.err kernel: [  634.999491] ieee80211 phy0: return code: 0x001d
Thu Jan 18 10:30:39 2018 kern.err kernel: [  635.004050] ieee80211 phy0: timeout: 0x001d
Thu Jan 18 10:30:43 2018 kern.err kernel: [  638.998331] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Thu Jan 18 10:30:43 2018 kern.err kernel: [  639.004199] ieee80211 phy0: return code: 0x001d
Thu Jan 18 10:30:43 2018 kern.err kernel: [  639.008748] ieee80211 phy0: timeout: 0x001d
Thu Jan 18 10:30:47 2018 kern.err kernel: [  643.022434] ieee80211 phy0: cmd 0x8050=broadcast_ssid_enable timed out
Thu Jan 18 10:30:47 2018 kern.err kernel: [  643.028992] ieee80211 phy0: return code: 0x0050
Thu Jan 18 10:30:47 2018 kern.err kernel: [  643.033560] ieee80211 phy0: timeout: 0x0050
Thu Jan 18 10:30:51 2018 kern.err kernel: [  647.040369] ieee80211 phy0: cmd 0x8127=SetInformationElements timed out
Thu Jan 18 10:30:51 2018 kern.err kernel: [  647.047014] ieee80211 phy0: return code: 0x0127
[... cut ...]
Thu Jan 18 10:31:39 2018 kern.err kernel: [  695.191577] ieee80211 phy0: timeout: 0x001d
Thu Jan 18 10:31:43 2018 kern.err kernel: [  699.179598] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Thu Jan 18 10:31:43 2018 kern.err kernel: [  699.185720] ieee80211 phy0: return code: 0x1122
Thu Jan 18 10:31:43 2018 kern.err kernel: [  699.190279] ieee80211 phy0: timeout: 0x1122
Thu Jan 18 10:31:43 2018 kern.err kernel: [  699.194480] wlan0: failed to set key (5, ff:ff:ff:ff:ff:ff) to hardware (-5)

yuhhaurlin, 2018-01-18:
1. Performance is not good previously due to you use TKIP.
2. Can you give me your setting (/etc/config/wireless)?

yuhhaurlin, 2018-01-18:
BTW, I will give you my LEDE image I used when I come to office tomorrow.

jacksonatl, 2018-01-18:
Wireless config:

config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11a'
	option path 'soc/soc:pcie-controller/pci0000:00/0000:00:01.0/0000:01:00.0'
	option htmode 'VHT80'
	option channel '140'
	option country 'US'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid <removed>
	option encryption 'psk2+ccmp'
	option key <removed>
	option ieee80211w '1'

config wifi-device 'radio1'
	option type 'mac80211'
	option channel '11'
	option hwmode '11g'
	option path 'soc/soc:pcie-controller/pci0000:00/0000:00:02.0/0000:02:00.0'
	option htmode 'HT20'
	option disabled '1'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'LEDE'
	option encryption 'none'

yuhhaurlin, 2018-01-18:
Please disable ieee802.11w.

yuhhaurlin, 2018-01-18:
You did not set SSID and key.

jacksonatl, 2018-01-18:
I replaced them with <removed> when posting, however when I saved thru web UI config changed slightly. Still no clients can connect

config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11a'
	option path 'soc/soc:pcie-controller/pci0000:00/0000:00:01.0/0000:01:00.0'
	option htmode 'VHT80'
	option country 'US'
	option channel 'auto'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid <removed>
	option encryption 'psk2+ccmp'
	option key <removed>
	option ieee80211w '1'

config wifi-device 'radio1'
	option type 'mac80211'
	option channel '11'
	option hwmode '11g'
	option path 'soc/soc:pcie-controller/pci0000:00/0000:00:02.0/0000:02:00.0'
	option htmode 'HT20'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'LEDE'
	option encryption 'none'
	option disabled '1'

yuhhaurlin, 2018-01-18:
Disable ieee80211w.

jacksonatl, 2018-01-18:
Appears github removes the formatting changes made to config file by UI and removes "<removed>"

jacksonatl, 2018-01-18:
it is:
iw dev
phy#0
	Interface wlan0
		ifindex 54
		wdev 0x17
		addr 60:38:e0:b9:c6:c2
		ssid Pope Alley
		type AP
		channel 116 (5580 MHz), width: 80 MHz, center1: 5610 MHz
		txpower 23.00 dBm

jacksonatl, 2018-01-18:
Ok, that fixed it and the CCMP does seem to have improved performance. Thank you. When will  ieee80211w be working again?

yuhhaurlin, 2018-01-18:
Don't set it unless you know what it does. Data path for this feature on 8964 still has problem.
I close this issue.

closed 2018-01-18 by yuhhaurlin