kaloz/mwlwifi #259: vfs_write dropped in linux kernel 4.14.14

cilix-lab, 2018-01-18:
I stumbled upon this issue when trying to update to current stable linux kernel 4.14.14, since mwlwifi wouldn't work, leaving this on my syslog:
```
Jan 18 11:34:40 Thor kernel: [    9.008723] mwlwifi: Unknown symbol vfs_write (err 0)
Jan 18 11:34:40 Thor kernel: [    9.053198] mwlwifi: Unknown symbol vfs_write (err 0)
Jan 18 11:43:47 Thor kernel: [    8.867520] mwlwifi: Unknown symbol vfs_write (err 0)
Jan 18 11:43:47 Thor kernel: [    8.891863] mwlwifi: Unknown symbol vfs_write (err 0)
```
After some brief research, I found that vfs_read/vfs_write is being dropped for kernel_read/kernel_write, so I fixed it with this patch to debugfs.c: [vfs_write.txt](https://github.com/kaloz/mwlwifi/files/1643440/vfs_write.txt)

yuhhaurlin, 2018-01-18:
Thanks. I will check and fix it.

cilix-lab, 2018-01-18:
It only happened on kernel 4.14.14. Until 4.14.13 it worked fine.

yuhhaurlin, 2018-01-18:
Understood. Thanks.

ValCher1961, 2018-01-20:
I have warnings and error when I build the driver module
```
root@wrt3200acm:/home/kernel/linux-4.14.14# make ARCH=arm CROSS_COMPILE=arm-none-eabi- -j3 modules
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  CHK     scripts/mod/devicetable-offsets.h
  CHK     include/generated/timeconst.h
  CHK     include/generated/bounds.h
  CHK     include/generated/asm-offsets.h
  CALL    scripts/checksyscalls.sh
  CC [M]  drivers/net/wireless/marvell/mwlwifi/core.o
  CC [M]  drivers/net/wireless/marvell/mwlwifi/mac80211.o
drivers/net/wireless/marvell/mwlwifi/mac80211.c: In function 'mwl_mac80211_sta_add':
drivers/net/wireless/marvell/mwlwifi/mac80211.c:488:23: warning: 'sta_stnid' may be used uninitialized in this function [-Wmaybe-uninitialized]
   sta_info->sta_stnid = sta_stnid;
                       ^
  CC [M]  drivers/net/wireless/marvell/mwlwifi/utils.o
  CC [M]  drivers/net/wireless/marvell/mwlwifi/hif/fwcmd.o
  CC [M]  drivers/net/wireless/marvell/mwlwifi/hif/pcie/pcie.o
  CC [M]  drivers/net/wireless/marvell/mwlwifi/hif/pcie/fwdl.o
  CC [M]  drivers/net/wireless/marvell/mwlwifi/hif/pcie/tx.o
  CC [M]  drivers/net/wireless/marvell/mwlwifi/hif/pcie/rx.o
  CC [M]  drivers/net/wireless/marvell/mwlwifi/hif/pcie/tx_ndp.o
  CC [M]  drivers/net/wireless/marvell/mwlwifi/hif/pcie/rx_ndp.o
  CC [M]  drivers/net/wireless/marvell/mwlwifi/thermal.o
  CC [M]  drivers/net/wireless/marvell/mwlwifi/debugfs.o
drivers/net/wireless/marvell/mwlwifi/debugfs.c: In function 'mwl_debugfs_core_dump_read':
drivers/net/wireless/marvell/mwlwifi/debugfs.c:1128:2: warning: 'cd' may be used uninitialized in this function [-Wmaybe-uninitialized]
  kfree(cd);
  ^
  LD [M]  drivers/net/wireless/marvell/mwlwifi/mwlwifi.o
  Building modules, stage 2.
  MODPOST 465 modules
ERROR: "vfs_write" [drivers/net/wireless/marvell/mwlwifi/mwlwifi.ko] undefined!
scripts/Makefile.modpost:92: recipe for target '__modpost' failed
make[1]: *** [__modpost] Error 1
Makefile:1215: recipe for target 'modules' failed
make: *** [modules] Error 2
root@wrt3200acm:/home/kernel/linux-4.14.14#
```

yuhhaurlin, 2018-01-20:
You can mark off vfs_write first.

ValCher1961, 2018-01-21:
Sorry, I don't really understand how to separate mark off Vfs_write first.
The first warning drivers/net/wireless/marvell/mwlwifi/mac80211.c:488:23; - appeared in previous versions of the driver and did not affect anything.
The second warning drivers/net/wireless/marvell/mwlwifi/debugfs.c:1128:2: - appeared in the latest version of the driver and leads to an error.

BrainSlayer, 2018-01-25:
vfs_write is not exported anymore on 4.14 (this i valid for all 4.14 versions)
you may use kernel_write instead.

BrainSlayer, 2018-01-25:
@ValCher1961 yuhhaurlin says you should simply remove the vfs_write calls. they are only for debugging purpose and will not be used for normal operation

yuhhaurlin, 2018-01-25:
Yes. That is the fix posted by cilix-lab.

ValCher1961, 2018-01-25:
@BrainSlayer , @yuhhaurlin 
Yes, thanks, I've got it. )))
@cilix-lab 
Thanks for the patch.

BrainSlayer, 2018-01-26:
consider that the patch wont work for older versions since the kernel_write api has been changed in 4.14. so final patch should take care about kernel versions

yuhhaurlin, 2018-01-27:
Yes. I will check how to do that.

tapper82, 2018-02-07:
Hi yuhhaurlin. Do you have any news about this bug pleas? OpenWRT are hoping to bump up to Kernel 4.14 soon. I don't want this target  to be left behind.

yuhhaurlin, 2018-02-07:
Not yet.

cilix-lab, 2018-02-07:
@tapper82 if you want to compile kernel 14.14 or above, just replace vfs_write with kernel_write doing something like:   
```
sed -i 's/vfs_write/kernel_write/g' debugfs.c
```

lucize, 2018-02-08:
on `<<Marvell Mac80211 Wireless PCIE Network Driver version 10.3.4.0-20171214>>`
 `OpenWrt SNAPSHOT, r6018-9a08c10`

```
 uname -a
Linux home 4.14.18 #0 SMP Thu Feb 8 18:04:38 2018 armv7l GNU/Linux
```

is working as expected, didn't tried latest commit

yuhhaurlin, 2018-03-06:
Fixed by https://github.com/kaloz/mwlwifi/commit/b1b9a9e1c1beee30a8cce4038f4109727362ebe0.

closed 2018-03-06 by yuhhaurlin