kaloz/mwlwifi #260: No Route to Host (LAN)

godlessfather, 2018-01-18:
When I ssh into boxes on my network, I have occasional "No route to host" errors on IPv4 and IPv6. `ifup wan6 && ifup lan` resolves the issue. I apologize if this is addressed elsewhere -- showing my ignorance perhaps.

yuhhaurlin, 2018-01-18:
Please report issue related to mwlwifi here.

closed 2018-01-18 by yuhhaurlin

godlessfather, 2018-01-18:
This issue only affects wireless clients on my network. Another way to resolve this issue is to simply click "Scan" on both wireless radios.

yuhhaurlin, 2018-01-18:
Please let me know your device and version of driver.

godlessfather, 2018-01-19:
3200ACM

`cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info`

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20171214
firmware version: 0x09030204
power table loaded from dts: no
firmware region code: 0x10
mac address: 
2g: disable
5g: enable
antenna: 4 4
irq number: 45
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e1080000
iobase1: e1300000
tx limit: 1024
rx limit: 16384

Current time without an issue is 5h35m with 9.3.2.4.

yuhhaurlin, 2018-01-19:
9.3.2.4 fixes DHCP IP address assignment problem. Maybe your problem is related to issue #243.