kaloz/mwlwifi #261: Bricked router after upgrading mwlwifi drivers

eduperez, 2018-01-18:
There are a couple of users complaining about bricked routers after updating to 09395179b1fe87c56f55257e09c46cd3ff70fa0c, and I have also bricked mine after updating to a149180fecec4fbaafb3f52ae46e33bd66501441 from 5955f5ea40a51f9ed4c6a85b4b94cfece1b4f716. I had to boot on the other partition, and will investigate the issue.

eduperez, 2018-01-18:
I have seen that the "9.3.2.4" firmware that I had downloaded from one of the threads here does not match the "9.3.2.4" firmware that is currently available on this repo (and it does not match any of the other previous firmwares). I will test again with the correct "9.3.2.4" firmware (after I upgrade back to the current LEDE version, reinstall everything, and recompile the packages).

yuhhaurlin, 2018-01-18:
I don't know what do you mean?

eduperez, 2018-01-18:
You published a 9.3.2.4 firmware in one of the issues here (can't remember which one), and I downloaded it; but that file is not the same (MD5 is different) as the 9.3.2.4 firmware that is currently on the repo.

yuhhaurlin, 2018-01-18:
Please use the one I push to repository.

yuhhaurlin, 2018-01-18:
I close this one. Thanks.

closed 2018-01-18 by yuhhaurlin

eduperez, 2018-01-18:
Definitively, I had downloaded a wrong firmware file; after downloading the right one, device is working again. Sorry about wasting your time.