kaloz/mwlwifi #262: Connection problems with Android clients

Alexander-r, 2018-01-18:
On wrt3200acm with 10.3.4.0-20180118 and 9.3.2.4 firmware I have problems with Android clients.
If i use wrt3200acm as a dhcp server and try to connect my phone then it may take around a minute to get an IP address and the client reconnects to the network a couple of times before it gets the address.
If I set up wrt3200acm as a bridge and use another routers dhcp for lan clients then it takes 10-20 seconds for my phone to connect but after that it shows "Connected, no Internet". If it's the first time I connect to the network it asks if I want to stay connected. If I stay connected the network starts to work after some time.
If on the phone I turn wi-fi off and on and it starts reconnecting automatically then after it gets "Connected, no Internet" it disconnects from the network and tries reconnecting to it again. It stays in this connect/disconnect loop until I connect to the network manually again.

In both setups clients have access to internet.

This problem only affects my Android devices. My notebooks connects to the network without any problems.
I did not experience this problem with official 1.0.6.186168 firmware in both when wrt3200acm was a dhcp server and when it was in bridge mode.
All the phones that I tested have at least Android 7 on them.

Alexander-r, 2018-01-18:
Small update. This only happens on 5GHz network. On 2.4GHz network clients connect normally and receive "Connected" status right after connection.

yuhhaurlin, 2018-01-18:
Please let me know the phone you used. Thanks.

yuhhaurlin, 2018-01-19:
1. Please help to confirm firmware 9.3.2.2 is all right.
2. Do you have lot of broadcast/multicast packets on your 5g band?

Thanks.

yuhhaurlin, 2018-01-19:
F.Y.I. http://community.linksys.com/t5/Wireless-Routers/WiFi-Drops-and-Disconnects-with-Android-Devices-on-your-network/td-p/1263827.

Alexander-r, 2018-01-19:
On firmware 9.3.2.2 it takes longer to connect to 5Ghz network then to 2.4GHz but the phone stays connected. After connection I still get "Connected, no Internet" for about 10 seconds before the phone sees internet.
I tested it with Oneplus 2 (Android 8.1.0 LineageOS 15.1) and Sony Xperia X Performance (stock Android 7.0 from Sony)

Almost all of my devices use 5GHz network as 2.4Ghz range is really crowded (I can see about a dozen 2.4GHz networks and only one other 5GHz network).
I've dumped traffic going through wlan0 on the router. For broadcast and multicast I see STP packets from wan router (one packet about every 5 seconds), ARP packets (usually 2 packets at the same time but not too often), ICMPv6 for router advertisement, solicitation and neighbor discovery and DHCPv6 as I have IPv6 set up (2-3 different requests at the same time about once every 10 seconds), SSDP packets (rare but sometimes a lot of packets at once), IGMPv2 (1-2 packets at once, rare), MDNS (5-6 packets at once about every 10 seconds).
MDNS packets are mostly from my Google Nexus Player (Android 8 stock from Google).
In 1209 seconds I got broadcast and multicast 6291 packets captured. I don't think that's a lot.

I know that there is a problem with some google devices. But that's usually affects network when the devices resumes from sleep and sends about 400 MDSN packets at once.
Also in the same environment my backup router does not experience this problems.

yuhhaurlin, 2018-01-19:
Can you install the fixes and check if the problem is gone? Thanks.
BTW, can you set DTIM to 1 and check test result before installation of the fixes?

Alexander-r, 2018-01-19:
Right now I have LEDE r5892-26045049ba with  mwlwifi a149180:
```
[    0.000000] Linux version 4.9.76 (sr@sr-pc) (gcc version 5.5.0 (OpenWrt GCC 5.5.0 r5892-26045049ba) ) #0 SMP Thu Jan 18 10:39:42 2018
...
[    9.494209] <<Marvell Mac80211 Wireless PCIE Network Driver version 10.3.4.0-20180118>>
[    9.502282] pci 0000:00:01.0: enabling device (0140 -> 0142)
[    9.508068] ieee80211 phy0: priv->pcmd_buf = dddfc000  priv->pphys_cmd_buf = 1ddfc000
[    9.548277] ieee80211 phy0: fw download start
[    9.548282] ieee80211 phy0: ddr init: download start
[    9.548660] ieee80211 phy0: ddr init: download complete
[    9.686033] ieee80211 phy0: FwSize = 449468 downloaded Size = 449468 curr_iteration 65522
[   12.557269] ieee80211 phy0: fw download complete
[   12.557308] ieee80211 phy0: device node: /soc/pcie-controller/pcie@1,0
[   12.573730] ieee80211 phy0: pcmd = dddfc000
[   12.590286] ieee80211 phy0: firmware version: 0x9030204
[   12.951271] ieee80211 phy0: firmware region code: 10
[   12.980890] ieee80211 phy0: 2G disabled, 5G enabled
[   12.985806] ieee80211 phy0: 4 TX antennas, 4 RX antennas
```
Or are there any newer fixes?

Do you want me to test changing DTIM  on 9.3.2.4 or on 9.3.2.2?

yuhhaurlin, 2018-01-19:
Tested on 9.3.2.4. I means fixes from Google release on 1/18.

godlessfather, 2018-01-19:
I've got two Chromecasts, two devices with Oreo, and two with Nougat. The
'Connected, no internet' issue happened for my Oreo devices prior to
switching to 9.3.2.4 and hasn't happened in the 14 hours that I've been
running the latest firmware. Previously, my two Oreo phones would take
longer than normal to connect to my router and find internet and would
occasionally lose their wireless connection during sleep. I've just
upgraded Google Play Services to the reported Cast fix.

Alexander-r, 2018-01-19:
Here is station dump for when my phone tries to connect:
```
Station c0:ee:fb:46:7a:24 (on wlan0)
        inactive time:  480 ms
        rx bytes:       2716
        rx packets:     25
        tx bytes:       1045
        tx packets:     13
        tx retries:     0
        tx failed:      0
        rx drop misc:   0
        signal:         0 dBm
        signal avg:     0 dBm
        tx bitrate:     292.6 MBit/s VHT-MCS 6 80MHz short GI VHT-NSS 1
        rx bitrate:     54.0 MBit/s
        authorized:     yes
        authenticated:  yes
        associated:     yes
        preamble:       long
        WMM/WME:        yes
        MFP:            no
        TDLS peer:      no
        DTIM period:    1
        beacon interval:100
        short slot time:yes
        connected time: 3 seconds
```

Here is config for 5GHz network:
```
config wifi-device 'radio0'
        option type 'mac80211'
        option hwmode '11a'
        option path 'soc/soc:pcie-controller/pci0000:00/0000:00:01.0/0000:01:00.0'
        option ht_coex '1'
        option noscan '1'
        option country 'US'
        option legacy_rates '0'
        option htmode 'VHT80'
        option txpower '23'
        option channel '36'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option mode 'ap'
        option ssid 'Test5'
        option encryption 'psk2+ccmp'
        option key '1234567890'
        option doth '0'
        option network 'wan'
        option dtim_period '1'
        option powersave '0'
```

I don't see any differences with DTIM set to 1.

So far Google Nexus Player tells me that there are no updates available.
It's not clear if it should be part of firmware update or just Google Play services update. Google Play services version on the device is 8.5.39.W-xhdpi [8] [PR] 178322352 (I think it was released on January 7, 2018) and it updates automatically.

bg4drl, 2018-01-20:
Google Pixel 2 XL OPM2.171019.016 is out OTA

bg4drl, 2018-01-20:
https://www.apkmirror.com/apk/google-inc/google-play-services/google-play-services-11-9-74-release/

adamcarter1, 2018-01-20:
Just my two cents:

Vizio m65-E0 with built in chromecast seems to always disconnect at random, along the lines of what everyone is saying here. 

I use dd-wrt, using the newest build jan 19th 2018 and had to restart the router to allow connection. 

My firestick (an android device) and apple devices are unaffected by this problem.

I can’t test your lede stuff as i am a loyal dd-wrt user, but hoping this will be fixed soon.

yuhhaurlin, 2018-01-22:
@Alexander-r Can I know which version of firmware is all right under your test environment?

yuhhaurlin, 2018-01-22:
@Alexander-r Please help to verify 10.3.4.0-20180122-debug:

Factory: https://drive.google.com/open?id=14xLOFzs5u-vSwlvFVx-Y5H1AaxLMhVug
Upgrade: https://drive.google.com/open?id=1XuTVj2vpyqt9_N0YtD5PCzbgSLokimJf

Thanks.

yuhhaurlin, 2018-01-22:
@Alexander-r I print out DHCPOFFER Broadcast packet and send it to different queue. Please help to check the LEDE image. Based on your test result, I can check how to debug the problem. Thanks.

Alexander-r, 2018-01-23:
As I mentioned I did not notice this problems on 9.3.2.2 firmware for as long as the client could still connect to the network. And there are no problems with official Linksys 1.0.6.186168 firmware.

I've tested your lede build. Android clients still have problems on it.

Here is the log for Android device:
```
Tue Jan 23 12:37:06 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 IEEE 802.11: associated (aid 3)
Tue Jan 23 12:37:06 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED c0:ee:fb:46:7a:24
Tue Jan 23 12:37:06 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 RADIUS: starting accounting session D091A5DF752B6ED2
Tue Jan 23 12:37:06 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 WPA: pairwise key handshake completed (RSN)
Tue Jan 23 12:37:07 2018 daemon.info dnsmasq-dhcp[2445]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24 
Tue Jan 23 12:37:07 2018 daemon.info dnsmasq-dhcp[2445]: DHCPOFFER(br-lan) 192.168.7.131 c0:ee:fb:46:7a:24 
Tue Jan 23 12:37:08 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 IEEE 802.11: authenticated
Tue Jan 23 12:37:12 2018 daemon.info dnsmasq-dhcp[2445]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24 
Tue Jan 23 12:37:12 2018 daemon.info dnsmasq-dhcp[2445]: DHCPOFFER(br-lan) 192.168.7.131 c0:ee:fb:46:7a:24 
Tue Jan 23 12:37:17 2018 daemon.info dnsmasq-dhcp[2445]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24 
Tue Jan 23 12:37:17 2018 daemon.info dnsmasq-dhcp[2445]: DHCPOFFER(br-lan) 192.168.7.131 c0:ee:fb:46:7a:24 
Tue Jan 23 12:37:24 2018 daemon.info dnsmasq-dhcp[2445]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24 
Tue Jan 23 12:37:24 2018 daemon.info dnsmasq-dhcp[2445]: DHCPOFFER(br-lan) 192.168.7.131 c0:ee:fb:46:7a:24 
Tue Jan 23 12:37:40 2018 daemon.info dnsmasq-dhcp[2445]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24 
Tue Jan 23 12:37:40 2018 daemon.info dnsmasq-dhcp[2445]: DHCPOFFER(br-lan) 192.168.7.131 c0:ee:fb:46:7a:24 
Tue Jan 23 12:37:43 2018 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED c0:ee:fb:46:7a:24
Tue Jan 23 12:37:43 2018 kern.debug kernel: [  393.688616] ieee80211 phy0: staid 3 deleted
Tue Jan 23 12:37:44 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 IEEE 802.11: associated (aid 3)
Tue Jan 23 12:37:44 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED c0:ee:fb:46:7a:24
Tue Jan 23 12:37:44 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 RADIUS: starting accounting session B8F8D03F874FA414
Tue Jan 23 12:37:44 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 WPA: pairwise key handshake completed (RSN)
Tue Jan 23 12:37:44 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 IEEE 802.11: authenticated
Tue Jan 23 12:37:45 2018 daemon.info dnsmasq-dhcp[2445]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24 
Tue Jan 23 12:37:45 2018 daemon.info dnsmasq-dhcp[2445]: DHCPOFFER(br-lan) 192.168.7.131 c0:ee:fb:46:7a:24 
Tue Jan 23 12:37:50 2018 daemon.info dnsmasq-dhcp[2445]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24 
Tue Jan 23 12:37:50 2018 daemon.info dnsmasq-dhcp[2445]: DHCPOFFER(br-lan) 192.168.7.131 c0:ee:fb:46:7a:24 
Tue Jan 23 12:37:55 2018 daemon.info dnsmasq-dhcp[2445]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24 
Tue Jan 23 12:37:55 2018 daemon.info dnsmasq-dhcp[2445]: DHCPOFFER(br-lan) 192.168.7.131 c0:ee:fb:46:7a:24 
Tue Jan 23 12:38:03 2018 daemon.info dnsmasq-dhcp[2445]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24 
Tue Jan 23 12:38:03 2018 daemon.info dnsmasq-dhcp[2445]: DHCPOFFER(br-lan) 192.168.7.131 c0:ee:fb:46:7a:24 
Tue Jan 23 12:38:20 2018 daemon.info dnsmasq-dhcp[2445]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24 
Tue Jan 23 12:38:20 2018 daemon.info dnsmasq-dhcp[2445]: DHCPOFFER(br-lan) 192.168.7.131 c0:ee:fb:46:7a:24 
Tue Jan 23 12:38:20 2018 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED c0:ee:fb:46:7a:24
Tue Jan 23 12:38:20 2018 kern.debug kernel: [  431.598025] ieee80211 phy0: staid 3 deleted
Tue Jan 23 12:38:22 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 IEEE 802.11: associated (aid 3)
Tue Jan 23 12:38:22 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED c0:ee:fb:46:7a:24
Tue Jan 23 12:38:22 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 RADIUS: starting accounting session 3F6D9F7FD4422A88
Tue Jan 23 12:38:22 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 WPA: pairwise key handshake completed (RSN)
Tue Jan 23 12:38:23 2018 daemon.info dnsmasq-dhcp[2445]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24 
Tue Jan 23 12:38:23 2018 daemon.info dnsmasq-dhcp[2445]: DHCPOFFER(br-lan) 192.168.7.131 c0:ee:fb:46:7a:24 
Tue Jan 23 12:38:23 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 IEEE 802.11: authenticated
Tue Jan 23 12:38:28 2018 daemon.info dnsmasq-dhcp[2445]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24 
Tue Jan 23 12:38:28 2018 daemon.info dnsmasq-dhcp[2445]: DHCPOFFER(br-lan) 192.168.7.131 c0:ee:fb:46:7a:24 
Tue Jan 23 12:38:33 2018 daemon.info dnsmasq-dhcp[2445]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24 
Tue Jan 23 12:38:33 2018 daemon.info dnsmasq-dhcp[2445]: DHCPOFFER(br-lan) 192.168.7.131 c0:ee:fb:46:7a:24 
Tue Jan 23 12:38:40 2018 daemon.info dnsmasq-dhcp[2445]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24 
Tue Jan 23 12:38:40 2018 daemon.info dnsmasq-dhcp[2445]: DHCPOFFER(br-lan) 192.168.7.131 c0:ee:fb:46:7a:24 
Tue Jan 23 12:38:40 2018 daemon.info dnsmasq-dhcp[2445]: DHCPREQUEST(br-lan) 192.168.7.131 c0:ee:fb:46:7a:24 
Tue Jan 23 12:38:40 2018 daemon.info dnsmasq-dhcp[2445]: DHCPACK(br-lan) 192.168.7.131 c0:ee:fb:46:7a:24 
```

And here is the log for my notebook:
```
[... cut ...]
Tue Jan 23 12:41:13 2018 daemon.info dnsmasq-dhcp[2445]: DHCPACK(br-lan) 192.168.7.86 64:80:99:98:f7:92 sr-note
Tue Jan 23 12:41:13 2018 kern.warn kernel: [  604.359708] DHCPOFFER MC for 64:80:99:98:f7:92
Tue Jan 23 12:41:13 2018 kern.warn kernel: [  604.364177] DHCPOFFER MC for 64:80:99:98:f7:92
Tue Jan 23 12:41:15 2018 daemon.info hostapd: wlan0: STA 64:80:99:98:f7:92 IEEE 802.11: authenticated
```

Even after android client connects to the network I don't see DHCPOFFER MC message from kernel log for it.

yuhhaurlin, 2018-01-23:
It means your Andriod device does not ask for broadcast DHCPOFFER. Can you get log with firmware 9.3.2.2 for me. Use the LEDE image I posted and replace the firmware to 9.3.2.2.
It is very strange, the modification of 9.3.2.4 is related to broadcast and multicast packets.

yuhhaurlin, 2018-01-23:
BTW, if you use DHCP server of LEDE, you can specify if DHCP server will reply broadcast DHCPOFFER.

vi /etc/config/dhcp 

add:

config host
    option ip 'IP for your Andriod phone'
    option mac 'MAC address of Andriod phone'
    option broadcast '0' or '1'

Please help to check if only unicast will have this problem. Thanks.

yuhhaurlin, 2018-01-23:
BTW, can you let me know the Andriod phone you use?

Alexander-r, 2018-01-24:
With option broadcast '1' set for client the client connects much faster and I don't see duplicate requests. But right after the phone connects it disconnects almost immediately and starts reconnecting.
I've also tested this with 9.3.2.2 firmware and the client also just connected and disconnected. Here is the log:
```
Wed Jan 24 20:51:38 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 IEEE 802.11: associated (aid 4)
Wed Jan 24 20:51:38 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED c0:ee:fb:46:7a:24
Wed Jan 24 20:51:38 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 RADIUS: starting accounting session CD3A53C3E72087F8
Wed Jan 24 20:51:38 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 WPA: pairwise key handshake completed (RSN)
Wed Jan 24 20:51:38 2018 daemon.info dnsmasq-dhcp[2007]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24 
Wed Jan 24 20:51:38 2018 daemon.info dnsmasq-dhcp[2007]: DHCPOFFER(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24 
Wed Jan 24 20:51:38 2018 kern.warn kernel: [  422.496349] DHCPOFFER MC for c0:ee:fb:46:7a:24
Wed Jan 24 20:51:38 2018 kern.warn kernel: [  422.500829] DHCPOFFER MC for c0:ee:fb:46:7a:24
Wed Jan 24 20:51:38 2018 daemon.info dnsmasq-dhcp[2007]: DHCPREQUEST(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24 
Wed Jan 24 20:51:38 2018 daemon.info dnsmasq-dhcp[2007]: DHCPACK(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24 OnePlus2
Wed Jan 24 20:51:38 2018 kern.warn kernel: [  422.571561] DHCPOFFER MC for c0:ee:fb:46:7a:24
Wed Jan 24 20:51:38 2018 kern.warn kernel: [  422.576033] DHCPOFFER MC for c0:ee:fb:46:7a:24
Wed Jan 24 20:51:40 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 IEEE 802.11: authenticated
Wed Jan 24 20:51:42 2018 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED c0:ee:fb:46:7a:24
Wed Jan 24 20:51:42 2018 kern.debug kernel: [  425.983204] ieee80211 phy0: staid 4 deleted
Wed Jan 24 20:51:43 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 IEEE 802.11: associated (aid 4)
Wed Jan 24 20:51:43 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED c0:ee:fb:46:7a:24
Wed Jan 24 20:51:43 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 RADIUS: starting accounting session 85E7569307AACA7C
Wed Jan 24 20:51:43 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 WPA: pairwise key handshake completed (RSN)
Wed Jan 24 20:51:43 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 IEEE 802.11: authenticated
Wed Jan 24 20:51:44 2018 daemon.info dnsmasq-dhcp[2007]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24 
Wed Jan 24 20:51:44 2018 daemon.info dnsmasq-dhcp[2007]: DHCPOFFER(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24 
Wed Jan 24 20:51:44 2018 kern.warn kernel: [  427.889556] DHCPOFFER MC for c0:ee:fb:46:7a:24
Wed Jan 24 20:51:44 2018 kern.warn kernel: [  427.894032] DHCPOFFER MC for c0:ee:fb:46:7a:24
Wed Jan 24 20:51:44 2018 daemon.info dnsmasq-dhcp[2007]: DHCPREQUEST(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24 
Wed Jan 24 20:51:44 2018 daemon.info dnsmasq-dhcp[2007]: DHCPACK(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24 OnePlus2
Wed Jan 24 20:51:44 2018 kern.warn kernel: [  427.995244] DHCPOFFER MC for c0:ee:fb:46:7a:24
Wed Jan 24 20:51:44 2018 kern.warn kernel: [  427.999720] DHCPOFFER MC for c0:ee:fb:46:7a:24
Wed Jan 24 20:51:47 2018 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED c0:ee:fb:46:7a:24
Wed Jan 24 20:51:47 2018 kern.debug kernel: [  431.422121] ieee80211 phy0: staid 4 deleted
```

The client was Oneplus 2 with Android 8.1.0.

yuhhaurlin, 2018-01-24:
Only this client has problem?

yuhhaurlin, 2018-01-24:
BTW, if you did not set broadcast reply, does firmware 9.3.2.2 work all right? You told me firmware 9.3.2.2 is all right for this phone. Can you get the log for unitcast  reply for firmware 9.3.2.2? Thanks.

Alexander-r, 2018-01-25:
Only this client disconnects immediately. Sony Xperia X Performance (Android 7.0) does not disconnect from the network but it takes up to a minute after connection for it to go from "Connected, no Internet" to "Connected" status and be able to access anything.

Oneplus 2 is able to connect if I first select forget network and set it up again and after first connection select stay connected. After that it hangs for about a minute on "Connected, no Internet" before switching to "Connected". Here is a log for that with broadcast '1':
```
Thu Jan 25 17:26:14 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED c0:ee:fb:46:7a:24
Thu Jan 25 17:26:14 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 RADIUS: starting accounting session 7A7F55FF639BCAD5
Thu Jan 25 17:26:14 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 WPA: pairwise key handshake completed (RSN)
Thu Jan 25 17:26:14 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 IEEE 802.11: authenticated
Thu Jan 25 17:26:15 2018 daemon.info dnsmasq-dhcp[2012]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24 
Thu Jan 25 17:26:15 2018 daemon.info dnsmasq-dhcp[2012]: DHCPOFFER(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24 
Thu Jan 25 17:26:15 2018 kern.warn kernel: [  204.340759] DHCPOFFER MC for c0:ee:fb:46:7a:24
Thu Jan 25 17:26:15 2018 kern.warn kernel: [  204.345233] DHCPOFFER MC for c0:ee:fb:46:7a:24
Thu Jan 25 17:26:15 2018 daemon.info dnsmasq-dhcp[2012]: DHCPREQUEST(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24 
Thu Jan 25 17:26:15 2018 daemon.info dnsmasq-dhcp[2012]: DHCPACK(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24 OnePlus2
Thu Jan 25 17:26:15 2018 kern.warn kernel: [  204.381076] DHCPOFFER MC for c0:ee:fb:46:7a:24
Thu Jan 25 17:26:15 2018 kern.warn kernel: [  204.385544] DHCPOFFER MC for c0:ee:fb:46:7a:24
```

And here is the log for unitcast connection:
```
Thu Jan 25 17:16:17 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED c0:ee:fb:46:7a:24
Thu Jan 25 17:16:17 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 RADIUS: starting accounting session 682AC9DBFBA230C9
Thu Jan 25 17:16:17 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 WPA: pairwise key handshake completed (RSN)
Thu Jan 25 17:16:17 2018 daemon.info dnsmasq-dhcp[2013]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24 
Thu Jan 25 17:16:17 2018 daemon.info dnsmasq-dhcp[2013]: DHCPOFFER(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24 
Thu Jan 25 17:16:22 2018 daemon.info dnsmasq-dhcp[2013]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24 
Thu Jan 25 17:16:22 2018 daemon.info dnsmasq-dhcp[2013]: DHCPOFFER(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24 
Thu Jan 25 17:16:27 2018 daemon.info dnsmasq-dhcp[2013]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24 
Thu Jan 25 17:16:27 2018 daemon.info dnsmasq-dhcp[2013]: DHCPOFFER(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24 
Thu Jan 25 17:16:36 2018 daemon.info dnsmasq-dhcp[2013]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24 
Thu Jan 25 17:16:36 2018 daemon.info dnsmasq-dhcp[2013]: DHCPOFFER(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24 
Thu Jan 25 17:16:41 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 IEEE 802.11: authenticated
Thu Jan 25 17:16:51 2018 daemon.info dnsmasq-dhcp[2013]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24 
Thu Jan 25 17:16:51 2018 daemon.info dnsmasq-dhcp[2013]: DHCPOFFER(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24 
Thu Jan 25 17:16:53 2018 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED c0:ee:fb:46:7a:24
Thu Jan 25 17:16:53 2018 kern.debug kernel: [  176.418274] ieee80211 phy0: staid 4 deleted
Thu Jan 25 17:16:55 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 IEEE 802.11: associated (aid 4)
Thu Jan 25 17:16:55 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED c0:ee:fb:46:7a:24
Thu Jan 25 17:16:55 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 RADIUS: starting accounting session AE5878DBC0FF76DC
Thu Jan 25 17:16:55 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 WPA: pairwise key handshake completed (RSN)
Thu Jan 25 17:16:55 2018 daemon.info dnsmasq-dhcp[2013]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24 
Thu Jan 25 17:16:55 2018 daemon.info dnsmasq-dhcp[2013]: DHCPOFFER(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24 
Thu Jan 25 17:16:55 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 IEEE 802.11: authenticated
Thu Jan 25 17:17:00 2018 daemon.info dnsmasq-dhcp[2013]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24 
Thu Jan 25 17:17:00 2018 daemon.info dnsmasq-dhcp[2013]: DHCPOFFER(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24 
Thu Jan 25 17:17:05 2018 daemon.info dnsmasq-dhcp[2013]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24 
Thu Jan 25 17:17:05 2018 daemon.info dnsmasq-dhcp[2013]: DHCPOFFER(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24 
Thu Jan 25 17:17:13 2018 daemon.info dnsmasq-dhcp[2013]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24 
Thu Jan 25 17:17:13 2018 daemon.info dnsmasq-dhcp[2013]: DHCPOFFER(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24 
Thu Jan 25 17:17:28 2018 daemon.info dnsmasq-dhcp[2013]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24 
Thu Jan 25 17:17:28 2018 daemon.info dnsmasq-dhcp[2013]: DHCPOFFER(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24 
Thu Jan 25 17:17:28 2018 daemon.info dnsmasq-dhcp[2013]: DHCPREQUEST(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24 
Thu Jan 25 17:17:28 2018 daemon.info dnsmasq-dhcp[2013]: DHCPACK(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24 OnePlus2
```

Is there a way to get any detailed logs on official Linksys firmware 1.0.6.186168? Because with that firmware this clients work without any problems.

Alexander-r, 2018-01-25:
I've tried to check what is happening on the network when the phone tries to connect.
On official firmware in bridge mode when the phone connects I see DHCP requests, ARP requests, DNS requests and HTTP request checking internet connection and after that there is normal traffic flow.
On LEDE (also in bridge mode so that I get similar setup) in addition to this I see a lot of MDNS requests right after connection. On first connection after reboot there were 29 MDNS packets and the phone disconnected that time. On successful connection there were 8 MDNS requests right after connection.
There were no MDNS packets on the network with official firmware ar all. It seems Linksys fix for the problem was to drop all MDNS traffic altogether.
Maybe this information might be useful.

yuhhaurlin, 2018-01-25:
1. So all firmware has this problem for LEDE?
2. The problem is related to the fix of Google.
3. I will check what is fixed by Google.

adamcarter1, 2018-01-25:
And chromecast devices. Like my vizio tv.

yuhhaurlin, 2018-01-25:
Thanks. BTW, if you install fix from Google, did you still encounter problem?

bbogey, 2018-01-26:
I haven't seen the problem with my wrt1900acs, but this is how it looks when I wake up my Android phone. So just by opening the lock screen, not even unlocking it.

I have two chromecasts, Google Play Services 11.9.51 updated on 30.11.2017.
It keeps spamming that multicast traffic all the time when screen on, about 20 packets / second.

````#  tcpdump -i eth1 -s0 -vv net 224.0.0.0/4 -tttt
tcpdump: listening on eth1, link-type EN10MB (Ethernet), capture size 262144 bytes
2018-01-26 02:04:54.729649 IP (tos 0x0, ttl 255, id 50038, offset 0, flags [DF], proto UDP (17), length 138)
    192.168.1.140.5353 > 224.0.0.251.5353: [udp sum ok] 20891 [3q] PTR (QM)? _%9E5E7C8F47989526C9BCD95D24084F6F0B27C5ED._sub._googlecast._tcp.local. PTR (QM)? _CC32E753._sub._googlecast._tcp.local. PTR (QM)? _googlecast._tcp.local. (110)
2018-01-26 02:04:54.729798 IP (tos 0x0, ttl 255, id 50039, offset 0, flags [DF], proto UDP (17), length 138)
    192.168.1.140.5353 > 224.0.0.251.5353: [udp sum ok] 16613 [3q] PTR (QM)? _%9E5E7C8F47989526C9BCD95D24084F6F0B27C5ED._sub._googlecast._tcp.local. PTR (QM)? _CA5E8412._sub._googlecast._tcp.local. PTR (QM)? _googlecast._tcp.local. (110)
2018-01-26 02:04:54.730561 IP (tos 0x0, ttl 255, id 50040, offset 0, flags [DF], proto UDP (17), length 122)
    192.168.1.140.5353 > 224.0.0.251.5353: [udp sum ok] 20213 [2q] PTR (QM)? _%9E5E7C8F47989526C9BCD95D24084F6F0B27C5ED._sub._googlecast._tcp.local. PTR (QM)? _googlecast._tcp.local. (94)
2018-01-26 02:04:54.730698 IP (tos 0x0, ttl 255, id 50041, offset 0, flags [DF], proto UDP (17), length 138)
    192.168.1.140.5353 > 224.0.0.251.5353: [udp sum ok] 15458 [3q] PTR (QM)? _%9E5E7C8F47989526C9BCD95D24084F6F0B27C5ED._sub._googlecast._tcp.local. PTR (QM)? _CA5E8412._sub._googlecast._tcp.local. PTR (QM)? _googlecast._tcp.local. (110)
2018-01-26 02:04:54.730774 IP (tos 0x0, ttl 255, id 50042, offset 0, flags [DF], proto UDP (17), length 122)
    192.168.1.140.5353 > 224.0.0.251.5353: [udp sum ok] 20423 [2q] PTR (QM)? _%9E5E7C8F47989526C9BCD95D24084F6F0B27C5ED._sub._googlecast._tcp.local. PTR (QM)? _googlecast._tcp.local. (94)
2018-01-26 02:04:54.730850 IP (tos 0x0, ttl 255, id 50043, offset 0, flags [DF], proto UDP (17), length 138)
    192.168.1.140.5353 > 224.0.0.251.5353: [udp sum ok] 16319 [3q] PTR (QM)? _%9E5E7C8F47989526C9BCD95D24084F6F0B27C5ED._sub._googlecast._tcp.local. PTR (QM)? _CA5E8412._sub._googlecast._tcp.local. PTR (QM)? _googlecast._tcp.local. (110)
2018-01-26 02:04:54.730927 IP (tos 0x0, ttl 255, id 50044, offset 0, flags [DF], proto UDP (17), length 138)
    192.168.1.140.5353 > 224.0.0.251.5353: [udp sum ok] 16550 [3q] PTR (QM)? _%9E5E7C8F47989526C9BCD95D24084F6F0B27C5ED._sub._googlecast._tcp.local. PTR (QM)? _CA5E8412._sub._googlecast._tcp.local. PTR (QM)? _googlecast._tcp.local. (110)
2018-01-26 02:04:54.742010 IP (tos 0x0, ttl 255, id 50092, offset 0, flags [DF], proto UDP (17), length 89)
    192.168.1.140.5353 > 224.0.0.251.5353: [udp sum ok] 15534 [2q] PTR (QM)? _CA5E8412._sub._googlecast._tcp.local. PTR (QM)? _googlecast._tcp.local. (61)
2018-01-26 02:04:54.742189 IP (tos 0x0, ttl 255, id 50093, offset 0, flags [DF], proto UDP (17), length 122)
    192.168.1.140.5353 > 224.0.0.251.5353: [udp sum ok] 4793 [2q] PTR (QM)? _%9E5E7C8F47989526C9BCD95D24084F6F0B27C5ED._sub._googlecast._tcp.local. PTR (QM)? _googlecast._tcp.local. (94)
2018-01-26 02:04:54.742275 IP (tos 0x0, ttl 255, id 50094, offset 0, flags [DF], proto UDP (17), length 138)
    192.168.1.140.5353 > 224.0.0.251.5353: [udp sum ok] 19817 [3q] PTR (QM)? _%9E5E7C8F47989526C9BCD95D24084F6F0B27C5ED._sub._googlecast._tcp.local. PTR (QM)? _CA5E8412._sub._googlecast._tcp.local. PTR (QM)? _googlecast._tcp.local. (110)
2018-01-26 02:04:54.744072 IP (tos 0x0, ttl 255, id 50095, offset 0, flags [DF], proto UDP (17), length 154)
    192.168.1.140.5353 > 224.0.0.251.5353: [udp sum ok] 13916 [4q] PTR (QM)? _%9E5E7C8F47989526C9BCD95D24084F6F0B27C5ED._sub._googlecast._tcp.local. PTR (QM)? _CA5E8412._sub._googlecast._tcp.local. PTR (QM)? _9CB11073._sub._googlecast._tcp.local. PTR (QM)? _googlecast._tcp.local. (126)
2018-01-26 02:04:54.744223 IP (tos 0x0, ttl 255, id 50096, offset 0, flags [DF], proto UDP (17), length 138)
    192.168.1.140.5353 > 224.0.0.251.5353: [udp sum ok] 6164 [3q] PTR (QM)? _%9E5E7C8F47989526C9BCD95D24084F6F0B27C5ED._sub._googlecast._tcp.local. PTR (QM)? _CA5E8412._sub._googlecast._tcp.local. PTR (QM)? _googlecast._tcp.local. (110)
2018-01-26 02:04:54.744299 IP (tos 0x0, ttl 255, id 50097, offset 0, flags [DF], proto UDP (17), length 89)
    192.168.1.140.5353 > 224.0.0.251.5353: [udp sum ok] 8873 [2q] PTR (QM)? _9CB11073._sub._googlecast._tcp.local. PTR (QM)? _googlecast._tcp.local. (61)
2018-01-26 02:04:54.744371 IP (tos 0x0, ttl 255, id 50098, offset 0, flags [DF], proto UDP (17), length 122)
    192.168.1.140.5353 > 224.0.0.251.5353: [udp sum ok] 2692 [2q] PTR (QM)? _%9E5E7C8F47989526C9BCD95D24084F6F0B27C5ED._sub._googlecast._tcp.local. PTR (QM)? _googlecast._tcp.local. (94)
2018-01-26 02:04:54.744442 IP (tos 0x0, ttl 255, id 50099, offset 0, flags [DF], proto UDP (17), length 154)
    192.168.1.140.5353 > 224.0.0.251.5353: [udp sum ok] 14909 [4q] PTR (QM)? _37F83649._sub._googlecast._tcp.local. PTR (QM)? _%9E5E7C8F47989526C9BCD95D24084F6F0B27C5ED._sub._googlecast._tcp.local. PTR (QM)? _CA5E8412._sub._googlecast._tcp.local. PTR (QM)? _googlecast._tcp.local. (126)
[... cut ...]
    192.168.1.140.5353 > 224.0.0.251.5353: [udp sum ok] 2696 [2q] PTR (QM)? _%9E5E7C8F47989526C9BCD95D24084F6F0B27C5ED._sub._googlecast._tcp.local. PTR (QM)? _googlecast._tcp.local. (94)
2018-01-26 02:05:04.145295 IP (tos 0x0, ttl 255, id 50699, offset 0, flags [DF], proto UDP (17), length 154)
    192.168.1.140.5353 > 224.0.0.251.5353: [udp sum ok] 14913 [4q] PTR (QM)? _37F83649._sub._googlecast._tcp.local. PTR (QM)? _%9E5E7C8F47989526C9BCD95D24084F6F0B27C5ED._sub._googlecast._tcp.local. PTR (QM)? _CA5E8412._sub._googlecast._tcp.local. PTR (QM)? _googlecast._tcp.local. (126)
^C
67 packets captured
165 packets received by filter
98 packets dropped by kernel
`

yuhhaurlin, 2018-01-26:
Thanks for your information.

yuhhaurlin, 2018-01-26:
@Alexander-r Please help to check if https://github.com/kaloz/mwlwifi/commit/2b99e150f954944d17607004416e1f2321a73510 fixes your problem. If you still have problem, I will check QA to reproduce the problem. Thanks.

Alexander-r, 2018-01-26:
@yuhhaurlin Yes the problem affects all firmwares with LEDE

I've tested 2b99e15 and it does not fix the problem. The phone still disconnects.

The problem is related to recent fix from Google.
At the start of the tests I had Google Play services 8.6.22-all [0] [PR] 179647274 installed on the phone. Right now I've installed version 8.7.10-all [0] [PR] 18179946. With this version with broadcast '1' the phone connects immediately but it disconnects after a couple of seconds as before. With unicast replies it takes longer to connect (it's stuck on "Obtaining IP address" for about a minute) but after that it no longer disconnects.
It seems that router has problems sending broadcast traffic right after connection but after some time everything works normally.

yuhhaurlin, 2018-01-26:
Can you show me the log? Thanks.

yuhhaurlin, 2018-01-26:
BTW, can you cat ampdu when the phone wait for IP.

bbogey, 2018-01-26:
@Alexander-r those versions seems to point to Google Play _Store_.
Did you check version and do upgrade for Google Play service?

spamtree, 2018-01-26:
I wonder if there are different versions of google play services for nougat and oreo. 
I have nougat the google play services on both my phones is 
version 11.9.51 (440-177350961) 
com.google.android.gms

bbogey, 2018-01-26:
I have the exact same version in Oreo. That @Alexander-r version was for Store what has nothing to do with this.

spamtree, 2018-01-26:
Sorry about that.

Alexander-r, 2018-01-26:
@yuhhaurlin Right now Google Play Services is 11.9.75 (948-182402865). Before the update the version was 11.9.51 (948-177350961) (that's according to the backup I did before upgrade).

Right now I don't have access to the device. I will be able to do more tests and provide logs on Monday. Do you want the logs for normal connection or for one with broadcast '1'?

yuhhaurlin, 2018-01-26:
1. It looks like your problem is not related to broadcast.
2. The phone will ask unicast reply.
3. Please do unicast and get log. I need to know if it is the same as before the phone can't get unicast reply.
4. You told me that it takes one minute to get IP address for the phone under unicast reply. Please cat ampdu when the phone try to get the IP address.

Thanks.

bg4drl, 2018-01-28:
Android three devices Home Max, Pixel 2 xl, Pixelbook2 using Google cast, Intel 9260 also connected in this wifi network to go 1733Mbps. VHT160SGI drive has been used for two days Crush, restart WRT3200, Channel will not up, Flash WRT3200, turn on VHT80 to see how long
pixel 2 xl ‘s google play beta 51 is installed.

yuhhaurlin, 2018-01-28:
We found 160MHz throughput problem with 9260. We work for It now.

Alexander-r, 2018-01-29:
@yuhhaurlin Here is the log for unicast dhcp:
```
Mon Jan 29 19:49:04 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 IEEE 802.11: associated (aid 3)
Mon Jan 29 19:49:04 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED c0:ee:fb:46:7a:24
Mon Jan 29 19:49:04 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 RADIUS: starting accounting session 97E69C0C532B586F
Mon Jan 29 19:49:04 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 WPA: pairwise key handshake completed (RSN)
Mon Jan 29 19:49:04 2018 daemon.info dnsmasq-dhcp[7315]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24
Mon Jan 29 19:49:04 2018 daemon.info dnsmasq-dhcp[7315]: DHCPOFFER(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24
Mon Jan 29 19:49:05 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 IEEE 802.11: authenticated
Mon Jan 29 19:49:09 2018 daemon.info dnsmasq-dhcp[7315]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24
Mon Jan 29 19:49:09 2018 daemon.info dnsmasq-dhcp[7315]: DHCPOFFER(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24
Mon Jan 29 19:49:14 2018 daemon.info dnsmasq-dhcp[7315]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24
Mon Jan 29 19:49:14 2018 daemon.info dnsmasq-dhcp[7315]: DHCPOFFER(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24
Mon Jan 29 19:49:22 2018 daemon.info dnsmasq-dhcp[7315]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24
Mon Jan 29 19:49:22 2018 daemon.info dnsmasq-dhcp[7315]: DHCPOFFER(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24
Mon Jan 29 19:49:38 2018 daemon.info dnsmasq-dhcp[7315]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24
Mon Jan 29 19:49:38 2018 daemon.info dnsmasq-dhcp[7315]: DHCPOFFER(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24
Mon Jan 29 19:49:40 2018 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED c0:ee:fb:46:7a:24
Mon Jan 29 19:49:40 2018 kern.debug kernel: [249398.921884] ieee80211 phy0: staid 3 deleted
Mon Jan 29 19:49:42 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 IEEE 802.11: associated (aid 3)
Mon Jan 29 19:49:42 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED c0:ee:fb:46:7a:24
Mon Jan 29 19:49:42 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 RADIUS: starting accounting session 562F499DE13D21FC
Mon Jan 29 19:49:42 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 WPA: pairwise key handshake completed (RSN)
Mon Jan 29 19:49:42 2018 daemon.info dnsmasq-dhcp[7315]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24
Mon Jan 29 19:49:42 2018 daemon.info dnsmasq-dhcp[7315]: DHCPOFFER(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24
Mon Jan 29 19:49:47 2018 daemon.info dnsmasq-dhcp[7315]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24
Mon Jan 29 19:49:47 2018 daemon.info dnsmasq-dhcp[7315]: DHCPOFFER(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24
Mon Jan 29 19:49:52 2018 daemon.info dnsmasq-dhcp[7315]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24
Mon Jan 29 19:49:52 2018 daemon.info dnsmasq-dhcp[7315]: DHCPOFFER(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24
Mon Jan 29 19:49:58 2018 daemon.info hostapd: wlan0: STA c0:ee:fb:46:7a:24 IEEE 802.11: authenticated
Mon Jan 29 19:50:00 2018 daemon.info dnsmasq-dhcp[7315]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24
Mon Jan 29 19:50:00 2018 daemon.info dnsmasq-dhcp[7315]: DHCPOFFER(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24
Mon Jan 29 19:50:15 2018 daemon.info dnsmasq-dhcp[7315]: DHCPDISCOVER(br-lan) c0:ee:fb:46:7a:24
Mon Jan 29 19:50:15 2018 daemon.info dnsmasq-dhcp[7315]: DHCPOFFER(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24
Mon Jan 29 19:50:15 2018 daemon.info dnsmasq-dhcp[7315]: DHCPREQUEST(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24
Mon Jan 29 19:50:15 2018 daemon.info dnsmasq-dhcp[7315]: DHCPACK(br-lan) 192.168.7.10 c0:ee:fb:46:7a:24 OnePlus2
```

And here is ampdu while phone waits for IP address:
```
stream: 6
idx: 6
state: 3
mac address: c0:ee:fb:46:7a:24
tid: 6
```

yuhhaurlin, 2018-01-30:
Thanks. Can you also cat stnid for me?

yuhhaurlin, 2018-01-30:
@Alexander-r Please help to verify https://github.com/kaloz/mwlwifi/commit/adae20af0ddac061807fbaadec6c3b8129290480. Thanks.

Alexander-r, 2018-01-30:
I've tried adae20a. It has same problems.
Here is stnid output:
```
stnid: 1 macid: 0 aid: 1
stnid: 2 macid: 0 aid: 2
```

yuhhaurlin, 2018-01-30:
It is very strange, I will try to reproduce the problem here.

yuhhaurlin, 2018-01-30:
I hope I can collect Andriod phone which has this issue:
OnePlus 2
Sony Xperia X Performance

Please let me know if other Andriod phone has this issue.

Samsung S6 edge does not have this issue.

adamcarter1, 2018-01-31:
I have a feeling today's the day this gets fixed.

adamcarter1, 2018-02-03:
[image]

yuhhaurlin, 2018-02-03:
Please only report problems here.

yuhhaurlin, 2018-02-04:
BTW, can I know what the problem you have? The same issue here, what kind of Android phone?

adamcarter1, 2018-02-04:
Vizio android tv with smartcast. M65-E0. Basically a glorrified chromecast.

I know google is working on their end. Linksys sent out an email saying they are working on it on their end.

adamcarter1, 2018-02-04:
I use dd-wrt. Couldnt test ur builds even if i wanted to.

Keltere, 2018-02-07:
Same problem here: with xiaomi mi5 and huawei p10 i can't connect to 5ghz and i have a very bad performance (30MB/s and disconnection) with usb wifi card Realtek 8812au. This is really a mess...
I've tried the new stock firmware and the new davidc502 firmware with lede and wifi driver 10.3.4.0-20180118

yuhhaurlin, 2018-02-07:
Please let me know your setting.

yuhhaurlin, 2018-02-07:
Issue here is some Android phones take time to connect and I did not hear low throughput problem.

Keltere, 2018-02-07:
@yuhhaurlin what settings? Wifi settings? I've tried channel 1-6-10-11, all width settings (20,40,80), and of course all auto. Same problem.

aaron1959, 2018-02-07:
With this driver and firmware dd-wrt is working fine with android clients including android Galaxy S7.
driver version: 10.3.4.0-20180118
firmware version: 0x09030204

yuhhaurlin, 2018-02-07:
@aaron1959 Yes. Same result as we test here.

[48 more replies]