kaloz/mwlwifi #264: Stuck on channel 36.

tapper82, 2018-01-24:
Hi when setting any other channel than 36 wi-fi will not work. I want to use channel 100 as in the UK you get up to 27 db on channels above 100.

yuhhaurlin, 2018-01-25:
1. Use "iw reg get" to check your regulatory information.
2. If channel 100 is allowable, you need to wait CAC (60 seconds) for the time to check radar signal. If no radar signal is detected, you can use the channel (using "iw dev" to check if SSID appears to make sure the AP is working).

closed 2018-01-25 by yuhhaurlin

tapper82, 2018-01-25:
Thu Jan 25 07:57:12 2018 daemon.notice hostapd: wlan0: interface state HT_SCAN->DFS
Thu Jan 25 07:57:12 2018 daemon.notice hostapd: wlan0: DFS-CAC-START freq=5500 chan=100 sec_chan=1, width=1, seg0=106, seg1=0, cac_time=60s
Thu Jan 25 07:57:12 2018 daemon.err hostapd: DFS start_dfs_cac() failed, -1
Thu Jan 25 07:57:12 2018 daemon.err hostapd: Interface initialization failed
Thu Jan 25 07:57:12 2018 daemon.notice hostapd: wlan0: interface state DFS->DISABLED
Thu Jan 25 07:57:12 2018 daemon.notice hostapd: wlan0: AP-DISABLED
Thu Jan 25 07:57:12 2018 daemon.notice hostapd: wlan0: interface state DISABLED->DISABLED
Thu Jan 25 07:57:12 2018 daemon.notice hostapd: wlan0: AP-DISABLED
Thu Jan 25 07:57:12 2018 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Thu Jan 25 07:57:12 2018 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0 wasn't started
Thu Jan 25 07:57:12 2018 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Thu Jan 25 07:57:12 2018 kern.debug kernel: [53637.585296] ieee80211 phy0: change: 0x60
Thu Jan 25 07:57:12 2018 kern.info kernel: [53637.625307] device wlan0 left promiscuous mode
Thu Jan 25 07:57:12 2018 kern.info kernel: [53637.629807] br-lan: port 2(wlan0) entered disabled state
Thu Jan 25 07:57:12 2018 kern.debug kernel: [53637.681433] ieee80211 phy0: change: 0x100
Thu Jan 25 07:57:12 2018 daemon.notice hostapd: ELOOP: remaining socket: sock=20 eloop_data=0xb6f886b0 user_data=0 handler=0x36c28
Thu Jan 25 07:57:12 2018 daemon.notice netifd: radio0 (9660): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process 27256 path ()
Thu Jan 25 07:57:12 2018 kern.debug kernel: [53637.850276] ieee80211 phy0: change: 0xffffffff
Thu Jan 25 07:57:13 2018 kern.info kernel: [53637.948837] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready

tapper82, 2018-01-25:
And this on channel 60. There should not even be DFS on channel 60.
Thu Jan 25 08:02:28 2018 daemon.notice hostapd: wlan0: DFS-CAC-START freq=5300 chan=60 sec_chan=1, width=1, seg0=58, seg1=0, cac_time=60s
Thu Jan 25 08:02:28 2018 daemon.err hostapd: DFS start_dfs_cac() failed, -1
Thu Jan 25 08:02:28 2018 daemon.err hostapd: Interface initialization failed
Thu Jan 25 08:02:28 2018 daemon.notice hostapd: wlan0: interface state DFS->DISABLED
Thu Jan 25 08:02:28 2018 daemon.notice hostapd: wlan0: AP-DISABLED
Thu Jan 25 08:02:28 2018 daemon.notice hostapd: wlan0: interface state DISABLED->DISABLED
Thu Jan 25 08:02:28 2018 daemon.notice hostapd: wlan0: AP-DISABLED
Thu Jan 25 08:02:28 2018 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Thu Jan 25 08:02:28 2018 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0 wasn't started
Thu Jan 25 08:02:28 2018 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Thu Jan 25 08:02:28 2018 kern.debug kernel: [53953.631018] ieee80211 phy0: change: 0x40
Thu Jan 25 08:02:28 2018 kern.info kernel: [53953.691050] device wlan0 left promiscuous mode
Thu Jan 25 08:02:28 2018 kern.info kernel: [53953.695564] br-lan: port 2(wlan0) entered disabled state
Thu Jan 25 08:02:28 2018 kern.debug kernel: [53953.730026] ieee80211 phy0: change: 0x100
Thu Jan 25 08:02:28 2018 daemon.notice hostapd: ELOOP: remaining socket: sock=20 eloop_data=0xb6fbc6b0 user_data=0 handler=0x36c28
Thu Jan 25 08:02:28 2018 daemon.notice netifd: radio0 (10871): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process 27256 path ()
Thu Jan 25 08:02:29 2018 kern.debug kernel: [53953.909030] ieee80211 phy0: change: 0xffffffff
Thu Jan 25 08:02:29 2018 kern.info kernel: [53954.014619] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready

yuhhaurlin, 2018-01-25:
Please check README to resolve your problem.

tapper82, 2018-01-25:
The wrt3200acm I have defaults to setting it's Country Code to FR I live in the UK so i set it to GB. When i do iw reg get" it still says that country is FR.

yuhhaurlin, 2018-01-25:
Thu Jan 25 08:02:28 2018 daemon.err hostapd: DFS start_dfs_cac() failed, -1 => Please check README to resolve this problem.

lantis1008, 2018-01-26:
@tapper82 set it to FR and try again. 
In general you can’t overwrite the hardcoded country setting (without patching the code and DTS)