kaloz/mwlwifi #265: Driver 10.3.4.0-20171214 does not support WDS

francispereira, 2018-01-25:
Driver version 10.3.4.0-20171214 does not support WDS required to extend the local lan in deployments where there is a single DHCP server. 

Notice how setting 4addr works on an older driver version:

```
root@wifi:~# cat /sys/kernel/debug/ieee80211/phy1/mwlwifi/info

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.2.0-20170110
firmware version: 0x0702091a
power table loaded from dts: no
firmware region code: 0x30
mac address: 00:25:9c:14:41:6c
2g: enable
5g: disable
antenna: 4 4
irq number: 45
iobase0: e1d80000
iobase1: e1f00000
tx limit: 768
rx limit: 64
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00010000
qe trigger number: 39178083

root@wifi:~#  iw dev wlan1 set 4addr on
root@wifi:~#
```

And it doesn't on the newer version:

```
root@MCDEBIAN:~# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.4.0-20171214
firmware version: 0x0702091a
power table loaded from dts: yes
firmware region code: 0x0
mac address: 00:25:9c:13:a6:e0
2g: enable
5g: disable
antenna: 4 4
irq number: 46
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: d2180000
iobase1: d2300000
tx limit: 768
rx limit: 64
qe trigger number: 887

root@MCDEBIAN:~#  iw dev wlp2s0 set 4addr on
command failed: Operation not supported (-95)
```

yuhhaurlin, 2018-01-25:
I don't know what you mean,  WDS works for the new driver. Only one group rekey issue for 8964.

closed 2018-01-25 by yuhhaurlin

francispereira, 2018-01-25:
@yuhhaurlin why can't I set 4addr on the interface ?

francispereira, 2018-01-25:
@yuhhaurlin I suggest you reopen this issue and try this out yourself against driver version 10.3.4.0-20171214

yuhhaurlin, 2018-01-25:
I test WRT1200AC with WRT3200ACM.  I can't find this kind of problem. BTW, you did not use LEDE. It looks like you build the driver with MCDEBIAN. Are you sure original driver can work on your environment?

BrainSlayer, 2018-01-25:
@francispereira wds 4 address mode is working. i'm using it every day

yuhhaurlin, 2018-01-25:
@BrainSlayer Thanks. I use 8864 to develop and debug WDS feature of 8964. WDS of 8864 works well without any problems. Even though for 8964, WDS only has one remaining problem: group rekey will let data can't work. However it can be workaround via disabling of group rekey of WDS AP.

francispereira, 2018-01-25:
@BrainSlayer @yuhhaurlin thanks for confirming that this is currently working. I will see how the driver is being build for McDebian and fix it accordingly