kaloz/mwlwifi #266: Whether it's Cisco, Linksys, ASUS

wifismashington, 2018-01-25:
Hey hey you know what it is:
WiFi and Darkwin bringing you brand new music all day baby all day
This one right here called "Smashed"

This one is straight out to Hack Forums

[Hook]
Whether it's Asus, Cisco, Linksys (smashed)
That's what I call a router breaking
Your Internet's slow I see you're raging (smashed)
I think I hear a modem breaking
[Repeat hook]

[Verse 1: WiFi]
You say your security's tight?
Well who you kidding dog?
My hack skills are way too 'leet for your firewalls

So I see that you port-forward to your Minecraft server
Let's fire up Cain and Abel up and get to port-sniffing
Similar to your chick, I know it's hard to admit it
It just takes a handshake for me to get all up in it
So kid, why you talking smack when your IP is static?
I could boot you anytime there's nothing you could do about it
Smash

Hacking up niggas already know I'm Backtracking them
LR's money I'm sacking them
They already know where the fuck I'm at
Stay getting money they taxing them
Full of capitals like an acronym
Me and WiFi lagging them, RATing them, and smacking them
Smash

[Repeat hook]

[Verse 2: Darkwin]
I see your router go boom
You're offline, so is everyone in the room
Your modem offline, no ping back
It's a fucking dish, your IP on track
Go get your money back, Best Buy you see when you try to get a refund
They going to fucking be, "your router's been smashed!"
Motherfucking stunned
Good luck trying to get a refund

You go home crying, yelling at your mom
When all you really need is a motherfucking bong
Smoke that fucking kush up in the clouds
Be so high that nothing can take you down

Clouds in the air, bitches in the seats
Shagging these bitches when our motherfucking meat
Cause your router got smashed
Now your router's in the trash
Somebody needs to take a motherfucking bath

[Repeat hook]

Hey yo, Noble, you're a fool for this one man

yuhhaurlin, 2018-01-26:
Welcome to report issue here.

closed 2018-01-26 by yuhhaurlin

yuhhaurlin, 2018-01-26:
I think I know why less and less vendors want to work for open source.

Skilmatix, 2018-01-26:
@yuhhaurlin, you are doing a great job!  Keep up the good work.  You can't fix ignorance.