kaloz/mwlwifi #267: Android phone not reconnecting

eduperez, 2018-01-26:
I have noticed that one of my Android phones is not reconnecting when I unlock it (the phone is configured to switch wifi off while locked). When this happens, all I see at the router is this line:

```
Fri Jan 26 23:11:59 2018 daemon.notice hostapd: handle_auth_cb: STA 80:65:6d:db:c9:a9 not found
```

Then I switch wifi off and back on on the phone, and this happens:

```
Fri Jan 26 23:12:31 2018 daemon.info hostapd: wlan_2g: STA 80:65:6d:db:c9:a9 IEEE 802.11: associated (aid 1)
Fri Jan 26 23:12:31 2018 daemon.notice hostapd: wlan_2g: AP-STA-CONNECTED 80:65:6d:db:c9:a9
Fri Jan 26 23:12:31 2018 daemon.info hostapd: wlan_2g: STA 80:65:6d:db:c9:a9 WPA: pairwise key handshake completed (RSN)
Fri Jan 26 23:12:32 2018 daemon.info dnsmasq-dhcp[4559]: DHCPDISCOVER(br-lan) 80:65:6d:db:c9:a9 
Fri Jan 26 23:12:32 2018 daemon.info dnsmasq-dhcp[4559]: DHCPOFFER(br-lan) 192.168.1.179 80:65:6d:db:c9:a9 
Fri Jan 26 23:12:32 2018 daemon.info dnsmasq-dhcp[4559]: DHCPREQUEST(br-lan) 192.168.1.179 80:65:6d:db:c9:a9 
Fri Jan 26 23:12:32 2018 daemon.info dnsmasq-dhcp[4559]: DHCPACK(br-lan) 192.168.1.179 80:65:6d:db:c9:a9 android-fe23c8561fc73c98
Fri Jan 26 23:12:32 2018 daemon.info hostapd: wlan_2g: STA 80:65:6d:db:c9:a9 IEEE 802.11: authenticated
```

I'm pretty sure this did not happen with older versions of the driver.

Phone is a Samsung Galaxy Core Prime SM-G361F running Android 5.1.1
Router is WRT3200ACM running LEDE plus mwlwifi 2b99e15

yuhhaurlin, 2018-01-26:
This modification should not have this king of problem. I will try my Samsung phone. Thanks for your information.

eduperez, 2018-01-28:
Just a clarification: with "this did not happen with _older versions_" I did not mean "the version prior to this one". I have just noticed this now, but it can have been happening for several versions. I will try to find out what commit introduced the bug later.

eduperez, 2018-01-30:
This is what I see when using adae20af0ddac061807fbaadec6c3b8129290480 after waking up the phone:

```
Tue Jan 30 22:45:23 2018 daemon.info hostapd: wlan_2g: STA 80:65:6d:db:c9:a9 IEEE 802.11: associated (aid 2)
Tue Jan 30 22:45:23 2018 daemon.info hostapd: wlan_2g: STA 80:65:6d:db:c9:a9 IEEE 802.11: authenticated
Tue Jan 30 22:45:28 2018 kern.debug kernel: [ 3166.739187] ieee80211 phy1: staid 3 deleted
Tue Jan 30 22:45:31 2018 daemon.info hostapd: wlan_2g: STA 80:65:6d:db:c9:a9 IEEE 802.11: deauthenticated due to local deauth request
Tue Jan 30 22:45:36 2018 daemon.info hostapd: wlan_2g: STA 80:65:6d:db:c9:a9 IEEE 802.11: associated (aid 2)
Tue Jan 30 22:45:41 2018 kern.debug kernel: [ 3179.435735] ieee80211 phy1: staid 3 deleted
Tue Jan 30 22:45:44 2018 daemon.info hostapd: wlan_2g: STA 80:65:6d:db:c9:a9 IEEE 802.11: deauthenticated due to local deauth request
Tue Jan 30 22:45:58 2018 daemon.notice hostapd: handle_auth_cb: STA 80:65:6d:db:c9:a9 not found
```

Then I switch wifi off and back on on the phone:

```
Tue Jan 30 22:46:10 2018 daemon.info hostapd: wlan_2g: STA 80:65:6d:db:c9:a9 IEEE 802.11: associated (aid 2)
Tue Jan 30 22:46:10 2018 daemon.notice hostapd: wlan_2g: AP-STA-CONNECTED 80:65:6d:db:c9:a9
Tue Jan 30 22:46:10 2018 daemon.info hostapd: wlan_2g: STA 80:65:6d:db:c9:a9 WPA: pairwise key handshake completed (RSN)
Tue Jan 30 22:46:11 2018 daemon.info odhcpd[1309]: Using a RA lifetime of 1800 seconds on br-lan
Tue Jan 30 22:46:11 2018 daemon.info dnsmasq-dhcp[3137]: DHCPREQUEST(br-lan) 192.168.1.179 80:65:6d:db:c9:a9 
Tue Jan 30 22:46:11 2018 daemon.info dnsmasq-dhcp[3137]: DHCPACK(br-lan) 192.168.1.179 80:65:6d:db:c9:a9 android-a376465fbf81566a
Tue Jan 30 22:46:12 2018 daemon.info odhcpd[1309]: Using a RA lifetime of 1800 seconds on br-lan
Tue Jan 30 22:46:13 2018 daemon.info hostapd: wlan_2g: STA 80:65:6d:db:c9:a9 IEEE 802.11: authenticated
```

I have not had time to test any previous versions, will try to do it soon.

eduperez, 2018-02-04:
I have been testing older releases, in reverse order.
So far, af210f0ef79d5c416dd455747a16d76bec5912ba (and everything later) has the issue.

yuhhaurlin, 2018-02-05:
So this issue is gone before this commit?

eduperez, 2018-02-05:
No, I still have not found the commit that introduced the bug; I know that it fails at af210f0ef79d5c416dd455747a16d76bec5912ba, but did not have time to test previous commits. Sorry if I have been dragging these tests, but this is my main router now, and I can only afford to make tests now and then.

eduperez, 2018-02-09:
Looks like commit 760f7bf1b8af6c899ccb14d47e8b987ed9a62adf works properly, so culprit is probably either 466368f9454250c2bc024795600d92564553d9bb or d5c5b05c538ad678929f1ed2891dde25a6fdd3eb; will try to keep testing tomorrow.

yuhhaurlin, 2018-02-09:
Thanks a lot.

rmandrad, 2018-02-10:
well ... whatever code has been commited this week .. has broken ... 5g/vht160 keeps dropping ... I am back to the code commited 2w ago

yuhhaurlin, 2018-02-10:
Yes. Don't use the latest commit now.

eduperez, 2018-02-12:
First tests indicate that it works on 760f7bf1b8af6c899ccb14d47e8b987ed9a62adf and fails on 466368f9454250c2bc024795600d92564553d9bb; so that is the prime suspect. I will reinstall both versions and try again, just to confirm.

eduperez, 2018-02-22:
I am getting crazy about this issue... sometimes some commit seems to work, but then it fails later; sometimes some commit seems to fail, but then it works later. I have disabled the "smart wifi" / "wifi+"feature on my phone, and now it seems to reconnect properly. But my wife's phone (also an Android) reconnects immediately even with "smart wifi" / "wifi+" activated.

I am not going to do any more tests, will just leave the "smart wifi" / "wifi+" feature disabled, and test this again when there is a new release. If nobody else is experiencing this issue, I think it can be closed.

yuhhaurlin, 2018-02-22:
The phone is all right with stock firmware and other AP?

eduperez, 2018-02-23:
I am only using stock Android 5.1.1; "smart wifi" / "wifi+" is a standard feature that auto-switches to 3G/4G when there is an AP available, but the phone decides it is "low quality". I am trying with another AP, will report back later.

eduperez, 2018-02-23:
Test done: with a different AP (TP-Link TL-WDR3600 + LEDE 17.01.4) the phone reconnects immediately, even if "smart wifi" / "wifi+" is enabled.

yuhhaurlin, 2018-02-26:
Thanks. After current job is done, I will try to reproduce this problem.

eduperez, 2018-02-27:
Could this have been solved in the latest release (51350a3e085a811179a270baddcae03ec9384f07)?
I cannot seem to be able to reproduce the issue after upgrading...

yuhhaurlin, 2018-02-27:
I don't know. But if it is all right now, I close this issue. Thanks for your help.

closed 2018-02-27 by yuhhaurlin