kaloz/mwlwifi #269: DFS channels do not last

XiaofengXu-GS, 2018-01-31:
Hi,

I have a WRT3200ACM running the latest Davidc502 snapshot. One problem I have is DFS channels do not stick. I can set to a DFS channel and several minutes later it will automatically go back to a non DFS channel and never goes back to DFS. Is this normal for the current firmware and driver?

Thanks.

Xiaofeng

yuhhaurlin, 2018-01-31:
If radar signal is detected, hostapd will change channel channel for you.

closed 2018-01-31 by yuhhaurlin

XiaofengXu-GS, 2018-02-02:
Should it try to go back to the DFS channel after 30 minutes if the interference is gone?

yuhhaurlin, 2018-02-02:
If the switch channel is also DFS channel, the same procedure will be used on it.
If the switch channel is not DFS channel, the channel will be used until you change it.