kaloz/mwlwifi #270: mwlwifi causes reboot on openwrt snapshot

e88z4, 2018-02-12:
Hi, 

I am experiencing random reboot issue with kernel 4.4 and kernel 4.9. The crashlog shows [this](https://gist.github.com/e88z4/c6f0a2d9232b16eb47282c28eb9c0619). This log copied from crashlog in kernel 4.9. 

Both kernels were compiled against:

PKG_NAME:=mwlwifi
PKG_VERSION:=10.3.4.0-20171214
PKG_RELEASE=1

PKG_LICENSE:=ISC
PKG_LICENSE_FILES:=

PKG_SOURCE_URL:=https://github.com/kaloz/mwlwifi
PKG_SOURCE_PROTO:=git
PKG_SOURCE_VERSION:=843d00cd9c134629b9dad7162831ec5f136399b3
PKG_MIRROR_HASH:=080c491473f0dbe9a7cac0fa3c6d004bf2dc3bd953a9f9d074eb8876a86437c2

I didn't remember having this issue when I compiled kernel 4.4 from 1 year ago. I am not sure if there is a recent commit in the code that causing this issue.

e88z4, 2018-02-13:
This issue can be reproduced on the latest commit of mwlwifi. 

Below is the Makefile that I use to compile
PKG_SOURCE_URL:=https://github.com/kaloz/mwlwifi
PKG_SOURCE_PROTO:=git
PKG_SOURCE_VERSION:=064e888e801d5e4c3638e92ee18a2cde06a6f2d0
PKG_MIRROR_HASH:=8cd5a0a3687c35cd163586bd1dcf97d042aa588d0ecf32fe6f609d904a5a0aab

Router crashed within 10 minutes after starting.

aaron1959, 2018-02-13:
Wow that's an old package, I think lot's been done since that one.

lister-wrt, 2018-02-13:
I have the same issue on ACv1 and ACSv2. Here's the one from ACS [link](https://gist.github.com/lister-wrt/e583ee11f2523f009f5b789df6d2f9df)

[064e888](https://github.com/kaloz/mwlwifi/commit/064e888e801d5e4c3638e92ee18a2cde06a6f2d0) seems especially bad, reboots every 10-15 minutes

e88z4, 2018-02-13:
Hi @aaron1959

I compiled several versions of mwlwifi driver. Below is the latest commit as of today. 
 
commit 064e888e801d5e4c3638e92ee18a2cde06a6f2d0
Author: David Lin <dlin@marvell.com>
Date:   Thu Feb 8 16:50:51 2018 +0800

    Added debugfs file tx_hist.
    
    Signed-off-by: David Lin <dlin@marvell.com>

The above version is even worse than 10.3.4.0-20171214. This version restarts the router less than 15 minutes. This version is basically useless for daily use. At least version 10.3.4.0-20171214 can survive up to 24-48 hours. 

As far as I remember when I compiled the lede package last year (probably early 2017), the driver didn't suffer reboot problem. 

I am going to compile several mwlwifi driver and test it to narrow down the issue.

e88z4, 2018-02-13:
@lister-wrt 

Did you ever try to revert the driver to an earlier version? What version is in your router now?

lister-wrt, 2018-02-13:
@e88z4 I'm running [4464371](https://github.com/kaloz/mwlwifi/commit/4464371fb45a2f86112b37a999a00355be128ca1) now as it was on my alt. It's not great but works for long enough not to irritate.

I tested a build with [adae20a](https://github.com/kaloz/mwlwifi/commit/adae20af0ddac061807fbaadec6c3b8129290480) last night but couldn't connect any (Android) devices.

aaron1959, 2018-02-13:
driver version: 10.3.4.0-20180118
firmware version: 0x09030204

This combo is working flawlessly on wrt3200acm everything seems to work fine all except DLNA.
PPPOE, NTP client, wifi, android devices etc..
Up for many days.

spamtree, 2018-02-13:
https://forum.openwrt.org/viewtopic.php?id=64949&p=174

Read the last comment on this page by davidc502. It appears that there is a patch to fix the reboot problem and this has been resolved. Read the complete thread for details. You may want to try this build. Hope this helps.

yuhhaurlin, 2018-02-13:
Thanks. I close it.

lister-wrt, 2018-02-13:
@spamtree isn't this just for ACv1? I'm having this issue on ACv1 and ACSv2. 

Also, it doesn't work for everyone.

e88z4, 2018-02-13:
@yuhhaurlin 

I have the CPU power management disabled in my configuration as being directed in this [patch](https://gist.github.com/NainKult/5658474291a02a992c68f2ceb587261d). The symptoms is completely different. 

The CPU idle power management issue symtomps are the following:
1. Affecting only kernel 4.9. 
2. Kernel 4.4 is not affected.
3. No error being logged anywhere. 
4. Affecting only Mamba device, 

My issue is a bit different:
1. Affecting both kernel 4.4 and kernel 4.9
2. Crashedlog identified that mwlwifi is the culprit as it is being identified [here](https://gist.github.com/e88z4/c6f0a2d9232b16eb47282c28eb9c0619)
3. My openwrt firmware has the CPU idle power management disabled. /proc/config.gz is provided [here](https://gist.github.com/e88z4/90ca38dfab4c00b5f07b4310c49f6d03)
4. @lister-wrt who has non mamba device is affected as well. 

I don't think this issue should be closed.

yuhhaurlin, 2018-02-13:
WRT1900AC:
Factory: https://drive.google.com/open?id=14RzI3GtXBHGAv9sk2Wwjafkj4roJnaYy
Upgrade: https://drive.google.com/open?id=1ryMOffud5hH10T_yNqNy1fc0Ch7PVVRK
WRT1900ACSv2: 
Factory: https://drive.google.com/open?id=1wuO3gPYKqrafnlPCyR4UxGtvMyK31SAa
Upgrade: https://drive.google.com/open?id=1as3oZ87_DvjGNU9y9BHE6wWy6HfQivNn

yuhhaurlin, 2018-02-13:
Please let me know if you have problem with above LEDE images. Thanks.

lister-wrt, 2018-02-13:
I tried to sysupgrade to your image and got this "The uploaded image file does not contain a supported format. Make sure that you choose the generic image format for your platform."

Is it safe to sysupgrade -F?

yuhhaurlin, 2018-02-13:
There are two partitions. If you have problem to upgrade image, you can use factory image and flash it to another partition. I think I will not  check upgrade problem.

lister-wrt, 2018-02-13:
@yuhhaurlin ACSv2 not ACv2. I think we both missed it. See my edited post^

yuhhaurlin, 2018-02-13:
Sorry. I update the links for WRT1900ACSv2.

lister-wrt, 2018-02-13:
@yuhhaurlin that's the same image. Hash is the same for both.

yuhhaurlin, 2018-02-13:
WRT1900ACS:
Factory: https://drive.google.com/open?id=14IQX0DrxC0zE09ruiZ4WoJtvrepVqoTx
Upgrade: https://drive.google.com/open?id=1XoO67XMYHA4gAOaOWPhFRRx0akoxQNvd

lister-wrt, 2018-02-13:
@yuhhaurlin installed + wiped configs. I'll report back if/when I get any errors.

Which commit is this based on?

yuhhaurlin, 2018-02-13:
https://github.com/kaloz/mwlwifi/commit/a149180fecec4fbaafb3f52ae46e33bd66501441#comments. Thanks for your help.

eduperez, 2018-02-13:
My two cents:

- LEDE 17.01.4 + adae20af0ddac061807fbaadec6c3b8129290480 : Rock solid.
- LEDE 17.01.4 + 064e888e801d5e4c3638e92ee18a2cde06a6f2d0 : Locks and reboots

There are only debugfs changes between those commits, so unless you need to do some debugging, I would just stick to the first one until the issue is fixed.

yuhhaurlin, 2018-02-13:
@eduperez Only the last commit has problem. I can reproduce it here with WRT3200ACM.

lister-wrt, 2018-02-13:
@eduperez I tested [adae20a](https://github.com/kaloz/mwlwifi/commit/adae20af0ddac061807fbaadec6c3b8129290480) yesterday and had issues connecting my wireless devices (all Android).  The error was something about receiving an empty packet. Tested with and without WPA2. 

Might have been a config issue I guess... I'll revert to that in my build and retest with clean configs in ~24hrs or so (when I'm done with @yuhhaurlin 's test image).

yuhhaurlin, 2018-02-13:
You have different devices. The last few commits have nothing to do with device which uses 8864.

lister-wrt, 2018-02-13:
@yuhhaurlin exactly what I thought. I've been bumping to latest commit for months presuming nothing would affect 8864.

eduperez, 2018-02-13:
@yuhhaurlin : So, 4464371fb45a2f86112b37a999a00355be128ca1 is safe? (I have not tested it, jumped directly to adae20af0ddac061807fbaadec6c3b8129290480).

yuhhaurlin, 2018-02-13:
Yes.

e88z4, 2018-02-13:
@yuhhaurlin 

When you said [4464371](https://github.com/kaloz/mwlwifi/commit/4464371fb45a2f86112b37a999a00355be128ca1) is safe, what does it mean? Will I expect this version to be bug free from constant reboot? Or will this commit suffer constant reboot but longer time (more than 24 hours)?

Sorry, I am not able to reply quickly with work and kids around. I will try to build this tonight.

jbsky, 2018-02-13:
For the wrt1900ac router, in order to get the best performance, I have disabled the following points:
CPU Frequency scaling
CPU idle PM support
DSA switch for MV88E6XXX
Indeed, there is a problem with the above list.

yuhhaurlin, 2018-02-13:
In fact, the recent commits are related to 8964. I don't understand why there is reboot problem happened for 8864. Please test the images I posted, so I can check here if you encounter reboot problem.

e88z4, 2018-02-13:
@jbsky 
I have disabled CPU frequency scaling, CPU idle PM support as well. The problem may not be related to the these items probably mwlwifi itsel. 

@yuhhaurlin 
I will try your image tonight and report to you. 
It may not be your recent commit that causes this problem. As my initial post mentioned that 843d00cd9c134629b9dad7162831ec5f136399b3 suffers the same reboot issue as well. This is commit is way back in December 2017.

yuhhaurlin, 2018-02-13:
I quickly check log history, it looks like commits after 2017/5 are not related to 8864. I did not check commits before 2017/5. However, you can try to reproduce this issue on the images I just posted.

tapper82, 2018-02-13:
A build with kernel 4.14 with latest driver here for testing.
https://drive.google.com/open?id=1LkbWyqDku1jJUbY6rOO5ZBFdwH5WG6F1

e88z4, 2018-02-14:
@yuhhaurlin 

I am currently using your firmware for my mamba device. It doesn't suffer the frequent 15 minutes reboot like 064e888e801d5e4c3638e92ee18a2cde06a6f2d0. I am testing the long term stability for this version.

lister-wrt, 2018-02-14:
Uptime at around 20 hrs on ACS here. I'm satisfied with that.

@yuhhaurlin are you interested in further testing on 8864? Currently building a new image from openwrt/master as the mwlwifi version is the same as your test image now. If that's also stable I could keep updating until I encounter issues. I'll probably do it for fun anyway...

yuhhaurlin, 2018-02-14:
Yes. Please stay with stable version. I close this issue first. Once if commits after this version is stable, I will change driver version. Thanks.

e88z4, 2018-02-14:
Using supplied firmware by yuhhaurlin, my uptime pasts 24 hours.

root@OpenWrt:~# uptime
 17:27:04 up 23:57,  load average: 0.01, 0.01, 0.00
root@OpenWrt:~# 

I am building the latest pull openwrt.git the same version as the supplied firmware. I will post the feedback here if reboot occurs.

e88z4, 2018-02-16:
@yuhhaurlin 

After switching back to latest openwrt version, I get the same kernel crash message like before. Please check the log [here](https://gist.github.com/e88z4/fa0735810553952d8d559da31d670e00)

My openwrt.git is using the same version as your image. 
`PKG_SOURCE_DATE:=2018-01-18
PKG_SOURCE_VERSION:=a149180fecec4fbaafb3f52ae46e33bd66501441
PKG_MIRROR_HASH:=e5a791182208eea4602692e5453dc67bfe87f080b6f03b1c6d1657b546f93b3`

I may be just lucky having 24 hours up time with your image. I am going to switch to your image again tomorrow and test it longer. In the meantime, it seems this issue is not yet resolved. 

@lister-wrt 
Do you have any update with your openwrt image? Did it experience crash issue mentioning mwlwifi driver in the crashlog like mine?

e88z4, 2018-02-17:
@yuhhaurlin 

I am currently using your supplied firmware for wrt1900ac v1. My router is recently reboot but I don't see crashlog. 

Can you reopen this ticket? It seems it is not yet resolved. Does your firmware have CPU power management and frequency scaling disabled?

thagabe, 2018-02-18:
@e88z4 

Documented check https://forum.lede-project.org/t/wrt1900acv1-reboots-kernel-4-9/2025/76

You need to disable the powermanagement as there is some kind of issue there. yahurlin maybe take a look as it is a persistent trouble with 4.9 + 4.14 or pls let us know who the maintainers of mvebu are

lister-wrt, 2018-02-19:
@e88z4 @yuhhaurlin it's still crashing on shelby too ([crashlog](https://gist.github.com/lister-wrt/63fc6531e95381e6f9fa8a90a4df6bb4)).

e88z4, 2018-02-19:
@thagabe 

All power management feature of the kernel have been disabled, so does CPU frequency scaling. I can reproduce the issue in all kernel version such as 4.4, 4.9, 4.14. Power management issue is affecting mamba device only. This issue is affecting mamba and shellby so far. 

@lister-wrt 

I get the same error message like yours.

yuhhaurlin, 2018-02-21:
Build OpenWrt (commit 9595a9e kernel: fix return value of usb_xhci_needs_pci_reset, got on 2018/2/21):

WRT1900AC:
Factory: https://drive.google.com/open?id=1FQPoQ-q-N_BFY1aIHca-fo2nv8vCBIaQ
Upgrade: https://drive.google.com/open?id=1WXV8r8OAKoRed7Y1xYDpFX8MFIqdZMJr
WRT1900ACS:
Factory: https://drive.google.com/open?id=1FQPoQ-q-N_BFY1aIHca-fo2nv8vCBIaQ
Upgrade: https://drive.google.com/open?id=1oGTawggwVejNLnFrw6X7eUMrlaYvEiQa

I disable PM, CPU FREQ and CPU IDLE, you can check /proc/config.gz to confirm that.
Please help to test these images and give me Oops once if the problem happened, so I can check the code to figure out the possible root cause.

e88z4, 2018-02-21:
@yuhhaurlin 

I will test your image tonight and report the Oops.

e88z4, 2018-02-22:
@yuhhaurlin 

Your image crashed after 1.5 hour. Oops can be found [here](https://gist.github.com/e88z4/fa0735810553952d8d559da31d670e00)

yuhhaurlin, 2018-02-22:
Thanks. The same as you sent to me before. I will check it.

p3x-robot, 2018-02-22:
Hi!

How are you?

Yes, I got the same error, with the latest version. Reboots the router randomly - because of the Wifi, it has never done it before (for like over 1 year) - I use @eduperez  's version (https://github.com/eduperez/mwlwifi_LEDE).

I hope the latest commit fixes this. :)

But thanks anyway for the effort and help!

Sincerely,
Patrik

p3x-robot, 2018-02-23:
For me, with the latest commit fixed it for me, it is not rebooting anymore.

gpiszczek, 2018-02-23:
In my setup this problem happens always when specific device (namely Samsung Galaxy S6) gets disconnected due to inactivity. Looking at the crashlogs problem is caused by bad ordering of calls in the `mwl_fwcmd_del_sta_streams()` . In code from head it looks like this:

```
                        if (stream->sta == sta) {
                                mwl_fwcmd_remove_stream(hw, stream);           // <-- first call
                                spin_unlock_bh(&priv->stream_lock);
                                mwl_fwcmd_destroy_ba(hw, stream,
                                                     BA_FLAG_DIRECTION_UP);   // <-- second call
                                spin_lock_bh(&priv->stream_lock);
                        }

```
`mwl_fwcmd_remove_stream()` is pretty straight forward:

```
void mwl_fwcmd_remove_stream(struct ieee80211_hw *hw,
                             struct mwl_ampdu_stream *stream)
{
        memset(stream, 0, sizeof(*stream));
}
```

After it stream structure is filled with zeros. Inside `mwl_fwcmd_destroy_ba()` there is code:

```
        pcmd->ba_info.destroy_params.flags = cpu_to_le32(ba_flags);
        pcmd->ba_info.destroy_params.fw_ba_context.context =
                cpu_to_le32(stream->idx);
        pcmd->ba_info.destroy_params.tid = stream->tid;
        ether_addr_copy(&pcmd->ba_info.destroy_params.peer_mac_addr[0],
                        stream->sta->addr);          // <-- (XXX)
```

In (XXX) line we have a dereference of `stream->sta pointer`, which was previously nulled by `memset()` in `mwl_fwcmd_remove_stream()`. This is where null pointer dereference / kernel oops happens.

Current order of call to `mwl_fwcmd_remove_stream()` and `mwl_fwcmd_destroy_ba()` was introduced by (according to git blame):

```
commit 3eadcb154d43d92e8a0b58207587ca9e0f2bd608
Author: David Lin <dlin@marvell.com>
Date:   Thu Apr 20 22:01:22 2017 +0800

    Avoid calling mutex under spin lock protection.

    Signed-off-by: David Lin <dlin@marvell.com>

```
It looks as a side effect of another change.

Before this commit it looked like this:

```
                        if (stream->sta == sta) {
                                mwl_fwcmd_destroy_ba(hw, stream,
                                                     BA_FLAG_DIRECTION_UP);
                                mwl_fwcmd_remove_stream(hw, stream);
                        }
```

First `mwl_fwcmd_destroy_ba()`, then `mwl_fwcmd_remove_stream()

To correct the issue the previous order of above calls should be restored, patch included [mwlwifi_fix270.txt](https://github.com/kaloz/mwlwifi/files/1751318/mwlwifi_fix270.txt) .

yuhhaurlin, 2018-02-23:
Thanks. I will check it.

p3x-robot, 2018-02-23:
@gpiszczek that's true, but at least it is not rebooting anymore!!! oh my god, it was a nightmare! 
it was rebooting every 1.5 hours and i was building and my builds were missing packages because of the reboots...

p3x-robot, 2018-02-23:
@gpiszczek @yuhhaurlin  yeah, i checked it, it is the wrong order, the stream is running when it is destroyed.
It should keep it remove the stream and destory then. That's very true.

p3x-robot, 2018-02-23:
@gpiszczek by the way, if you are so good, you can create a pull request... just any way.... why not... when you got some time... I dont know so much the code driver as you searched in it, but i can see the bug, that's sure.. Plus I didn't get the Android error. But it's true, Android sometimes turns off, I remember it for sure.

gpiszczek, 2018-02-23:
Submitted PR - https://github.com/kaloz/mwlwifi/pull/274 :).

yuhhaurlin, 2018-02-26:
Please help to check https://github.com/kaloz/mwlwifi/commit/51350a3e085a811179a270baddcae03ec9384f07. Thanks.

p3x-robot, 2018-02-26:
I tested, on Android, it is twice speed.

p3x-robot, 2018-02-26:
You can test here on LEDE 17.01.4:  
https://cdn.corifeus.com/archive/

Building a full re-build OpenWRT master and LEDE 17.01.4.

e88z4, 2018-02-27:
I built my own using the pull request from @gpiszczek three days ago. System is up for more than 2 days. 

19:54:15 up 2 days,  7:51,  load average: 0.01, 0.01, 0.00

I will report stability if the system passes 5 days up time.

yuhhaurlin, 2018-02-27:
From the log, the patch should fix this issue. I close this issue first. Thanks.

[1 more reply]

closed 2018-02-27 by yuhhaurlin