kaloz/mwlwifi #272: mlwifi crash

pvanamst, 2018-02-15:
I am running dd-wrt r34578 and believe the mlwifi driver crashed. I logged into the box with ssh and found hostapd not running with dmesg showing the following:

(Note the 'phy0: staid ? deleted' repeat all the time so likely can be ignored) 

[2092099.544848] ieee80211 phy0: staid 3 deleted
[2092420.584912] ieee80211 phy0: staid 3 deleted
[2092804.482629] Unable to handle kernel NULL pointer dereference at virtual address 0000000c
[2092804.490954] pgd = d4df4000
[2092804.493845] [0000000c] *pgd=144bc831, *pte=00000000, *ppte=00000000
[2092804.500351] Internal error: Oops: 17 [#1] PREEMPT SMP ARM
[2092804.505947] Modules linked in: ebt_ip6 ebt_ip ebtable_filter ebtables shortcut_fe btmrvl_sdio btmrvl bluetooth mwifiex_sdio mvsdio sdhci_pxav3 sdhci_pltfm sdhci mmc_block mmc_core mwifiex mwlwifi mac80211 cfg80211 compat mii tmp421 pwm_fan leds_pca963x leds_tlc591xx orion_wdt
[2092804.530925] CPU: 1 PID: 1229 Comm: hostapd Not tainted 4.9.77 #94
[2092804.537217] Hardware name: Marvell Armada 380/385 (Device Tree)
[2092804.543336] task: df47f900 task.stack: d5bde000
[2092804.548066] PC is at _165+0x98/0xf0 [mwlwifi]
[2092804.552618] LR is at _165+0x34/0xf0 [mwlwifi]
[2092804.557167] pc : [<bf0c45c4>]    lr : [<bf0c4560>]    psr: 60000013
[2092804.557167] sp : d5bdfc88  ip : 00000000  fp : 00000000
[2092804.569044] r10: de780bc0  r9 : da5b0dd0  r8 : de5b0060
[2092804.574464] r7 : de7828a4  r6 : de7815a0  r5 : 00000000  r4 : de56c000
[2092804.581194] r3 : 00000000  r2 : 00000002  r1 : 00000000  r0 : 00000000
[2092804.587924] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
[2092804.595264] Control: 10c5387d  Table: 14df404a  DAC: 00000051
[2092804.601208] Process hostapd (pid: 1229, stack limit = 0xd5bde210)
[2092804.607500] Stack: (0xd5bdfc88 to 0xd5be0000)
[2092804.612050] fc80:                   00000008 de7835a0 da5b0dd0 de784560 de5b0060 bf0c491c
[2092804.620438] fca0: da5b0dd0 de7815a0 de780bc0 d50bec68 00000000 00000014 de450000 bf0bfba8
[2092804.628826] fcc0: da5b0800 00000002 d50be4c0 d3a55800 00000000 00000014 de450000 bf060004
[2092804.637214] fce0: da5b0800 bf087930 c0911d00 da5b0800 00000002 bf062eb0 00000002 00000000
[2092804.645602] fd00: 00000014 da5b0800 de780bc0 bf06384c d3a55800 00000000 00000000 da5b0800
[2092804.653990] fd20: d3a55420 bf0639b8 d50be4c0 00000454 d3a55420 bf063a30 de780000 d50be000
[2092804.662378] fd40: d5bdfd68 bf0296c0 d3a55420 0002000c bf0523a4 bf048600 d3a55414 c045b828
[2092804.670767] fd60: bf047ed0 d3a55800 00045386 000004ca d3a55400 d3a55410 d3a55414 de450000
[2092804.679155] fd80: de780000 d50be000 d3a55400 d3a55800 c045b508 00000028 d5bdfdcc 00000000
[2092804.687543] fda0: 00000000 c045aaec c0927b84 d3a55800 de510c00 c045b4f8 df4fe000 d3a55800
[2092804.695931] fdc0: de510c00 c045a490 00000008 7fffffff 00000028 d5bdff68 00000028 de510c00
[2092804.704319] fde0: 00000008 d3a55800 024000c0 c045a92c 00000000 c02c5420 d5bdfe70 d5bdfe04
[2092804.712707] fe00: 0000000c d44d7780 00000000 000004cd 00000000 00000000 00000000 d5bdff68
[2092804.721095] fe20: 00000000 00000000 df020480 d5bdfe50 00000000 d5bdfe50 000000b2 c03f0834
[2092804.729483] fe40: d5bdff68 c03f1270 ffffe000 00000000 d5bdff68 00000044 00000000 beb91394
[2092804.737871] fe60: 00000000 d5bdfeb8 df020480 c03f1384 01ad2020 00000028 01ad3030 00004000
[2092804.746258] fe80: 00000000 00000001 00000002 00000000 00000002 0000002c de593d80 0000002c
[2092804.754646] fea0: de593d88 d50be000 c0925b00 beb91424 00000010 00000000 00000000 00000000
[2092804.763034] fec0: 00000000 de54bc00 60000013 00000000 00000001 00000000 de54ba5c de54b9c0
[2092804.771422] fee0: 00000000 de54b9c0 00000007 de695548 df04740c 00000007 00000000 00000000
[2092804.779810] ff00: df47f900 c0158ca8 d5bdff08 d5bdff08 d5bde000 de695548 df0473a0 df04740c
[2092804.788198] ff20: dfbe0680 c016d034 df0473a0 00000000 df1667f8 beb913f4 00000000 df020480
[2092804.796586] ff40: 00000128 c0107a04 d5bde000 00000000 000000b2 c03f1f20 00000000 00000000
[2092804.804974] ff60: 00000000 fffffff7 d5bdfeb0 0000000c 00000001 00000000 00000000 d5bdfe78
[2092804.813362] ff80: 00000000 c013b2ec 00000000 00000000 00000000 00000000 00000000 00000000
[2092804.821750] ffa0: b6f4edd0 c0107840 00000000 00000000 00000005 beb913f4 00000000 00000000
[2092804.830138] ffc0: 00000000 00000000 b6f4edd0 00000128 b6f4edd0 00000000 b6f049b0 000000b2
[2092804.838526] ffe0: beb91380 beb9136c b6fb22d0 b6fb17c4 60000010 00000005 ffffffff ffffffd7
[2092804.846927] [<bf0c45c4>] (_165 [mwlwifi]) from [<bf0c491c>] (_164+0x88/0x2e8 [mwlwifi])
[2092804.855146] [<bf0c491c>] (_164 [mwlwifi]) from [<bf0bfba8>] (_240+0x38/0x1ec [mwlwifi])
[2092804.863381] [<bf0bfba8>] (_240 [mwlwifi]) from [<bf060004>] (_17+0x150/0x198 [mac80211])
[2092804.871702] [<bf060004>] (_17 [mac80211]) from [<bf062eb0>] (_813+0x210/0x260 [mac80211])
[2092804.880109] [<bf062eb0>] (_813 [mac80211]) from [<bf06384c>] (_822+0x8b8/0xa04 [mac80211])
[2092804.888603] [<bf06384c>] (_822 [mac80211]) from [<bf0639b8>] (_801+0x20/0x28 [mac80211])
[2092804.896921] [<bf0639b8>] (_801 [mac80211]) from [<bf063a30>] (_803+0x2c/0x44 [mac80211])
[2092804.905242] [<bf063a30>] (_803 [mac80211]) from [<bf0296c0>] (_308+0xe8/0xf8 [cfg80211])
[2092804.913554] [<bf0296c0>] (_308 [cfg80211]) from [<c045b828>] (genl_rcv_msg+0x320/0x3ac)
[2092804.921768] [<c045b828>] (genl_rcv_msg) from [<c045aaec>] (netlink_rcv_skb+0x58/0xbc)
[2092804.929807] [<c045aaec>] (netlink_rcv_skb) from [<c045b4f8>] (genl_rcv+0x24/0x34)
[2092804.937497] [<c045b4f8>] (genl_rcv) from [<c045a490>] (netlink_unicast+0x17c/0x22c)
[2092804.945361] [<c045a490>] (netlink_unicast) from [<c045a92c>] (netlink_sendmsg+0x32c/0x36c)
[2092804.953839] [<c045a92c>] (netlink_sendmsg) from [<c03f0834>] (sock_sendmsg+0x14/0x3c)
[2092804.961880] [<c03f0834>] (sock_sendmsg) from [<c03f1270>] (___sys_sendmsg+0x208/0x21c)
[2092804.970007] [<c03f1270>] (___sys_sendmsg) from [<c03f1f20>] (__sys_sendmsg+0x40/0x64)
[2092804.978050] [<c03f1f20>] (__sys_sendmsg) from [<c0107840>] (ret_fast_syscall+0x0/0x48)
[2092804.986178] Code: e5840010 e5d81004 e5c41014 e598c000 (e59c300c) 
[2092804.992650] ---[ end trace 91be088a3b4cde43 ]---

Looking for input if this is a mlwifi issue, or a dd-wrt kernel issue..

BrainSlayer, 2018-02-15:
could you use a more recent version? 02-12-2018-r34929 is latest published.

pvanamst, 2018-02-15:
Would be happy to.. I'll let you know if it happens again..

yuhhaurlin, 2018-02-15:
Please don't use latest commit.

yuhhaurlin, 2018-02-15:
So what the issue is? Can you give me function call?

aaron1959, 2018-02-15:
I can't reproduce on wrt3200acm with 34929.

yuhhaurlin, 2018-02-15:
Thanks. I close this one. If you have the the issue with latest release, you can create another issue to track it.

closed 2018-02-15 by yuhhaurlin