kaloz/mwlwifi #273: Small question Linksys WRT 1900ACS => opkg remove kmod-mwifiex-sdio and opkg remove mwifiex-sdio-firmware

p3x-robot, 2018-02-18:
Ciao !

How are you?
Because I have a default build I removed from the firwmare not only 3200ACM but the 1900ACS as well.

```bash
opkg remove kmod-mwifiex-sdio
opkg remove mwifiex-sdio-firmware
```
basically it is not in the ```.config``` and i added in the ```image builder``` like this ``` -mwifiex-sdio-firmware -kmod-mwifiex-sdio```.

Is it a problem for the 1900ACS?

Thanks,
Patrik

yuhhaurlin, 2018-02-19:
Mwifiex only works for WRT3200ACM (phy2).

closed 2018-02-19 by yuhhaurlin

p3x-robot, 2018-02-19:
Ciao!
How are you?

But the ```readme.md``` says I should disable ```mwifiex```. So now I should use it or not?
I disabled for the Linksys WRT completely, so is that right?

Thanks,
Patrik

yuhhaurlin, 2018-02-19:
You only needs to do this for WRT3200ACM if you want to make sure DFS can work. This is one way to let DFS can work for WRT3200ACM.

p3x-robot, 2018-02-19:
ok, thanks so much! i guess it is working, both radar and DFS,