kaloz/mwlwifi #278: Wireless camera unable to connect (2.4ghz)

iganeshk, 2018-03-03:
I am trying to setup the wireless camera over 2.4ghz access point on 3200ACM running openwrt r6302/r6365 using the latest driver on 3200ACM with all the possible encryption and channels but it just fails after authentication and hence dhcp is unable to detect and allocate an IP to it. Here are the logs that I observed on various settings.

```
Fri Mar  2 14:07:05 2018 daemon.info hostapd: wlan1: STA 88:83:5d:XX:XX:XX IEEE 802.11: associated (aid 4)
Fri Mar  2 14:07:06 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 88:83:5d:XX:XX:XX
Fri Mar  2 14:07:06 2018 daemon.info hostapd: wlan1: STA 88:83:5d:XX:XX:XX WPA: pairwise key handshake completed (RSN)
Fri Mar  2 14:07:17 2018 daemon.info hostapd: wlan1: STA 88:83:5d:XX:XX:XX IEEE 802.11: authenticated
Fri Mar  2 14:07:51 2018 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 88:83:5d:XX:XX:XX
Fri Mar  2 14:07:51 2018 kern.debug kernel: [43300.431211] ieee80211 phy1: staid 4 deleted
```

```
Sat Mar  3 02:53:37 2018 kern.debug kernel: [43405.708782] ieee80211 phy1: staid 4 deleted
Sat Mar  3 02:53:40 2018 daemon.info hostapd: wlan1: STA 88:83:5d:b1:f8:38 IEEE 802.11: deauthenticated due to local deauth request
Sat Mar  3 02:53:41 2018 daemon.notice hostapd: handle_auth_cb: STA 88:83:5d:b1:f8:38 not found
Sat Mar  3 02:53:41 2018 daemon.notice hostapd: handle_auth_cb: STA 88:83:5d:b1:f8:38 not found
Sat Mar  3 02:53:41 2018 daemon.notice hostapd: handle_auth_cb: STA 88:83:5d:b1:f8:38 not found
Sat Mar  3 02:53:43 2018 daemon.info hostapd: wlan1: STA 88:83:5d:b1:f8:38 IEEE 802.11: associated (aid 4)
Sat Mar  3 02:53:43 2018 daemon.notice hostapd: wlan1: AP-STA-POSSIBLE-PSK-MISMATCH 88:83:5d:b1:f8:38
Sat Mar  3 02:53:44 2018 daemon.notice hostapd: wlan1: AP-STA-POSSIBLE-PSK-MISMATCH 88:83:5d:b1:f8:38
Sat Mar  3 02:53:45 2018 daemon.notice hostapd: wlan1: AP-STA-POSSIBLE-PSK-MISMATCH 88:83:5d:b1:f8:38
```

After allocating static ip:
```
Fri Mar  2 14:10:22 2018 daemon.info hostapd: wlan1-1: STA 88:83:5d:XX:XX:XX IEEE 802.11: associated (aid 1)
Fri Mar  2 14:10:23 2018 daemon.notice hostapd: wlan1-1: AP-STA-CONNECTED 88:83:5d:XX:XX:XX
Fri Mar  2 14:10:23 2018 daemon.info hostapd: wlan1-1: STA 88:83:5d:XX:XX:XX WPA: pairwise key handshake completed (WPA)
Fri Mar  2 14:10:23 2018 daemon.info hostapd: wlan1-1: STA 88:83:5d:XX:XX:XX WPA: group key handshake completed (WPA)
Fri Mar  2 14:10:27 2018 daemon.warn dnsmasq-dhcp[3775]: DHCP packet received on wlan1-1 which has no address
Fri Mar  2 14:10:30 2018 daemon.warn dnsmasq-dhcp[3775]: DHCP packet received on wlan1-1 which has no address
Fri Mar  2 14:10:33 2018 daemon.warn dnsmasq-dhcp[3775]: DHCP packet received on wlan1-1 which has no address
Fri Mar  2 14:10:37 2018 daemon.warn dnsmasq-dhcp[3775]: DHCP packet received on wlan1-1 which has no address
Fri Mar  2 14:10:40 2018 daemon.warn dnsmasq-dhcp[3775]: DHCP packet received on wlan1-1 which has no address
Fri Mar  2 14:10:43 2018 daemon.warn dnsmasq-dhcp[3775]: DHCP packet received on wlan1-1 which has no address
Fri Mar  2 14:10:43 2018 daemon.info hostapd: wlan1-1: STA 88:83:5d:XX:XX:XX IEEE 802.11: authenticated
Fri Mar  2 14:10:48 2018 daemon.warn dnsmasq-dhcp[3775]: DHCP packet received on wlan1-1 which has no address
Fri Mar  2 14:10:51 2018 daemon.warn dnsmasq-dhcp[3775]: DHCP packet received on wlan1-1 which has no address
Fri Mar  2 14:10:54 2018 daemon.warn dnsmasq-dhcp[3775]: DHCP packet received on wlan1-1 which has no address
Fri Mar  2 14:10:59 2018 daemon.warn dnsmasq-dhcp[3775]: DHCP packet received on wlan1-1 which has no address
Fri Mar  2 14:11:02 2018 daemon.warn dnsmasq-dhcp[3775]: DHCP packet received on wlan1-1 which has no address
Fri Mar  2 14:11:05 2018 daemon.warn dnsmasq-dhcp[3775]: DHCP packet received on wlan1-1 which has no address
Fri Mar  2 14:11:08 2018 daemon.notice hostapd: wlan1-1: AP-STA-DISCONNECTED 88:83:5d:XX:XX:XX
```

At first I though it was an issue with dnsmasq, but then I was able to connect the camera with radio2 over 2.4ghz.

yuhhaurlin, 2018-03-03:
1. AP-STA-POSSIBLE-PSK-MISMATCH?
2. If static ip is used, why DHCP packets still sent out? BTW, wlan1-1 is used for multi-BSSID? Can you give your setting for the router?

iganeshk, 2018-03-03:
Log 1: Openwrt r6302 (davidc502 build), Kernel 4.9.82
Log 3: Openwrt r6302 (davidc502 build), Kernel 4.9.82, multi-wlan1 bssid to test multiple encryption is working
Log 2: Openwrt r6365 (snapshot), Kernel 4.14.23

Wireless config:
```
config wifi-device 'radio0'
	option type 'mac80211'
	option channel '36'
	option hwmode '11a'
	option path 'soc/soc:pcie-controller/pci0000:00/0000:00:01.0/0000:01:00.0'
	option htmode 'VHT80'
	option disabled '1'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'OpenWrt'
	option encryption 'none'

config wifi-device 'radio1'
	option type 'mac80211'
	option hwmode '11g'
	option path 'soc/soc:pcie-controller/pci0000:00/0000:00:02.0/0000:02:00.0'
	option country 'US'
	option legacy_rates '1'
	option htmode 'HT20'
	option txpower '30'
	option channel '2'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid '2_4_SSID'
	option key 'SECURE_PASS'
	option wmm '0'
	option encryption 'psk2'
```
5Ghz disabled (different router), 2.4ghz opened for IoT devices (Espressif)

iganeshk, 2018-03-03:
radio2 went out shortly (12hrs), is it because it's used for DFS scan?

```Sat Mar  3 16:49:49 2018 kern.warn kernel: [42473.599540] ------------[ cut here ]------------
Sat Mar  3 16:49:49 2018 kern.warn kernel: [42473.604192] WARNING: CPU: 1 PID: 0 at net/sched/sch_generic.c:320 dev_watchdog+0x16c/0x248
Sat Mar  3 16:49:49 2018 kern.info kernel: [42473.612497] NETDEV WATCHDOG: wlan2 (mwifiex_sdio): transmit queue 1 timed out
Sat Mar  3 16:49:49 2018 kern.warn kernel: [42473.619665] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_CT slhc rfcomm nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack mwifiex_sdio mwifiex iptable_mangle iptable_filter ip_tables hidp hci_uart crc_ccitt btusb btmrvl_sdio btmrvl btintel bnep bluetooth hid evdev input_core mwlwifi mac80211 cfg80211 compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ecdh_generic kpp ecb cmac gpio_button_hotplug
Sat Mar  3 16:49:49 2018 kern.warn kernel: [42473.687105] CPU: 1 PID: 0 Comm: swapper/1 Not tainted 4.14.23 #0
Sat Mar  3 16:49:49 2018 kern.warn kernel: [42473.693137] Hardware name: Marvell Armada 380/385 (Device Tree)
Sat Mar  3 16:49:49 2018 kern.warn kernel: [42473.699097] [<c010f3d0>] (unwind_backtrace) from [<c010b134>] (show_stack+0x10/0x14)
Sat Mar  3 16:49:49 2018 kern.warn kernel: [42473.706880] [<c010b134>] (show_stack) from [<c060e4c4>] (dump_stack+0x8c/0xa0)
Sat Mar  3 16:49:49 2018 kern.warn kernel: [42473.714137] [<c060e4c4>] (dump_stack) from [<c0121e1c>] (__warn+0xe4/0x100)
Sat Mar  3 16:49:49 2018 kern.warn kernel: [42473.721130] [<c0121e1c>] (__warn) from [<c0121e70>] (warn_slowpath_fmt+0x38/0x48)
Sat Mar  3 16:49:49 2018 kern.warn kernel: [42473.728646] [<c0121e70>] (warn_slowpath_fmt) from [<c053b518>] (dev_watchdog+0x16c/0x248)
Sat Mar  3 16:49:49 2018 kern.warn kernel: [42473.736864] [<c053b518>] (dev_watchdog) from [<c016f340>] (call_timer_fn+0x24/0x98)
Sat Mar  3 16:49:49 2018 kern.warn kernel: [42473.744556] [<c016f340>] (call_timer_fn) from [<c016f510>] (run_timer_softirq+0x15c/0x1e8)
Sat Mar  3 16:49:49 2018 kern.warn kernel: [42473.752857] [<c016f510>] (run_timer_softirq) from [<c01015f0>] (__do_softirq+0xd0/0x21c)
Sat Mar  3 16:49:49 2018 kern.warn kernel: [42473.760986] [<c01015f0>] (__do_softirq) from [<c01265f4>] (irq_exit+0xb0/0xd8)
Sat Mar  3 16:49:49 2018 kern.warn kernel: [42473.768245] [<c01265f4>] (irq_exit) from [<c015d388>] (__handle_domain_irq+0x9c/0xac)
Sat Mar  3 16:49:49 2018 kern.warn kernel: [42473.776111] [<c015d388>] (__handle_domain_irq) from [<c0101438>] (gic_handle_irq+0x58/0x98)
Sat Mar  3 16:49:49 2018 kern.warn kernel: [42473.784499] [<c0101438>] (gic_handle_irq) from [<c010bdcc>] (__irq_svc+0x6c/0x90)
Sat Mar  3 16:49:49 2018 kern.warn kernel: [42473.792013] Exception stack(0xdf465f80 to 0xdf465fc8)
Sat Mar  3 16:49:49 2018 kern.warn kernel: [42473.797087] 5f80: 00000001 00000000 00000000 c0114b60 df464000 c0946410 df465fd8 00000000
Sat Mar  3 16:49:49 2018 kern.warn kernel: [42473.805301] 5fa0: 00000000 414fc091 00000000 00000000 df465fc8 df465fd0 c01082fc c0108300
Sat Mar  3 16:49:49 2018 kern.warn kernel: [42473.813513] 5fc0: 60000013 ffffffff
Sat Mar  3 16:49:49 2018 kern.warn kernel: [42473.817017] [<c010bdcc>] (__irq_svc) from [<c0108300>] (arch_cpu_idle+0x34/0x38)
Sat Mar  3 16:49:49 2018 kern.warn kernel: [42473.824452] [<c0108300>] (arch_cpu_idle) from [<c01540f4>] (do_idle+0xd0/0x188)
Sat Mar  3 16:49:49 2018 kern.warn kernel: [42473.831795] [<c01540f4>] (do_idle) from [<c0154408>] (cpu_startup_entry+0x18/0x1c)
Sat Mar  3 16:49:49 2018 kern.warn kernel: [42473.839398] [<c0154408>] (cpu_startup_entry) from [<001017cc>] (0x1017cc)
Sat Mar  3 16:49:49 2018 kern.warn kernel: [42473.846223] ---[ end trace 13e6eee65b64e4a8 ]---
Sat Mar  3 16:49:49 2018 kern.info kernel: [42473.850868] mwifiex_sdio mmc0:0001:1: 4215929 : Tx timeout(#1), bss_type-num = 1-0
Sat Mar  3 16:49:49 2018 daemon.info hostapd: wlan2: STA dc:4f:22:c3:c4:8f IEEE 802.11: associated
Sat Mar  3 16:49:52 2018 daemon.info hostapd: wlan2: STA dc:4f:22:c3:c4:8f IEEE 802.11: disassociated
Sat Mar  3 16:49:54 2018 daemon.info hostapd: wlan2: STA dc:4f:22:c3:c4:8f IEEE 802.11: associated
Sat Mar  3 16:49:55 2018 kern.info kernel: [42479.681635] mwifiex_sdio mmc0:0001:1: 4216512 : Tx timeout(#2), bss_type-num = 1-0
Sat Mar  3 16:49:57 2018 daemon.info hostapd: wlan2: STA dc:4f:22:c3:c4:8f IEEE 802.11: disassociated
Sat Mar  3 16:49:59 2018 daemon.info hostapd: wlan2: STA dc:4f:22:c3:c4:8f IEEE 802.11: associated
Sat Mar  3 16:50:03 2018 daemon.info hostapd: wlan2: STA dc:4f:22:c3:c4:8f IEEE 802.11: disassociated
Sat Mar  3 16:50:05 2018 kern.info kernel: [42489.605060] mwifiex_sdio mmc0:0001:1: 4217504 : Tx timeout(#3), bss_type-num = 1-0
Sat Mar  3 16:50:06 2018 daemon.info hostapd: wlan2: STA dc:4f:22:c3:c4:8f IEEE 802.11: associated
Sat Mar  3 16:50:09 2018 daemon.info hostapd: wlan2: STA dc:4f:22:c3:c4:8f IEEE 802.11: disassociated
Sat Mar  3 16:50:11 2018 daemon.info hostapd: wlan2: STA dc:4f:22:c3:c4:8f IEEE 802.11: associated
Sat Mar  3 16:50:14 2018 daemon.info hostapd: wlan2: STA dc:4f:22:c3:c4:8f IEEE 802.11: disassociated
[... cut ...]
Sat Mar  3 16:50:56 2018 kern.info kernel: [42540.567703] mwifiex_sdio mmc0:0001:1: last_event_index = 1
Sat Mar  3 16:50:56 2018 kern.info kernel: [42540.573215] mwifiex_sdio mmc0:0001:1: last_event: 43 00 76 00 2c 00 2d 00 2c 00
Sat Mar  3 16:50:56 2018 kern.info kernel: [42540.580559] mwifiex_sdio mmc0:0001:1: data_sent=1 cmd_sent=1
Sat Mar  3 16:50:56 2018 kern.info kernel: [42540.586245] mwifiex_sdio mmc0:0001:1: ps_mode=0 ps_state=0
Sat Mar  3 16:50:56 2018 kern.info kernel: [42540.591899] mwifiex_sdio mmc0:0001:1: info: _mwifiex_fw_dpc: unregister device
```

yuhhaurlin, 2018-03-04:
1. Radio 2 is not mwlwifi.
2. Can you use the same version to test DHCP and static IP? Please get the log for me.
3. Don't use WEP or TKIP.

iganeshk, 2018-03-04:
Log 1 repeats for DHCP and static IP. If I create a secondary radio on wlan1, I get Log 3. Logread doesn't output anything else, its just keep repeating.

iganeshk, 2018-03-04:
On a single session, after making several changes to the AP(2.4)'s settings, the rest of the devices doesn't seem to get discovered as well. Here are 2 other ESP's log:

```
Sun Mar  4 00:13:37 2018 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED dc:4f:22:xx:xx:xx
Sun Mar  4 00:13:37 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: disassociated
Sun Mar  4 00:13:37 2018 kern.debug kernel: [69110.826839] ieee80211 phy1: staid 3 deleted
Sun Mar  4 00:13:38 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sun Mar  4 00:13:40 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: associated (aid 3)
Sun Mar  4 00:13:40 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED dc:4f:22:xx:xx:xx
Sun Mar  4 00:13:40 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx WPA: pairwise key handshake completed (RSN)
Sun Mar  4 00:13:40 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: authenticated
Sun Mar  4 00:13:40 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: authenticated
Sun Mar  4 00:13:59 2018 kern.debug kernel: [69132.918797] ieee80211 phy1: staid 3 deleted
Sun Mar  4 00:14:03 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: associated (aid 3)
Sun Mar  4 00:14:03 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED dc:4f:22:xx:xx:xx
Sun Mar  4 00:14:03 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx WPA: pairwise key handshake completed (RSN)
Sun Mar  4 00:14:06 2018 kern.debug kernel: [69140.408993] ieee80211 phy1: staid 1 deleted
Sun Mar  4 00:14:06 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: associated (aid 1)
Sun Mar  4 00:14:06 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED dc:4f:22:xx:xx:xx
Sun Mar  4 00:14:06 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx WPA: pairwise key handshake completed (RSN)
Sun Mar  4 00:14:15 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: authenticated
Sun Mar  4 00:14:15 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: authenticated
Sun Mar  4 00:14:15 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: authenticated
Sun Mar  4 00:14:22 2018 kern.debug kernel: [69156.009745] ieee80211 phy1: staid 3 deleted
Sun Mar  4 00:14:22 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: associated (aid 3)
Sun Mar  4 00:14:22 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED dc:4f:22:xx:xx:xx
Sun Mar  4 00:14:22 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx WPA: pairwise key handshake completed (RSN)
Sun Mar  4 00:14:38 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: authenticated
Sun Mar  4 00:14:42 2018 authpriv.info dropbear[23493]: Child connection from 192.168.1.151:54099
Sun Mar  4 00:14:48 2018 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED dc:4f:22:xx:xx:xx
Sun Mar  4 00:14:48 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: disassociated
Sun Mar  4 00:14:48 2018 kern.debug kernel: [69181.811196] ieee80211 phy1: staid 3 deleted
Sun Mar  4 00:14:49 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sun Mar  4 00:14:51 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: associated (aid 3)
Sun Mar  4 00:14:51 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED dc:4f:22:xx:xx:xx
Sun Mar  4 00:14:51 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx WPA: pairwise key handshake completed (RSN)
Sun Mar  4 00:15:05 2018 kern.debug kernel: [69199.324638] ieee80211 phy1: staid 1 deleted
Sun Mar  4 00:15:05 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: associated (aid 1)
Sun Mar  4 00:15:05 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED dc:4f:22:xx:xx:xx
Sun Mar  4 00:15:05 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx WPA: pairwise key handshake completed (RSN)
Sun Mar  4 00:15:07 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: authenticated
Sun Mar  4 00:15:07 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: authenticated
Sun Mar  4 00:15:25 2018 kern.debug kernel: [69219.207861] ieee80211 phy1: staid 3 deleted
Sun Mar  4 00:15:25 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: associated (aid 3)
Sun Mar  4 00:15:25 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED dc:4f:22:xx:xx:xx
Sun Mar  4 00:15:25 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx WPA: pairwise key handshake completed (RSN)
```

after successful handshake, does mwlwifi notify about the new device to DHCP?

iganeshk, 2018-03-04:
Second update: Disabling encryption allows all the three devices to login (discovered by DHCP).

```
Sun Mar  4 00:32:15 2018 daemon.err hostapd: Using interface wlan1 with hwaddr 60:38:e0:bc:fd:c9 and ssid "BSSID_NAME"
Sun Mar  4 00:32:15 2018 kern.debug kernel: [70228.922074] ieee80211 phy1: change: 0x100
Sun Mar  4 00:32:15 2018 kern.debug kernel: [70228.935141] ieee80211 phy1: change: 0x42
Sun Mar  4 00:32:15 2018 kern.info kernel: [70229.105066] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
Sun Mar  4 00:32:15 2018 daemon.notice hostapd: wlan1: interface state COUNTRY_UPDATE->ENABLED
Sun Mar  4 00:32:15 2018 daemon.notice hostapd: wlan1: AP-ENABLED
Sun Mar  4 00:32:15 2018 daemon.notice netifd: Network device 'wlan1' link is up
Sun Mar  4 00:32:16 2018 daemon.notice hostapd: Station dc:4f:22:xx:xx:xx trying to disassociate, but it is not associated
Sun Mar  4 00:32:16 2018 daemon.notice hostapd: Station dc:4f:22:xx:xx:xx trying to disassociate, but it is not associated
Sun Mar  4 00:32:16 2018 daemon.notice hostapd: Station dc:4f:22:xx:xx:xx trying to disassociate, but it is not associated
Sun Mar  4 00:32:16 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: associated (aid 1)
Sun Mar  4 00:32:16 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED dc:4f:22:xx:xx:xx
Sun Mar  4 00:32:16 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: associated (aid 2)
Sun Mar  4 00:32:16 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED dc:4f:22:xx:xx:xx
Sun Mar  4 00:32:16 2018 daemon.info dnsmasq-dhcp[30747]: DHCPDISCOVER(br-lan) dc:4f:22:xx:xx:xx
Sun Mar  4 00:32:16 2018 daemon.info dnsmasq-dhcp[30747]: DHCPOFFER(br-lan) 192.168.1.235 dc:4f:22:xx:xx:xx
Sun Mar  4 00:32:16 2018 daemon.info dnsmasq-dhcp[30747]: DHCPREQUEST(br-lan) 192.168.1.235 dc:4f:22:xx:xx:xx
Sun Mar  4 00:32:16 2018 daemon.info dnsmasq-dhcp[30747]: DHCPACK(br-lan) 192.168.1.235 dc:4f:22:xx:xx:xx ESP_C0F513
Sun Mar  4 00:32:16 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: authenticated
Sun Mar  4 00:32:16 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: authenticated
Sun Mar  4 00:32:16 2018 daemon.info dnsmasq-dhcp[30747]: DHCPDISCOVER(br-lan) dc:4f:22:xx:xx:xx
Sun Mar  4 00:32:16 2018 daemon.info dnsmasq-dhcp[30747]: DHCPOFFER(br-lan) 192.168.1.223 dc:4f:22:xx:xx:xx
Sun Mar  4 00:32:16 2018 daemon.info dnsmasq-dhcp[30747]: DHCPREQUEST(br-lan) 192.168.1.235 dc:4f:22:xx:xx:xx
Sun Mar  4 00:32:16 2018 daemon.info dnsmasq-dhcp[30747]: DHCPACK(br-lan) 192.168.1.235 dc:4f:22:xx:xx:xx ESP_C0F513
Sun Mar  4 00:32:16 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: associated (aid 3)
Sun Mar  4 00:32:16 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED dc:4f:22:xx:xx:xx
Sun Mar  4 00:32:18 2018 daemon.info dnsmasq-dhcp[30747]: DHCPDISCOVER(br-lan) dc:4f:22:xx:xx:xx
Sun Mar  4 00:32:18 2018 daemon.info dnsmasq-dhcp[30747]: DHCPOFFER(br-lan) 192.168.1.223 dc:4f:22:xx:xx:xx
Sun Mar  4 00:32:19 2018 kern.debug kernel: [70233.755306] ieee80211 phy1: staid 3 deleted
Sun Mar  4 00:32:19 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: associated (aid 3)
Sun Mar  4 00:32:19 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED dc:4f:22:xx:xx:xx
Sun Mar  4 00:32:19 2018 daemon.info dnsmasq-dhcp[30747]: DHCPDISCOVER(br-lan) dc:4f:22:xx:xx:xx
Sun Mar  4 00:32:19 2018 daemon.info dnsmasq-dhcp[30747]: DHCPOFFER(br-lan) 192.168.1.249 dc:4f:22:xx:xx:xx
Sun Mar  4 00:32:19 2018 daemon.info dnsmasq-dhcp[30747]: DHCPREQUEST(br-lan) 192.168.1.249 dc:4f:22:xx:xx:xx
Sun Mar  4 00:32:19 2018 daemon.info dnsmasq-dhcp[30747]: DHCPACK(br-lan) 192.168.1.249 dc:4f:22:xx:xx:xx ESP_84C1B5
Sun Mar  4 00:32:22 2018 daemon.info dnsmasq-dhcp[30747]: DHCPDISCOVER(br-lan) dc:4f:22:xx:xx:xx
Sun Mar  4 00:32:22 2018 daemon.info dnsmasq-dhcp[30747]: DHCPOFFER(br-lan) 192.168.1.223 dc:4f:22:xx:xx:xx
Sun Mar  4 00:32:22 2018 daemon.info dnsmasq-dhcp[30747]: DHCPREQUEST(br-lan) 192.168.1.223 dc:4f:22:xx:xx:xx
Sun Mar  4 00:32:22 2018 daemon.info dnsmasq-dhcp[30747]: DHCPACK(br-lan) 192.168.1.223 dc:4f:22:xx:xx:xx ESP_C3C48F
Sun Mar  4 00:32:27 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: authenticated
Sun Mar  4 00:32:27 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: authenticated
Sun Mar  4 00:32:55 2018 kern.debug kernel: [70269.223629] ieee80211 phy1: staid 1 deleted
Sun Mar  4 00:32:55 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: associated (aid 1)
Sun Mar  4 00:32:55 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED dc:4f:22:xx:xx:xx
Sun Mar  4 00:32:55 2018 daemon.info dnsmasq-dhcp[30747]: DHCPDISCOVER(br-lan) dc:4f:22:xx:xx:xx
Sun Mar  4 00:32:55 2018 daemon.info dnsmasq-dhcp[30747]: DHCPOFFER(br-lan) 192.168.1.223 dc:4f:22:xx:xx:xx
Sun Mar  4 00:32:57 2018 daemon.info dnsmasq-dhcp[30747]: DHCPDISCOVER(br-lan) dc:4f:22:xx:xx:xx
Sun Mar  4 00:32:57 2018 daemon.info dnsmasq-dhcp[30747]: DHCPOFFER(br-lan) 192.168.1.223 dc:4f:22:xx:xx:xx
Sun Mar  4 00:33:01 2018 daemon.info dnsmasq-dhcp[30747]: DHCPDISCOVER(br-lan) dc:4f:22:xx:xx:xx
Sun Mar  4 00:33:01 2018 daemon.info dnsmasq-dhcp[30747]: DHCPOFFER(br-lan) 192.168.1.223 dc:4f:22:xx:xx:xx
Sun Mar  4 00:33:01 2018 daemon.info dnsmasq-dhcp[30747]: DHCPREQUEST(br-lan) 192.168.1.223 dc:4f:22:xx:xx:xx
Sun Mar  4 00:33:01 2018 daemon.info dnsmasq-dhcp[30747]: DHCPACK(br-lan) 192.168.1.223 dc:4f:22:xx:xx:xx ESP_C3C48F
[... cut ...]
Sun Mar  4 00:33:21 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED dc:4f:22:xx:xx:xx
Sun Mar  4 00:33:21 2018 daemon.info dnsmasq-dhcp[30747]: DHCPDISCOVER(br-lan) dc:4f:22:xx:xx:xx
Sun Mar  4 00:33:21 2018 daemon.info dnsmasq-dhcp[30747]: DHCPOFFER(br-lan) 192.168.1.249 dc:4f:22:xx:xx:xx
Sun Mar  4 00:33:21 2018 daemon.info dnsmasq-dhcp[30747]: DHCPREQUEST(br-lan) 192.168.1.249 dc:4f:22:xx:xx:xx
Sun Mar  4 00:33:21 2018 daemon.info dnsmasq-dhcp[30747]: DHCPACK(br-lan) 192.168.1.249 dc:4f:22:xx:xx:xx ESP_84C1B5
```

yuhhaurlin, 2018-03-04:
Please don't disable WMM.

yuhhaurlin, 2018-03-04:
And use CCMP only.

iganeshk, 2018-03-04:
I was trying to find out if any of those were conflicting with this issue, apparently they do not. I have set the wlan to AES only (not auto nor TKIP)

yuhhaurlin, 2018-03-04:
1. Don't disable WMM and use CCMP only.
2. Only test with problem client.
3. Get log for DHCP and static IP. The logs should be different and you should check ping for static IP if connection is established.

iganeshk, 2018-03-04:
[image]

yuhhaurlin, 2018-03-04:
This is DHCP, how about static IP?

yuhhaurlin, 2018-03-04:
BTW, aid 2 and 3 are the same kind of client?

iganeshk, 2018-03-04:
I've allocated static ip for all the 3 clients. They're all same kind of client (3) and one camera which is offline.

iganeshk, 2018-03-04:
In this image [image]

I believe the clients have IP on them only because I had them on static list. There is no ip on the hostname if the client does not have a static ip

iganeshk, 2018-03-04:
Removing them from static list, the log:
[image]
```
Sun Mar  4 04:07:41 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED dc:4f:22:xx:xx:xx
Sun Mar  4 04:07:41 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx WPA: pairwise key handshake completed (RSN)
Sun Mar  4 04:07:50 2018 authpriv.info dropbear[3371]: Child connection from 192.168.1.151:57119
Sun Mar  4 04:08:07 2018 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED dc:4f:22:xx:xx:xx
Sun Mar  4 04:08:07 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: disassociated
Sun Mar  4 04:08:07 2018 kern.debug kernel: [  105.684705] ieee80211 phy1: staid 3 deleted
Sun Mar  4 04:08:08 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sun Mar  4 04:08:10 2018 kern.debug kernel: [  107.923936] ieee80211 phy1: staid 2 deleted
Sun Mar  4 04:08:10 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: associated (aid 2)
Sun Mar  4 04:08:10 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED dc:4f:22:xx:xx:xx
Sun Mar  4 04:08:10 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx WPA: pairwise key handshake completed (RSN)
Sun Mar  4 04:08:13 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: associated (aid 3)
Sun Mar  4 04:08:13 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED dc:4f:22:xx:xx:xx
Sun Mar  4 04:08:13 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx WPA: pairwise key handshake completed (RSN)
Sun Mar  4 04:08:16 2018 daemon.err uhttpd[1810]: luci: accepted login on / for root from 192.168.1.1
Sun Mar  4 04:08:26 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: authenticated
Sun Mar  4 04:08:26 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: authenticated
Sun Mar  4 04:08:26 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: authenticated
Sun Mar  4 04:08:26 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: authenticated
Sun Mar  4 04:08:44 2018 kern.debug kernel: [  141.932637] ieee80211 phy1: staid 2 deleted
Sun Mar  4 04:08:44 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: associated (aid 2)
Sun Mar  4 04:08:44 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED dc:4f:22:xx:xx:xx
Sun Mar  4 04:08:44 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx WPA: pairwise key handshake completed (RSN)
Sun Mar  4 04:08:44 2018 kern.debug kernel: [  142.629431] ieee80211 phy1: staid 3 deleted
Sun Mar  4 04:08:48 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: associated (aid 3)
Sun Mar  4 04:08:48 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED dc:4f:22:xx:xx:xx
Sun Mar  4 04:08:48 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx WPA: pairwise key handshake completed (RSN)
Sun Mar  4 04:09:22 2018 kern.debug kernel: [  180.723186] ieee80211 phy1: staid 3 deleted
Sun Mar  4 04:09:22 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: associated (aid 3)
Sun Mar  4 04:09:23 2018 kern.debug kernel: [  181.044324] ieee80211 phy1: staid 2 deleted
Sun Mar  4 04:09:23 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: associated (aid 2)
Sun Mar  4 04:09:23 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED dc:4f:22:xx:xx:xx
Sun Mar  4 04:09:23 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx WPA: pairwise key handshake completed (RSN)
Sun Mar  4 04:09:26 2018 kern.debug kernel: [  184.838938] ieee80211 phy1: staid 3 deleted
Sun Mar  4 04:09:26 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: associated (aid 3)
Sun Mar  4 04:09:28 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED dc:4f:22:xx:xx:xx
Sun Mar  4 04:09:28 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx WPA: pairwise key handshake completed (RSN)
Sun Mar  4 04:09:34 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: authenticated
Sun Mar  4 04:09:34 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: authenticated
Sun Mar  4 04:09:34 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: authenticated
Sun Mar  4 04:09:34 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: authenticated
Sun Mar  4 04:09:34 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: authenticated
Sun Mar  4 04:09:34 2018 kern.debug kernel: [  191.989466] ieee80211 phy1: staid 3 deleted
Sun Mar  4 04:09:34 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: authenticated
Sun Mar  4 04:10:06 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: associated (aid 3)
Sun Mar  4 04:10:06 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED dc:4f:22:xx:xx:xx
Sun Mar  4 04:10:06 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx WPA: pairwise key handshake completed (RSN)
Sun Mar  4 04:10:07 2018 kern.debug kernel: [  225.251038] ieee80211 phy1: staid 2 deleted
Sun Mar  4 04:10:07 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: associated (aid 2)
Sun Mar  4 04:10:07 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED dc:4f:22:xx:xx:xx
Sun Mar  4 04:10:07 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx WPA: pairwise key handshake completed (RSN)
```

yuhhaurlin, 2018-03-04:
1. For static IP, it means you set IP to the client, so the client won't ask IP from DHCP server. After connection, if ping is all right, we can make sure data encryption is all right.
2. Can you just use one client to do the test？

iganeshk, 2018-03-04:
1. The terminal screenshot in my previous post had devices allocated with static ip and that is when I get "DHCP packet received from wlan1 which has no address"

2. Setup for only ESP device with static ip and log after a reboot.
cat /etc/config/dhcp
```
config dnsmasq
    option domainneeded '1'
    option localise_queries '1'
    option rebind_protection '1'
    option rebind_localhost '1'
    option local '/lan/'
    option domain 'lan'
    option expandhosts '1'
    option authoritative '1'
    option readethers '1'
    option leasefile '/tmp/dhcp.leases'
    option resolvfile '/tmp/resolv.conf.auto'
    option nonwildcard '1'
    option localservice '1'

config dhcp 'lan'
    option interface 'lan'
    option start '100'
    option limit '150'
    option leasetime '12h'
    list dhcp_option '6,8.8.8.8,8.8.4.4'

config dhcp 'wan'
    option interface 'wan'
    option ignore '1'

config odhcpd 'odhcpd'
    option maindhcp '0'
    option leasefile '/tmp/hosts/odhcpd'
    option leasetrigger '/usr/sbin/odhcpd-update'
    option loglevel '4'

config host
    option name 'DESKTOP-XX'
    option dns '1'
    option mac 'xx:xx:xx:xx:xx'
    option ip '192.168.1.199'
    option leasetime 'infinite'

config host
    option name 'VelopNode1'
    option dns '1'
    option mac '58:ef:xx:xx:xx:xx'
    option ip '192.168.1.151'
    option leasetime 'infinite'

config host
    option name 'Hikam-0505'
    option dns '1'
    option mac '88:83:5d:xx:xx:xx'
    option ip '192.168.1.200'
    option leasetime 'infinite'

config host
    option name 'ESP_C0F513'
    option dns '1'
    option mac 'dc:4f:22:xx:xx:xx'
    option ip '192.168.1.235'
    option leasetime 'infinite'
```
cat /etc/config/wireless

```
config wifi-device 'radio0'
	option type 'mac80211'
	option channel '36'
	option hwmode '11a'
	option path 'soc/soc:pcie/pci0000:00/0000:00:01.0/0000:01:00.0'
	option htmode 'VHT80'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'OpenWrt'
	option encryption 'none'
	option disabled '1'

config wifi-device 'radio1'
	option type 'mac80211'
	option hwmode '11g'
	option path 'soc/soc:pcie/pci0000:00/0000:00:02.0/0000:02:00.0'
	option country 'US'
	option channel '11'
	option txpower '29'
	option htmode 'HT20'
	option legacy_rates '1'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'SSID'
	option key 'PASSWORD'
	option encryption 'psk2+ccmp'
	option disabled '1'
```

Logs:
```
Sun Mar  4 12:45:37 2018 daemon.err uhttpd[1810]: luci: accepted login on /admin/network/wireless for root from 192.168.1.1
Sun Mar  4 12:45:48 2018 daemon.notice netifd: radio1 (11006): command failed: Not supported (-95)
Sun Mar  4 12:45:48 2018 daemon.err hostapd: Configuration file: /var/run/hostapd-phy1.conf
Sun Mar  4 12:45:48 2018 kern.debug kernel: [  546.593951] ieee80211 phy1: change: 0xffffffff
Sun Mar  4 12:45:48 2018 kern.info kernel: [  546.683234] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
Sun Mar  4 12:45:48 2018 kern.info kernel: [  546.690075] br-lan: port 2(wlan1) entered blocking state
Sun Mar  4 12:45:48 2018 kern.info kernel: [  546.695453] br-lan: port 2(wlan1) entered disabled state
Sun Mar  4 12:45:48 2018 kern.info kernel: [  546.700905] device wlan1 entered promiscuous mode
Sun Mar  4 12:45:48 2018 daemon.notice hostapd: wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
Sun Mar  4 12:45:48 2018 daemon.err hostapd: Using interface wlan1 with hwaddr 60:38:e0:xx:xx:xx and ssid "SSID"
Sun Mar  4 12:45:49 2018 kern.debug kernel: [  546.756698] ieee80211 phy1: change: 0x100
Sun Mar  4 12:45:49 2018 kern.debug kernel: [  546.769749] ieee80211 phy1: change: 0x42
Sun Mar  4 12:45:49 2018 kern.info kernel: [  546.937221] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
Sun Mar  4 12:45:49 2018 kern.info kernel: [  546.943666] br-lan: port 2(wlan1) entered blocking state
Sun Mar  4 12:45:49 2018 kern.info kernel: [  546.949012] br-lan: port 2(wlan1) entered forwarding state
Sun Mar  4 12:45:49 2018 daemon.notice hostapd: wlan1: interface state COUNTRY_UPDATE->ENABLED
Sun Mar  4 12:45:49 2018 daemon.notice hostapd: wlan1: AP-ENABLED
Sun Mar  4 12:45:49 2018 kern.debug kernel: [  547.027698] ieee80211 phy1: change: 0x20
Sun Mar  4 12:45:49 2018 daemon.notice netifd: Network device 'wlan1' link is up
Sun Mar  4 12:45:53 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: associated (aid 1)
Sun Mar  4 12:45:54 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED dc:4f:22:xx:xx:xx
Sun Mar  4 12:45:54 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx WPA: pairwise key handshake completed (RSN)
Sun Mar  4 12:46:26 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: authenticated
```

Ping: (no response, not even timeout)
```
root@XXXXX:~# ping 192.168.1.235
PING 192.168.1.235 (192.168.1.235): 56 data bytes
```

rs-se, 2018-03-04:
Have you not tried to set a static IP address inside the wireless clients
themselves and not within the router that still talks to DHCP.

Sounds like you keep setting a static DHCP lease within the router and not
client.

If you set a static IP within the wireless clients and not the router, do
you still have these problems?

iganeshk, 2018-03-04:
@rs-se the problem is these are ESP8285 based devices and configuring static ip is not possible because they run proprietary firmware. I am not sure what the camera's wifi module but all these devices can configured from their respective apps to join 2.4Ghz networks. 

As of now the devices connect without issues to the radio2 on 3200acm (which is not mwlwifi, dhcp discovers it)  and my velop mesh nodes.

yuhhaurlin, 2018-03-05:
1. So this issue is related to ESP8285?
2. Can you let me know version of mwlwifi and firmware.
3. Please use monitor mode to check if DHCP packets can be received. You can use open to check it first, then check via security.
Thanks.

iganeshk, 2018-03-05:
1. I don't think so because these are able to join other networks.
2. 
```
kmod-mwlwifi - 4.14.23+2018-02-26-51350a3e-1
mwlwifi-firmware-88w8964 - 2018-02-26-51350a3e-1
```
3. I did not get you, how do I capture DHCP packets in monitor mode again?

PS: I was going through this issue https://github.com/kaloz/mwlwifi/issues/242, I see similar issue with authentication.

yuhhaurlin, 2018-03-06:
1. I just want to make sure the client is ESP8285.
2. Using following commands to create monitor interface:
iw wlan0/wlan1 interface add mon0 type monitor
ifconfig mon0 up
3. The way to dump dhcp packets:
tcpdump -vvvi mon0 -n port 67 and port 68
4. The way to dump icmp packets:
tcpdump -vvvi mon0 icmp

Please use open to test and get log:
1. Prepare mon0 to dump dhcp packets and then connect client to AP.
2. Prepare mon0 to dump icmp packets, try to ping AP from client after connection is established.
Run the same test under security mode and get log.

iganeshk, 2018-03-06:
Had one ESP client running, right after enabling wlan1 and setting up the monitor mode it connected! I tried turned on all the clients as of now and DHCP allocated the static ip. Had the clients connected over wlan2 for the previous 18hrs. This is bizarre.

Will continue to monitor them and report with monitor logs if it occurs.

iganeshk, 2018-03-06:
Apologies, due to my absent mindedness, I forgot to look at the encryption being disabled. So I followed your steps to get mon0 up and monitoring tcpdump.

```
Mon Mar  5 23:55:53 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: associated (aid 2)
Mon Mar  5 23:55:53 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED dc:4f:22:xx:xx:xx
Mon Mar  5 23:55:53 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx WPA: pairwise key handshake completed (RSN)
Mon Mar  5 23:56:17 2018 daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: authenticated
Mon Mar  5 23:56:52 2018 kern.debug kernel: [111031.348942] ieee80211 phy1: staid 2 deleted
```
```
root@HOST:~# tcpdump -vvvi mon0 icmp
tcpdump: listening on mon0, link-type IEEE802_11_RADIO (802.11 plus radiotap header), capture size 262144 bytes
```
```
root@HOST:~# tcpdump -vvvi mon0 -n port 67 and port 68
tcpdump: listening on mon0, link-type IEEE802_11_RADIO (802.11 plus radiotap header), capture size 262144 bytes

```

yuhhaurlin, 2018-03-06:
Please use open mode to do the test first:
1. Set AP to open mode.
2. Issue "tcpdump -vvvi mon0 -n port 67 and port 68" and connect client to AP (get the log from tcpdump).
3. Stop tcpdump (ctrl-C).
4. Issue "tcpdump -vvvi mon0 icmp"
5. Ping from cilent and get log from tcpdump).

Since open mode is all right according to your description, tcpdump should get DHCP and Ping packets.

Run the same test for security mode, I want to know if DHCP packets can be received by the router.

Finally, run the test for security mode but set static IP to client.

iganeshk, 2018-03-06:
LOGGGGG
```
root@openwrt:~# iw wlan1 interface add mon0 type monitor
root@openwrt:~# ifconfig mon0 up
root@openwrt:~# iw wlan1 info
Interface wlan1
    ifindex 7343
    wdev 0x1000007ab
    addr 60:38:e0:xx:xx:xx
    ssid SSID
    type AP
    wiphy 1
    channel 11 (2462 MHz), width: 20 MHz, center1: 2462 MHz
    txpower 30.00 dBm
root@openwrt:~# uci show wireless.radio1
wireless.radio1=wifi-device
wireless.radio1.type='mac80211'
wireless.radio1.hwmode='11g'
wireless.radio1.path='soc/soc:pcie/pci0000:00/0000:00:02.0/0000:02:00.0'
wireless.radio1.country='US'
wireless.radio1.channel='11'
wireless.radio1.htmode='HT20'
wireless.radio1.legacy_rates='1'
wireless.radio1.txpower='30'
root@openwrt:~# uci show wireless.default_radio1
wireless.default_radio1=wifi-iface
wireless.default_radio1.device='radio1'
wireless.default_radio1.network='lan'
wireless.default_radio1.ssid='SSID'
wireless.default_radio1.key='PASSWORD'
wireless.default_radio1.mode='ap'
wireless.default_radio1.encryption='psk2'

# changing encryption to nothing

root@openwrt:~# uci set wireless.default_radio1.encryption=''
root@openwrt:~# uci commit wireless; wifi
root@openwrt:~# iw wlan1 interface add mon0 type monitor
root@openwrt:~# ifconfig mon0 up
```

```
root@openwrt:~# tcpdump -vvvi mon0 -n port 67 and port 68
tcpdump: listening on mon0, link-type IEEE802_11_RADIO (802.11 plus radiotap header), capture size 262144 bytes
tcpdump: pcap_loop: The interface went down
0 packets captured
2 packets received by filter
0 packets dropped by kernel

# bringing mon0 up again after commit

root@openwrt:~# tcpdump -vvvi mon0 -n port 67 and port 68
tcpdump: listening on mon0, link-type IEEE802_11_RADIO (802.11 plus radiotap header), capture size 262144 bytes
09:51:27.370789 1.0 Mb/s 2462 MHz 11b 0dBm signal antenna 0 IP (tos 0x0, ttl 255, id 9992, offset 0, flags [none], proto UDP (17), length 336)
    0.0.0.0.68 > 255.255.255.255.67: [udp sum ok] BOOTP/DHCP, Request from dc:4f:22:xx:xx:xx, length 308, xid 0xc4daac6a, Flags [none] (0x0000)
      Client-Ethernet-Address dc:4f:22:xx:xx:xx
      Vendor-rfc1048 Extensions
        Magic Cookie 0x63825363
        DHCP-Message Option 53, length 1: Discover
        MSZ Option 57, length 2: 1500
        Hostname Option 12, length 10: "ESP_C3C48F"
        Parameter-Request Option 55, length 12: 
          Subnet-Mask, Default-Gateway, BR, Domain-Name-Server
          Domain-Name, Netbios-Name-Server, Netbios-Node, Netbios-Scope
          Router-Discovery, Static-Route, Classless-Static-Route, Vendor-Option
        END Option 255, length 0
        PAD Option 0, length 0, occurs 34
09:51:27.371232 6.0 Mb/s [bit 15] IP (tos 0xc0, ttl 64, id 17190, offset 0, flags [none], proto UDP (17), length 329)
    192.168.1.1.67 > 192.168.1.223.68: [udp sum ok] BOOTP/DHCP, Reply, length 301, xid 0xc4daac6a, Flags [none] (0x0000)
      Your-IP 192.168.1.223
[... 293 lines cut ...]
daemon.notice netifd: radio1 (10837): command failed: Not supported (-95)
daemon.err hostapd: Configuration file: /var/run/hostapd-phy1.conf
kern.debug kernel: [147208.985220] ieee80211 phy1: change: 0xffffffff
kern.info kernel: [147209.074433] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
kern.info kernel: [147209.081237] br-lan: port 2(wlan1) entered blocking state
kern.info kernel: [147209.086678] br-lan: port 2(wlan1) entered disabled state
kern.info kernel: [147209.092222] device wlan1 entered promiscuous mode
kern.info kernel: [147209.097060] br-lan: port 2(wlan1) entered blocking state
kern.info kernel: [147209.102490] br-lan: port 2(wlan1) entered forwarding state
daemon.notice hostapd: wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
daemon.err hostapd: Using interface wlan1 with hwaddr 60:38:e0:xx:xx:xx and ssid "SSID"
kern.debug kernel: [147209.108943] ieee80211 phy1: change: 0x100
kern.debug kernel: [147209.122078] ieee80211 phy1: change: 0x42
kern.info kernel: [147209.272310] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
daemon.notice hostapd: wlan1: interface state COUNTRY_UPDATE->ENABLED
daemon.notice hostapd: wlan1: AP-ENABLED
daemon.notice netifd: Network device 'wlan1' link is up
daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: associated (aid 1)
daemon.notice hostapd: wlan1: AP-STA-CONNECTED dc:4f:22:xx:xx:xx
daemon.info dnsmasq-dhcp[3326]: DHCPDISCOVER(br-lan) dc:4f:22:xx:xx:xx
daemon.info dnsmasq-dhcp[3326]: DHCPOFFER(br-lan) 192.168.1.223 dc:4f:22:xx:xx:xx
daemon.info dnsmasq-dhcp[3326]: DHCPREQUEST(br-lan) 192.168.1.223 dc:4f:22:xx:xx:xx
daemon.info dnsmasq-dhcp[3326]: DHCPACK(br-lan) 192.168.1.223 dc:4f:22:xx:xx:xx ESP_C3C48F
daemon.info hostapd: wlan1: STA dc:4f:22:xx:xx:xx IEEE 802.11: authenticated
```

iganeshk, 2018-03-06:
I also flashed factory image (FW_WRT3200ACM_1.0.6.186168_prod) to check if there was nothing wrong with the hardware and the clients were able to join 2.4ghz radio AP.

schm0, 2018-03-08:
Hi!
Can you check the arp tables on the clients?
On my wrt1200 with 802.11w enabled arp broadcast are not delivered after some time.
But im also not sure if this a client bug.
Maybe this is somehow related?

iganeshk, 2018-03-08:
@schm0 I'm afraid, I can't. These clients run embedded programs and do not have an interface to debug. You can either flash the firmware with config stating the WiFi credentials or configure them web interface if supported. I suspect the device state after authentication. Clients connect to factory firmware on 3200acm and same SSID/encryption on a different routers.

iganeshk, 2018-03-08:
[image]
[image]
[image]

schm0, 2018-03-09:
For me it is bit different.
With 802.11w enabled:
The clients can always ping the router.
Also other clients but only for a short time after they associated.
When they ARP entries expire, the communication under the clients breaks because the ARP broadcasts are not delivered anymore. However the router's ARP table is always well populated and clients can browse the web just fine.

forlotto, 2018-04-12:
Esp8266 it is possible to run static with ESP8266 basic I do it all the time actually. I did a fairly cool home automation project with these and use static IP's.

Aside from that my lack of knowledge of this project my comment not having logs etc no raw data I understand it may be a useless comment.

Something I have noticed is DDWRT has a problem with mixed mode encryption on the WRT3200ACM it does effect the ability to connect. AES from my experience is best in the case.

1 of my access points is seen with my phone. But when I run wifi analyzer the access points show up but the SSID's show up as asteriks.

The other issue I should share is if you do any daily reboots they always force the router into a condition where you have to manually get the router back up and running on a couple occasions I had to reset the router completely to get it running. So there is something that is being missed somewhere as to why these things happen. 

RECAP OF WHAT I AM SHARING:
Daily scheduled reboots = Cause issues. DDWRT latest version
Cryptographic Method Mixed = Cause issues. DDWRT latest version
ESP8266 Basic = Allows for static IP Configuration Just fine

BrainSlayer, 2018-04-12:
@forlotto hey i'm the dd-wrt developer. one comment for mixed connections. with mixed connections (aes+tkip) for instance group keys will be alway tkip. otherwise broadcast cannot work between clients. but tkip will always disable 802.11n and 802.11ac operation. so never use tkip+aes, but aes only for full 802.11n and 802.11ac operation. by specification tkip is not allowed for 802.11n and 802.11ac. this will also explain any dhcp issues since dhcp packets are broadcast based. so use only AES only for 802.11n and 802.11ac.

BrainSlayer, 2018-04-12:
in addition. dd-wrt does ignore this specification and does allow tkip in 802.11n and 802.11ac, but your esp8266 will not. so this will create incompatiblities. so this special type of operation will only work between dd-wrt based devices (in client and ap modes). but tkip encryption does prevent packet aggregation so its not recommended to use tkip in any way. its obsolete and a insecure protocol

forlotto, 2018-04-12:
@brainslayer thx for this oddly I never had these problems on my EA6500 or other ac routers in mixed mode same with the E4200 older solid wireless N performer mixed mode I can scan with wifi analyzer zero issues the SSID's don't turn from SSIDNAME to **** it is an odd problem that only exhibits it's self on the wrt3200ACM (RM2 in my case) err I feel as if I am hijacking the thread I just was curious if the same things were consistent with this project as well as a host of other oddities. 
Enlightening though I did subscribe to the TKIP fallacy based on too many people pushing it for compatibility I looked into it but wasn't aware of links in speed or that it was insecure I do know that it essentially salts the SSID with the passphrase beyond that I knew very little. After you posting this I feel stupid for knowing so much and missing this based on a fallacy. I was unaware that it was specification driven rather than device driven compatibility more or less which kills me that people promote it so openly as a compatibility must have. Nearly all instances I know of personally be it business or household it is pointless to use TKIP these days after knowing this! I cannot think of a single scenario where I would suggest the use of TKIP anymore! I feel dumb here. It makes me wonder if other people are not experiencing the same issues due to this heavily promoted fallacy. 
I will say it is however very obvious there are still problems with this router scheduled reboots with this router cause issues and it may be due to TKIP or it may not be I am not sure now but there is a lot of strangeness going on with this router. 

Oddly in the case of ESP devices, it has caused me zero issues when using the EA6500 and the E4200 to use "compatibility mode" or mixed mode WPA2 everything worked as planned 3 AP's zero issues 6 ESP's about 10 devices many wired devices. But I upgraded now that I have a 1GB/1GB connection. Including this, a switch and an edge router to my network to achieve 1GB speeds real 1GB speeds. https://www.speedtest.net/my-result/d/4e7e2f10-9d8e-4648-832f-e0f2b428e2c3

But never the less I wanted better LAN throughput essentially so I opted for this router. As a side question in this hijack, what router do you recommend for best heavy traffic multiple connections to run DD-WRT at this point in time and distance. I figured it would be this router but there are far too many issues with it and I live down the road from a small town airport so it could be part of my issue with this router I suppose but when daily reboots cause issues non recoverable from just removing power it makes me wonder what else is up.

But I thank you for your help very revealing a huge fallacy I had kept in my education in the case of TKIP time to fix this and fix my brain. I have a great dislike for fallacies more and more I am less inclined to take peoples word for anything even with things like OpenVPN for instance. I feel stupid for not seeing what was obvious in speed during my testing but I assumed changes may have been linked to location and beamforming bad assumption how foolish man I feel a bit gullible for this one a huge mistake on my part. But this fallacy was plastered everywhere and preached by even the likes of Cisco back in 2010 ish which makes sense there were more G devices in operation back then.

There are still problems with this router clearly I am very inclined to return it and opt for a different model. Another interesting thing from a security standpoint it would be nice to somehow mask the hardware being used in DD-WRT to make it difficult for attackers targeting specific equipment, for example, the latest Microtik exploits and so forth. It would be great to mask this info somehow to the outside I so wish it was possible obscurity, in this case, may help network security to a large degree. So many attacks now hardware focused out in the open are a new angle to worry about now that the cat is out of the bag with vault 7 release and so on.

Anyways I thank you very much for your response.

And Sincerely apologize to all for the hijack potential of this all I just figured the experiences I had may be of help it was not my intention to get to this point.

marc-o94, 2018-08-19:
Hello,

i am having the same issue with my ESP8266 devices on my WRT3200ACM on 2G wlan1 interface. They seem to connect but do not get an IP address via DHCP.

Do you have an solution for this issue?

Thanks.

marc-o94, 2018-08-19:
I read your posts that you are also having issues with radio2 when DFS kicks in. I am having the same issue with this radio. How do you solve this issue? Reboot the router every time it is disconnected?

iganeshk, 2018-08-19:
I experienced radio2 issues a lot when devices disconnect. I couldn't pin point the root cause of the radio malfunction but if you keep the devices connected and prevent them from disconnecting, it would be stable enough.

yuhhaurlin, 2018-08-24:
[image]

yuhhaurlin, 2018-08-24:
Or anyone can take picture of your device and post here, so I can try to get one to do test. Thanks.

iganeshk, 2018-08-24:
@yuhhaurlin You could try these:

NodeMCU (Original/Compatible) using ESP8266 - [AliExpress](https://www.aliexpress.com/item/New-Wireless-module-NodeMcu-Lua-WIFI-Internet-of-Things-development-board-based-ESP8266-with-pcb-Antenna/32656775273.html)

WLAN Adapter using RTL8188EUS - [AliExpress](https://www.aliexpress.com/item/Comfast-RTL8188EUS-150M-USB-2-0-WiFi-Wireless-Network-Card-802-11-b-g-n-LAN/32826876939.html)

mrvltest, 2018-08-24:
Thanks for your information.

spamtree, 2018-09-10:
Hello,

Had the exact same error code with a device that uses an esp8266. My search for an answer went to this thread and I have an WRT3200ACM. However, I was using an old netgear router (ar71xx) running openwrt. It was the exact same error listed on the logs posted above. Of course the device would not connect to the WRT3200ACM and had the same error. 

The exact chip in the device is an [ESP-WROOM-02](https://www.espressif.com/en/products/hardware/esp-wroom-02/overview) this chip is based on the esp8266. 

Found the solution when reading the forums for some users and it was recommended to turn Wireless "N" mode off and use only Wireless "G" mode.  That worked but not really happy about the solution. The esp8266 is supposed to support b/g/n. 

I guess it could be something else going on other than the mwlwifi driver. This  is above my knowledge level but wanted to throw this out there as another data point. I would be glad to get additional information if needed. 

Thanks

eduperez, 2018-09-20:
I recently acquired a wifi-enabled RGB LED strip. This device seems to be based on a ESP chip (as implied by the name sent to the DHCP server). It connects fine to radio2, but on radio1 DHCP packets seem to be lost: hostapd accepts the connection, but then dnsmasq does not log any request.

The ESP is embedded in the device, I cannot confirm its firmware version, or even wether it's a ESP8266 or a ESP32.

Phil795, 2018-09-20:
If it is a MagicHome RGB controller, an ESP12S or ESP8285 (older version) will be installed (I have the same device running here). If the WMM option is switched off, the device runs perfectly on radio-0.

But I flashed the firmware with Tasmota. Would be possible that the compatibility increased thereby.

eduperez, 2018-09-20:
Yes, it's just a "Magic Home" device, relabeled to something else. But this device only works on the 2.4GHz band (radio1 on the WRT32000ACM, if I am not mistaken).

I do not think I want to disable WMM, however.

I'll probably have a look to that firmware, thanks for the tip.

Phil795, 2018-09-20:
(Of course, radio-1 not 0)

I operate an additional WLAN network on the radio-1 where my IoT devices are, with the WMM option turned off.

eduperez, 2018-09-20:
Oh, I assumed WMM was configured at device level, not interface level. Thanks!

Anyway, I'm happy with radio2 for my IOT devices.

yuhhaurlin, 2018-09-20:
Per module, so phy0 and phy1 can have different WMM setting.

yuhhaurlin, 2018-09-20:
But WMM is disable, you can only use g rate.

Phil795, 2018-09-20:
Correct, but the G-Standard is completely sufficient for IoT devices. Since a switched on WMM often leads to problems, it should only be deactivated in networks intended for IoT.

yuhhaurlin, 2018-09-20:
How about other AP?

thecode, 2018-09-20:
I have various ESP82XX based devices, some of them are using the original firmware and some custom based firmware.

The custom based firmware (such as Tasmota) using SDK 2.3 connects only when WMM is disabled.
Some other devices which I don't have info about the SDK are having problems to connect even when WMM is off.

Sometimes to connect I switch the WMM on and off few times.
I'm also using a separate WLAN on radio1 with WMM disabled for IOT devices only.
radio2 is not an option since it is sometimes used by the DFS and it also uses an internal antenna which is less sensitive.

@yuhhaurlin I have tried different routers (d-link based N router, adb AC router, tp-link N router) they all works without any problem with the ESP82XX devices.

chaseadam, 2018-09-20:
@yuhhaurlin if you can point us to some debugging or data collection techniques, I may be able to poke around and find an anomaly. Use a separate second wifi card in promiscuous mode?

diabl0w, 2018-12-14:
so is everyone still just disabling wmm for this issue or has someone found another better way? some of my iot things only connect with wmm disabled as is the fix in this issue, some of my other iot clients however can only connect if wmm mode is enabled. I've tried disabling N mode by setting htmode  to 'NOHT' but it doesn't help esp8266 devices as they seem to require wmm to be off. 

off topic sort of but is this an issue of linksys not providing a working open source code to their open source router? just curious as to whether or not I could call customer service and bring this up as an argument. this router is out of warranty but I'd like to get my money back if it doesn't work as advertised

iganeshk, 2018-12-14:
Disabled both the radios, bridged velops to the network.

diabl0w, 2018-12-14:
expensive workaround ;) but interesting

forlotto, 2018-12-15:
I just went backwards with my wifi for the IOT to an N300 wifi appliance using OPENWRT on a Linksys Repeater turned into a wired range extender instead. 

Got my cash back for the router and said screw that thing Linksys is no longer a Cisco company anymore anyways and I believe Netgear was sold to a chinese company as well who purchased Linksys. I do not believe they will ever be the company they once were. 

What I did was purchase the last router made by Cisco for consumers before the sellout to Netgear. 

These things are super stable with insane uptimes and very little conflict and the kicker is for now they are very cheap I suggest Tomato Hyzoom firmware for them all wired connections run through good quality 1GB switches so I don't lose any pipe the routers only function as AP mode and the repeaters are used where needed but they function now as wired access points. 

Works like a charm I think all hold off on the fancy new stuff for some time and stick with what works.

I personally can't wait for the long range low bandwidth wireless standards to be used and ratified.
802.11ah (HaLow), created for low data rate, long-range sensors and controllers.
802.11af, created for applications similar to HaLow. (This network option relies on unused TV spectrums instead of 2.4 GHz or 5 GHz bands for transmission.)

Surely not compatible out of the box but what about disabling the wifi on the esp and having a serial module hooked to it pulling the switches or something similar of that nature who knows maybe devices like the ESP will be created just for this!

[117 more replies]