kaloz/mwlwifi #280: Unable to enable 160 Mhz on Linksys WRT 3200ACM

p3x-robot, 2018-03-06:
Hello @yuhhaurlin !

How are you today?

This is something, I found and this is what is weird for me, given it is the latest ```LEDE 17.01.4``` and your latest driver:
**Mar  6 19:37:11 192.168.78.1 digi.router hostapd: ACS: Only VHT20/40/80 is supported now**

How come it is wrong?

The driver is built from:
https://github.com/kaloz/mwlwifi/commit/4591708bfd92b99102976b7ae2571101cbf8518e - 2018. March 5. (there is only one new commit about a fix for the 4.14 Kernel - Replaced vfs_write with __kernel_write - this is not built still).

The ```hostpad ``` is udpated from the stable ```LEDE 17.01.4``` as it is exactly:
```Makefile
PKG_NAME:=hostapd
PKG_RELEASE:=6

PKG_SOURCE_URL:=http://w1.fi/hostap.git
PKG_SOURCE_PROTO:=git
PKG_SOURCE_DATE:=2016-12-19
PKG_SOURCE_VERSION:=ad02e79d12fd70ed6bd5fbaf64001a2851e5bb7b
PKG_MIRROR_HASH:=7a0983f004b156d46911765c113754a4c00f56fb889430620bbd061b3b1fcf33
```
The ```README.md``` says it should be after:
```Note: After hostapd package 2016-06-15, this commit is already included.```

So it should be working or is there a configuration?

I am using ```wpad``` instead of ```wpad-mini```, is that the problem? (I needed for the ```Enable WPS pushbutton``` and is only works with the full ```wpad```).

I updated likes hits:
```bash
opkg update
opkg remove wpad-mini
# if you remove it, it will not work
opkg install wpad hostapd-utils
reboot && exit
```

My wifi ```/etc/config/wireless```:
```config
config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11a'
	option path 'soc/soc:pcie-controller/pci0000:00/0000:00:01.0/0000:01:00.0'
	option channel 'auto'
	option country 'FR'
	option htmode 'VHT80'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'patrikx3-digi'
	option encryption 'psk2+ccmp'
	option key '12345678'
	option wps_pushbutton '0'

config wifi-device 'radio1'
	option type 'mac80211'
	option hwmode '11g'
	option path 'soc/soc:pcie-controller/pci0000:00/0000:00:02.0/0000:02:00.0'
	option channel '1'
	option htmode 'HT20'
	option txpower '20'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
[... 133 lines cut ...]
Mar  6 19:37:11 192.168.78.1 digi.router kernel: [39821.531501] ieee80211 phy0: change: 0x40
Mar  6 19:37:11 192.168.78.1 digi.router hostapd: ACS: Only VHT20/40/80 is supported now
Mar  6 19:37:11 192.168.78.1 digi.router hostapd: wlan0: ACS-COMPLETED freq=5620 channel=124
Mar  6 19:37:11 192.168.78.1 digi.router hostapd: wlan0: interface state ACS->HT_SCAN
Mar  6 19:37:11 192.168.78.1 digi.router kernel: [39821.741493] ieee80211 phy0: change: 0x60
Mar  6 19:37:11 192.168.78.1 digi.router kernel: [39821.837498] ieee80211 phy0: change: 0x100
Mar  6 19:37:11 192.168.78.1 digi.router kernel: [39821.850548] ieee80211 phy0: change: 0x100
Mar  6 19:37:11 192.168.78.1 digi.router kernel: [39821.863703] ieee80211 phy0: change: 0x60
Mar  6 19:37:11 192.168.78.1 digi.router kernel: [39822.081481] ieee80211 phy0: change: 0x40
Mar  6 19:37:11 192.168.78.1 digi.router hostapd: Interface initialization failed
Mar  6 19:37:11 192.168.78.1 digi.router hostapd: wlan0: interface state HT_SCAN->DISABLED
Mar  6 19:37:11 192.168.78.1 digi.router hostapd: wlan0: AP-DISABLED 
Mar  6 19:37:11 192.168.78.1 digi.router hostapd: wlan0: interface state DISABLED->DISABLED
Mar  6 19:37:11 192.168.78.1 digi.router hostapd: wlan0: AP-DISABLED 
Mar  6 19:37:11 192.168.78.1 digi.router hostapd: wlan0: CTRL-EVENT-TERMINATING 
Mar  6 19:37:11 192.168.78.1 digi.router hostapd: hostapd_free_hapd_data: Interface wlan0 wasn't started
Mar  6 19:37:11 192.168.78.1 digi.router hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Mar  6 19:37:11 192.168.78.1 digi.router kernel: [39822.291477] ieee80211 phy0: change: 0x60
Mar  6 19:37:11 192.168.78.1 digi.router kernel: [39822.331542] device wlan0 left promiscuous mode
Mar  6 19:37:11 192.168.78.1 digi.router kernel: [39822.336045] br-lan: port 3(wlan0) entered disabled state
Mar  6 19:37:11 192.168.78.1 digi.router kernel: [39822.389504] ieee80211 phy0: change: 0x100
Mar  6 19:37:11 192.168.78.1 digi.router hostapd: ELOOP: remaining socket: sock=22 eloop_data=0xb6ed5c70 user_data=0 handler=0x3e7c4
Mar  6 19:37:12 192.168.78.1 digi.router kernel: [39822.573691] ieee80211 phy0: change: 0xffffffff
Mar  6 19:37:12 192.168.78.1 digi.router kernel: [39822.683528] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
```

p3x-robot, 2018-03-06:
Ok, my bad, I guess @BrainSlayer is in Europe and once I selected the 160 Mhz channel 100, it all started working, and the increase on Android from 280 Mbit went to 400 MBit, awesome!!!

But besides, so do you think I can keep ```phy2``` and ```radio2``` if I disable the ```radio0``` and ```radio1``` country and only fix it in ```radio2``` so it is fixed ```US``` and fix it with the ```option country 'FS'```.

Is that correct?

Reference: https://github.com/kaloz/mwlwifi/issues/226

BrainSlayer, 2018-03-06:
@p3x-robot yes i'm in europe. and 160 mhz works only with channel 100 upwards. for the region setting. it works gobally in cfg80211/mac80211. so you cannot use independend regions per radio as far as i know. but you can maybe mod the driver to hardcode a own region setting (except for mwlwifi, this one is fixed by the chipset)

p3x-robot, 2018-03-06:
@BrainSlayer  Yesh, it works, you already helped me in a reference in this issue.

My only question now if I can configure to keep the ```phy2``` and ```radio2``` by reconfigure the ```US``` country to ```FR```, then I can keep the ```kmod-mwifiex-sdio``` and ```mwifiex-sdio-firmware``` without having to remove, with some tweaks that @yuhhaurlin said it might work without the remove packages. I think if I can work out the ```phy2``` with ```radio2```, I think it might be much closer to the ```mwlwifi``` original closed source driver.

Will test though after the full build is done.

Thanks so much!

Though, it would be cool, if @yuhhaurlin could understand my poor English what I am trying to achive (with this mixed up regions from Europe but still ```phy2``` is fixed with ```US```, but the ```wireless``` config could be overridden). I suppose and hope.

yuhhaurlin, 2018-03-06:
This will be done later.

p3x-robot, 2018-03-07:
Ok, the readme says either i keep the country (code 00 - unset) or remove mwifiex-sdio-firmware kmod-mwifiex-sdio, but the truth is it only works if I remove phy2 and remove mwifiex-sdio-firmware kmod-mwifiex-sdio and then set the country by hand for Europe France.

Than it works. So it either a problem with the chiip (fixed phy2 US, cannot override now way, I set it, luci, /etc/config/wireless), but it always US when i check ```iw reg get``.

Besides need to remove the radio2 as well.

Only solutions is 
```bash
opkg remove mwifiex-sdio-firmware kmod-mwifiex-sdio
# remove radio2 from /etc/config/network
# set the wifi wireless in LUCI to France and the it is cool
```

With the Europan version cannot set ```mwlwifi``` at all.

yuhhaurlin, 2018-03-07:
The limitation will be removed later. The code will be modified as follows:
1. Check if there is power table in DTS. If power table is existed, mwlwifi will load the power table to firmware.
2. Due to firmware can check if region code and power table are programmed in EEPROM, it will reject the power setting from mwlwifi if EEPROM is programmed with region code and power table.
3. Mapping region code to country code and rejecting setting of country code will be removed from mwlwifi.
4. If user's setting is conflicted with region code and power table programmed in EEPROM, firmware won't let the setting work. In this way, it won't violate FCC rules and user has the flexibility to set country code.

This job will be done later.

p3x-robot, 2018-03-07:
Hey @yuhhaurlin !

How are you?

So one more question, the ```phy2``` I guess will be enabled.
But what we will do with the ```radio2```? Is it gonna be enabled (which is weird, because it is open and anyone can connect and works) so we remove the ```radio2``` or ```disable``` or how is it going to go?

yuhhaurlin, 2018-03-07:
I have no idea about phy2. I only take care of mwlwifi.

p3x-robot, 2018-03-07:
Aha, so I still have to remove
```opkg remove mwifiex-sdio-firmware kmod-mwifiex-sdio```, is that right?
So is ```mwlwifi``` use the 3rd ```radio``` that is actually the ```radar```, in my thinking?

yuhhaurlin, 2018-03-07:
Phy2 is mwifiex which is another open source driver from Marvell. Mwlwifi is used for 88W8964 of WRT3200ACM (phy0 and phy1). You can use phy2, but I can't support it.

p3x-robot, 2018-03-07:
1. Hm, so given there is no support and is not needed, in the firmware, i actually can just remove totally ```mwifiex-sdio-firmware``` , ```kmod-mwifiex-sdio``` and freely remove the ```radio2``` in the ```/etc/config/wireless``` and that's how it will works with ```mwlwifi``` and still ```mwlwifi``` will use that 3rd radio for radar detection but in the driver instead of the ```/etc/config/wireless``` ?

1. Now in my understanding is that ```mwlwifi``` will do everything including the ```radar```, ```MU-MIMO``` and ```DTS```, so only ```mwlwifi``` is required for the ```3200ACM``` (no ```phy2``` and no ```radio2```).

yuhhaurlin, 2018-03-07:
Mwlwifi can do everything. Phy2 is used for some purposes, but I have no idea of that.
I think what I need to do is let it can work with mwlwifi (running DFS).

BrainSlayer, 2018-03-07:
@p3x-robot phy2 can be used as additional wifi interface. it has no influence to mwlwifi (phy0 / phy1)

p3x-robot, 2018-03-07:
@BrainSlayer you mean remove the mwifiex-sdio-firmware kmod-mwifiex-sdio, but keep the 3rd wifi radio?
or how do you complete it so that the DFS/DTS works?

eduperez, 2018-03-07:
@BrainSlayer I am currently using phy2 as an "auxiliary channel"; I have both `mwifiex-sdio-firmware` and `kmod-mwifiex-sdio` installed. On the 5GHz band at phy0 I only use the DFS-free channels.

p3x-robot, 2018-03-07:
@eduperez, you are not in the EU, right?  In USA, that's why it works?

p3x-robot, 2018-03-07:
@eduperez  or using on the 5Ghz 80Mhz?

p3x-robot, 2018-03-07:
@eduperez  yes, i confirmed, that 80 Mhz works, while 160Mhz is not :)

eduperez, 2018-03-07:
@p3x-robot In EU, 80MHz only.

p3x-robot, 2018-03-07:
yes, so it is only non DTS frequencies. Got it.
What is weird is that radio2 is like half slower than the normal.

p3x-robot, 2018-03-07:
i guess until it is completed, 160 Mhz and DTS is not worth dealing with...

p3x-robot, 2018-03-07:
@yuhhaurlin @eduperez  @BrainSlayer 
I have added some info in the European version in the readme:  
https://github.com/patrikx3/mwlwifi#for-the-european-version-also-requires-the-following-if-you-want-160mhz-and-dts

For both options, if you want use 160 Mhz and DTS (dropping ```phy2``` and ```radio2```) or use only 80 Mhz and only non DTS frequencies - by keeping the ```kmod-mwifiex-sdio``` and ```mwifiex-sdio-firmware``` including ```phy2``` and ```radio2```.

howl, 2018-04-04:
@yuhhaurlin if what in manage to understand in your explanation about what are you going to do with the country setting being locked by mwlwifi to a permissive one that will solve a dispute we had time ago about one this should be done.
If you do what you say letting mwlwifi to take country codes with same fcc rules like most of the european countries as for example FR, DE, ES... will be great.

howl, 2018-08-10:
> The limitation will be removed later. The code will be modified as follows:
> 
>     Check if there is power table in DTS. If power table is existed, mwlwifi will load the power table to firmware.
>     Due to firmware can check if region code and power table are programmed in EEPROM, it will reject the power setting from mwlwifi if EEPROM is programmed with region code and power table.
>     Mapping region code to country code and rejecting setting of country code will be removed from mwlwifi.
>     If user's setting is conflicted with region code and power table programmed in EEPROM, firmware won't let the setting work. In this way, it won't violate FCC rules and user has the flexibility to set country code.
> 
> This job will be done later.

Hi @yuhhaurlin do you have an estimated date to make this change?

p3x-robot, 2018-08-12:
@BrainSlayer @yuhhaurlin  @eduperez i bought a Intel® Wireless-AC 9260 for my laptop. what is weird, on if i set up 160Mhz on Windows it works, but on Linux i can get like 1-2 Mbit, but once I set it to 80 Mhz, I get the speed cool like 700Mbit.

My question, is MU-MIMO is only 160Mhz or that is independent from MU-MIMU?

PS: I think maybe the Windows driver is already using not 160Mhz but 80Mhz otherwise it would be  weird. How can with 160Mhz in Windows works and in Linux not, but once I am 80Mhz 5Ghz, it flies...

BrainSlayer, 2018-08-13:
@p3x-robot  no this is not related to mu-mimo in any way. the reason why 160 doesnt work in linux sounds more like a bad or buggy linux driver. personally i avoid intel chipsets in windows and linux systems since intel is known for good cpus but also known for having the worst wireless chipsets ever seen. i tested 160 mhz with qca and broadcom chipsets in linux and it worked without any issues

p3x-robot, 2018-08-13:
@BrainSlayer you are right, the driver will be supported in 4.16... i have a back-port and use 80mhz, i dont have MU-MIMO any was so the speed is ok... i get the 800 mbits....

p3x-robot, 2018-08-13:
@BrainSlayer always buying the wrong wifi chipsets :-1:

ryan-dev, 2018-08-13:
@p3x-robot <https://github.com/p3x-robot> don't we all at some point, Intel
has always wanted their hands in everything but have done poorly for
networking.

You cannot go wrong with a solid Broadcom chipset -;)

p3x-robot, 2018-08-13:
@BrainSlayer I was already like asking before I buy and talk to you, but I was like Intel is the bomb. And nah, wrong again. Never mind, 80 Mhz is ok for now.... 🔢

BrainSlayer, 2018-08-13:
@p3x-robot often you dont have a choice. these intel chipsets are often builtinto laptops and forced by intel to be used and intel wasnt bad at all for networking. remember the old e1000 cards. but they never build a good wifi chipset. i have seen troubles like that cards dont support channel greater than 9 in europe, broken wpa support which leads to constant disconnects etc.

ryan-dev, 2018-08-13:
I do agree Intel for networking is good in general, just not wifi chipsets.

Have had a lot of gigabit intel cards that did amazing in windblows or
linux (redhat in my early teen years).

Apology for the broad mark on all of Intel networking.

lantis1008, 2018-11-20:
Hi @yuhhaurlin 
Any update on this issue, and in particular, power table loading for WRT32X devices?
https://github.com/kaloz/mwlwifi/issues/305#issuecomment-415702157

UsoriNomen, 2019-02-07:
> Hi @yuhhaurlin
> Any update on this issue, and in particular, power table loading for WRT32X devices?
> [#305 (comment)](https://github.com/kaloz/mwlwifi/issues/305#issuecomment-415702157)

Bump. I'm looking to maybe buy a WRT32X because they are a really good deal if OpenWrt works well  on them.

@lantis1008 Is it worth buying one, or are they still in need of further development?

lantis1008, 2019-02-07:
It’s a good device, but if you have certain wireless chipsets (e.g. ESP) you’ll have a bit of trouble. 
It could do with a bit of spring cleaning, but it performs admirably in my network setup.

intdev32, 2019-02-07:
Plus,

WRT3200ACM(and WRT32X) mwlwifi driver and Intel 8260 wireless chip have some issues.

Sometimes(first connection, within 3-15min) unstable speed, high ping spike, ping drops.

Stock firmware and other routers works ok.

Turn off Intel 8260 driver advanced option U-APSD and use old intel wireless driver helps a lot. But not enough to me.

UsoriNomen, 2019-02-13:
I'm brand new to router flashing, but is there anything I can do to help expedite full OpenWrt support for WRT32X / WRT3200ACM?

I've got a WRT32X spun up with up-to-date OEM firmware, and everything seems great so far, although I've done no diagnostics or testing besides speed tests. 

Also, looks like dd-wrt supports the WRT32X, but it's not clear if there are any issues. Does anyone know if dd-wrt runs pretty well?

lantis1008, 2019-02-13:
@UsoriNomen WRT32X is already supported. What makes you think otherwise?

UsoriNomen, 2019-02-13:
Sorry, my phrasing was incorrect. What I meant is: how can an OpenWrt novice help you guys fix the remaining bugs?

And also, I would like to know whether OpenWrt or dd-wrt is more usable, as in which has the least remaining bugs?

lantis1008, 2019-02-13:
Unless you had access to the source code for the closed source binary firmware, you’re pretty much at the whim of the developer. 

Subjectively i would say they have a similar number of bugs, but in different areas. 
It’s more what you like or are used to which makes a decision as to what to use.

UsoriNomen, 2019-02-13:
Well, currently I'm more interested in usability than having access to every single possible router setting. But obviously I want more options and control than stock firmware.

UsoriNomen, 2019-02-13:
For instance, I don't know what effect the ipv6 bug might have on the devices connected to my network.

https://openwrt.org/unsupported/ipv6_until_bug_fixed

lantis1008, 2019-02-13:
This is now pretty off topic to this issue.
If you want to know more you should continue at either the dd wrt forums or Openwrt forums. 

No ipv6 would only affect wan connections not the internal network in general.

lantis1008, 2019-04-12:
Hi @yuhhaurlin , still looking for any updates on this issue. If you don't have time to develop it yourself, are you able to provide us any hints on the what this new format for the wrt32x power tables is and how we could try loading it ourselves?

yuhhaurlin, 2019-04-12:
Sorry. I am not familiar with WRT32X.

lantis1008, 2019-04-12:
You said you would get one to test with a while ago? WRT32X is able to use mwlwifi, so it should be interesting to you.
88W8964.bin must be able to load an external power table, we just need to know how.

howl, 2019-04-12:
@lantis1008 seriously, you still have hope that will ever exists a proper kernel module for these marvel wifi modules?

p3x-robot, 2019-06-27:
i doubt they ever will develop more on this driver

httpstorm, 2020-01-10:
OpenWRT on WRT3200ACM: to accept user region on devices where the region is hard-coded, the wireless drivers can be modified not to inform the system about the wireless chip's region. The following patches can be applied to r11829. Use for educational purposes only is advised.

With that in place the 5 GHz radio will work at 160 MHz. All radios are available to the user. TX power on radio0 and radio1 is still controlled by the wireless chip, so regulations are still followed. radio2 is limited to 8-15 dBm by the chip.
 
[009-mwlwifi-2019-03-02-DFS-free.patch](https://pastebin.com/1LRctnm2)
[609-mwifiex-4.19.91-DFS-free.patch](https://pastebin.com/JU6mSJxC)

p3x-robot, 2020-01-10:
this repo is dead, no one will pull requests... u know ...

howl, 2020-01-10:
@httpstorm this could be interesting for you https://github.com/openwrt/openwrt/pull/2397

eduperez, 2020-01-29:
I noticed that `phy#2` can be forced to a region: 
```
root@router:~# cat /etc/modules.d/mwifiex
mwifiex reg_alpha2=FR
```

Now, all three radios are on the same region, and even the global region is properly configured:
```
root@router:~# iw reg get
global
country FR: DFS-ETSI
	(2402 - 2482 @ 40), (N/A, 20), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
	(5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
	(5470 - 5725 @ 160), (N/A, 27), (0 ms), DFS
	(5725 - 5875 @ 80), (N/A, 13), (N/A)
	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

phy#2
country FR: DFS-ETSI
	(2402 - 2482 @ 40), (N/A, 20), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
	(5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
	(5470 - 5725 @ 160), (N/A, 27), (0 ms), DFS
	(5725 - 5875 @ 80), (N/A, 13), (N/A)
	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

phy#1
country FR: DFS-ETSI
	(2402 - 2482 @ 40), (N/A, 20), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
	(5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
	(5470 - 5725 @ 160), (N/A, 27), (0 ms), DFS
	(5725 - 5875 @ 80), (N/A, 13), (N/A)
	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

phy#0
country FR: DFS-ETSI
	(2402 - 2482 @ 40), (N/A, 20), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
	(5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
	(5470 - 5725 @ 160), (N/A, 27), (0 ms), DFS
	(5725 - 5875 @ 80), (N/A, 13), (N/A)
	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

```

This way, `phy#2` remains fully functional, and `phy#0` can be used at 160MHz:

```
root@router:~# iw dev
phy#2
	Interface wlan_iot
		ifindex 28
		wdev 0x200000001
		addr 60:38:e0:bc:34:e3
		ssid acuario_iot
		type AP
		channel 11 (2462 MHz), width: 20 MHz, center1: 2462 MHz
		txpower 17.00 dBm
phy#1
	Interface wlan_kid1
		ifindex 35
		wdev 0x100000004
		addr 62:38:e0:bc:34:ef
		ssid acuario_kid
		type AP
		channel 1 (2412 MHz), width: 40 MHz, center1: 2422 MHz
		txpower 20.00 dBm
	Interface wlan_ext1
		ifindex 32
		wdev 0x100000003
		addr 62:38:e0:bc:34:e0
		ssid acuario_ext
		type AP
		channel 1 (2412 MHz), width: 40 MHz, center1: 2422 MHz
		txpower 20.00 dBm
	Interface wlan_lan1
		ifindex 27
		wdev 0x100000002
		addr 60:38:e0:bc:34:e1
		ssid acuario
		type AP
		channel 1 (2412 MHz), width: 40 MHz, center1: 2422 MHz
		txpower 20.00 dBm
phy#0
	Interface wlan_kid0
		ifindex 44
		wdev 0x5
		addr 62:38:e0:bc:34:e4
		ssid acuario_kid
		type AP
		channel 104 (5520 MHz), width: 160 MHz, center1: 5570 MHz
		txpower 27.00 dBm
	Interface wlan_ext0
		ifindex 43
		wdev 0x4
		addr 62:38:e0:bc:34:e3
		ssid acuario_ext
		type AP
		channel 104 (5520 MHz), width: 160 MHz, center1: 5570 MHz
		txpower 27.00 dBm
	Interface wlan_lan0
		ifindex 42
		wdev 0x3
		addr 60:38:e0:bc:34:e2
		ssid acuario
		type AP
		channel 104 (5520 MHz), width: 160 MHz, center1: 5570 MHz
		txpower 27.00 dBm

```

Hope this helps!

p3x-robot, 2020-01-30:
wow it is awesome! insider solution!

howl, 2020-01-30:
We still will be broadcasting the wrong country code in many places. But at least with your appointment plus the reg free patch is possible to enforce the country code where one is by only compiling mwlwifi and not both mwlwifi and mwifiex.

Thanks @eduperez.

thecode, 2020-01-30:
I wonder if there is similar workaround for the mwlwifi to set the region code in similar way without a path.
Anyway I think the regfree patch should be included in next openwrt releases. It doesn't override TX regulations, it will just allow setting the right country code.

eduperez, 2020-01-30:
@howl Well, it's not the wrong **region** here... (it's the wrong **country**, but so far I cannot force the country to the mwlwifi driver, so the situation is not worse).

eduperez, 2020-01-30:
@thecode I think that the mwlwifi kernel module does not accept any parameter at all, or at least I could not find any reference in the source code.

howl, 2020-01-31:
@eduperez it's only possible to avoid forcing the country code by patching the module.

p3x-robot, 2020-01-31:
@eduperez this hack is awesome, but my question, it does not work on 160mhz , correct?

howl, 2020-01-31:
@p3x-robot the parameter is only to adjust the same country code you have hardcoded for the mwifiex module phy2, so the result is you can use 160 mhz over phy0 without being necesary to deactivate phy2.

[78 more replies]