kaloz/mwlwifi #287: I received an english message from my site about the 3200ACM Winbond chip W29N02GV

p3x-robot, 2018-03-16:
Ciao!

How are you?

So I got this message:
```text
I have a NEW Linksys WRT3200ACM with the new Winbond chip W29N02GV.  
Do you have support for the new Winbond chip W29N02GV in your image of LEDE Insomnia?  
The LEDE reboot image. 
https://downloads.openwrt.org/releases/17.01.4/targets/mvebu/generic/ lede-17.01.4-mvebu-linksys-wrt3200acm-squashfs-factory.img  
Do not have support for new Winbond chip W29N02GV.  
I also have a older Linksys WRT3200ACM running LEDE 17.01.04 (image above).  
Have problem with the WiFi range.  
The range is much shorter then a old ASUS RT-AC66U.  
Do you have any recommendations?   
I have upgraded the WiFi driver to the latest with:  
kmod-mwlwifi_4.4.92.10.3.4.0-20180305-b1b9a9e-9325_arm_cortex-a9_vfpv3.ipk   
And radio2 is disable. 
Still very short range, What can I do?  
Best regards Per-Ove Carlsson
```

# My question is
Is it only working with a new Kernel or is it possible with the ```kaloz/mwlwifi``` to make it to work with a stable current ```LEDE 17.01.4```?

Sincerely,
Patrik

p3x-robot, 2018-03-16:
I see Kaloz has updated his staging source at here:
https://git.openwrt.org/openwrt/staging/kaloz.git  

I guess I try to create a built from this source.

thagabe, 2018-03-16:
mwlwifi is a kernel driver, no? Should be platform independent but the target mvebu should be the one that deals with the hardware topology.

p3x-robot, 2018-03-16:
Yeah, I guess, it is important to update the kernel, so it is independent of the kernel. 
i am building the kaloz's staging openwrt repo. i hope that will be the sweet paradise.

yuhhaurlin, 2018-03-16:
Yes，This is not related with mwlwifi.

closed 2018-03-16 by yuhhaurlin

p3x-robot, 2018-03-17:
https://cdn.corifeus.com/openwrt/SNAPSHOT/targets/mvebu/generic/