kaloz/mwlwifi #291: Question: does lowering the tx power actually work?

kubrickfr, 2018-04-04:
Hi,
I'm trying to reduce the tx power of my 2.4GHz network, as apparently by default it's blasting at 1000mw in the US.
But even when I set it to something ridiculously low, like 1mw, it still appears at the same level on my devices and they still manage to connect to it.
I don't have a proper scientific way of measuring the power output, so I'm asking: is it supposed to work?

AjkayAlan, 2018-04-04:
It depends on what device you have. If you have a WRT3200ACM then the power levels can not be adjusted as they are programmed with a non-changeable power table to comply with FCC regulations.

kubrickfr, 2018-04-04:
I do have the WRT3200ACM indeed.
But surely it would still comply with FFC regulations to adjust the power level down.

This is actually a big problem, because the 2.4GHz signal is so strong compared to the 5GHz that devices do not choose it. And in a dense environment 2.4GHz at 1W is just not usable.

kubrickfr, 2018-04-04:
I guess issue #165 answers my question, and this is really retarded (from the FCC, I know) 
Can you confirm that this also applies to the 2.4GHz band? I was under the impression that the "new" regulation only applied to the 5GHz band.

thagabe, 2018-04-05:
@AjkayAlan That is not true. It's the lazy way out.

aaron1959, 2018-04-10:
It's been a rule since the communications act of 1934, preceding the F.C.C. as we know it today. 'To use the minimum power level to maintain reliable communication'. 
Being a ham radio operator since the late 60's I know better then use 1.5 Kw to talk to a guy across town on 40 meters. 
It's clearly against the rules to not be able to reduce the levels, you folks at Marvell have it all mixed up.

yuhhaurlin, 2018-04-10:
Sorry. If device is programmed with EEPROM power table, power can't be changed.

closed 2018-04-10 by yuhhaurlin

dengqf6, 2019-01-01:
Is there any way to change the power table?
Hardware specs of WRT1900ACv2/ACSv1 and WRT1900ACSv2 are identical, and I've tried flashing ACS OpenWrt into an ACv2 and it worked fine. The only difference is power table location
Ref link: https://community.linksys.com/t5/Wireless-Routers/WRT1900ACS-V1-vs-V2-hardware-changes/m-p/1048334/highlight/true#M326255

BrainSlayer, 2019-01-01:
no. there is no documented way to access the power table in any way. its likelly stored on a otp flash with a efuse. so once programmed it cannot be changed anymore

BrainSlayer, 2019-01-25:
@dengqf6 its stored on chip and nowhere on a accessible flash. so you need to modify the on chip eeprom which has usually a efuse to be non writeable. other option is to reverse engineer the firmware which is a very hard task