kaloz/mwlwifi #293: Changing LEDs settings causes WIFI crash

fbafkis, 2018-04-30:
Hi! I have LEDE Reboot 17.01.4 with firmware 4.4.92. I have the latest mwlwifi drivers installed (10.3.4.0-20180330 with 9.3.2.6 Marvell firmware). If I try to change (or better to override) the invisible, at least under the LED configuration page in LEDE, default configuration (I want that the led stay firmly on when the corresponding interface is turned on and not that the leds blink only if a device is connected to the interface) then the WIFI crashes (all the interfaces related to the led-setting changes). Is it related to the driver/firmware of the Marvell module, or is it a LEDE's problem? 

Thank you for the support!

ryan-dev, 2018-04-30:
I am no expert but this is a OpenWRT issue, I have also noticed that when I
first played with those settings.

yuhhaurlin, 2018-05-01:
Sorry. Please focus issues of mwlwifi here.

closed 2018-05-01 by yuhhaurlin

metronidazole, 2018-09-09:
@ryan-dev 

How do you know this is an OpenWRT issue? You keep on posting this yet provide no evidence.

The error in syslog clearly shows error occurring in the kernel, specifically the wireless driver. Userspace should not be able to break kernel space. Ergo, the driver is at fault.

ryan-dev, 2018-09-09:
There is no syslog provided, lots of issues that have happened have been at
the hands of faulty code from within OpenWRT.

I have been watching and experiencing mwlwifi with both DD-WRT and
OpenWRT/LEDE since the start, lots of problems including crashes have been
caused by the driver as well.

You speak that I have failed to provide proof, however you haven't provided
any proof aside from that you are a troll posting to a 4 month old topic.

metronidazole, 2018-09-09:
> You speak that I have failed to provide proof, however you haven't provided any proof 

You made the claim first (and on multiple occasions) so the burden of proof is on you. Secondly, the syslogs have already been posted on other issues here, the same issues that you yourself has posted on. You yourself have commented on experiencing the same issue. So don't pull some bullshit "there are no logs".

Truncated syslog:

> Wed Aug 30 22:28:36 2017 kern.err kernel: [15918.613620] ieee80211 phy1: buffer is NULL for tx done ring
> Wed Aug 30 22:28:36 2017 kern.err kernel: [15918.793616] ieee80211 phy1: buffer is NULL for tx done ring
> Wed Aug 30 22:28:36 2017 kern.err kernel: [15918.853615] ieee80211 phy1: buffer is NULL for tx done ring
> Wed Aug 30 22:28:36 2017 kern.err kernel: [15919.003613] ieee80211 phy1: buffer is NULL for tx done ring
> Wed Aug 30 22:28:36 2017 kern.err kernel: [15919.063612] ieee80211 phy1: buffer is NULL for tx done ring
> Wed Aug 30 22:28:37 2017 kern.err kernel: [15919.229806] ieee80211 phy1: adapter does not exist
> Wed Aug 30 22:28:37 2017 kern.err kernel: [15919.234629] ieee80211 phy1: adapter does not exist
> Wed Aug 30 22:28:37 2017 kern.err kernel: [15919.239440] ieee80211 phy1: adapter does not exist
> 

> aside from that you are a troll posting to a 4 month old topic.

This just makes you sound stupid.

ryan-dev, 2018-09-09:
Interesting that you picked a 4 month old issue to dispute the merits of me
being wrong.

Where were you 4 months ago, interesting that instead of you choosing to
assist in having the problem fixed you rather be a troll.

I am intrigued that you are offended that fingers "may" been pointed at
other causes than mwlwifi (this has not always been so).

This so called LED issue, you claim is caused by the driver only affects
OpenWRT. If you don't play with default settings nothing bad happens and
this happened before the driver was properly reporting traffic data.

Nice log you posted, there is no proof that is caused by changing LEDs.

Everyone makes mistakes, instead of pointing guilt at posts regarding
fingers pointed away from OpenWRT (lots of times that was justified),
perhaps you should look and judge the posts of mine about when I was
pointing fingers at mwlwifi for driver issues.

You say you are not a troll, yet you have made yourself look stupid
disputing the merits of a 4 month old issue -:)

>

metronidazole, 2018-09-09:
I'm not sure why you keep on bringing up the issue being 4 months old as having any significance. It doesn't. I didn't have the device four months old, so it wasn't an issue for me then.

> Interesting that you picked a 4 month old issue to dispute the merits of me being wrong. 

You know what your issue is? You assume I'm out to get you. That's a character flaw on you. All i want is the bug fixed. Please keep your self-esteem issues to yourself.

> I am intrigued that you are offended that fingers "may" been pointed at other causes than mwlwifi (this has not always been so). This so called LED issue, you claim is caused by the driver only [...]

I'm not offended. You said it wasn't caused by the driver and I wanted to know what made you think that.

>  Nice log you posted, there is no proof that is caused by changing LEDs.

I never said it was proof. It's evidence. You know there's a difference right? 
The kernel errors in mwlfwifi I posted only occur when you set the LEDs to trigger on wireless activity. It's 100% reproducible.

> there is no proof that is caused by changing LEDs. 

It's already been established on multiple threads both here and on the OpenWRT forums that changing the LED triggers is what triggers the issue to occur. You yourself have posted experiencing the same bug in #212 . Now whether or not OpenWRT is at fault or mwlwifi is at fault is another matter, and is the subject of my first reply here, before you derailed the thread.

>  You say you are not a troll, yet you have made yourself look stupid disputing the merits of a 4 month old issue -:)

The only person who looks stupid is you. You keep on parroting the issue is 4 month old, as if it had any significance whatsoever.

yuhhaurlin, 2018-09-09:
I don’t know what do you mean LED trigger. Activity LED of wireless is done by firmware with specific GPIO. Mwlwifi does not control it.

metronidazole, 2018-09-10:
Mwlwifi doesn't control the LEDs, but the kernel errors that occur when you set the LEDs to trigger based off wireless activity DO occur in mwlwifi.

https://github.com/kaloz/mwlwifi/search?q=adapter+does+not+exist&unscoped_q=adapter+does+not+exist

https://github.com/kaloz/mwlwifi/search?q=buffer+is+NULL+for+tx+done&unscoped_q=buffer+is+NULL+for+tx+done

https://openwrt.org/docs/guide-user/base-system/led_configuration

ryan-dev, 2018-09-10:
@metronidazole

What is really interesting here is that you created a special github
account to troll me. If you really wanted it fixed, you would have work
harder to prove your case and move on.

Sounds like character attacks is all you are here for, so are you a user,
contributor, or lead developer at OpenWRT?

I haven't followed the OpenWRT forums since the suspicious server failure
at Digital Ocean, LEDE forums have never been my strong suite.