kaloz/mwlwifi #295: Very Slow 5G WiFi Connections for Laptop and Tablet - MAC80211 version 10.3.4.0-20180330

2Torr, 2018-05-15:
Hi Everyone,

I've been successfully running David's LEDE distributions on my Linksys WRT3200ACM router since Fall 2017 with no issues until Feb 24, 2018 (r6302). The last firmware that worked well for me is r5917. Every version after that, including the latest r6868, causes my WiFi connection for both my laptop (HP Envy 17) and Tablet (Samsung Galaxy-Tab-A-2016) to come to a crawl. I have very strong WiFi signal (computer in same room as router). When the adapter first connects, it says it's connected at 433Mb/s speed, however, in the LEDE GUI wireless status, it shows under Rx / Tx Rate:
18.0 Mbit/s, 20MHz
13.0 Mbit/s, 20MHz, VHT-MCS 1, VHT-NSS 1
Yuk!!
Reverting back to r5917 fixes everything. My laptop is running Windows 8.1 Pro with Intel Dual Band Wireless-AC 3160 card with latest drivers. I've played with router and adapter settings and channel frequencies. Nothing helps.

Here's a brief summary:

Build  |   Download Speed (Mb/s)  |   Upload Speed (Mb/s)
-----------------------------------------------------------------
5917     145+                                       16
6302      <= 2                                       12
6520      <= 1.3                                    <= 6
6565      <= 1.2                                    <= 8.6
6868      <= 2.4				     <= 9

Kernel log (r6868):
9.474634] <<Marvell Mac80211 Wireless PCIE Network Driver version 10.3.4.0-20180330>>
[    9.482704] pci 0000:00:01.0: enabling device (0140 -> 0142)
[    9.488490] ieee80211 phy0: priv->pcmd_buf = dde50000  priv->pphys_cmd_buf = 1de50000
[    9.540434] ieee80211 phy0: fw download start
[    9.540440] ieee80211 phy0: ddr init: download start
[    9.540817] ieee80211 phy0: ddr init: download complete
[    9.678603] ieee80211 phy0: FwSize = 450732 downloaded Size = 450732 curr_iteration 65522
[   12.546417] ieee80211 phy0: fw download complete
[   12.546442] ieee80211 phy0: device node: /soc/pcie/pcie@1,0
[   12.561543] ieee80211 phy0: pcmd = dde50000
[   12.579435] ieee80211 phy0: firmware version: 0x9030206
[   12.939422] ieee80211 phy0: firmware region code: 10
[   12.969046] ieee80211 phy0: 2G disabled, 5G enabled
[   12.973959] ieee80211 phy0: 4 TX antennas, 4 RX antennas
[   12.979426] pci 0000:00:02.0: enabling device (0140 -> 0142)
[   12.985233] ieee80211 phy1: priv->pcmd_buf = d9aa8000  priv->pphys_cmd_buf = 19aa8000
[   12.990447] ieee80211 phy1: fw download start
[   12.990453] ieee80211 phy1: ddr init: download start
[   12.990834] ieee80211 phy1: ddr init: download complete
[   13.133706] ieee80211 phy1: FwSize = 450732 downloaded Size = 450732 curr_iteration 65527
[   14.326421] ieee80211 phy1: fw download complete
[   14.326464] ieee80211 phy1: device node: /soc/pcie/pcie@2,0
[   14.341602] ieee80211 phy1: pcmd = d9aa8000
[   14.359434] ieee80211 phy1: firmware version: 0x9030206
[   14.719422] ieee80211 phy1: firmware region code: 10
[   14.743203] ieee80211 phy1: 2G enabled, 5G disabled
[   14.748110] ieee80211 phy1: 4 TX antennas, 4 RX antennas

What should I try next? I believe this issue is related to the MAC80211 WiFi driver / fw for the Marvell 88W8964 chip.
Any help would be much appreciated!
Thanks,
Ed

Keltere, 2018-05-16:
Me too same problem but with my android phone i get some random disconnection even behind a wireless repeater. With my wrt1900acs no problem at all. It's a problem just for wrt3200 5ghz

eduperez, 2018-05-17:
My two cents:
- 50/50 Mbps FFTH internet connection
- WRT3200ACM + LEDE 17.01.4 + mwlwifi fcaea79
- Xiaomi Mi A1 + Android 8.0, connected to the 5GHz band, a couple of meters away from the router
- SpeedTest Android App, latest version, and default options
- Results: 14ms ping, 40ms jitter, 45.9 Mbps down, 46.4 Mbps up

I can saturate the internet connection at 45Mbps using the 5GHz band (far from the speeds reported here); however, I have not done any tests recently using only internal traffic. If you feel I can be of any help, just tell me.

yuhhaurlin, 2018-05-21:
I think you need to list version of mwlwifi instead of version of LEDE.

2Torr, 2018-05-21:
Yes, sorry about that.
The last version of mwlwifi that works for me is 10.3.4.0-20180118-1
(kmod-mwlwifi - 4.9.77+10.3.4.0-20180118-1)

With that driver, all my 5GHz connections work great, including my HP Envy laptop and Samsung Galaxy tablet.
Here are some results:
wlan0    ...   Envy-TS17	-70 / -96 dBm	   54.0 Mbit/s, 20MHz
 								   433.3 Mbit/s, 80MHz,VHT-MCS 9, VHT-NSS 	Short GI

wlan0    ....    android-xxx1      -76 / -96 dBm	150.0 Mbit/s, 40MHz, MCS 7, Short GI
									150.0 Mbit/s, 40MHz, MCS 7, Short GI

wlan0    ...     Galaxy-Tab-A       0 / -96 dBm	54.0 Mbit/s, 20MHz
									292.6 Mbit/s, 80MHz, VHT-MCS 6, VHT-NSS 1, Short GI

Comcast speed test with this driver has result: 160 Mb/s down, 12 Mb/s up (I have 150 Mb/s service). 
With newer mwlwifi driver, Comcast speed test is < 10 Mb/s down (16x slower!), < 10 Mb/s up for 5GHz devices.

yuhhaurlin, 2018-05-22:
1. Please user iperf to test throughput between router and client.
2. Please swap firmware to see if this issue is related to firmware (/lib/firmware/mwlwifi/88W8964.bin).

eduperez, 2018-05-22:
@2Torr I just tried to use iperf3 between the phone mentioned above and a desktop PC connected to the router by an ethernet cable, and I'm getting about 250 - 275 Mbps with fcaea79. Will try with other wireless devices later.

2Torr, 2018-05-22:
@eduperez: I experience the extreme slowdown for 5 GHz connections only for 2 of my wireless devices:
HP Envy-17ts laptop
Samsung Galaxy-Tab-A-2016

all of my other 5GHz connections (e.g. android phones, Samsung TV, etc.) seem fine with the latest mwlwifi driver.

eduperez, 2018-05-23:
@2Torr I'm trying to get my hands on more 5GHz devices to test

bbogey, 2018-05-24:
I got OnePlus 6 and have problem with upload speed from OP3 to wrt1900acs.
This happens when roaming from Archer C5 to wrt1900acs and slowness is only on 5GHz. 2.4GHz is ok.
For roaming FT / 802.11r is enabled.
Other way around no issue, so when roaming back to Archer or other routers speed is OK.

Also if I just restart WiFi on the phone, speed is OK again.

I'm at
r7020-43d70a7a9d
10.3.4.0-20180330

Before I was in 
r5329-769e1bac7b
10.3.4.0-20170810
and it had the same issue

```
jukka@bogey:~/openwrt$ iperf3 -c 192.168.1.181 -R
Connecting to host 192.168.1.181, port 5201
Reverse mode, remote host 192.168.1.181 is sending
[  4] local 192.168.1.33 port 42946 connected to 192.168.1.181 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec   123 KBytes  1.01 Mbits/sec                  
[  4]   1.00-2.00   sec   132 KBytes  1.08 Mbits/sec                  
[  4]   2.00-3.00   sec   177 KBytes  1.45 Mbits/sec                  
[  4]   3.00-4.00   sec  63.6 KBytes   521 Kbits/sec                  
[  4]   4.00-5.00   sec  69.3 KBytes   568 Kbits/sec                  
[  4]   5.00-6.00   sec   158 KBytes  1.30 Mbits/sec                  
[  4]   6.00-7.00   sec   112 KBytes   915 Kbits/sec                  
[  4]   7.00-8.00   sec   113 KBytes   927 Kbits/sec                  
[  4]   8.00-9.00   sec   119 KBytes   973 Kbits/sec                  
[  4]   9.00-10.00  sec   148 KBytes  1.22 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.00  sec  2.03 MBytes  1.70 Mbits/sec                  sender
[  4]   0.00-10.00  sec  1.19 MBytes   995 Kbits/sec                  receiver

iperf Done.
jukka@bogey:~/openwrt$ iperf3 -c 192.168.1.181 
Connecting to host 192.168.1.181, port 5201
[  4] local 192.168.1.33 port 42950 connected to 192.168.1.181 port 5201
[ ID] Interval           Transfer     Bandwidth       Retr  Cwnd
[  4]   0.00-1.00   sec  54.7 MBytes   459 Mbits/sec    0   1.35 MBytes       
[  4]   1.00-2.00   sec  57.9 MBytes   486 Mbits/sec    1   1.35 MBytes       
[  4]   2.00-3.00   sec  56.3 MBytes   472 Mbits/sec    1   1.35 MBytes       
[  4]   3.00-4.00   sec  65.8 MBytes   552 Mbits/sec    0   1.35 MBytes       
[  4]   4.00-5.00   sec  66.7 MBytes   559 Mbits/sec    0   1.35 MBytes       
[  4]   5.00-6.00   sec  66.4 MBytes   557 Mbits/sec    0   1.35 MBytes       
[  4]   6.00-7.00   sec  64.6 MBytes   542 Mbits/sec    1   1.05 MBytes       
[  4]   7.00-8.00   sec  66.7 MBytes   560 Mbits/sec    0   1.15 MBytes       
[  4]   8.00-9.00   sec  67.0 MBytes   562 Mbits/sec    0   1.22 MBytes       
[  4]   9.00-10.00  sec  65.5 MBytes   549 Mbits/sec    0   1.27 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-10.00  sec   632 MBytes   530 Mbits/sec    3             sender
[  4]   0.00-10.00  sec   632 MBytes   530 Mbits/sec                  receiver

iperf Done.
```

After restarting WiFi on the phone speed is better

```
jukka@bogey:~/openwrt$ iperf3 -c 192.168.1.181 -R
Connecting to host 192.168.1.181, port 5201
Reverse mode, remote host 192.168.1.181 is sending
[  4] local 192.168.1.33 port 42972 connected to 192.168.1.181 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  28.7 MBytes   241 Mbits/sec                  
[  4]   1.00-2.00   sec  39.0 MBytes   327 Mbits/sec                  
[  4]   2.00-3.00   sec  32.4 MBytes   272 Mbits/sec                  
[  4]   3.00-4.00   sec  41.9 MBytes   351 Mbits/sec                  
[  4]   4.00-5.00   sec  43.2 MBytes   362 Mbits/sec                  
[  4]   5.00-6.00   sec  43.5 MBytes   365 Mbits/sec                  
[  4]   6.00-7.00   sec  44.8 MBytes   376 Mbits/sec                  
[  4]   7.00-8.00   sec  43.7 MBytes   366 Mbits/sec                  
[  4]   8.00-9.00   sec  44.1 MBytes   370 Mbits/sec                  
[  4]   9.00-10.00  sec  42.6 MBytes   357 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.00  sec   407 MBytes   342 Mbits/sec                  sender
[  4]   0.00-10.00  sec   404 MBytes   339 Mbits/sec                  receiver

iperf Done.
jukka@bogey:~/openwrt$ iperf3 -c 192.168.1.181 
Connecting to host 192.168.1.181, port 5201
[  4] local 192.168.1.33 port 42976 connected to 192.168.1.181 port 5201
[ ID] Interval           Transfer     Bandwidth       Retr  Cwnd
[  4]   0.00-1.00   sec  1.64 MBytes  13.8 Mbits/sec    1   66.5 KBytes       
[  4]   1.00-2.00   sec  32.3 MBytes   271 Mbits/sec    0    221 KBytes       
[  4]   2.00-3.00   sec  48.5 MBytes   407 Mbits/sec    0    337 KBytes       
[  4]   3.00-4.00   sec  53.9 MBytes   452 Mbits/sec    0    436 KBytes       
[  4]   4.00-5.00   sec  60.8 MBytes   510 Mbits/sec    0    529 KBytes       
[  4]   5.00-6.00   sec  57.3 MBytes   481 Mbits/sec    0    601 KBytes       
[  4]   6.00-7.00   sec  50.9 MBytes   427 Mbits/sec    0    656 KBytes       
[  4]   7.00-8.00   sec  55.8 MBytes   468 Mbits/sec    0    713 KBytes       
[  4]   8.00-9.00   sec  50.9 MBytes   427 Mbits/sec    0    762 KBytes       
[  4]   9.00-10.00  sec  50.3 MBytes   422 Mbits/sec    0    809 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-10.00  sec   462 MBytes   388 Mbits/sec    1             sender
[  4]   0.00-10.00  sec   462 MBytes   387 Mbits/sec                  receiver

iperf Done.
```

bbogey, 2018-05-24:
In station list only difference what I see when it is slow that preamble is short.
Is it possible to force to long for testing?

Station 64:a2:f9:xx:xx:xx (on wlan0-2)
        inactive time:  0 ms
        rx bytes:       4684
        rx packets:     66
        tx bytes:       2500
        tx packets:     19
        tx retries:     0
        tx failed:      0
        rx drop misc:   0
        signal:         -59 dBm
        signal avg:     -67 dBm
        tx bitrate:     866.7 MBit/s VHT-MCS 9 80MHz short GI VHT-NSS 2
        rx bitrate:     48.0 MBit/s
        authorized:     yes
        authenticated:  yes
        associated:     yes
        preamble:       short
        WMM/WME:        yes
        MFP:            no
        TDLS peer:      no
        DTIM period:    2
        beacon interval:100
        short slot time:yes
        connected time: 35 seconds

bbogey, 2018-05-25:
Here steps to reproduce the issue 100%

AP1 = Linksys WRT1900ACS, Openwrt r7020, mwlwifi 10.3.4.0-20180330
AP2 = Archer C5 v1, Openwrt r7021
Phone = OnePlus 6, OOS 5.1.5

Have same SSID configured on both APs on 5GHz and place so that they overlap a bit.

1. Connect Phone to AP1. Upload speed is OK.
2. Walk near AP2 so it gets connected to it. Upload still OK.
3. Walk back to AP1 so it gets connected to it, upload speed is very bad.
4. Switch phone wifi OFF ON to reconnect to AP1, upload speed is good again.

bbogey, 2018-05-27:
Any other logs I could collect?
I have this issue only with wrt1900acs..

bbogey, 2018-05-28:
I monitored the AP wlan interface while roaming, seems OnePlus 6 capabilities gets changed while roaming.
I have also collected tcpdumps from both AP and Phone, but not sure what to look from them.
At least there is a lot of TCP Dup Acks, TCP Re-transmissions and TCP Previous segment not captured.

Attached screenshot below for the capabilities.

[image]

bbogey, 2018-05-29:
@yuhhaurlin mwlwifi does not support Short Preamble?

2Torr, 2018-05-29:
@yuhhaurlin, @eduperez:

I ran iperf3 tests between device and router for both devices mentioned before:

Driver: mwlwifi 10.3.4.0-20180118-1

1. Device HP Envy-17 laptop:
------------------------------------
root@OLAN:~# iperf3 -c 192.168.1.143 -R
Connecting to host 192.168.1.143, port 5201
Reverse mode, remote host 192.168.1.143 is sending
[  5] local 192.168.1.1 port 37326 connected to 192.168.1.143 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  12.0 MBytes   101 Mbits/sec
[  5]   1.00-2.00   sec  14.7 MBytes   123 Mbits/sec
[  5]   2.00-3.00   sec  14.6 MBytes   123 Mbits/sec
[  5]   3.00-4.00   sec  16.0 MBytes   134 Mbits/sec
[  5]   4.00-5.00   sec  16.4 MBytes   137 Mbits/sec
[  5]   5.00-6.00   sec  10.1 MBytes  85.1 Mbits/sec
[  5]   6.00-7.00   sec  4.55 MBytes  38.1 Mbits/sec
[  5]   7.00-8.00   sec  4.65 MBytes  39.0 Mbits/sec
[  5]   8.00-9.00   sec  10.4 MBytes  87.4 Mbits/sec
[  5]   9.00-10.00  sec  15.9 MBytes   133 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.04  sec   120 MBytes  99.8 Mbits/sec                  sender
[  5]   0.00-10.00  sec   119 MBytes   100 Mbits/sec                  receiver

root@OLAN:~# iperf3 -c 192.168.1.143
Connecting to host 192.168.1.143, port 5201
[  5] local 192.168.1.1 port 37330 connected to 192.168.1.143 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  10.1 MBytes  84.5 Mbits/sec    0    132 KBytes
[  5]   1.00-2.00   sec  10.9 MBytes  91.4 Mbits/sec    0    132 KBytes
[  5]   2.00-3.00   sec  13.6 MBytes   114 Mbits/sec    0    132 KBytes
[  5]   3.00-4.00   sec  13.6 MBytes   114 Mbits/sec    0    132 KBytes
[  5]   4.00-5.00   sec  13.3 MBytes   112 Mbits/sec    0    132 KBytes
[  5]   5.00-6.00   sec  13.2 MBytes   111 Mbits/sec    0    132 KBytes
[  5]   6.00-7.00   sec  13.4 MBytes   112 Mbits/sec    0    132 KBytes
[  5]   7.00-8.00   sec  13.7 MBytes   115 Mbits/sec    0    132 KBytes
[  5]   8.00-9.00   sec  13.4 MBytes   112 Mbits/sec    0    132 KBytes
[  5]   9.00-10.00  sec  13.4 MBytes   112 Mbits/sec    0    132 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   128 MBytes   108 Mbits/sec    0             sender
[  5]   0.00-10.05  sec   128 MBytes   107 Mbits/sec                  receiver

2. Device Samsung Galaxy-Tab-A-2016:
------------------------------------
root@OLAN:~# iperf3 -c 192.168.1.182 -R
Connecting to host 192.168.1.182, port 5201
Reverse mode, remote host 192.168.1.182 is sending
[  5] local 192.168.1.1 port 46650 connected to 192.168.1.182 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  29.6 MBytes   249 Mbits/sec
[  5]   1.00-2.00   sec  26.9 MBytes   226 Mbits/sec
[  5]   2.00-3.00   sec  30.1 MBytes   252 Mbits/sec
[  5]   3.00-4.00   sec  29.5 MBytes   248 Mbits/sec
[  5]   4.00-5.00   sec  28.0 MBytes   235 Mbits/sec
[  5]   5.00-6.00   sec  27.8 MBytes   233 Mbits/sec
[  5]   6.00-7.00   sec  29.4 MBytes   246 Mbits/sec
[  5]   7.00-8.00   sec  26.8 MBytes   225 Mbits/sec
[  5]   8.00-9.00   sec  23.9 MBytes   200 Mbits/sec
[  5]   9.00-10.00  sec  29.7 MBytes   249 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.00  sec   283 MBytes   237 Mbits/sec                  sender
[  5]   0.00-10.00  sec   283 MBytes   237 Mbits/sec                  receiver

root@OLAN:~# iperf3 -c 192.168.1.182
[... 82 lines cut ...]
[  5]   0.00-10.00  sec   211 MBytes   177 Mbits/sec                  sender
[  5]   0.00-10.00  sec   210 MBytes   176 Mbits/sec                  receiver

root@OLAN:~# iperf3 -c 192.168.1.182
Connecting to host 192.168.1.182, port 5201
[  5] local 192.168.1.1 port 34116 connected to 192.168.1.182 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  25.7 MBytes   215 Mbits/sec    0    513 KBytes
[  5]   1.00-2.00   sec  30.1 MBytes   252 Mbits/sec    0    513 KBytes
[  5]   2.00-3.00   sec  31.7 MBytes   266 Mbits/sec    0    513 KBytes
[  5]   3.00-4.00   sec  30.3 MBytes   254 Mbits/sec    0    513 KBytes
[  5]   4.00-5.00   sec  30.0 MBytes   252 Mbits/sec    0    540 KBytes
[  5]   5.00-6.00   sec  33.0 MBytes   277 Mbits/sec    0    540 KBytes
[  5]   6.00-7.00   sec  30.4 MBytes   255 Mbits/sec    0    540 KBytes
[  5]   7.00-8.00   sec  30.7 MBytes   258 Mbits/sec    0    540 KBytes
[  5]   8.00-9.00   sec  32.7 MBytes   274 Mbits/sec    0    540 KBytes
[  5]   9.00-10.00  sec  30.6 MBytes   257 Mbits/sec    0    540 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   305 MBytes   256 Mbits/sec    0             sender
[  5]   0.00-10.00  sec   304 MBytes   255 Mbits/sec                  receiver

Summary:
HP Envy-17 device experiences ~10x slower download, ~2x slower upload speeds (between device and router) with latest mwlwifi 2018-03-30-fcaea79a-1 driver, compared to mwlwifi 10.3.4.0-20180118-1 driver.  Samsung Galaxy-Tab-A-2016 device experiences only 5% slower download, ~25% slower upload speeds now (after software update), even though it still associates with router as: 54.0 Mbit/s, 20MHz; 6.0 Mbit/s, 80MHz
----------

bbogey, 2018-05-29:
Tested also with davidc502 build r6952-5399de754d  and it has the same issue.

p3x-robot, 2018-05-29:
i can confirm that something is weird, i never had wifi issues for a long time on 3200acm, i think, the last commit did this. i am on lede 17.01.4, totally something cannot register itself, or wifi goes away, and must be the last commit, i think, before always working.

p3x-robot, 2018-05-29:
just, i have no idea, when it started going wrong, because, i was rebuilding every time, i saw a new commit, that's sad...  but in the last weeks, it was going bad. sometimes, i couldn't find the wifi, sometimes, even after reboot. i will shoot the error log, if it goes bad again.

p3x-robot, 2018-05-29:
though on android i can hit 200-500 mbits. so weird, but i sometimes not working with mwlwifi latest.

bbogey, 2018-05-29:
No one else has OnePlus 5 or 6 could test this?

yuhhaurlin, 2018-05-30:
@2Torr You can wait for future release to check it again.

yuhhaurlin, 2018-05-30:
@bbogey If stock firmware has the same issue, please report your problem for stock firmware.

yuhhaurlin, 2018-05-30:
@p3x-robot I have no idea what the problem you have.

p3x-robot, 2018-05-30:
sorry,it was stable for long time, then it was weird, but given we have like 50 devices in the apartment, i have no idea.

@yuhhaurlin i think you are awesome, i was questionig like others as well right now... no problem at all. ciao

bbogey, 2018-05-30:
@yuhhaurlin I don't think they bother this old device. I'll try check with OnePlus as the issue is only with OnePlus devices. Because it will most probably never be fixed their end either I ordered new AP with ath10k and maybe return the phone also...

yuhhaurlin, 2018-05-30:
@bbogey The strange thing is that reboot the phone will fix the problem. Unless using sniffer to capture air packets, I have no idea how to check it.

bbogey, 2018-05-30:
@derosier it is valid address. For different APs I have created custom MAC  addresses.

Yes screenshot I have provided to show up the odd thing I noticed from phone assoc request like wrote above. So phone capatibilities are different in association and re-association.
The thing is that wrt1900acs together with OnePlus does not work together. 
Archer C5 (ath10k) works fine with OnePlus. 
And with any other device combinations no issues...

Kherby, 2018-05-30:
Same problem here, 5ghz wifi speed is very irregular, drops often down to 3-5mbps and stays there, my wifi card (Intel AC8260) shows 390mbps and Luci shows 29mbps @20mhz.
I cant enjoy my  WRT3200acm because of the wifi performance, it's realy frustrating !!

Pls fix this, it's ridiciolous how bad this marvel wifi is...
Sometimes i do think my WRT3200acm has a faulty wifi module but it seems that im not the only one with these issues.... :/
My router from 2012 is performing way better and it really shouldn't be like this and that's not a joke.

```
root@OpenWrt:~# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20180330
firmware version: 0x09030206
power table loaded from dts: no
firmware region code: 0x30
mac address: 60:38:e0:b6:72:1a
2g: disable
5g: enable
antenna: 4 4
irq number: 49
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000003
radio: enable
iobase0: e1100000
iobase1: e1380000
tx limit: 1024
rx limit: 16384

```

yuhhaurlin, 2018-05-31:
@Kherby @bbogey's issue is related to WRT1900ACS and it happened after reassociated. What problem that you want me to fix? Please run iperf and show me the number.

bbogey, 2018-05-31:
And one more thing
In hostapd-phy0.conf there is
preamble=1
And in AP probe / (re)assoc response

.... .... ..0. .... = Short Preamble: Not Allowed

For phone it shows after associated

preamble: long

And after re-association

preamble: short

For any other device it isn't short ever.

But I don't think it matters as it is used only for b and g standards?

Kherby, 2018-05-31:
@yuhhaurlin I would love it if you finaly would fix the Wifi firmware, so my 5Ghz doesnt drop to 3.5-4mbps anymore when downloading something over the internet.
The problem will happen on my WRT3200acm as well, that's why im writing here.
Why do u want me to run iperf ? It's clear that the wifi troughput is completely out of line....
If i switch to my other Wifi AP and i'm getting full speed immediately, if i reboot the WRT3200acm im getting full speed as well for some time....

Or there any Magic wifi config that will work/fix the problem ?
I'm running iperf later, can't access any client at the moment as im not at home.

yuhhaurlin, 2018-05-31:
Please use iperf and show me the number.

Kherby, 2018-05-31:
I'll do that !!
I guess u want 5ghz WLAN >> LAN test right ?

bbogey, 2018-06-02:
Ok contacted Linksys.

http://community.linksys.com/t5/Wireless-Routers/After-reassociating-to-wrt1900acs-with-OnePlus-5-or-6-upload-is/m-p/1317289#M35242

And actually there's the source for OP6 wifi and it is pretty recent fork

https://github.com/OnePlusOSS/android_vendor_qcom-opensource_wlan_qcacld-3.0/tree/oneplus/SDM845_O_8.1?files=1

If someone has Pixel or Essential or some other Qualcomm device with Oreo 8.1 could you please test the if you could reproduce the reassociation issue.

bbogey, 2018-06-06:
Is it normal that client acts as beamformer also? I only read that it is used in APs to send beamformed frames.

bbogey, 2018-06-07:
Got Netgear r7800 and bridged to wrt1900acs directly on LAN port. No issues with speed after reassociation.

Next maybe I'll take mon capture from both at the same time and see if something found.

In mean time OnePlus has corrected the Short Preable and VHT capabilities in reassociation, not sure what other changes they did, but no help in this issue.

bbogey, 2018-06-07:
After reassoc I started upload from the phone via wrt1900acs.

Here is left monitor capture from wrt1900acs, right from r7800.
I don't see all frames in wrt1900acs capture what I can see in r7800.

[image]

Here attached the frames

[missing packets.zip](https://github.com/kaloz/mwlwifi/files/2082022/missing.packets.zip)

bbogey, 2018-06-08:
Looks like the culprit was these too large frames sent by the phone. OnePlus made a new version for me and there is no these large frames anymore and it seems to fix this issue.

bbogey, 2018-06-09:
Not sure about that, wrt1900acs work very fast and responsive even with the smallest frame size.
But yes for misbehaving clients it would help.

gufus, 2018-06-22:
@yuhhaurlin 

On my wrt1900ac I'm haveing problem. :-

yuhhaurlin, 2018-06-23:
For WRT1200 or WRT1900 (which uses 88W8864), the maximum A-MSDU size is 3839.

yuhhaurlin, 2018-06-23:
I wonder if I can close this issue?

yuhhaurlin, 2018-06-23:
Bit 11 (Maximum A-MSDU Length) of HT Capability Information field:
 
0: 3839 octets.
1: 7935 octets.
 
Bit 0..1 (Maximum MPDU length) of VHT Capability Information filed:
 
00: 3895 octets.
01: 7991 octets.
10: 11454 octets.
 
A VHT STA that sets the value of the Maximum MPDU Length subfield in the VHT Capabilities
Information field of the VHT Capabilities element to indicate 3895 octets shall set the Maximum A-MSDU
Length in the HT Capabilities element to indicate 3839 octets. A VHT STA that sets the Maximum MPDU
Length in the VHT Capabilities element to indicate 7991 octets or 11 454 octets shall set the Maximum
A-MSDU Length in the HT Capabilities element to indicate 7935 octets.
 
The length of an A-MSDU transmitted in a VHT PPDU is limited by the maximum MPDU size supported
by the recipient STA (see 10.13.5).

yuhhaurlin, 2018-06-23:
If you use WRT3200ACM (88W8964), you will not encounter this kind of thing.

gufus, 2018-06-23:
@yuhhaurlin

So is there a fix For WRT1200 or WRT1900 ?

yuhhaurlin, 2018-06-24:
WRT1200 and WRT1900 does the right thing. Maybe you can disable AMSDU to WAR the problem of client.

gufus, 2018-06-24:
@yuhhaurlin

Sorry, I forgot how to.

How?

yuhhaurlin, 2018-06-25:
Sorry. From debugfs file, you can only disable AMSDU for Tx.

For Rx, you need to mark off following line from core.c (function mwl_set_ht_caps()):

ieee80211_hw_set(hw, SUPPORTS_AMSDU_IN_AMPDU);

yuhhaurlin, 2018-06-25:
I close this issue first. If you have problem with normal client, please create another issue and describe the specific test setup and steps to reproduce the issue. Thanks.

closed 2018-06-25 by yuhhaurlin

gufus, 2018-06-25:
@yuhhaurlin 

Just a quick note

I got everything working with normal rates

yuhhaurlin, 2018-06-25:
What is the issue you have? Same as problem with phone OnePlus?