kaloz/mwlwifi #296: Please Mu-Mimo.

adamcarter1, 2018-05-18:
Thank you

Keltere, 2018-05-20:
yuhhaurlin wrote this 17 days ago

`It is under working. However, other jobs will interrupt this working sometimes, but this feature will be added finally.`

Based on the complexity of the job, this statement and the long wait for previous updates i think we will wait several month if not even a year before we get Mu-Mimo.
Anyway some updates on the status of works will be appreciate by the community.

thagabe, 2018-05-21:
@adamcarter1 I recommend picking up a better supported Open Source router. Marvell has consistently shown their complacency regarding this driver. They have only assigned one person for this project and now they are cutting into his time to make him work in another project. This driver will never be feature parity with the closed source driver. After following and testing this driver since the original WRT1900AC and now with a WRT3200ACM, I have finally given up on this driver being truly open sourced or being on par with the closed driver. It's unfortunate as the WRT line was so beloved by the open source community. Belkin tried to revive it but chose Marvell who clearly do not care about open source. If you can sell or return this router I suggest you do. Best of luck- G

yuhhaurlin, 2018-05-21:
No. Marvell will support this driver.

yuhhaurlin, 2018-05-21:
Trace of MU-MIMO is done. We will complete it ASAP.

BrainSlayer, 2018-05-21:
@thagabe : but you should consider that no opensource router yet does support MU-MIMO. so marvell would be still the first

yuhhaurlin, 2018-05-21:
That is true. For AP mode, no open source router supports MU-MIMO. For mwlwifi, we will port simple grouping rule to support MU-MIMO for AP mode.

adamcarter1, 2018-05-24:
I have stability.

tapper82, 2018-05-24:
On the wrt3200acm? The 5ghz is a crock and keeps changing chanels. the 2.4 ghz just stops ded and needs rebooting.

adamcarter1, 2018-05-24:
Yikes, i dont know your settings, or which firmware you are running. Buy my 3200acm dd-wrt build is stable, i dont use a lot of settings. Just vpn client, policy based routing, basically it. I also dont combine my 2.4ghz and 5 ghz into one ssid, like many folks do.

Regardless. I only need to reboot if my openvpn drops.

I grant you it isnt perfect, but, i dont have your problems.

supersebbo, 2018-05-25:
5Ghz does change channels occasionally for me, but this is due to DFS which is unavoidable.  However I do feel that the radar detection is a little sensitive on this firmware, I have not seen it happen so much with the stock firmware.  2.4Ghz is rock solid however.

howl, 2018-05-27:
@BrainSlayer MU-MIMO isn't supported in ip806x routers with ath10k driver?

Chadster766, 2018-05-28:
@BrainSlayer MU-MIMO is supported on Linux PC with the below hardware spec. In the old days wireless routers were just low profile Linux PC with wireless adapters. I don't think Marvell is the first to support MU-MIMO with there long development times.

https://www.snbforums.com/threads/qnap-wi-fi-adapter-adds-wireless-features-to-linux-based-nas-and-pcs.46800/

yuhhaurlin, 2018-05-28:
What is the MU-MIMO grouping rule used by QNAP for AP mode? For ath10, it only supports MU-MIMO for client mode when I checked it last year. If you find ath10 supports MU-MIMO for AP mode, please let me know.

Chadster766, 2018-05-28:
Quote from: 
https://www.qnap.com/en/product/qwa-ac2600

> The QWA-AC2600 supports Dual Band Dual Concurrent (DBDC) for acting as an access point or a base station, adding more possibilities to your investment.

BrainSlayer, 2018-05-28:
@howl : the ath10k firmware blob does support it, but the ath10k driver does not.

BrainSlayer, 2018-05-28:
@Chadster766 : sure the hw does support it and the windows drivers do support it as well, but ath10k does not

p3x-robot, 2018-05-29:
3200acm is so low to buy, once mumimo is implemented, it will be so cheap, i love in it, just mumimo is not implemented, it will be in one two years done, then u can use it like 20 years, so little router with big antenna

Chadster766, 2018-05-31:
@BrainSlayer : The below document seems to indicate that ath10k has mu-mimo with hostapd

https://wireless.wiki.kernel.org/en/users/drivers/ath10k/configuration#mu-mimo_configuration

eduperez, 2018-05-31:
Perhaps we should stop arguing about what other drivers can o cannot do,
and focus on this one.

Chadster766, 2018-05-31:
@eduperez : We aren't arguing as far as I know. Just discussing mu-mimo. You never know maybe the ath10k source code could help speed this drivers development and stability.

BrainSlayer, 2018-06-01:
@Chadster766 : this just enables the vht cap flags for mu-mimo, but you need more than just this to run mu-mimo. this is just tx beamforming for mu-mimo

yuhhaurlin, 2018-06-01:
Yes. MU-MIMO needs explicit beamforming: VHT sounding protocol. Except for that, you still needs rule to group MU clients for AP mode.

BrainSlayer, 2018-06-01:
@yuhhaurlin : thanks for further explanation

yuhhaurlin, 2018-06-01:
In fact, I already get enough information to implement MU-MIMO for mwlwifi with simple static grouping rule. Thanks for all your information.

adamcarter1, 2018-06-28:
Ok, we have waited long enough. Please yuhhaurlin give us the power.... the power of mu-mimo..

p3x-robot, 2018-06-28:
hay he will do it, goes other work as well, i was inpatient as well. just let him be,

davidc502, 2018-06-28:
I'm not sure what kind of a difference people are expecting with mu-mimo. I hope they aren't expecting too much because they are likely to be disappointed.

p3x-robot, 2018-06-28:
We cannot even use it the bandwidth with MU-MIMO, unless u are in business but then you probable use Cisco and not OpenWrt. It will be on time. I GUESS.

adamcarter1, 2018-06-28:
Hey david i agree with you. 

It’s just that i paid for a router whn it first came out that touted open-source ready and mu-mimo. And i kinda like to get what was advertised. I paid for it. We all did.

PalebloodSky, 2018-06-28:
@davidc502 fair enough but it would be great exposure for the Linksys WRT series to be the first OpenWrt router with MU-MIMO. It would almost make up for the absolutely horrific support they had for most of 2017. Things are finally shaping up to be in a good place open source wise.

BrainSlayer, 2018-06-29:
@PalebloodSky and DD-WRT :-)

p3x-robot, 2018-06-29:
but guys, can we use up all that MU-MIMO bandwidth??? so few MU-MIMO wifi cards, right? by the time it is implemented, it will be available everywhere, so it is not a problem now, isnt it?

p3x-robot, 2018-06-29:
if you want to sell your 3200ACM i buy it! it is still the 2nd best!!!

PalebloodSky, 2018-06-29:
@BrainSlayer yes and thanks for all your hard work on DD-WRT, it's run great on my R7000 for a while now. For my WRT32X using OpenWRt 18.06 :)

Skilmatix, 2018-06-29:
@BrainSlayer, is there a plan to support 802.11r on DD-WRT?  That’s the only reason I switched to LEDE. And thanks for all your work on the firmware. I’ve enjoyed them both.

BrainSlayer, 2018-06-29:
@Skilmatix i recently released a working version for the WRT32X. originally a version was available since a long time but only flashable by commandline. the linksys gui has a file size restriction so i was able to create a overtake image for dd-wrt as well. which can be flashed as initial image with limited features. then you can upgrade to the full version. its basicly identical with the wrt3200acm version since the hw is identical and its always up to date with drivers unlike openwrt

BrainSlayer, 2018-06-29:
but its no debian style system for sure. its designed for use as is, but has a commandline for custom things. but not that flexible as openwrt. but robust

BrainSlayer, 2018-06-29:
@Skilmatix 802.11r was not big thing they. i dont believe in mesh. its more like a low performance use as is system and its also not supported by all chipsets. i believe more in dedicated long range links and local ap installations. i played with 802.11r but never found any use in it. performance sucks for professional applications

Skilmatix, 2018-06-29:
@BrainSlayer: got it and thanks!

PalebloodSky, 2018-06-29:
@BrainSlayer yes I've been running davidc502's build of OpenWrt Master which has the latest drivers. Ran DD-WRT on my R7000 and you do great work, but decided at least for the time being to go OpenWrt on my new WRT32X. I really like this router (overall look, performance, etc.) I hope open source development for it continues.

BrainSlayer, 2018-06-30:
@bbogey you mean fast transission. this is basicly no problem to implement in dd-wrt. the curious thing is that i never had issues with roaming even with normal wpa authentication. seamless handover can be handled by the client without connection loss. dont know if it really is usefull. but might be for some people. but another problem is the constent code increase in hostapd and wpa supplicant and i still have to support smaller devices. its not just about the wrt1900 series

bbogey, 2018-06-30:
Yes, I mean fast bss transition. Yes also in openwrt it requires full wpad to have it working.

My OnePlus doesn't support fast bss transition, but still while roaming it is not disconneting. But for example iPad air 2 SSH gets disconnected while roaming without 802.11r enabled. I haven't tested for a while without, it might not be the case anymore.

adamcarter1, 2018-08-29:
Looks like 802.11 ax comes out in september with asus rog gt-11000.  I think i was correct when i said this wouldn’t get fixed before 802.11ax .

PalebloodSky, 2018-09-20:
Sadly they don't Linksys lied and doesn't support their routers like they claim. They haven't' even updated the stock firmware in 6+ months let alone supporting open source.

s-pimenta, 2018-09-26:
I'm tired of waiting and waiting... Please finish the backlog of 88W8864 and do not pass to 88W8997! 

You are tring to let old things to die and be forgotten...

Support from Marvell is horrendous! They will pass to 88W8997 and also not finish all the features... 

So all in the end Marvell support is all of incomplete drivers...

Maybe I will sell my WRT3200ACM and still get some money back...

p3x-robot, 2018-09-26:
hey, forget loosing your precious router, linksys 3200acm is awesome and you cannot even use up the bandwidth, it is just too new.
check out, with the latest wifi:  
https://github.com/patrikx3/openwrt-insomnia  
https://cdn.corifeus.com/openwrt/18.06.1/targets/mvebu/cortexa9/  
https://cdn.corifeus.com/openwrt/18.06.1/targets/mvebu/cortexa9/insomnia-openwrt-18.06.1-mvebu-cortexa9-linksys-wrt3200acm-squashfs-factory.img

s-pimenta, 2018-09-26:
@yuhhaurlin Sorry, but this feature is being promissed since JANUARY! 

10 MONTHS later and still nothing... That's too much... I'm DONE!

@p3x-robot The WRT3200ACM has very good specs... 

But if the software (drivers) are SHIT and you cannot extract the full potential of the harware... is not worth to keep it... 

Better to buy a cheaper one with the same features that are working on the WRT3200ACM...

p3x-robot, 2018-09-26:
hey, can you sell me cheap? i wanna replace 1900acs to 3200acm ...

yuhhaurlin, 2018-09-26:
There is no open source router which supports this feature for AP mode.

s-pimenta, 2018-09-26:
Where are you from? I'm from Portugal

p3x-robot, 2018-09-26:
have you tried the latest wifi installed? i have no issues and it flights like a unicorn. which software is not working for you out of the 5K packages in OpenWrt 18.06.1?

s-pimenta, 2018-09-26:
It's not only the MU-MIMO, and fix for esp32 devices, slow speeds on intel wifi cards, and I cannot use 5ghz on channel 100 and up... And the list of things to be fixed is being dragged to die and be forgotten....

p3x-robot, 2018-09-26:
the whole router that is not perfect is the MU-MIMU, but i cannot use the bandwidth, so it is weird everyone is having problems with it, hating it. i even added a huge antenna and it is like a Unicorn :1st_place_medal:

p3x-robot, 2018-09-26:
i use an intel 9260 on 800 Mbit 5GHZ. intel wifi cards as @BrainSlayer says, the intel wifi cards are not good. not only the router wifi is bad, but u are using an intel wifi card, which is not the best totally.

p3x-robot, 2018-09-26:
yes the, 100 channels for me too it reverts itself to 36 channel, but the issue is that there is no issue, i can fly on 800 mbit easy on 36 channel and we have like 40 routers in the apartment complex and the other packages like VPN and all others is awesome, i use overlay via USB3, like a big router.

if you befriend your router, it will be nicer to you!!!

yuhhaurlin, 2018-09-26:
Remaining issues will be checked later.

p3x-robot, 2018-09-26:
ok, so as we get is that, it is still comming ... later :) :+1:

derosier, 2018-09-26:
Hey guys, let's not forget something: _you_ aren't paying yuhhaurlin's salary. He works for Marvell, and thus his priority list is what his bosses want it to be, not what you want it to be. Perhaps Marvell is at fault in advertising something it's not, though, IIRC the devices worked out-of-the-box with the specs that were on the box. Any open-source-work that yuhhaurlin does is either on his own time and/or comes directly from his work. Unless your goal is to get him to quit on this project, be appreciative of any effort in a positive direction.

aaron1959, 2018-09-26:
@derosier You don't recall correctly the devices didn't meet the claims made by the manufacture out of the box. None of the devices will meet those claims today!!!

PalebloodSky, 2018-09-26:
Yes these complaints are not and should not be against @yuhhaurlin he has done a great job. The open source support from Marvel is pathetic and there is absolutely no reason to move on to the 88W8997 which has no meaningful improvement nor does it help any of these high end routers with the 88W8864 (WRT3200ACM, WRT32X, etc.).

I will be voting with my wallet and not buying a Marvel based product next time because of this lack of support. These drivers should have met or exceeded the proprietary drivers by now. Linksys doesn't even provide firmware updates anymore. The last update from them is nearly 10 months ago.

[100 more replies]