kaloz/mwlwifi #297: crash dump

Ansuel, 2018-05-27:
i got this crash dump. What do you need? 

```
[  477.411738] ------------[ cut here ]------------
[  477.416436] WARNING: CPU: 1 PID: 14800 at backports-2017-11-01/net/mac80211/ieee80211_i.h:1412 sta_info_alloc+0x310/0x3e8 [mac80211]
[  477.428415] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CLASSIFY ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda slhc rfcomm nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_netlink libcrc32c iptable_mangle iptable_filter ipt_ECN ip_tables hidp hci_uart crc_ccitt btusb btmrvl_sdio btmrvl btintel bnep bluetooth
[  477.499782]  sch_cake act_connmark nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress hid evdev input_core mwlwifi mac80211 cfg80211 compat ledtrig_gpio cryptodev xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ifb ecdh_generic sha512_generic sha256_generic seqiv jitterentropy_rng drbg md5 kpp hmac fcrypt echainiv ecb ctr cmac cbc authenc uhci_hcd gpio_button_hotplug
[  477.570791]  btrfs xor zstd_decompress zstd_compress xxhash xor_neon raid6_pq
[  477.577967] CPU: 1 PID: 14800 Comm: hostapd Tainted: G        W       4.14.43 #0
[  477.585394] Hardware name: Marvell Armada 380/385 (Device Tree)
[  477.591351] [<c010f64c>] (unwind_backtrace) from [<c010b348>] (show_stack+0x10/0x14)
[  477.599132] [<c010b348>] (show_stack) from [<c063f9a8>] (dump_stack+0x88/0x9c)
[  477.606388] [<c063f9a8>] (dump_stack) from [<c0121fd4>] (__warn+0xe4/0x100)
[  477.613381] [<c0121fd4>] (__warn) from [<c01220a0>] (warn_slowpath_null+0x20/0x28)
[  477.621008] [<c01220a0>] (warn_slowpath_null) from [<bf36b048>] (sta_info_alloc+0x310/0x3e8 [mac80211])
[  477.630486] [<bf36b048>] (sta_info_alloc [mac80211]) from [<bf381e64>] (ieee80211_add_station+0x80/0x120 [mac80211])
[  477.641105] [<bf381e64>] (ieee80211_add_station [mac80211]) from [<bf33e600>] (nl80211_new_station+0x330/0x370 [cfg80211])
[  477.652216] [<bf33e600>] (nl80211_new_station [cfg80211]) from [<c056dcac>] (genl_rcv_msg+0x2e8/0x388)
[  477.661566] [<c056dcac>] (genl_rcv_msg) from [<c056ce8c>] (netlink_rcv_skb+0x8c/0xfc)
[  477.669431] [<c056ce8c>] (netlink_rcv_skb) from [<c056d9b4>] (genl_rcv+0x24/0x34)
[  477.676947] [<c056d9b4>] (genl_rcv) from [<c056c660>] (netlink_unicast+0x158/0x208)
[  477.684638] [<c056c660>] (netlink_unicast) from [<c056cb04>] (netlink_sendmsg+0x334/0x37c)
[  477.692944] [<c056cb04>] (netlink_sendmsg) from [<c051d598>] (sock_sendmsg+0x14/0x24)
[  477.700810] [<c051d598>] (sock_sendmsg) from [<c051dacc>] (___sys_sendmsg+0x204/0x22c)
[  477.708763] [<c051dacc>] (___sys_sendmsg) from [<c051e8b8>] (__sys_sendmsg+0x40/0x64)
[  477.716630] [<c051e8b8>] (__sys_sendmsg) from [<c0107980>] (ret_fast_syscall+0x0/0x54)
[  477.724605] ---[ end trace 0dafb8486082d14b ]---
[  478.412706] ------------[ cut here ]------------
[  478.417384] WARNING: CPU: 1 PID: 14800 at backports-2017-11-01/net/mac80211/ieee80211_i.h:1412 sta_info_alloc+0x310/0x3e8 [mac80211]
[  478.429358] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CLASSIFY ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda slhc rfcomm nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_netlink libcrc32c iptable_mangle iptable_filter ipt_ECN ip_tables hidp hci_uart crc_ccitt btusb btmrvl_sdio btmrvl btintel bnep bluetooth
[  478.500717]  sch_cake act_connmark nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress hid evdev input_core mwlwifi mac80211 cfg80211 compat ledtrig_gpio cryptodev xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ifb ecdh_generic sha512_generic sha256_generic seqiv jitterentropy_rng drbg md5 kpp hmac fcrypt echainiv ecb ctr cmac cbc authenc uhci_hcd gpio_button_hotplug
[  478.571726]  btrfs xor zstd_decompress zstd_compress xxhash xor_neon raid6_pq
[  478.578900] CPU: 1 PID: 14800 Comm: hostapd Tainted: G        W       4.14.43 #0
[  478.586326] Hardware name: Marvell Armada 380/385 (Device Tree)
[  478.592281] [<c010f64c>] (unwind_backtrace) from [<c010b348>] (show_stack+0x10/0x14)
[  478.600060] [<c010b348>] (show_stack) from [<c063f9a8>] (dump_stack+0x88/0x9c)
[  478.607315] [<c063f9a8>] (dump_stack) from [<c0121fd4>] (__warn+0xe4/0x100)
[  478.614307] [<c0121fd4>] (__warn) from [<c01220a0>] (warn_slowpath_null+0x20/0x28)
[  478.621934] [<c01220a0>] (warn_slowpath_null) from [<bf36b048>] (sta_info_alloc+0x310/0x3e8 [mac80211])
[  478.631412] [<bf36b048>] (sta_info_alloc [mac80211]) from [<bf381e64>] (ieee80211_add_station+0x80/0x120 [mac80211])
[  478.642020] [<bf381e64>] (ieee80211_add_station [mac80211]) from [<bf33e600>] (nl80211_new_station+0x330/0x370 [cfg80211])
[  478.653131] [<bf33e600>] (nl80211_new_station [cfg80211]) from [<c056dcac>] (genl_rcv_msg+0x2e8/0x388)
[  478.662479] [<c056dcac>] (genl_rcv_msg) from [<c056ce8c>] (netlink_rcv_skb+0x8c/0xfc)
[  478.670345] [<c056ce8c>] (netlink_rcv_skb) from [<c056d9b4>] (genl_rcv+0x24/0x34)
[  478.677861] [<c056d9b4>] (genl_rcv) from [<c056c660>] (netlink_unicast+0x158/0x208)
[  478.685551] [<c056c660>] (netlink_unicast) from [<c056cb04>] (netlink_sendmsg+0x334/0x37c)
[  478.693856] [<c056cb04>] (netlink_sendmsg) from [<c051d598>] (sock_sendmsg+0x14/0x24)
[  478.701721] [<c051d598>] (sock_sendmsg) from [<c051dacc>] (___sys_sendmsg+0x204/0x22c)
[  478.709673] [<c051dacc>] (___sys_sendmsg) from [<c051e8b8>] (__sys_sendmsg+0x40/0x64)
[  478.717540] [<c051e8b8>] (__sys_sendmsg) from [<c0107980>] (ret_fast_syscall+0x0/0x54)
[  478.725622] ---[ end trace 0dafb8486082d14c ]---
[  479.434909] ------------[ cut here ]------------
[  479.439613] WARNING: CPU: 1 PID: 14800 at backports-2017-11-01/net/mac80211/ieee80211_i.h:1412 sta_info_alloc+0x310/0x3e8 [mac80211]
[  479.451595] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CLASSIFY ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda slhc rfcomm nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_netlink libcrc32c iptable_mangle iptable_filter ipt_ECN ip_tables hidp hci_uart crc_ccitt btusb btmrvl_sdio btmrvl btintel bnep bluetooth
[  479.523003]  sch_cake act_connmark nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress hid evdev input_core mwlwifi mac80211 cfg80211 compat ledtrig_gpio cryp [... cut ...]
[  479.707927] [<c056c660>] (netlink_unicast) from [<c056cb04>] (netlink_sendmsg+0x334/0x37c)
[  479.716233] [<c056cb04>] (netlink_sendmsg) from [<c051d598>] (sock_sendmsg+0x14/0x24)
[  479.724099] [<c051d598>] (sock_sendmsg) from [<c051dacc>] (___sys_sendmsg+0x204/0x22c)
[  479.732052] [<c051dacc>] (___sys_sendmsg) from [<c051e8b8>] (__sys_sendmsg+0x40/0x64)
[  479.739918] [<c051e8b8>] (__sys_sendmsg) from [<c0107980>] (ret_fast_syscall+0x0/0x54)
[  479.747971] ---[ end trace 0dafb8486082d14d ]---
```

yuhhaurlin, 2018-05-27:
This is warning message. I will check it.

bg4drl, 2018-05-28:
Google home max speaker not work

yuhhaurlin, 2018-08-24:
Did you still get this warning message with latest version of driver. If yes, I will try to reproduce and fix it.

yuhhaurlin, 2018-09-10:
I close this issue first. If you encounter any warning messages later, please create another issue to track it. Thanks.

closed 2018-09-10 by yuhhaurlin