kaloz/mwlwifi #298: IOT can't connect to 2,4Ghz

daloki83, 2018-05-28:
Other Router with default settings = OK
WRT3200 with Linksys OS = OK
WRT3200 OpenWrt SNAPSHOT r6865-419238fdb3 / LuCI Master (git-18.148.50949-a013967) =NOK
Device is a Gosund Smart Plug (https://www.amazon.de/gp/product/B07B911Y6V/ref=oh_aui_detailpage_o03_s00?ie=UTF8&psc=1)
I tried all different kinds of Encription/Cipher/Key/WifiName/Channel etc...but no way. The only working way is without encryption. If you need anything else, just let me know.

**SYSLOG**
`Mon May 28 21:03:02 2018 user.notice root: testLog #################################### CUT #################################################
Mon May 28 21:04:14 2018 daemon.info hostapd: wlan1: STA ec:fa:bc:93:57:c5 IEEE 802.11: associated (aid 3)
Mon May 28 21:04:14 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED ec:fa:bc:93:57:c5
Mon May 28 21:04:14 2018 daemon.info hostapd: wlan1: STA ec:fa:bc:93:57:c5 WPA: pairwise key handshake completed (RSN)
Mon May 28 21:04:14 2018 daemon.info hostapd: wlan1: STA ec:fa:bc:93:57:c5 IEEE 802.11: authenticated
Mon May 28 21:04:27 2018 kern.debug kernel: [13130.714789] ieee80211 phy1: staid 3 deleted
Mon May 28 21:04:27 2018 daemon.info hostapd: wlan1: STA ec:fa:bc:93:57:c5 IEEE 802.11: associated (aid 3)
Mon May 28 21:04:27 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED ec:fa:bc:93:57:c5
Mon May 28 21:04:27 2018 daemon.info hostapd: wlan1: STA ec:fa:bc:93:57:c5 WPA: pairwise key handshake completed (RSN)
Mon May 28 21:04:29 2018 daemon.info hostapd: wlan1: STA ec:fa:bc:93:57:c5 IEEE 802.11: authenticated
Mon May 28 21:04:33 2018 daemon.warn odhcpd[1729]: DHCPV6 SOLICIT IA_NA from 00030001f8461c81370d on br-lan: ok fd7d:4600:2d4e::763/128
Mon May 28 21:04:43 2018 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED ec:fa:bc:93:57:c5
Mon May 28 21:04:43 2018 daemon.info hostapd: wlan1: STA ec:fa:bc:93:57:c5 IEEE 802.11: disassociated
Mon May 28 21:04:43 2018 kern.debug kernel: [13146.623454] ieee80211 phy1: staid 3 deleted
Mon May 28 21:04:44 2018 daemon.info hostapd: wlan1: STA ec:fa:bc:93:57:c5 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Mon May 28 21:04:46 2018 daemon.info hostapd: wlan1: STA ec:fa:bc:93:57:c5 IEEE 802.11: associated (aid 3)
Mon May 28 21:04:46 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED ec:fa:bc:93:57:c5
Mon May 28 21:04:46 2018 daemon.info hostapd: wlan1: STA ec:fa:bc:93:57:c5 WPA: pairwise key handshake completed (RSN)
Mon May 28 21:04:46 2018 daemon.info hostapd: wlan1: STA ec:fa:bc:93:57:c5 IEEE 802.11: authenticated
Mon May 28 21:05:10 2018 kern.debug kernel: [13173.807477] ieee80211 phy1: staid 3 deleted
Mon May 28 21:05:10 2018 daemon.info hostapd: wlan1: STA ec:fa:bc:93:57:c5 IEEE 802.11: associated (aid 3)
Mon May 28 21:05:11 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED ec:fa:bc:93:57:c5
Mon May 28 21:05:11 2018 daemon.info hostapd: wlan1: STA ec:fa:bc:93:57:c5 WPA: pairwise key handshake completed (RSN)
Mon May 28 21:05:11 2018 daemon.info hostapd: wlan1: STA ec:fa:bc:93:57:c5 IEEE 802.11: authenticated
Mon May 28 21:05:39 2018 kern.debug kernel: [13202.914951] ieee80211 phy1: staid 3 deleted
Mon May 28 21:05:39 2018 daemon.info hostapd: wlan1: STA ec:fa:bc:93:57:c5 IEEE 802.11: associated (aid 3)
Mon May 28 21:05:40 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED ec:fa:bc:93:57:c5
Mon May 28 21:05:40 2018 daemon.info hostapd: wlan1: STA ec:fa:bc:93:57:c5 WPA: pairwise key handshake completed (RSN)
Mon May 28 21:05:40 2018 daemon.info hostapd: wlan1: STA ec:fa:bc:93:57:c5 IEEE 802.11: authenticated
`

**WIRELESS_CONF**
`config wifi-device 'radio0'
        option type 'mac80211'
        option country 'FR'
        option channel '48'
        option hwmode '11a'
        option htmode 'VHT160'
        option path 'soc/soc:pcie/pci0000:00/0000:00:01.0/0000:01:00.0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'CR4P_2_5G'
        option encryption 'psk2'
        option key '14digits'

config wifi-device 'radio1'
        option type 'mac80211'
        option country 'FR'
        option channel '8'
        option hwmode '11g'
        option htmode 'HT40'
        option noscan '1'
        option txpower '20'
        option path 'soc/soc:pcie/pci0000:00/0000:00:02.0/0000:02:00.0'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'CR4P_2_24G'
        option encryption 'psk2'
        option key '14digits'

config wifi-device 'radio2'
        option type 'mac80211'
        option channel '36'
        option hwmode '11a'
        option path 'platform/soc/soc:internal-regs/f10d8000.sdhci/mmc_host/mmc0/mmc0:0001/mmc0:0001:1'
        option htmode 'VHT80'

config wifi-iface 'default_radio2'
        option device 'radio2'
        option network 'lan'
        option mode 'ap'
        option ssid 'LEDE'
        option encryption 'none'
        option disabled '1'
`
**MWFWIFI**
`driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20180330
firmware version: 0x09030206
power table loaded from dts: no
firmware region code: 0x30
mac address: 60:38:e0:bd:94:b9
2g: enable
5g: disable
antenna: 4 4
irq number: 50
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e1200000
iobase1: e1480000
tx limit: 1024
rx limit: 16384
`

kubrickfr, 2018-05-29:
Don't set the country code to FR

https://github.com/kaloz/mwlwifi/issues/280

Leave it unset (00)

daloki83, 2018-05-29:
Then DFS will not work and my 5Ghz WiFi will be down. However, i tried it for both Wifi, and the result is still the same.

config wifi-device 'radio0'
        option type 'mac80211'
        option channel '48'
        option hwmode '11a'
        option htmode 'VHT160'
        option path 'soc/soc:pcie/pci0000:00/0000:00:01.0/0000:01:00.0'
        option country '00'
        option legacy_rates '1'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'CR4P_2_5G'
        option encryption 'psk2'
        option key '14digits'

config wifi-device 'radio1'
        option type 'mac80211'
        option channel '8'
        option hwmode '11g'
        option htmode 'HT40'
        option noscan '1'
        option txpower '20'
        option path 'soc/soc:pcie/pci0000:00/0000:00:02.0/0000:02:00.0'
        option country '00'
        option legacy_rates '1'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'CR4P_2_24G'
        option encryption 'psk2'
        option key '14digits'

config wifi-device 'radio2'
        option type 'mac80211'
        option channel '36'
        option hwmode '11a'
        option path 'platform/soc/soc:internal-regs/f10d8000.sdhci/mmc_host/mmc0/mmc0:0001/mmc0:0001:1'
        option htmode 'VHT80'

config wifi-iface 'default_radio2'
        option device 'radio2'
        option network 'lan'
        option mode 'ap'
        option ssid 'LEDE'
        option encryption 'none'
        option disabled '1'

iganeshk, 2018-05-31:
I assume you're running devices powered by ESP8266 or similar. I have been battling past 2 months here: https://github.com/kaloz/mwlwifi/issues/278

iganeshk, 2018-05-31:
For the time being, you can use the 3rd radio for your IoT/Embedded devices. If you don't have it installed, install "kmod-mwifiex-sdio" from the package manager. The only issue with this radio is DFS kicks in or you frequently disconnect/connect devices, it will stop working and rebooting the 3200ACM is the only way.

daloki83, 2018-06-01:
@yuhhaurlin maybe this can help you to fix this issue ?
https://www.espressif.com/sites/default/files/documentation/0a-esp8266ex_datasheet_en.pdf

yuhhaurlin, 2018-06-01:
I will check if we can get this device to test. However, it should be done after MU-MIMO.

iganeshk, 2018-06-01:
@yuhhaurlin the camera with a Realtek RTL8188EUS chip 
[image]

eduperez, 2018-06-01:
@yuhhaurlin <https://github.com/yuhhaurlin> You just need an ESP8266-01
module (can be bought them on eBay for a few bucks) a serial adapter (you
must have a bunch of them around there), and half an hour to prepare the
environment to compile and install a test program.

daloki83, 2018-06-14:
FYI - 10.3.8.0-20180614 - Still not working

yuhhaurlin, 2018-06-15:
Yes. This version of code is for 88W8997 with few minor fixes of original code.

p3x-robot, 2018-06-26:
@yuhhaurlin this latest commit is it worth building it? it is a bigfix so it is better right?

daloki83, 2018-07-04:
@yuhhaurlin will this commit fix that issue ?
https://github.com/kaloz/mwlwifi/commit/d9daa1dfd418a0a790188b805c95c99dea23f1bb

yuhhaurlin, 2018-07-04:
No. Your client is ESP8266? I think I will check remaining issues after MU-MIMO is done.

daloki83, 2018-07-04:
yep, its ESP8266

yuhhaurlin, 2018-07-04:
Once if MU-MIMO is done. I will check our QA to get this client and check it.
BTW, I remember @eduperez can connect this client to WRT series router.

daloki83, 2018-07-09:
@eduperez could you manage it with the ESP8266 ? If yes, would you please be so kind and tell me/us how ?

eduperez, 2018-07-10:
@daloki83 Please, browse my posts at issue #170, I explained all my tests and the results there.

iganeshk, 2018-07-10:
@daloki83 discussion over there is mostly about trying to connect your ESPs to radio2 rather than radio1.

So this issue has been on-going since June 2017 but why was the issue #170 marked solved?

eduperez, 2018-07-10:
@iganeshk I tested the ESP on both radio1 and radio2, in hopes that will give more information and make the issue easier to debug; but in my tests the ESP could connect to radio1 (albeit with some weird log messages).

iganeshk, 2018-07-10:
@eduperez with the latest driver, I've tried all the possible combinations of encryption here #278. The only time it connects is when I the encryption is set to None.

eduperez, 2018-07-11:
@iganeshk In my tests I did use encryption on radio1, and it worked. However, that was a long time ago, I now use radio2 for the ESP. Give me a couple of days, and I will repeat my tests with a more recent version of the drivers.

I want to clarify that I am not trying to defend Marvel or dismiss this issue, I just want to try to help fix it, by providing more tests.

iganeshk, 2018-07-11:
@eduperez do take your time. I've also found out it's not just the ESP being affected. The low power Realtek RTL8188EUS exhibits a similar case. But the problem with radio2 is that it gets disabled if devices connect and disconnect in short interval of time and can only be fixed by restarting the whole router, bringing down and up the interface doesn't fix it.

FWDInnovations, 2018-07-15:
Also having trouble with ESP8266 units here. Simply not connecting to wifi on my wrt32x at all (running [davidc502's latest OpenWRT build](https://forum.lede-project.org/t/davidc502-wrt1200ac-wrt1900acx-wrt3200acm-wrt32x-builds/15839/218)).

The device MAC shows up in Associated Stations list on the wireless settings page, but it does not get an IP address. Setting a static address also does not work.

Some people suggest disabling legacy b rates, but that does not work for me either.

Tried multiple alternative firmwares on the ESP chips, but none solve the issue.

Phil795, 2018-07-17:
With the settings shown below I can now connect my ESP devices. WMM has to be dispensed with, which is no problem in my case, as I provide my own radio network for the IoT devices.  

The following configuration is active on radio1, but this is a temporary solution and no problem solving. 

[image]

[image]

BrainSlayer, 2018-07-17:
if you disable you effectivily disable 802.11n in a indirect way. without wmm 802.11n is not operational. its well known that the issue an be resolved if you run the interface in 802.11g only

yuhhaurlin, 2018-08-24:
This one is duplicated as #278. I close it and use #278 to track issue with ESP8266.

closed 2018-08-24 by yuhhaurlin