kaloz/mwlwifi #299: 88W8864 firmware 7.2.9.27 available?

Gelated-Thoughts, 2018-06-05:
Looks like a new version of the 88W8864 firmware might be available.  I saw reference to it in:

http://downloads.linksys.com/downloads/releasenotes/WRT1900ACS_Customer_Release_Notes_1.0.3.187766.txt

Looks like Linksys has made the GPL source of this available at:

http://downloads.linksys.com/downloads/gpl/WRT1900ACS_v1.0.3.187766_GPL.tar.gz

BrainSlayer, 2018-06-07:
i dumped the firmware from the driver binary of the latest linksys firmware. (and yes it is different from the one used in mwlwifi) 
if someone is willing to test it ftp://ftp.dd-wrt.com/others/test/linksys_wrt1900_fw/88W8864.bin

BrainSlayer, 2018-06-07:
old mwlwifi fw
firmware version: 0x0702091a
new dumped from linksys
firmware version: 0x0702091b

Gelated-Thoughts, 2018-06-08:
It's far from a formal/proper test, but I installed the new FW image on my WRT1900ACS v1, which just gets casual/home use.  I'm running davidc502's [current Shelby build](https://davidc502sis.dynamic-dns.net/snapshots/r7093/).

After reboot, 'dmesg' shows the new firmware has been loaded:

[   10.931854] ieee80211 phy0: firmware version: 0x702091b
[   12.001855] ieee80211 phy1: firmware version: 0x702091b

Everything seems to be working just fine.

Gelated-Thoughts, 2018-06-08:
Oops, sorry.  Accidentally closed the issue, just reopened.

BrainSlayer, 2018-06-11:
the firmware rip i just posted works for all of these devices. just try it, use it as is. dont know if @yuhhaurlin will update it here as well. i dont expect much changes, maybe some regulatory related things

yuhhaurlin, 2018-06-12:
Yes. This version of firmware only affects regulatory rule. I don't think we need it for mwlwifi. Thanks.

closed 2018-06-12 by yuhhaurlin